geerlingguy/raspberry-pi-pcie-devices #587: Add 52Pi PD Power  HAT

tltangliang, 2024-01-16:
The Raspberry Pi 5 power requirements are 5V 5A PDO, which common PD power sources cannot satisfy. This PD HAT from 52pi has a built-in DC-DC converter, supporting a wide voltage input of 9-24V or a PD power input of 9-20V. It outputs a maximum power of 5V 8A and supports the 5V 5A PDO of the Raspberry Pi 5. It seems that there's no need to purchase a dedicated power adapter for the Pi 5 separately.

[image]

geerlingguy, 2024-01-16:
Thanks! I have the board listed on the site now: https://pipci.jeffgeerling.com/hats/52pi-pd-power-hat.html

And with that extra circuitry, it's kinda no wonder Pi didn't want to try to build in full PD support for 9v/12v/etc on the board!

closed 2024-01-16 by geerlingguy

YunXi1027, 2024-03-12:
Where can I buy it ?

tltangliang, 2024-03-13:
52pi hasn't hit the shelves yet, guess it's soon, should be in the next couple weeks

YunXi1027, 2024-03-13:
Ok, thank you for your reply

marekcygan, 2024-07-03:
Is it possible to buy?

dhlavaty, 2025-01-27:
Photos here and at https://pipci.jeffgeerling.com/hats/52pi-pd-power-hat.html are different. Seems like different HATs to me.