geerlingguy/raspberry-pi-pcie-devices #588: Add NVPI5-2242T and NVPI5-2280B HAT

tyanbiao, 2024-01-16:
NVPI5-2242T and NVPI5-2280BSSD HAT are new SSD peripheral Board for Raspberry Pi 5.

It comes from the PCBWay store. It looks similar to the Pineberry design, but the price is much cheaper.

https://www.pcbway.com/project/gifts_detail/NVPI5_2242T_8ef7a248.html

[image]
[image]

tyanbiao, 2024-01-18:
I'v created a pull request for it.

tltangliang, 2024-01-19:
There website even provide the schematic of the board, it looks quite simple.

[SCH_NVPI5-2280B.pdf](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/13989435/SCH_NVPI5-2280B.pdf)