geerlingguy/raspberry-pi-pcie-devices #595: Waveshare PCIe To M.2 Adapter for Raspberry Pi 5 Testing

tltangliang, 2024-01-21:
I can only give 3 ratings, STUPID! STUPID! STUPID! 

# Default soldered 2230 screw 

 The default welding is 2230 screws. The 2242 hard drive with flash welded on the back cannot be used at all. There are a lot of such hard drives, and Waveshare's NVMe hat can't even be called a product.
[image]

[image]

# soldering deficiency

This female connector is not soldered at all, it can be seen with the naked eye. Don't they check at the factory?

[image]

# Strange Resistors

The product I received seems to have a capacitor burned here, and a resistor is soldered at an angle. I dare not power it on for fear of burning my Raspberry Pi 5.

[image]

geerlingguy, 2024-01-21:
Looks like a mixture of poor design choice (having the screw adapters soldered onto the board instead of a removable/movable option), and poor quality control (the resistor popped off, header floating... looks like factory pick and place AOI failure).

TCIII, 2024-03-23:
@tltangliang,
"The product I received seems to have a capacitor burned here, and a resistor is soldered at an angle. I dare not power it on for fear of burning my Raspberry Pi 5."

Where did you purchase your Waveshare HAT from?

tltangliang, 2024-03-25:
Taobao.com

tltangliang, 2024-03-25:
> @tltangliang, "The product I received seems to have a capacitor burned here, and a resistor is soldered at an angle. I dare not power it on for fear of burning my Raspberry Pi 5." “我收到的产品这里好像烧了一个电容，还有一个斜焊的电阻，不敢开机，怕烧坏我的树莓派5。”
> 
> Where did you purchase your Waveshare HAT from?您从哪里购买 Waveshare HAT？

taobao.com

TCIII, 2024-03-27:
@tltangliang 
I just purchased a Waveshare PCIe To M.2 Adapter for my Raspberry Pi 5 from Amazon Prime, added a Transcend 256GB SSD module and had it up and running in about 1/2 an hour.
The quality of the Waveshare Adapter was excellent, unlike yours, and I had no problem inserting and mounting the Transcend 256GB SSD module on the Adapter.
The Bookworm OS now boots up off of the SSD module and I have upped the PCIe speed to the 'PCIe 3' mode from the 'PCIe 2' mode without any issues.

jibin2706, 2024-10-03:
I'm not able to remove the screw from 2242 hole (tried a lot)

krzysztofmaslak3, 2025-01-13:
> I'm not able to remove the screw from 2242 hole (tried a lot)

you need to use soldering tool, when you heat it up it jumps out