geerlingguy/raspberry-pi-pcie-devices #597: Add Waveshare PoE HAT (F) for Raspberry Pi 5

geerlingguy, 2024-01-29:
The [Waveshare PoE HAT (F) for Raspberry Pi 5](https://amzn.to/3vWYrLZ) is a $30 PoE HAT that should power the Raspberry Pi 5 off any PoE or PoE+ network switch/adapter.

[image]

It comes with a thin heatsink and three thermal pads, and that heatsink snaps into place using the same mounting holes the Raspberry Pi Active Cooler uses.

While this is not strictly a PCIe HAT, it does have cutouts for both PCIe port access and CSI/DSI connections, which is nice. Plus, I've tested it with Pimoroni's NVMe BASE and found no problems booting off a Kioxia XG6 NVMe SSD. See: #565.

geerlingguy, 2024-01-29:
The lowest I could get the voltage to drop under load (with an NVMe SSD benchmark running concurrently with `stress-ng` on all cores) was 5.04V. The voltage averaged 5.10V over the course of my testing.

[image]

geerlingguy, 2024-01-29:
It is on the site now: https://pipci.jeffgeerling.com/hats/waveshare-poe-hat-f.html

I have it in my rack and will be testing longer term.

closed 2024-01-29 by geerlingguy

geerlingguy, 2024-01-29:
Note: It does _not_ fit inside the official Pi case. It's about 5-6mm too long on the port-side edge:

[image]

geerlingguy, 2024-01-31:
I just noticed, if I plug in a high power USB device, it doesn't seem to be getting full functionality.

```
[   38.654342] usb 3-1: new high-speed USB device number 2 using xhci-hcd
[   38.808367] usb 3-1: New USB device found, idVendor=0bda, idProduct=9210, bcdDevice=21.00
[   38.808373] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   38.808374] usb 3-1: Product: RTL9210
[   38.808376] usb 3-1: Manufacturer: Realtek
[   38.808378] usb 3-1: SerialNumber: 012345679296
[   38.858734] usb-storage 3-1:1.0: USB Mass Storage device detected
[   38.858881] scsi host0: usb-storage 3-1:1.0
[   39.894582] scsi 0:0:0:0: Direct-Access     Realtek  RTL9210 NVME     1.00 PQ: 0 ANSI: 6
[   39.898786] sd 0:0:0:0: Attached scsi generic sg0 type 0
[   39.906048] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=DRIVER_OK
[   39.906054] sd 0:0:0:0: [sda] Sense Key : 0x5 [current] 
[   39.906055] sd 0:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[   39.906059] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[   39.906060] sd 0:0:0:0: [sda] 0-byte physical blocks
[   39.909323] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[   39.912572] sd 0:0:0:0: [sda] Asking for cache data failed
[   39.912574] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   39.912793] sd 0:0:0:0: [sda] Attached SCSI disk
```

And it shows up but is a little funky (notice the 0B):

```
pi@pi5base:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0     0B  0 disk 
nvme0n1     259:0    0 953.9G  0 disk 
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
└─nvme0n1p2 259:2    0 953.4G  0 part /
```

Then I checked:

```
pi@pi5base:~ $ vcgencmd get_config usb_max_current_enable
usb_max_current_enable=0
```

So I will have to force that (since Waveshare says I should have 2.4A at 5V, so around 12W) and see if I can power the Pi and _two_ NVMe drives...

Edit `sudo nano /boot/firmware/config.txt` and add to the bottom:

```
usb_max_current_enable=1
```

TODO.

sswc15, 2024-02-05:
@geerlingguy Using the Waveshare POE HAT with the HatDrive! Bottom, every time I boot I get the following error on my Pi5 desktop - "This power supply is not capable of supplying 5A. Power to peripherals will be restricted". Running the HatDrive! Bottom as Gen 2 in case that makes a difference. Does this affect functionality of the HatDrive?

geerlingguy, 2024-02-05:
@sswc15 - Have you tried adding: `usb_max_current_enable=1` to the bottom of your `/boot/firmware/config.txt` file? That should fix that issue, and I will be releasing a video with more on my testing soon—possibly tomorrow morning :)

geerlingguy, 2024-02-06:
And a video: https://www.youtube.com/watch?v=OEctB0HOpZ8

Molgaard78, 2024-02-07:
@geerlingguy: I have the same issue as @sswc15.

I have added usb_max_current_enable=1, but I still get the message on launch.

My guess it is because I have hooked it into a Ubiquiti US-8-60W because it is only capable to supply PoE and **not** PoE+?

sswc15, 2024-02-07:
> @geerlingguy: I have the same issue as @sswc15.
> 
> I have added usb_max_current_enable=1, but I still get the message on launch.
> 
> My guess it is because I have hooked it into a Ubiquiti US-8-60W because it is only capable to supply PoE and **not** PoE+?

I watched the video @geerlingguy - good post! I am still getting an issue even after changing the usb_max_current. I don't think that would make a difference though since all I'm running is the NVMe bottom via PCIe Gen 2 set. 

Using a TP-Link TL-SG1008P which is supposed to provide PoE+ up to 64 W and up to 30 W per port. Since I only have a Pi 3 and the Pi 5 running on PoE, I'd expect 30 W per. When I run vcgencmd pmic_read_adc I also see just above 5 for the EXT5AV_V value. Anything else we could check to see if this is just a warning or if it's really restricting power?

Molgaard78, 2024-02-07:
I agree it is a very good post. :-)

For me personally it is not so much of an issue, since I am just using it for Unifi Network Server and DNS.

After all: The Raspberry boots and thats enough for me. The error message is just a momentarily annoyance. :-)

What I really want to use it for is Kubernetes (k3s), but I have other issues there, that is outside the of this post.

ssalonen, 2024-03-01:
I understand that the fan is constantly on. Is there a way hack/mod to turn it on/off based on cpu temp?

geerlingguy, 2024-03-01:
@ssalonen - Nope, not on this board.

ssalonen, 2024-03-01:
Thanks for responding! Do you think it would be possible to connect fan via something like this? https://thepihut.com/products/fan-controller-for-raspberry-pi

What is the voltage?

Molgaard78, 2024-03-01:
@ssalonen: The fan takes 5V.

See the pin in left side on the picture.

(Sorry for the bad light). :-)

The other pin is the 12V output for external devices.

[image]

It will get tricky connecting fan controller to the PoE fan, since the wire for the fan and the 5V connection is placed underneath the hat.

My own issue with the fan at the moment, is that the fan is ever so slightly toching the board, so it is rather audible when connected to my Raspberry Pi 5. 

I guess I have to bring out my tools to tweak the the screws holding the fan in place.

uofirob, 2024-04-05:
I am using this with batocera... I've noticed that when I use the shutdown command, the device reboots in under 5 seconds. Is there any way to make the device stay off when we've turned it off?

Molgaard78, 2024-04-05:
@uofirob: That sounds a bit strange, but it is not the first time I have seen issues with shutdown/halt/reboot scripts.

The idea with shutdown is that it is switching the system to runlevel 0.

So is it possible to shutdown using the command `init 0`? It should have same effect as `shutdown now`.

For reference: Reboot is runlevel 6, so calling `init 6` should have the have the same effect as `reboot now`.

uofirob, 2024-04-05:
In Batocera via the UI I'm not exactly sure what command is being called but when I call init 0 from SSH, it seems to stay off.  Now if only I could get it to work via the UI so that it stays off.  Thanks!

Schaggo, 2024-04-09:
I got a somewhat related issue. The Pi does not reboot via `sudo reboot` it seems to get stuck at the insufficient PSU notice, regardless of the max current flag.

fraannk, 2024-05-06:
Two questions: 

- What happens when powering the Pi with USB-C while this PoE HAT is connected? Using the official Pi 4 PoE+ HAT, on a Pi 4, made a horrendous sound when powering with USB-C, as also mentioned in @geerlingguy's video on that HAT. 
- What are the dimensions of the fan?

siberex, 2024-05-06:
> * What are the dimensions of the fan?

3007 sticker means it is 30×30×7mm

Molgaard78, 2024-05-07:
> * What happens when powering the Pi with USB-C while this PoE HAT is connected?

Well... I have done that because my Pi was unexpectly rebooting while I was using `dd` to clone a harddrive over USB.

Long story short: Your Raspberry will be fine.

colts45s, 2024-06-05:
Can you remove the fan and use the active cooler? Just curious if it fits and if thermals would be affected since the active cooler uses pwm instead of always on.

leander091, 2024-06-05:
@colts45s, I am not sure this would be possible without some serious modding of the PoE hat.  The hat is connected using the GPIO pins and the 4 pin header on the Pi.  But to make sure it fits on top of the  active cooler the GPIO header of the Waveshare hat needs to be removed.

siberex, 2024-06-08:
> This power supply is not capable of supplying 5A. Power to peripherals will be restricted

Looks like when powering from PoE you have to explicitly tell Pi that you indeed connected a high-amp PSU:

```bash
sudo rpi-eeprom-config --edit
```

```
PSU_MAX_CURRENT=5000
```

siberex, 2024-06-08:
> Can you remove the fan and use the active cooler? Just curious if it fits and if thermals would be affected since the active cooler uses pwm instead of always on.

Active Cooler will not fit as a drop-in replacement. But in terms of connectivity you can safely disconnect the hat’s fan and either use it fanless or connect any PWM 5V fan to the Pi’s designated socket

Bahtiyar57, 2024-06-10:
Did everyone test it with 64bit system?

Molgaard78, 2024-06-10:
> Did everyone test it with 64bit system?

I did and I am fairly certain everyone else did, considering the CPU involved.

No reason to use 32bit version.

fili, 2024-06-14:
> Can you remove the fan and use the active cooler? Just curious if it fits and if thermals would be affected since the active cooler uses pwm instead of always on.

Yes, this is possible but with some caveats. 

You will need to increase the distance between the HAT and the Raspberry Pi 5. So you need 25mm spacers and extensions for the 4 pin and GPIO pins.

Here is an example of the WaveShare PoE HAT on a Raspberry Pi 5

[image]

And here is an example (sorry low quality photo) of the extensions for the pins. I recommend 6,5mm length of the pins.

[image]

And here is an example of two Raspberry Pi 5 with the official Raspberry Pi 5 active cooler and with WaveShare PoE HATs in a rack, powered by PoE (Netgear switch).

[image]

colts45s, 2024-06-15:
> You will need to increase the distance between the HAT and the Raspberry Pi 5. So you need 25mm spacers and extensions for the 4 pin and GPIO pins.
> 

Thanks! That's exactly what I was thinking.

Molgaard78, 2024-06-15:
> > This power supply is not capable of supplying 5A. Power to peripherals will be restricted
> 
> Looks like when powering from PoE you have to explicitly tell Pi that you indeed connected a high-amp PSU:
> 
> ```shell
> sudo rpi-eeprom-config --edit
> ```
> 
> ```
> PSU_MAX_CURRENT=5000
> ```

Actually: According to [Waveshare website](https://www.waveshare.com/PoE-HAT-F.htm) it can deliver 5V 4.5A at most via GPIO so that should translate to `PSU_MAX_CURRENT=4500`. And on top of that it can delivery 12V 2A via 2P header on bottom side of the HAT.

I tried changing the value to 4500, which resulted in the `This power supply is not capable of supplying 5A` message appearing on screen.

I tried connecting a NVMe drive to the USB3 port with a powered USB3 hub in between.

That did not go well as desktop became unresponsive and the Raspberry stopped responding to ping packages.

fallenguru, 2024-08-15:
Does the PoE HAT (F) have no fan control as in a) it's running full tilt all the time; or b) the fan control curve is fixed (and does not go all the way down to "off")?

fili, 2024-08-16:
I am not sure if it is a or b (I think a), but that is why you might want to try the (G) version with the original raspberry pi 5 fan. Note that you will need three 20mm spacers and an extension for the header.

Roiskia, 2024-08-27:
Has anyone used the 12v 2pins? I have two hats and on both of them i am not able to get a 12v noctua fan to spin. The fan works. Am i missing something?

max06, 2025-07-24:
One out of 4 fans on the (F) version failed on me after 6,5 months, now producing noise. It's probably only a matter of time for the other 3. I might have to think about modifications...