geerlingguy/raspberry-pi-pcie-devices #599: Add HAT: Mcuzone MPW7 E-Key WiFi 7 HAT

geerlingguy, 2024-02-02:
As seen on [CNX](https://www.cnx-software.com/2024/02/02/m2-e-key-wifi-7-hat-for-raspberry-pi-5-google-tpu/), Mcuzone just introduced the [MPW7](https://www.aliexpress.us/item/3256806224989235.html), available with SMA antenna connections for $12, and without for $10, right now only on AliExpress.

[image]

For WiFi 7, you would need to buy an appropriate WiFi 7 adapter... which I haven't really seen any yet (much less one that has drivers you could get running on a Pi). But WiFi 6E is definitely possible, as well as some expensive 5G chips.

geerlingguy, 2024-02-06:
It's on the site: https://pipci.jeffgeerling.com/hats/mcuzone-mpw7-wifi-7.html

closed 2024-02-10 by geerlingguy

thewade, 2024-04-16:
As commented in issue #120 I was able to get both the Intel AX210 (Wifi6E) and Intel BE200 (Wifi7) to work with Ubuntu 24.04 beta (kernel 6.8.0) with the Mcuzone E-key M.2 hat and the Raspberry Pi 5.

I also tested the MediaTek 7921k/RZ608 and Qualcomm QCNFA765 and although both were detected in lspci, their respective kernel modules failed to load (mt7921 and ath11k respectively).

For other Wifi7 devices Qualcomm has QCNCM825 and QCNCM865. I don't have one yet, but I suspect it would be the same result as the QCNFA765.

Anyway, the two Intel cards confirm this adapter works. I may open issues for RZ608 and QCNFA765 later if I have more time to document and debug.

ddodds-nlplogix, 2024-04-23:
I have been trying for the life of me to get this running on RaspbiOS 6.6.y and am finding a ton of compatibility issues with firmware versions. Everything from file size mismatching, to timeouts. I may try Ubuntu 24.04 now.

`[    5.137174] iwlwifi: unknown parameter 'debug' ignored
[    5.137840] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
[    5.151127] iwlwifi 0000:01:00.0: Detected crf-id 0x400410, cnv-id 0x400410 wfpm id 0x80000000
[    5.151153] iwlwifi 0000:01:00.0: PCI dev 2725/0024, rev=0x420, rfid=0x10d000
[    5.151241] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-83.ucode failed with error -2
[    5.151263] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-82.ucode failed with error -2
[    5.151282] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-81.ucode failed with error -2
[    5.151295] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-80.ucode failed with error -2
[    5.151308] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-79.ucode failed with error -2
[    5.151321] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-78.ucode failed with error -2
[    5.151334] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-77.ucode failed with error -2
[    5.151348] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-76.ucode failed with error -2
[    5.151361] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-75.ucode failed with error -2
[    5.151376] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-74.ucode failed with error -2
[    5.151388] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-73.ucode failed with error -2
[    5.151403] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-72.ucode failed with error -2
[    5.151416] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-71.ucode failed with error -2
[    5.151429] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-70.ucode failed with error -2
[    5.151441] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-69.ucode failed with error -2
[    5.151454] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-68.ucode failed with error -2
[    5.151466] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-67.ucode failed with error -2
[    5.151479] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-66.ucode failed with error -2
[    5.151494] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-65.ucode failed with error -2
[    5.151506] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-64.ucode failed with error -2
[    5.151520] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-63.ucode failed with error -2
[    5.151536] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-62.ucode failed with error -2
[    5.151551] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-61.ucode failed with error -2
[    5.151564] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-60.ucode failed with error -2
[    5.174510] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
[    5.174571] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 93.8.63.28
[    5.175746] iwlwifi 0000:01:00.0: loaded firmware version 59.601f3a66.0 ty-a0-gf-a0-59.ucode op_mode iwlmvm
[    5.561061] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[    6.614959] WARNING: CPU: 0 PID: 518 at drivers/net/wireless/intel/iwlwifi/pcie/trans.c:2220 __iwl_trans_pcie_grab_nic_access+0x168/0x178 [iwlwifi]
[    6.615006] Modules linked in: algif_hash algif_skcipher af_alg bnep iwlmvm(+) spidev mac80211 brcmfmac_wcc binfmt_misc vc4 libarc4 hci_uart snd_soc_hdmi_codec drm_display_helper aes_ce_blk aes_ce_cipher ghash_ce btusb gf128mul btrtl sha2_ce btintel sha256_arm64 btbcm iwlwifi sha1_ce bluetooth brcmfmac cec ecdh_generic raspberrypi_hwmon drm_dma_helper ecc brcmutil libaes drm_kms_helper cfg80211 snd_soc_core rpivid_hevc(C) snd_compress snd_pcm_dmaengine i2c_brcmstb v4l2_mem2mem pisp_be snd_pcm videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videodev rfkill snd_timer v3d snd spi_bcm2835 videobuf2_common mc gpio_keys gpu_sched drm_shmem_helper raspberrypi_gpiomem rp1_adc nvmem_rmem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[    6.615101] pc : __iwl_trans_pcie_grab_nic_access+0x168/0x178 [iwlwifi]
[    6.615131] lr : __iwl_trans_pcie_grab_nic_access+0x168/0x178 [iwlwifi]
[    6.615215]  __iwl_trans_pcie_grab_nic_access+0x168/0x178 [iwlwifi]
[    6.615244]  iwl_trans_pcie_grab_nic_access+0x2c/0x68 [iwlwifi]
[    6.615272]  iwl_read_prph+0x2c/0x90 [iwlwifi]
[    6.615452]  _iwl_op_mode_start.isra.12+0x6c/0xc0 [iwlwifi]
[    6.615482]  iwl_opmode_register+0x84/0x110 [iwlwifi]
[    6.615595] iwlwifi 0000:01:00.0: iwlwifi transaction failed, dumping registers
[    6.615599] iwlwifi 0000:01:00.0: iwlwifi device config registers:
[    6.615719] iwlwifi 0000:01:00.0: 00000000: 27258086 00100000 0280001a 00000000 00000004 00000000 00000000 00000000
[    6.615723] iwlwifi 0000:01:00.0: 00000020: 00000000 00000000 00000000 00248086 00000000 000000c8 00000000 00000100
[... cut ...]
[   14.214250] iwlwifi 0000:01:00.0: WRT: Failed to dump region: id=21, type=10
[   14.297402] iwlwifi 0000:01:00.0: WRT: Failed to dump region: id=1, type=10
[   14.395601] iwlwifi 0000:01:00.0: Failed to get monitor header
[   14.395603] iwlwifi 0000:01:00.0: WRT: Failed to fill region header: id=16, type=2
[   14.461033] iwlwifi 0000:01:00.0: WRT: Failed to dump region: id=21, type=10
[   16.185347] iwlwifi 0000:01:00.0: Failed to run INIT ucode: -110
[   16.198593] iwlwifi 0000:01:00.0: retry init count 2`

me-peng, 2024-04-24:
I have the same problem as "ddodds-nlplogix"
"Failed to run INIT ucode: -110"

ddodds-nlplogix, 2024-04-24:
> I have the same problem as "ddodds-nlplogix" "Failed to run INIT ucode: -110"

I recently retested this using Ubuntu 24.04 beta which uses kernel version 6.8, while the latest on raspbian uses 6.6. iwlwifi comes baked into the 24.04 beta image, and the firmware is already on it. All I had to do was run `sudo modprobe iwlwifi` to get everything working. After a reboot. I found the wlps10 interface, and was able to connect to a wifi network with it. Will do more extensive testing as we work with the chip.

astrizhkin, 2024-08-02:
I was able to run Coral TPU with Mcuzone MPW7 adapter but only at PCIe Gen1 speed.
lspci won't list the TPU in Gen2/Gen3 modes

```
dtparam=pciex1
dtparam=pciex1_gen=1
kernel=kernel8.img
dtoverlay=pineboards-hat-ai
```

jmdurant, 2024-10-03:
> As commented in issue #120 I was able to get both the Intel AX210 (Wifi6E) and Intel BE200 (Wifi7) to work with Ubuntu 24.04 beta (kernel 6.8.0) with the Mcuzone E-key M.2 hat and the Raspberry Pi 5.6
> 
> I also tested the MediaTek 7921k/RZ608 and Qualcomm QCNFA765 and although both were detected in lspci, their respective kernel modules failed to load (mt7921 and ath11k respectively).
> 
> For other Wifi7 devices Qualcomm has QCNCM825 and QCNCM865. I don't have one yet, but I suspect it would be the same result as the QCNFA765.
> 
> Anyway, the two Intel cards confirm this adapter works. I may open issues for RZ608 and QCNFA765 later if I have more time to document and debug.

Trying with MediaTek 7925 card with the Mcuzone MPW7 and the card is detected in lspci, but kernel modules aren't loading as was your experience.  Running 6.12.rc1 and have the latest kernel firmware downloaded.  Any ideas why it's not loading the kernel modules?  Have tried the settings below with no success either.

`dtparam=pciex1
dtparam=pciex1_gen=1
kernel=kernel8.img
`

ItsVRK, 2025-04-02:
This hat with a BE200 nic is visible for me on a cold boot of Raspberry Pi OS, however not visible at all on a warm reboot.:

```
uname -a
Linux TEST 6.12.21-v8+ #1868 SMP PREEMPT Mon Mar 31 17:09:25 BST 2025 aarch64 GNU/Linux

lspci
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0001:01:00.0 Network controller: Intel Corporation Wi-Fi 7(802.11be) AX1775*/AX1790*/BE20*/BE401/BE1750* 2x2 (rev 1a)
0002:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0002:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge

dmesg | grep -i iwlwifi
[   14.335245] iwlwifi 0001:01:00.0: BAR 0 [mem 0x1b00000000-0x1b00003fff 64bit]: assigned
[   14.335419] iwlwifi 0001:01:00.0: enabling device (0000 -> 0002)
[   14.337036] iwlwifi 0001:01:00.0: Detected crf-id 0x2001910, cnv-id 0x2001910 wfpm id 0x80000000
[   14.337105] iwlwifi 0001:01:00.0: PCI dev 272b/00f4, rev=0x472, rfid=0x112200
[   14.337110] iwlwifi 0001:01:00.0: Detected Intel(R) Wi-Fi 7 BE200 320MHz
[   14.337264] iwlwifi 0001:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-93.ucode failed with error -2
[   14.337282] iwlwifi 0001:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-92.ucode failed with error -2
[   14.337296] iwlwifi 0001:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-91.ucode failed with error -2
[   14.337309] iwlwifi 0001:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-90.ucode failed with error -2
[   14.337313] iwlwifi 0001:01:00.0: no suitable firmware found!
[   14.337315] iwlwifi 0001:01:00.0: minimum version required: iwlwifi-gl-c0-fm-c0-90
[   14.337318] iwlwifi 0001:01:00.0: maximum version supported: iwlwifi-gl-c0-fm-c0-93
[   14.337321] iwlwifi 0001:01:00.0: check git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```

thewade, 2025-04-03:
The error says you need to update the firmware. Likely what happened was you updated the kernel and the firmware requirements changed.

See https://pipci.jeffgeerling.com/cards_network/intel-be200-wifi-7.html for info on how install the new firmware.

ItsVRK, 2025-04-03:
> The error says you need to update the firmware. Likely what happened was you updated the kernel and the firmware requirements changed.
> 
> See https://pipci.jeffgeerling.com/cards_network/intel-be200-wifi-7.html for info on how install the new firmware.

Yeah I got further, but firmware v93 isn't available, and getting the folloiwng error:
```
dmesg | grep -i iwlwifi
[   17.761455] iwlwifi 0001:01:00.0: BAR 0 [mem 0x1b00000000-0x1b00003fff 64bit]: assigned
[   17.761602] iwlwifi 0001:01:00.0: enabling device (0000 -> 0002)
[   17.784726] iwlwifi 0001:01:00.0: Detected crf-id 0x2001910, cnv-id 0x2001910 wfpm id 0x80000000
[   17.784836] iwlwifi 0001:01:00.0: PCI dev 272b/00f4, rev=0x472, rfid=0x112200
[   17.784842] iwlwifi 0001:01:00.0: Detected Intel(R) Wi-Fi 7 BE200 320MHz
[   17.786006] iwlwifi 0001:01:00.0: Direct firmware load for iwlwifi-gl-c0-fm-c0-93.ucode failed with error -2
[   17.905010] iwlwifi 0001:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.182.4.1
[   17.905764] iwlwifi 0001:01:00.0: loaded firmware version 92.67ce4588.0 gl-c0-fm-c0-92.ucode op_mode iwlmvm
[   20.224832] iwlwifi 0001:01:00.0: SecBoot CPU1 Status: 0xffffffff, CPU2 Status: 0xffffffff
[   20.224848] iwlwifi 0001:01:00.0: WFPM_LMAC1_PD_NOTIFICATION: 0xffffffff
[   20.224857] iwlwifi 0001:01:00.0: HPM_SECONDARY_DEVICE_STATE: 0xffffffff
[   20.224865] iwlwifi 0001:01:00.0: WFPM_MAC_OTP_CFG7_ADDR: 0xffffffff
[   20.224873] iwlwifi 0001:01:00.0: WFPM_MAC_OTP_CFG7_DATA: 0xffffffff
[   20.224876] iwlwifi 0001:01:00.0: UMAC CURRENT PC: 0xd05c18
[   20.224879] iwlwifi 0001:01:00.0: LMAC1 CURRENT PC: 0xd05c20
[   20.224881] iwlwifi 0001:01:00.0: LMAC2 CURRENT PC: 0xd05c24
[   20.224884] iwlwifi 0001:01:00.0: UMAC CURRENT PC 1: 0xd05c1c
[   20.224886] iwlwifi 0001:01:00.0: TCM1 CURRENT PC: 0xa0444c
[   20.224889] iwlwifi 0001:01:00.0: RCM1 CURRENT PC: 0xa0bd7c
[   20.224891] iwlwifi 0001:01:00.0: RCM2 CURRENT PC: 0xb0bd7c
[   20.224895] iwlwifi 0001:01:00.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
[   20.224941] iwlwifi 0001:01:00.0: Start IWL Error Log Dump:
[   20.224943] iwlwifi 0001:01:00.0: Transport status: 0x00000042, valid: -1
[   20.224947] iwlwifi 0001:01:00.0: Loaded firmware version: 92.67ce4588.0 gl-c0-fm-c0-92.ucode
[   20.224950] iwlwifi 0001:01:00.0: 0xFFFFFFFF | ADVANCED_SYSASSERT
[   20.224953] iwlwifi 0001:01:00.0: 0xFFFFFFFF | trm_hw_status0
[   20.224956] iwlwifi 0001:01:00.0: 0xFFFFFFFF | trm_hw_status1
[   20.224958] iwlwifi 0001:01:00.0: 0xFFFFFFFF | branchlink2
[   20.224961] iwlwifi 0001:01:00.0: 0xFFFFFFFF | interruptlink1
[   20.224963] iwlwifi 0001:01:00.0: 0xFFFFFFFF | interruptlink2
[   20.224965] iwlwifi 0001:01:00.0: 0xFFFFFFFF | data1
[   20.224968] iwlwifi 0001:01:00.0: 0xFFFFFFFF | data2
[   20.224970] iwlwifi 0001:01:00.0: 0xFFFFFFFF | data3
[   20.224972] iwlwifi 0001:01:00.0: 0xFFFFFFFF | beacon time
[   20.224975] iwlwifi 0001:01:00.0: 0xFFFFFFFF | tsf low
[   20.224977] iwlwifi 0001:01:00.0: 0xFFFFFFFF | tsf hi
[   20.224979] iwlwifi 0001:01:00.0: 0xFFFFFFFF | time gp1
[   20.224982] iwlwifi 0001:01:00.0: 0xFFFFFFFF | time gp2
[   20.224984] iwlwifi 0001:01:00.0: 0xFFFFFFFF | uCode revision type
[   20.224987] iwlwifi 0001:01:00.0: 0xFFFFFFFF | uCode version major
[   20.224989] iwlwifi 0001:01:00.0: 0xFFFFFFFF | uCode version minor
[   20.224991] iwlwifi 0001:01:00.0: 0xFFFFFFFF | hw version
[   20.224994] iwlwifi 0001:01:00.0: 0xFFFFFFFF | board version
[   20.224996] iwlwifi 0001:01:00.0: 0xFFFFFFFF | hcmd
[   20.224998] iwlwifi 0001:01:00.0: 0xFFFFFFFF | isr0
[   20.225000] iwlwifi 0001:01:00.0: 0xFFFFFFFF | isr1
[   20.225003] iwlwifi 0001:01:00.0: 0xFFFFFFFF | isr2
[   20.225005] iwlwifi 0001:01:00.0: 0xFFFFFFFF | isr3
[   20.225007] iwlwifi 0001:01:00.0: 0xFFFFFFFF | isr4
[   20.225010] iwlwifi 0001:01:00.0: 0xFFFFFFFF | last cmd Id
[   20.225012] iwlwifi 0001:01:00.0: 0xFFFFFFFF | wait_event
[   20.225014] iwlwifi 0001:01:00.0: 0xFFFFFFFF | l2p_control
[   20.225017] iwlwifi 0001:01:00.0: 0xFFFFFFFF | l2p_duration
[   20.225019] iwlwifi 0001:01:00.0: 0xFFFFFFFF | l2p_mhvalid
[   20.225021] iwlwifi 0001:01:00.0: 0xFFFFFFFF | l2p_addr_match
[   20.225023] iwlwifi 0001:01:00.0: 0xFFFFFFFF | lmpm_pmg_sel
[   20.225026] iwlwifi 0001:01:00.0: 0xFFFFFFFF | timestamp
[   20.225028] iwlwifi 0001:01:00.0: 0xFFFFFFFF | flow_handler
[   20.225067] iwlwifi 0001:01:00.0: Start IWL Error Log Dump:
```

thewade, 2025-04-03:
From the log it says I accepts versions 90 to 93. Try deleting 92 and try iwlwifi-gl-c0-fm-c0-90.ucode.

thewade, 2025-04-03:
If that doesn't work try kernel 6.6.y per the install guide.