geerlingguy/raspberry-pi-pcie-devices #600: ASMedia Technology Inc. Device 1064 M.2 NVME to Mini SAS Expansion Card Support 4 Ports SATA3.0 6Gbps HDD SSD SATA Controller SFF8087

ezaul, 2024-02-05:
I bought it at the end of the year on Aliexpress to test with my Raspberry pi 5 + HatDrive! **Bottom**

SATA controller: ASMedia Technology Inc. Device 1064 (rev 02) (prog-if 01 [AHCI 1.0])
[image]
https://es.aliexpress.com/item/1005006050617325.html?spm=a2g0o.order_list.order_list_main.5.4f40194dhsM2q1&gatewayAdapt=glo2esp

HatDrive! **Bottom**
[image]
https://pineberrypi.com/products/hatdrive-bottom-2230-2242-2280-for-rpi5?pr_prod_strat=jac&pr_rec_id=aa41a0aa3&pr_rec_pid=8729962250573&pr_ref_pid=8729958023501&pr_seq=uniform

**lspci -k**

0000:01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02)
        Subsystem: ASMedia Technology Inc. Device 2116
        Kernel driver in use: ahci
        Kernel modules: ahci

**lspci -v**

0000:01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02) (prog-if 01 [AHCI 1.0])
        Subsystem: ASMedia Technology Inc. Device 2116
        Flags: bus master, fast devsel, latency 0, IRQ 170
        Memory at 1b00080000 (32-bit, non-prefetchable) [size=8K]
        Memory at 1b00082000 (32-bit, non-prefetchable) [size=8K]
        Expansion ROM at 1b00000000 [virtual] [disabled] [size=512K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [80] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [130] Secondary PCI Express
        Capabilities: [170] Latency Tolerance Reporting
        Capabilities: [200] L1 PM Substates
        Kernel driver in use: ahci
        Kernel modules: ahci

**lspci -vvv**

0000:01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02) (prog-if 01 [AHCI 1.0])
        Subsystem: ASMedia Technology Inc. Device 2116
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 170
        Region 0: Memory at 1b00080000 (32-bit, non-prefetchable) [size=8K]
        Region 5: Memory at 1b00082000 (32-bit, non-prefetchable) [size=8K]
        Expansion ROM at 1b00000000 [virtual] [disabled] [size=512K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
                Address: 000000ffffffe000  Data: 0008
        Capabilities: [80] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0W
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 256 bytes, MaxReadReq 256 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L0s L1 Enabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 8GT/s, Width x1
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR+
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
[... 321 lines cut ...]
[ 59.611841] pcieport 0000:00:00.0: [18] MalfTLP (First)
[ 59.618656] pcieport 0000:00:00.0: AER: TLP Header: 60000001 0100000f 000000ff ffffe000
[ 59.836820] ahci 0000:01:00.0: AHCI controller unavailable!
[ 59.836823] ahci 0000:01:00.0: AER: can't recover (no error_detected callback)
[ 59.849775] pcieport 0000:00:00.0: AER: device recovery failed
[ 59.855676] pcieport 0000:00:00.0: AER: Multiple Uncorrected (Non-Fatal) error received: 0000:00:00.0
[ 59.864986] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[ 59.876192] pcieport 0000:00:00.0: device [14e4:2712] error status/mask=00044000/00400000
[ 59.930810] pcieport 0000:00:00.0: [14] CmpltTO
[ 59.936942] pcieport 0000:00:00.0: [18] MalfTLP (First)
[ 59.943758] pcieport 0000:00:00.0: AER: TLP Header: 60000001 0100000f 000000ff ffffe000
[ 59.951975] ahci 0000:01:00.0: AER: can't recover (no error_detected callback)
[ 59.959240] pcieport 0000:00:00.0: AER: device recovery failed
[ 59.965096] pcieport 0000:00:00.0: AER: Uncorrected (Non-Fatal) error received: 0000:00:00.0
[ 59.973573] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[ 59.984753] pcieport 0000:00:00.0: device [14e4:2712] error status/mask=00044000/00400000
[ 59.993139] pcieport 0000:00:00.0: [14] CmpltTO
[ 59.999258] pcieport 0000:00:00.0: [18] MalfTLP (First)
[ 60.006072] pcieport 0000:00:00.0: AER: TLP Header: 60000001 0100000f 000000ff ffffe000
[ 60.014287] ahci 0000:01:00.0: AER: can't recover (no error_detected callback)

I'm using
Linux pi-nas 6.1.0-rpi8-rpi-2712 https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

**Would anyone have any idea what I could be doing wrong?**

ezaul, 2024-02-14:
**Problem solved, everything works normally, fix these changes**

```
# Back up the current dtb
sudo cp /boot/firmware/bcm2712-rpi-5-b.dtb /boot/firmware/bcm2712-rpi-5-b.dtb.bak

# Decompile the current dtb (ignore warnings)
dtc -I dtb -O dts /boot/firmware/bcm2712-rpi-5-b.dtb -o ~/test.dts

# Edit the file
nano ~/test.dts

# Change the line: msi-parent = <0x2f>; (under `pcie@110000`)
# To: msi-parent = <0x66>;
# Then save the file.

# Recompile the dtb and move it back to the firmware directory
dtc -I dts -O dtb ~/test.dts -o ~/test.dtb
sudo mv ~/test.dtb /boot/firmware/bcm2712-rpi-5-b.dtb
```

```
[    7.721041] ata4.00: ATA-10: ST1000LM035-1RK172, LCM2, max UDMA/133
[    7.730158] ata4.00: 1953525168 sectors, multi 16: LBA48 NCQ (depth 32), AA
[    7.779478] ata4.00: configured for UDMA/133
[    7.779606] scsi 3:0:0:0: Direct-Access     ATA      ST1000LM035-1RK1 LCM2 PQ: 0 ANSI: 5
[    7.779911] sd 3:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    7.779916] sd 3:0:0:0: [sda] 4096-byte physical blocks
[    7.779943] sd 3:0:0:0: [sda] Write Protect is off
[    7.779948] sd 3:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    7.779987] sd 3:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    7.780011] sd 3:0:0:0: [sda] Preferred minimum I/O size 4096 bytes
[    7.923830]  sda: sda1
```

```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0 931.5G  0 disk
└─sda1        8:1    0 931.5G  0 part
```

```
Disk /dev/sda: 931.51 GiB, 1000204886016 bytes, 1953525168 sectors
Disk model: ST1000LM035-1RK1
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: gpt
Disk identifier: 5712D292-AB27-442A-977E-E283C0790993
Device Start End Sectors Size Type
/dev/sda1 2048 1953525134 1953523087 931.5G Linux filesystem
```

 ```
hdparm -Ttv /dev/sda

/dev/sda:
 multcount     = 16 (on)
 IO_support    =  1 (32-bit)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 121601/255/63, sectors = 1953525168, start = 0
 Timing cached reads:   3958 MB in  2.00 seconds = 1979.00 MB/sec
 Timing buffered disk reads: 382 MB in  3.01 seconds = 126.76 MB/sec

```

closed 2024-02-14 by ezaul

albertz, 2024-06-01:
I think I have a similar problem. The SATA controller seems to be detected. I already tried several:
* Via SATA to M.2 to RP PCI.
* Via SATA to PCIe to M.2 to RP PCI.
* Via SATA to PCIe to RP PCI.

Although these are two different SATA controllers, in all cases `lspci` shows it as `SATA controller: ASMedia Technology Inc. ASM1166 Serial ATA Controller (rev 02)`, which is a bit weird? But maybe it's really the same chip in both cases. And also, the behavior in all three cases seems almost identical. Whenever I attach some SATA drive, I get very similar errors as you (but I think not exactly the same, not sure) (see log below). Also the boot time is much slower, it seems to wait for them until it probably hits some timeout or so.

Can you explain a bit your solution? I have no idea what you do there. I don't know what this dtb file is. Maybe the RP firmware? I don't really understand what it means to decompile it. I don't know this `dtc` tool. (But I guess, so far I can just read a bit about that myself.) But then most crucially, how did you get the idea to change `msi-parent = <0x2f>;` (under `pcie@110000`) to `msi-parent = <0x66>;`? What does this mean, and what is the reasoning behind this change?

Should I just try to do the same? Is this safe to try out?

---

`dmesg`, when the SATA drive is attached:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.8.0-1004-raspi (buildd@bos03-arm64-035) (aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #4-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 02:29:55 UTC 2024 (Ubuntu 6.8.0-1004.4-raspi 6.8.1)
...
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=2C:CF:67:1D:E3:DA vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 multipath=off dwc_otg.lpm_enable=0 console=tty1 root=LABEL=writable rootfstype=ext4 rootwait fixrtc cfg80211.ieee80211_regdom=GB
...
[    2.337222] ahci 0000:01:00.0: version 3.0
[    2.372148] ahci 0000:01:00.0: enabling device (0000 -> 0002)
[    2.377943] ahci 0000:01:00.0: ASM1166 has only six ports
[    2.383366] ahci 0000:01:00.0: forcing port_map 0x33f -> 0x3f
[    2.389303] ahci 0000:01:00.0: SSS flag set, parallel bus scan disabled
[    2.396008] ahci 0000:01:00.0: AHCI 0001.0301 32 slots 32 ports 6 Gbps 0x3f impl SATA mode
[    2.404315] ahci 0000:01:00.0: flags: 64bit ncq sntf stag pm led only pio sxs deso sadm sds apst 
[    2.435601] scsi host0: ahci
[    2.438732] scsi host1: ahci
[    2.441805] scsi host2: ahci
[    2.444860] scsi host3: ahci
[    2.447907] scsi host4: ahci
[    2.450959] scsi host5: ahci
...
[    2.525188] scsi host29: ahci
[    2.528296] scsi host30: ahci
[    2.531408] scsi host31: ahci
[    2.534445] ata1: SATA max UDMA/133 abar m8192@0x1b00082000 port 0x1b00082100 irq 43 lpm-pol 0
[    2.543096] ata2: SATA max UDMA/133 abar m8192@0x1b00082000 port 0x1b00082180 irq 43 lpm-pol 0
[    2.551745] ata3: SATA max UDMA/133 abar m8192@0x1b00082000 port 0x1b00082200 irq 43 lpm-pol 0
[    2.560399] ata4: SATA max UDMA/133 abar m8192@0x1b00082000 port 0x1b00082280 irq 43 lpm-pol 0
[    2.569049] ata5: SATA max UDMA/133 abar m8192@0x1b00082000 port 0x1b00082300 irq 43 lpm-pol 0
[    2.577703] ata6: SATA max UDMA/133 abar m8192@0x1b00082000 port 0x1b00082380 irq 43 lpm-pol 0
[    2.586354] ata7: DUMMY
[    2.588804] ata8: DUMMY
[    2.591254] ata9: DUMMY
[    2.593702] ata10: DUMMY
...
[    6.186644] ata1: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[    6.450379] ahci 0000:01:00.0: AHCI controller unavailable!
[    6.745463] ahci 0000:01:00.0: AHCI controller unavailable!
[   13.851180] ata2: failed to resume link (SControl FFFFFFFF)
[   18.486511] ata2: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   20.229121] ahci 0000:01:00.0: AHCI controller unavailable!
[   20.813486] ahci 0000:01:00.0: AHCI controller unavailable!
[   28.787205] ata3: failed to resume link (SControl FFFFFFFF)
[   33.422507] ata3: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   35.165114] ahci 0000:01:00.0: AHCI controller unavailable!
[   35.749467] ahci 0000:01:00.0: AHCI controller unavailable!
[   43.723197] ata4: failed to resume link (SControl FFFFFFFF)
[   48.358487] ata4: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   50.101096] ahci 0000:01:00.0: AHCI controller unavailable!
[   50.685459] ahci 0000:01:00.0: AHCI controller unavailable!
[   58.659196] ata5: failed to resume link (SControl FFFFFFFF)
[   63.294474] ata5: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   65.037082] ahci 0000:01:00.0: AHCI controller unavailable!
[   65.621433] ahci 0000:01:00.0: AHCI controller unavailable!
[   73.595195] ata6: failed to resume link (SControl FFFFFFFF)
[   78.230476] ata6: SATA link down (SStatus FFFFFFFF SControl FFFFFFFF)
[   79.973077] ahci 0000:01:00.0: AHCI controller unavailable!
...
```

Running `lspci` is also somewhat slow and takes a couple of seconds to complete.

ezaul, 2024-06-02:
**Hi Albertz!**

Now everything is working normally, I can update my raspiberry 5, the only thing I always have to do, after updating and recompiling dtb, I'll give you a great recipe!

First, update to the new kernel, I currently have 6.6.31-v8-16k+

`#rpi-update`

After this, restarting the system will take a while, as you mentioned, always leave your HDDs already connected and turned on...

Now sir, you must follow this!

# Back up the current dtb
sudo cp /boot/firmware/bcm2712-rpi-5-b.dtb /boot/firmware/bcm2712-rpi-5-b.dtb.bak

# Decompile the current dtb (ignore warnings)
dtc -I dtb -O dts /boot/firmware/bcm2712-rpi-5-b.dtb -o ~/test.dts

# Edit the file
nano ~/test.dts

now look for pcie@110000 {

change the value of the msi-parent = parameter to the same value as the phandle = parameter below
**example:** 
**if your value is phandle = <0x6e>** 
**put the same in msi-parent = <0x6e>**

and recompile the file following this
Then save the file

# Recompile the dtb and move it back to the firmware directory
dtc -I dts -O dtb ~/test.dts -o ~/test.dtb
sudo mv ~/test.dtb /boot/firmware/bcm2712-rpi-5-b.dtb

**reboot**

**Mine looked like this in the last update**
```
 pcie@110000 {
                        compatible = "brcm,bcm2712-pcie";
                        reg = <0x10 0x110000 0x00 0x9310>;
                        device_type = "pci";
                        max-link-speed = <0x02>;
                        #address-cells = <0x03>;
                        #interrupt-cells = <0x01>;
                        #size-cells = <0x02>;
                        interrupt-parent = <0x01>;
                        interrupts = <0x00 0xdf 0x04 0x00 0xe0 0x04>;
                        interrupt-names = "pcie\0msi";
                        interrupt-map-mask = <0x00 0x00 0x00 0x07>;
                        interrupt-map = <0x00 0x00 0x00 0x01 0x01 0x00 0xdb 0x04 0x00 0x00 0x00 0x02 0x01 0x00 0xdc 0x04 0x00 0x00 0x00 0x03 0x01 0x00 0xdd 0x04 0x00 0x00 0x00 0x04 0x01 0x00 0xde 0x04>;
                        resets = <0x2c 0x07 0x2c 0x2b 0x2d>;
                        reset-names = "swinit\0bridge\0rescal";
                        msi-controller;
                        msi-parent = <0x6e>;
                        ranges = <0x2000000 0x00 0x00 0x1b 0x00 0x00 0xfffffffc 0x43000000 0x04 0x00 0x18 0x00 0x03 0x00>;
                        dma-ranges = <0x3000000 0x10 0x00 0x00 0x00 0x10 0x00>;
                        brcm,enable-l1ss;
                        status = "disabled";
                        phandle = <0x6e>;

```

```
NAME        MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
sda           8:0    0 931.5G  0 disk
└─md0         9:0    0   1.7T  0 raid5 /srv/dev-disk-by-uuid-d62a4292-231b-45c6-89aa-bb21b62915ac
sdb           8:16   0 931.5G  0 disk
└─md0         9:0    0   1.7T  0 raid5 /srv/dev-disk-by-uuid-d62a4292-231b-45c6-89aa-bb21b62915ac
sdc           8:32   0 894.3G  0 disk
└─md0         9:0    0   1.7T  0 raid5 /srv/dev-disk-by-uuid-d62a4292-231b-45c6-89aa-bb21b62915ac
sdd           8:48   0   1.8T  0 disk
└─sdd1        8:49   0   1.8T  0 part  /srv/dev-disk-by-uuid-21e197d5-42b6-461a-b50f-5f3955b3b80a

```
[image]

[image]

albertz, 2024-06-02:
Hi Ezaul! Thanks a lot for your answer!

> **Hi Albertz!**

(It's just Albert. :) Albert Zeyer is my full name, I just liked albertz as a good fitting username here.)
 
> Now everything is working normally, I can update my raspiberry 5, the only thing I always have to do, after updating and recompiling dtb, I'll give you a great recipe!

But I wonder, the latest message [in the forum](https://forums.raspberrypi.com/viewtopic.php?t=363682) sounded like the problems are back? Although I realize now, that post is quite old, I guess this is outdated.

> First, update to the new kernel, I currently have 6.6.31-v8-16k+
> 
> `#rpi-update`

I don't have that command. I installed Ubuntu server 24.04 on it. I have `rpi-eeprom-update` though. Is that the same? I will not do that for now. But I have recently done `apt update` and `apt upgrade`. That should also update the kernel.

I have kernel 6.8.0-1004-raspi.

But maybe that should be fine? I will try to follow the steps.

> # Back up the current dtb
> sudo cp /boot/firmware/bcm2712-rpi-5-b.dtb /boot/firmware/bcm2712-rpi-5-b.dtb.bak

Strangely, there was already such a bak file. I made another bak2 file.

> # Decompile the current dtb (ignore warnings)
> dtc -I dtb -O dts /boot/firmware/bcm2712-rpi-5-b.dtb -o ~/test.dts
> 
> # Edit the file
> nano ~/test.dts
> 
> now look for pcie@110000 {

Ok... Can you explain a bit, why this section? Why not pcie@100000 or pcie@120000 or so?

> change the value of the msi-parent = parameter to the same value as the phandle = parameter below **example:** **if your value is phandle = <0x6e>** **put the same in msi-parent = <0x6e>**

Ok. My old values were:
```
msi-parent = <0x2c>;
phandle = <0x67>;
```
I now changed `msi-parent` to `0x67`.

I have no idea what this does. Can you explain a bit?

> and recompile the file following this Then save the file
> 
> # Recompile the dtb and move it back to the firmware directory
> dtc -I dts -O dtb ~/test.dts -o ~/test.dtb
> sudo mv ~/test.dtb /boot/firmware/bcm2712-rpi-5-b.dtb

The `sudo mv` triggered this warning/error:
```
mv: failed to preserve ownership for '/boot/firmware/bcm2712-rpi-5-b.dtb': Operation not permitted
```
But I guess I can also ignore this?

> **reboot**

Wow, I think it worked! :) I already recognized that the green LED on the SATA controller next to the attached SATA cable was blinking a bit different than before. And startup was much faster. And in `dmesg`, I now see this:
```
[    8.101051] ata1: link is slow to respond, please be patient (ready=0)
[    8.367081] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    8.374843] ata1.00: ATA-11: ST4000NE001-2MA101, EN01, max UDMA/133
[    8.394013] ata1.00: 7814037168 sectors, multi 16: LBA48 NCQ (depth 32), AA
[    8.401503] ata1.00: Features: NCQ-sndrcv
[    8.420145] ata1.00: configured for UDMA/133
[    8.424652] scsi 0:0:0:0: Direct-Access     ATA      ST4000NE001-2MA1 EN01 PQ: 0 ANSI: 5
[    8.433197] sd 0:0:0:0: Attached scsi generic sg0 type 0
[... 89 lines cut ...]
        Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:                                                    
        Used: unknown (minor revision code 0xffff) 
        Supported: 11 10 9 8 7 6 5       
        Likely used: 11                     
...
Capabilities:
        LBA, IORDY(can be disabled)
        Queue depth: 32
        Standby timer values: spec'd by Standard, no device specific minimum
        R/W multiple sector transfer: Max = 16  Current = 16
        Recommended acoustic management value: 254, current value: 0
        DMA: mdma0 mdma1 mdma2 udma0 udma1 udma2 udma3 udma4 udma5 *udma6 
             Cycle time: min=120ns recommended=120ns
        PIO: pio0 pio1 pio2 pio3 pio4 
             Cycle time: no flow control=120ns  IORDY flow control=120ns
...                 
```

My plan is now to attach at least another HDD and make a ZFS RAID.

Additionally, I wonder if I should keep using the micro SD card for the OS, or maybe use a small SSD instead?

My chaotic setup:
[image]

albertz, 2024-06-09:
I still don't really understand those changes in the dtb (dts) file. In any case, to me this sounds like some upstream bug of the Raspberry Pi Linux kernel (and/or the firmware?) and it should not be needed that we always edit this file? So I reported this upstream: https://github.com/raspberrypi/linux/issues/6214 (not sure if this was the right place...)