geerlingguy/raspberry-pi-pcie-devices #606: Delock NVME Ethernet I225-V Adapter on RPI5

bildobodo, 2024-02-18:
Hey everyone,

I recently bought got a Raspberry Pi5 which I'd like to use for LinuxCNC. To have an additional NIC to interface with my machine I thought I'd be nice to combine it with the Delock Converter M.2 Key B+M male to 1 x RJ45 2.5 Gigabit LAN which utilizes an Intel I225-V chip.

https://www.delock.com/produkt/62985/merkmale.html

I compiled the kernel on the RPI5 with the Intel igc module selected in menuconfig `(> Device Drivers > Network device support > Ethernet driver support > Intel(R) Ethernet Controller I225-LM/I225-V support`) and installed it. However I am facing some issues to get it up and running.

What I have got so far:

`uname-a`
 
Linux raspberrypi 6.1.77-rt24-v8-16k+ #7 SMP PREEMPT Sun Feb 18 14:40:31 CET 2024 aarch64 GNU/Linux

`lspci -nnk`

0000:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries Device [14e4:2712] (rev 21)
	Kernel driver in use: pcieport
0000:01:00.0 Ethernet controller [0200]: Intel Corporation Ethernet Controller I225-V [8086:15f3] (rev 03)
	Subsystem: Intel Corporation Ethernet Controller I225-V [8086:0000]
	Kernel modules: igc
0001:00:00.0 PCI bridge [0604]: Broadcom Inc. and subsidiaries Device [14e4:2712] (rev 21)
	Kernel driver in use: pcieport
0001:01:00.0 Ethernet controller [0200]: Device [1de4:0001]
	Kernel driver in use: rp1

And here is the error:

`dmesg | grep igc`

[    2.991025] igc 0000:01:00.0: enabling device (0000 -> 0002)
[    2.991051] igc 0000:01:00.0: PCIe PTM not supported by PCIe bus/controller
[    3.047275] igc: probe of 0000:01:00.0 failed with error -13

So there seems to be no kernel driver in use for the I225-V and dmseg shows an error -13. 
Anyone who knows what the issue might be and can point me in the right direction?

Thank you :)

6by9, 2024-02-19:
-13 is [EACCES](https://elixir.bootlin.com/linux/latest/C/ident/EACCES), which is an unusual error to get.

Can you post the output from `sudo lspci -vv` (just the complete bit for this card is sufficient - we know what the PCIe host and RP1 look like).

bildobodo, 2024-02-19:
Okay I have added now `dtoverlay=pcie-32bit-dma` to `/boot/firmware/config.txt`, but the error -13 persists.

bildobodo, 2024-02-20:
What I have also tried: Activate the Intel LXT PHYs and Intel XWAY PHYs in menuconfig. Not sure if it is required. I have also read through this issue: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/494

Here it seems they've got the I225V up and running, no mention of the PHYs though.

For this example Aquantia PHYs were required: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/15

geerlingguy, 2024-04-02:
@bildobodo - Did you add `pcie-32bit-dma` or `pcie-32bit-dma-pi5`? It seems like @6by9's comment had the CM4/Pi 4 overlay but the linked commit added `pcie-32bit-dma-pi5` instead.

6by9, 2024-04-02:
> @bildobodo - Did you add `pcie-32bit-dma` or `pcie-32bit-dma-pi5`? It seems like @6by9's comment had the CM4/Pi 4 overlay but the [linked commit added `pcie-32bit-dma-pi5` instead](https://github.com/raspberrypi/linux/commit/1cdbd99f402b76c61632d09a49b20ce90af0cc72).

The magic of [overlay_map.dts](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm/boot/dts/overlays/overlay_map.dts#L190-L193) means that specifying `dtoverlay=pcie-32bit-dma` on a bcm2712 device will load pcie-32bit-dma-pi5, and on a Pi0-3 it will be ignored as irrelevant.

geerlingguy, 2024-04-05:
Note that OCP-TAP was trying to integrate an i225 into their TimeHAT for the Pi 5 (see #619), but ran into the same driver issues—it worked on CM4 (even Intel's own i225 PCIe cards), but won't work on Pi 5, which is a bit strange... usually things are the other way around!

lasselj, 2024-04-27:
> ```
> 	Region 0: Memory at 1b00000000 (32-bit, non-prefetchable) [virtual] [size=1M]
> 	Region 3: Memory at 1b00200000 (32-bit, non-prefetchable) [size=16K]
> ```
> 
> At first glance that looks like your card can only address 32bits of address space.
> 
> You can try adding `dtoverlay=pcie-32bit-dma` (which should automatically map to pcie-32bit-dma-pi5) which rejigs the PCIe addressing windows. It was only added a [couple of weeks ago](https://github.com/raspberrypi/linux/commit/1cdbd99f402b76c61632d09a49b20ce90af0cc72), but you should be OK with 6.1.77.

"Maybe", but look.... : https://elixir.bootlin.com/linux/v6.6.28/source/drivers/net/ethernet/intel/igc/igc_main.c#L6736

lasselj, 2024-04-27:
> -13 is [EACCES](https://elixir.bootlin.com/linux/latest/C/ident/EACCES), which is an unusual error to get.

It's not EACCES, it's IGC_ERR_SWFW_SYNC : 

https://elixir.bootlin.com/linux/v6.6.28/source/drivers/net/ethernet/intel/igc/igc_i225.c#L120

lasselj, 2024-04-28:
This is the bit of the code that doesn't work:

https://elixir.bootlin.com/linux/v6.6.28/source/drivers/net/ethernet/intel/igc/igc_i225.c#L81

Irrespective of the timeout value the driver is unable to acquire the FW semaphore and the function returns -[IGC_ERR_NVM](https://elixir.bootlin.com/linux/v6.6.28/source/drivers/net/ethernet/intel/igc/v6.6.28/C/ident/IGC_ERR_NVM). I am not sure why yet...

lasselj, 2024-04-30:
For reference, I've submitted this issue: raspberrypi/linux#6134

ahmadexp, 2024-08-11:
thanks to @lasselj and @geerlingguy I can confirm this problem is resolved. 
You need to get the latest Raspbian and add the "dtoverlay=pciex1-compat-pi5,mmio-hi" to your config.txt
and importantly you need to run the rpi-update to make sure the latest patches are applied.

geerlingguy, 2024-09-20:
Just wanting to note that I'll be doing some additional testing on an Intel i226 in the TimeHAT V2 over in #674 (for anyone wanting to keep following along!).