geerlingguy/raspberry-pi-pcie-devices #609: Add HAT: NVMe Base Duo

geerlingguy, 2024-03-06:
The [NVMe Base Duo](https://shop.pimoroni.com/products/nvme-base-duo-for-raspberry-pi-5?variant=41434434895955) from Pimoroni is like their first effort... but double! It uses an asmedia PCIe Gen 2 switch (ASM1182e), but that means the drives/devices behind it will need to run at Gen 2 or Gen 1 speeds—can't uprate it to Gen 3.

[image]

Also, until https://github.com/raspberrypi/firmware/issues/1833 is resolved, you can't boot off an NVMe SSD behind a switch/bridge.

geerlingguy, 2024-03-15:
On the site: https://pipci.jeffgeerling.com/hats/pimoroni-nvme-base-duo.html

radutomy, 2024-05-14:
Can you stack two of these boards to get x4 NVMe slots in total? Want to build a small RAID 5 NAS.

killermosi, 2025-08-04:
Also, the board is commercially available, it no longer a `Prototype or Non-Commercial HATs` item.

geerlingguy, 2025-08-04:
Just updated it, thanks!

geerlingguy, 2026-01-19:
@killermosi - Ah, right you are! Updated the page.