geerlingguy/raspberry-pi-pcie-devices #612: Add Pineberry Pi HatBRICK! Commander

geerlingguy, 2024-03-15:
The [Pineberry Pi HatBRICK! Commander](https://pineberrypi.com/products/hatbrick-commander-2-ports-gen2-for-raspberry-pi-5) splits the Pi's single PCIe Gen 2.0 lane into two PCIe Gen 2.0 ports, via an integrated ASMedia ASM8211e PCIe Gen 2.0 switch chip.

[image]

geerlingguy, 2024-03-15:
It's on the site: https://pipci.jeffgeerling.com/hats/pineberry-pi-hatdrive-commander.html

geerlingguy, 2024-04-04:
It seems to work well; I've used it along with the HatNET! 2.5G (see #617) and Radxa's Penta SATA HAT (see #615).

Dreadles, 2024-09-17:
I am trying to configure a HatBrick Commander to split the Gen 2 pcie between a 2.5Gb/s Hatnet and HatDrive NVMe device.

When all connected, I am getting both devices limited to 2Gb/s due to the Commander unit and I do not know why.

Brand new Raspberry Pi 5 with 8 GB RAM. Pi os Lite fully patched to latest firmware and bootloader.

When each device is connected directly on their own, I am getting full Gen 2 speeds and nothing reported otherwise in lspci.

But the second I introduce the Commander.

dmesg:
[ 1.916448] pci 0000:03:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:02:03.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[ 1.975519] pci 0000:04:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:02:07.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)

and lspci shows a downgrade.

I get no more than 2Gb/s through the network card, and ~250MB/s on the NVMe.

I have tried a separate Commander module to no avail, as have I flashed the Pi and also used a separate Power Supply.

PCIe enabled with Gen 2 speeds through /boot/firmware/config - and the rest is vanilla.

What on earth am I doing wrong, or could be the issue?

@geerlingguy - you're wisdom is greatly sought after. Pineboards are fobbing me off directly with their chat bot.

bhouseski, 2025-05-25:
Have you tested any non Pineboard hat's with the Commander?  I have tried a Pimoroni NVME base as well as a Waveshare E-key hat, with no success.  The Waveshare hat shows power, but nothing is listed in lspci.  Since the Pimoroni does not have a power LED, I am not sure if it has power or not, but also nothing shows in lspci.  The commander itself has power.