geerlingguy/raspberry-pi-pcie-devices #616: Intel M.2 Dual Band Wireless-AC 3168NGW card with raspberry pi

PabasaraDilshan, 2024-03-30:
- I just tried install wifi drivers for  Intel Corporation Dual Band Wireless-AC 3168NGW. 
```
wget https://wireless.wiki.kernel.org/_media/en/users/drivers/iwlwifi-3168-ucode-22.361476.0.tgz
tar xvf iwlwifi-3168-ucode-22.361476.0.tgz
cd iwlwifi-3168-ucode-22.361476.0
sudo cp iwlwifi* /lib/firmware
```

- Output of `lspci -vvv`

```
01:00.0 Network controller: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak] (rev 10)
	Subsystem: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at 600000000 (64-bit, non-prefetchable) [disabled] [size=8K]
	Capabilities: <access denied>
```

- Output of `dmesg` regarding pcie. Also there are no outputs that related to `iwlwifi`
```
[    1.226561] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.226603] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.226645] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.226681] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x007fffffff -> 0x0400000000
[    1.228023] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.228070] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.228086] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.228149] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.228246] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.231904] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.232162] pci_bus 0000:01: supply vpcie3v3 not found, using dummy regulator
[    1.232321] pci_bus 0000:01: supply vpcie3v3aux not found, using dummy regulator
[    1.232378] pci_bus 0000:01: supply vpcie12v not found, using dummy regulator
[    1.341364] brcm-pcie fd500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.341493] pci 0000:01:00.0: [8086:24fb] type 00 class 0x028000
[    1.341551] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00001fff 64bit]
[    1.341830] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    1.351384] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.351449] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.351469] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600001fff 64bit]
[    1.351505] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.351517] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    1.351881] pcieport 0000:00:00.0: enabling device (0000 -> 0002)
[    1.352066] pcieport 0000:00:00.0: PME: Signaling with IRQ 27
[    1.352469] pcieport 0000:00:00.0: AER: enabled with IRQ 27
[    1.352749] brcm-pcie fd500000.pcie: clkreq control enabled
```

- Output of `name -a`
```
Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
```

Can you please help me to get this work with raspberry pi os?

thewade, 2024-04-19:
You also need to recompile the kernel, see:
https://pipci.jeffgeerling.com/cards_network/intel-ax210-wifi-6e.html

ddodds-nlplogix, 2024-04-24:
I just dealt with getting intel drivers to work on the ax210, recompiled the kernel and everything. None of the firmware on the linux firmware kernel is going to work with the latest version of the raspberry pi's kernel. I've tried every version of the firmware available, and even tried to go back in the git history for find older versions of the firmware, and drove myself crazy trying to figure out how to match versions. 1 week ago someone on this repo happened to post that they managed to get the ax210 working with the mcuzone wifi 6e m2 hat on the raspberry pi 5 on the ubuntu 24.04 beta image. I went ahead and tried it out. The iwlwifi module already comes baked into the ubuntu 24.04 beta image, you do not have to recompile the kernel for those intel drivers, and ubuntu comes with the firmware, you do not need to install it. The only thing I had to do was run `sudo modprobe iwlwifi` and everything worked. I highly recommend trying the ubuntu 24.04 beta image.