geerlingguy/raspberry-pi-pcie-devices #618: Geekworm X1011 PCIe to Four M.2 NVMe SSD HAT

geerlingguy, 2024-04-04:
[image]

Geekworm's X1011 is a 4-drive M.2 NVMe SSD carrier which uses an ASmedia PCIe Gen 2 switch to bridge four NVMe drives to a Raspberry Pi 5.

It looks like it includes some Pogo pins to provide power to (or leech power from) the Pi 5's USB-C supply input. It has a warning to not plug in both the 5V barrel plug and Pi 5 USB-C power input at the same time (and it may not have any safety in place to prevent a short circuit if you do!).

It supports M.2 drives in 2230/2242/2260/2280 lengths, and does not support NVMe boot (because of the switch), at least not until Raspberry Pi resolves the issue [Can't boot Pi 5 via NVMe behind PCIe switch / bridge](https://github.com/raspberrypi/firmware/issues/1833).

geerlingguy, 2024-04-04:
I ordered one of these today, hopefully it'll come in a few weeks.

It's on the site now: https://pipci.jeffgeerling.com/hats/geekworm-x1011-4-drive-nvme.html

paulwratt, 2024-04-05:
Hmmm.. shame noone had thought to use PCIe v3 bridge chips (or even cheap v4) on such devices ..

.. when you get this in the studio, I noticed you had one of those PCIe "splitter" Hats, can you give this board a try this with that, like you did with those 2 boards in the latest video (with RPi5 Penta SATA Hat).

Onthat note, is that PCIe "spitter" HAT on loan as a review sample, as I would really like to an experiment with 2x of them (ie 4x PCIe HAT pssibilities)

geerlingguy, 2024-04-05:
@paulwratt - Maybe even _inside_ your laptop! Haha.

I would really like to see someone incorporate PCIe Gen 3 switch chips, though from what I hear, unlike the ASM1182e, they are harder to acquire in any bulk, and with any datasheets, for smaller companies.

hipboi, 2024-04-06:
Radxa is working on one HAT with PCIe Gen3 switch - ASM2806, we had used it in other projects already. The HAT will be 2x PCIe Gen 3 uplink(Pi5 can only use 1x, some Radxa boards can use 2x)

4x PCIe Gen3 downlink:
- 1x 2.5G, 
- 1x 2.5G, 
- 1x M.2 M Key, 
- 1x Pi5 PCIe FPC

geerlingguy, 2024-04-09:
@hipboi - Oh cool! Would love to test it out. And hopefully the board's Wiki/product page has a compatibility matrix, that can show which models get which lane support (x1 or x2)!

luix93, 2024-04-22:
Hello, any updates on this product? I'm planning a little project with and this looks like exactly what I need, but there are no reviews whatsoever online

geerlingguy, 2024-05-01:
[image]

It comes together nicely and includes a few standoffs you screw into the bottom to hold the NVMe drives just off the surface of a desk. There's no case yet, which is a shame. It'd be nice to have a case with some ventilation, and maybe a fan that blows some air over the ASM1182e chip, since that's the hottest part of this setup:

[image]

It's not bad, at 55°C or so after an hour of writes (a few TB written). I chose some cheap Inland NVMes because... they were cheap. They're TLC so probably not the best for write-heavy workloads, but in ZFS RAIDZ1, I was getting over 1 GB/sec writes and reads, even writing 120 GB of files, so ZFS caching was certainly in play. I haven't tested any other modes yet, nor did I set up Samba for network file copy testing, but it should be able to saturate the Pi's 1 Gbps port no problem.

I like the right-angle FFC, because it gives easy access to the microSD card:

[image]

And you can use either a 5V barrel plug power supply, or a USB-C power adapter. Power is supplied to the Pi (or _from_ the Pi / USB-C connector) through pogo pins that touch test points on the bottom of the Pi 5. This setup is... interesting. I don't think I'd trust it for a permanent installation, but if you clean the contacts and maybe jiggle the springs every year or so, maybe it won't be an issue.

I didn't have any power warnings in my testing, nor throttling, even writing 120 GB to the NVMe array multiple times in a row (with iozone, doing 4K and 1024K cycles), and I was running it on an Argon GaN 27W USB-C power adapter:

[image]

There are four blinkenlights on top, which I like, though they're a slightly bright blue. Put some sort of cover over them if they get annoying:

[image]

And as luck would have it, the Inland SSDs also have their own green blinkenlights, so you get a light show no matter the board's orientation!

[image]

Upsides: very compact, fast enough for 1 Gbps use, everything worked without even tweaking the Pi OS settings (I didn't have to modify the boot config, it picked up the NVMe drives right away with latest Pi OS—this was on a brand new Pi 5 4GB I bought today).

Downsides: No official case (or any 3D printed designs I could find yet), PCIe Switch chip gets hot (55°C, so not terrible, but I'd at least throw a heatsink on it), PCIe Gen 2 switch limits total throughput, and pogo pins for power are a little sketchy, especially if you put in higher-power SSDs (some use 5W+ under load...).

Overall, if you have a lighter use NAS, or if you just want some redundancy with low latency storage, this isn't a bad option. I bought mine for $51 from Geekworm's website. Total cost for this build was around $225, including 4 256 GB NVMe SSDs, a 32GB microSD card for boot, and the power adapter.

And... it sounds like https://github.com/raspberrypi/firmware/issues/1833 is going to be resolved at some point in the near future (fingers crossed!), so you could even maybe boot from an NVMe drive behind the switch in the future! That'd be nice because you could have RAIDZ1 on 3 NVMe disks, and boot on another, for a speedy little NAS.

The biggest limiting factor using this for pure storage is the price of NVMe SSDs. Cost per terabyte goes _way_ up as you get into the 4 and 8TB range.

luix93, 2024-05-01:
This is great! Thank you for doing this. Any chance you have nice picture of those pogo pins?

geerlingguy, 2024-05-01:
Pogo power pins:

[image]

geerlingguy, 2024-05-01:
Testing with an mdadm RAID0 array of all four drives, to get a feel for the performance without ZFS caching in front (with ZFS, the numbers are all amazing, over 1 GB/sec lol):

| Benchmark | Result |
| --- | --- |
| iozone 4K random read | 46.43 MB/s |
| iozone 4K random write | 138.80 MB/s |
| iozone 1M random read | 418.14 MB/s |
| iozone 1M random write | 393.81 MB/s |
| iozone 1M sequential read | 430.45 MB/s |
| iozone 1M sequential write | 398.13 MB/s |

luix93, 2024-05-01:
Thank you so much this is amazing. Would you be able to test RAID5 if that's not too much to ask?

geerlingguy, 2024-05-01:
RAID 5 results:

```
| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 48.21 MB/s |
| iozone 4K random write     | 71.84 MB/s |
| iozone 1M random read      | 408.17 MB/s |
| iozone 1M random write     | 159.94 MB/s |
| iozone 1M sequential read  | 414.59 MB/s |
| iozone 1M sequential write | 162.81 MB/s |
```

luix93, 2024-05-01:
That's great, thank you so much man. Want to build a tiny nas and this seems like a good setup for what I have in mind. This has been very helpful! Glad to have discovered your channel :)

waszak, 2024-06-04:
According to wiki it nows support "Support NVMe boot with the latest firmware EEPROM 2024/05/17 version"
https://wiki.geekworm.com/X1011

mainLink0435, 2024-07-13:
Just sharing my experience in case anyone has a similar issue. I was running this with OMV, and booting to a 256gd NVME in slot 0, with a 4tb NVME in slot 1. I had constant issues (daily) with the device becoming non-responsive. If i was already SSH'd in, any command gave me an I error. No web pages served, no SMB. Also, nothing in the logs (probably because it couldn't write to the logs).

I switched back to microSD and have had no issues.

paulwratt, 2024-07-14:
@geerlingguy I would not worry too much about the _pogo pin connection_ issues. Years ago, maybe, but for the last decade or so they have proven to be more than reliable. That said if a device with pogo pings has issues (after years of reliable use) I too would first clean and test _springy-ness_ as my first solution choice.

BTW I am glad to see RPi OS now supports booting NVME behind a PCIe switch .. maybe they just didn't have the _actual_ hardware to test it against (at the time)

jedi58, 2024-07-30:
> Just sharing my experience in case anyone has a similar issue. I was running this with OMV, and booting to a 256gb NVME in slot 0, with a 4tb NVME in slot 1. I had constant issues (daily) with the device becoming non-responsive. If i was already SSH'd in, any command gave me an IO error. No web pages served, no SMB. Also, nothing in the logs (probably because it couldn't write to the logs).
> 
> I switched back to microSD and have had no issues.

I'm having the same issue, and if I run `sudo nvme list` both of the drives I've got connected have disappeared. Reboot, and they're back

mainLink0435, 2024-07-30:
@jedi58 It's definitely something to do with using the nvme as a boot disk (although I'm not smart enough to know any more detail than that). I switched back to sd card booting, and I'm using both nvme drives as storage - haven't had any errors, warnings, lockups, nothing. Been running for about 2 weeks without an issue.

jedi58, 2024-07-31:
I'm not convinced - I'm not using it as a boot disk, and still experience it - all connected NVMe drives just disappear with the I/O error until I reboot. Yesterday it did it during a 6hr file copy, today it did it on cold boot, but worked after a `shutdown -r now` so I could finish off the file copy that got interrupted yesterday

jedi58, 2024-07-31:
Digging into it a little further

```$ sudo nvme list
Node                  Generic               SN                   Model                                    Namespace Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------

$ sudo nvme list-subsys -vvv
scan controller nvme0
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme0
scan controller nvme0 namespace nvme0n1
failed to scan namespace nvme0n1
scan controller nvme1
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys0/nvme1
lookup subsystem /sys/class/nvme-subsystem/nvme-subsys1/nvme1
scan controller nvme1 namespace nvme1n1
failed to scan namespace nvme1n1
scan subsystem nvme-subsys0
scan subsystem nvme-subsys1
nvme-subsys1 - NQN=nqn.2018-01.com.wdc:guid:redacted
\
 +- nvme1 pcie 0000:05:00.0 dead
nvme-subsys0 - NQN=nqn.2018-01.com.wdc:guid:redacted
\
 +- nvme0 pcie 0000:03:00.0 dead```

After a reboot, I sat with `dmesg -wH` running, and just used a watch command on it reporting on smartlog for both drives, monitoring temperature - no actual writes happening to either nvme device. After a while I see a lof of brcmfmac errors and then:

```[  +7.227194] nvme nvme1: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0xffff
[  +0.000009] nvme nvme1: Does your device have a faulty power saving mode enabled?
[  +0.000003] nvme nvme1: Try "nvme_core.default_ps_max_latency_us=0 pcie_aspm=off" and report a bug
[  +0.083994] nvme 0000:05:00.0: Unable to change power state from D3cold to D0, device inaccessible
[  +0.000018] nvme nvme1: Disabling device after reset failure: -19
[  +0.015994] EXT4-fs (nvme1n1): shut down requested (2)
[  +0.005168] Aborting journal on device nvme1n1-8.
[  +0.000007] kworker/u12:2: attempt to access beyond end of device
              nvme1n1: rw=169985, sector=1950613504, nr_sectors = 8 limit=0
[  +0.000005] Buffer I/O error on dev nvme1n1, logical block 243826688, lost sync page write
[  +0.000005] JBD2: I/O error when updating journal superblock for nvme1n1-8.```

That I'm not sure how to resolve

jedi58, 2024-07-31:
Eurgh… ignore the above. Found a page on the Geekworm site that lists the **WD Black SN770** as being incompatible. At over £200 for the drives can't afford to replace them, so the X1011 and the drives were a waste of money for me then. I may as well spend ~£15 and get an M.2 PCIe adapter for my PC so I can at least use the drives, and yet another Raspberry Pi I've bought will be forgotten about 😂

mainLink0435, 2024-07-31:
OMG! Guess which disk I had as my boot drive.. WD SN740. Which appears on the 'incompatability list'. For me though I haven't had any issues reading/writing from the drives if the OS is on the SD card.

Pretty pathetic on Geekworm's part. Doesn't support Phison controller, doesn't support Polaris controller...

jedi58, 2024-07-31:
Geekworm suggest it's usable due to someone's comment:

> I can confirm updating the firmware seems to work on the pi5. Have a WD black SN770. Worked with USB, didn't work with the NVMe hat. Originally would boot very slow, and couldn't reliably launch a terminal, or a web browser at all. System would freeze or go to gray screen/black screen.
>
> Booted with the drive in a USB enclosure and went to raspi-config and loaded new bootloader/firmware through the advanced options, reinstalled the drive into the hat and now boots fine, fast, and can launch and run programs without issue.

But then that same user reports that it still fails with heavy usage. So if you're not booting off the drive, or using the drive much, should potentially be okay. I think I'll stick to accessing it over USB

mainLink0435, 2024-07-31:
oh great.. so it works.. if you don't use it much. Pretty pathetic on their part. Like you, didn't expect to buy this nvme hat that works with nvme drives (*asterisk* only SOME drives)

Already sunk cost at this point, so like you can't really justify starting again. Probably what Geekworm are banking on (literally)

waszak, 2024-10-30:
> @geerlingguy I tried with your build step by step but I cannot get any power at Geekworm HAT. I ordered twice both of them have the same issue. Also tried RaspberryPI 27W adapter and another try I also try 5V/5A barrel adapter. Still not getting any power or light blinking on HAT. Any solution will be appreciated.
> 
> Raspberry Pi 5, 8GB RAM Raspberry Pi 27W USB-C Power Supply Raspberry Pi Active Cooler SanDisk Extreme micro SDHC A1 UHS-I - boot Ediloca EN600 PRO Internal SSD 1TB PCle 3.0x4, NVMe M.2 2280 - NVMe
> 
> ![IMG_4948](https://private-user-images.githubusercontent.com/88280950/381601546-80826ddd-1747-47be-90d7-8d79017950f1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzMDE4NjQsIm5iZiI6MTczMDMwMTU2NCwicGF0aCI6Ii84ODI4MDk1MC8zODE2MDE1NDYtODA4MjZkZGQtMTc0Ny00N2JlLTkwZDctOGQ3OTAxNzk1MGYxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDE1MTkyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNmNmU2NzVlMzhkM2ZlNjkxY2YxMmIyMDRmMDZjNWIyZTg0YTZkZmRmN2RlMmFmODE5MzRmNDE1ZGYxOWYwM2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xIWWMouT1oLYnTg_cXJjq1IjTfqQHdmaXSuRg7Rp8IU) ![IMG_4949](https://private-user-images.githubusercontent.com/88280950/381601573-33c2b451-014d-4ed2-8e3a-9e5570ffe49a.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzMDE4NjQsIm5iZiI6MTczMDMwMTU2NCwicGF0aCI6Ii84ODI4MDk1MC8zODE2MDE1NzMtMzNjMmI0NTEtMDE0ZC00ZWQyLThlM2EtOWU1NTcwZmZlNDlhLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAzMFQxNTE5MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NTI4Yjc5OWI1MDAyZDNkOTNiMTllMGQ0MDQ5YTA0OGRjZjcyOTYyNDQ4MmFkZDQwZjQ3MWE2NzNlNjRjM2UxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.C7O-AZFN9CDR-57FNBeRKI4OH-85XeCvxAIpZa0igak)

You are not delivering power because pogo pins are not connecting pi5

hasibulmalik, 2024-10-30:
> > @geerlingguy I tried with your build step by step but I cannot get any power at Geekworm HAT. I ordered twice both of them have the same issue. Also tried RaspberryPI 27W adapter and another try I also try 5V/5A barrel adapter. Still not getting any power or light blinking on HAT. Any solution will be appreciated.
> > Raspberry Pi 5, 8GB RAM Raspberry Pi 27W USB-C Power Supply Raspberry Pi Active Cooler SanDisk Extreme micro SDHC A1 UHS-I - boot Ediloca EN600 PRO Internal SSD 1TB PCle 3.0x4, NVMe M.2 2280 - NVMe
> > ![IMG_4948](https://private-user-images.githubusercontent.com/88280950/381601546-80826ddd-1747-47be-90d7-8d79017950f1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzMDE4NjQsIm5iZiI6MTczMDMwMTU2NCwicGF0aCI6Ii84ODI4MDk1MC8zODE2MDE1NDYtODA4MjZkZGQtMTc0Ny00N2JlLTkwZDctOGQ3OTAxNzk1MGYxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMwVDE1MTkyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNmNmU2NzVlMzhkM2ZlNjkxY2YxMmIyMDRmMDZjNWIyZTg0YTZkZmRmN2RlMmFmODE5MzRmNDE1ZGYxOWYwM2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xIWWMouT1oLYnTg_cXJjq1IjTfqQHdmaXSuRg7Rp8IU) ![IMG_4949](https://private-user-images.githubusercontent.com/88280950/381601573-33c2b451-014d-4ed2-8e3a-9e5570ffe49a.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzMDE4NjQsIm5iZiI6MTczMDMwMTU2NCwicGF0aCI6Ii84ODI4MDk1MC8zODE2MDE1NzMtMzNjMmI0NTEtMDE0ZC00ZWQyLThlM2EtOWU1NTcwZmZlNDlhLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAzMFQxNTE5MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NTI4Yjc5OWI1MDAyZDNkOTNiMTllMGQ0MDQ5YTA0OGRjZjcyOTYyNDQ4MmFkZDQwZjQ3MWE2NzNlNjRjM2UxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.C7O-AZFN9CDR-57FNBeRKI4OH-85XeCvxAIpZa0igak)
> 
> You are not delivering power because pogo pins are not connecting pi5
> 
> You have to mount Pi5 on top of x1011 then pins will connect

Thank you so much for this. I almost forgot this.

Madwonk, 2024-12-03:
Something I noticed buying the full X1011 kit (with the metal case and all) - they now appear to include a small adhesive heatsink for the ASM1184 chip. Wonder if they had reports of overheating and decided to throw it in just in case.

I'll be reporting back as I get further into setting it up (work keeps me pretty busy these days so it may take some time) but I noticed the Geekworm wiki [claims](https://wiki.geekworm.com/X1011?_gl=1*1qahcij*_gcl_au*NjkzNjE0NDAwLjE3MzMxOTQ1ODM.*_ga*MTIzMzE1MDE0MS4xNzMzMTk0NTgz*_ga_WCBGYZY3V4*MTczMzE5NDU4Mi4xLjAuMTczMzE5NDU4Mi42MC4wLjA.#Tips) that the board does not support ZFS due to "driver compatibility issues". Any idea where this comes from? There isn't any issues (Github or similar) linked so I have nothing to go off of.

wolfnether, 2024-12-20:
i have x1011 with 4 ssds on zfs, i _had_ a problem that the ssds getting disconnected and rpy needed a reboot.
the real problem is that the ssds was overheating. i put a heatsink on them and voila no more "driver compatibility issues".

Harvie, 2025-01-19:
I have question that might seem bit crazy, but have anyone tried to plug one of those PCIe SATA converters based on ASM1166 to one of the slots on X1011? I really wonder if that works, because i want system on SSD, but i don't trust SSDs with long term storage of my data, so i am really looking for solution that would allow me to have at least 2x bootable SSD and 2x SATA HDD on RPi.

But this seems almost too good to be true with X1011:

[image]

waszak, 2025-01-19:
There is new version that could be more reliable because doesn't use pogo pins but 
no one reviewed this variant.
https://wiki.geekworm.com/X1011-12V

My experience after many months:
I run ZFS Pool in mirror mode and had no issue with ZFS except for temps in scrubing

I don't recommend official case "X1011-C1" if you have ZFS 
you need  something bigger that its easy to install heatsinks on ssds and one extra on chip

Sadly one of the port (SSD3)  is to tight(maybe dmged) so I never put forth ssd. 

I have 2x790NM 4TB and one 970 evo plus 1TB. Everything runs fast

Ps. I plan to upgrade to something better still but except few issues it works well.