geerlingguy/raspberry-pi-pcie-devices #620: Confusion about Current/Voltage from Geekworm Hats

00lex, 2024-04-08:
Confusion about Current/Voltage from Geekworm Hats

Hi, maybe someone can help me understand this better. 

Geekworm write on their Wiki about their HATs: 

* Can power the x100x shield directly from the FFC PCIe ribbon (providing maximum 5W of continuous power)
* Maximum current for pcie connector is 1A, and voltage is 5V

* x1001:  Integrated with XH2.54 5V power connector, just use it if you are using a ultra large capacity SSD if the current is not enough
* x1002: With 40pin pin header, it can automatically supply power from GPIO, avoiding the problem of insufficient power supply for ultra-large capacity SSDs.

and show this explanation: 
[image]

Geekpi and Pineberry write about their Hats: 

* M.2 Standard-Compliant Power System: Integrated voltage regulator delivering up to 3A for the 3.3V power rail, compliant with M.2 (NGFF) standard.

Raspberry Pi Foundation say 3.3V is generated from 5V but doesnt say anything about current. 
https://datasheets.raspberrypi.com/pcie/pcie-connector-standard.pdf 

the info from geekworm about the 5V seems superfluous to me, since NVME hard disks require 3.3V and therefore 1A is also superfluous. But how much current does the hard disk receive with the geekworm HATs? At the moment I have a 4TB SN700 with an x1001 HAT running flawlessly, but I would feel better if I knew what current is being supplied, as my NVME can draw 2.8A at its peak according to WD. 

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     6.00W       -        -    0  0  0  0        0       0
 1 +     3.50W       -        -    1  1  1  1        0       0
 2 +     3.00W       -        -    2  2  2  2        0       0
 3 -   0.1000W       -        -    3  3  3  3     4000   12000
 4 -   0.0035W       -        -    4  4  4  4     4000   45000
 

because of missing informations im planning switching to Pineberry.

mikegapinski, 2024-04-22:
@00lex the power delivery system is the only place where you can "save a buck" on those NVMe boards. The second part is the FPC ribbon cable (it carries power and can cause problems if impedance is not matched correctly).

There are 3 companies that do it "right" to my knowledge:
- Pimoroni with their NVMe base (although the ribbon cable is thinner than I'd like and it interferes with the Wi-Fi antenna slightly)
- Pineberry (full disclosure, I am the CEO of this company)
- Raspberry Pi (the pictures of the official HAT were leaked a while back here https://www.reichelt.com/pt/en/raspberry-pi-m-2-hat-for-raspberry-pi-5-rasp-m-2-hat--p360314.html?CCOUNTRY=451&LANGUAGE=en&&r=1)

All 3 vendors listed above use the same buck converter for generating the 3.3V rail (or very similar from the same manufacturer, we used to install a more expensive version that is rated for more than 3A when the cheaper one was out of stock)

00lex, 2024-04-23:
thank you. I stay with geekworm x1001 (without extra power) and my wd red sn700 4tb. after 3 weeks no issues or errors found on various scenarios. x1001 works for me best because of temperature. all other HATs are to close to the rpi an temperatures are 5-10°C higher.