geerlingguy/raspberry-pi-pcie-devices #621: Add 52Pi N07 M.2 2280 PCIe to NVMe Bottom

Damilko, 2024-04-10:
Im trying to get a PCIe 52pi N07 with a WD Black SN770 2TB working but it wont recognise the m.2 card. I see on another post that the 52pi N04 & the N05 boards are compatible so one would assume the n07 would tick that box as well. I also see that the SN770 SE series are also tested OK (as per #526 post) but I cant determine if my WD is a SE series or not. Firstly im hoping the N07 can be added to check that box then ill need to find out more info on my m.2 sn770 if also compatible

Damilko, 2024-04-11:
Ok just an update where im at.
I had Pi OS Bookworm 64bit with desktop. with a 16gb sd
I tried doing all the updates but had not enough Disk space to complete updates. I got a 64gb sd and conducted a sd card copier thru usb port.
booted up on new 64gb did all the updates, updated eeprom etc
then tried viewing the NVMe but it just couldn't see it by an lsblk command.
Went down a rabbit hole with google and doing everything under the sun that I could find. still it just could not identify the m.2 sitting there.
I was II close to pushing my fist thru the monitor and decided to grab another 64gb SD card fresh image using imager 
booted up on new image on sd did a lsblk and the NVme was listed in the list. you ripper.
put a fresh image (thru imager) on the NVMe did all the updates and setting up to boot from NVMe and its now booting on the NVMe  So I can now confirm that the 52pi N07 m.2 2280 PCIe to NVMe Bottom works with a WD Black SN770 2TB currently on gen 2 but im about to look at changing it to gen 3 (cos I can).  
this is the cat print...
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Operating System: Debian GNU/Linux 12 (bookworm)  
          Kernel: Linux 6.6.20+rpt-rpi-2712
    Architecture: arm64

if there is any other info you would like to know just let me know.

yoyojacky, 2024-05-15:
> Ive followed last nights sucss line and still not so now Im just gunna go back down that rabbit hole again and try to work out why it worked last night and tonight its cactus.

Could you please show me your /boot/firmware/config.txt file ? 
```bash
cat /boot/firmware/config.txt 
```
copy and paste it here .
and please execute this command in terminal : 
```bash
lsblk
```