geerlingguy/raspberry-pi-pcie-devices #624: Read over I2C to a PCIe device

JohnGoogl, 2024-04-17:
Hi Team,
I'm relatively new to Raspberry Pi and embedded systems, so please forgive me if this question seems basic.

I'm working with a scenario where I want to read data over the I2C protocol to a device connected via PCIe.
However, I'm unsure if this is a directly achievable approach.

Is it possible to communicate with a PCIe device using I2C on the Raspberry Pi?  If not, are there any alternative approaches I should consider?
Any insights or resources you can share on this topic would be incredibly helpful!

Thanks in advance for your support!

Best Regards,
Paul Johns

JohnGoogl, 2024-04-22:
Would appreciate any thoughts on this please!!

6by9, 2024-04-22:
Your query makes little sense.

If you have a PCIe device, then the interface to it is PCIe, not I2C.

If the PCIe device exposes an I2C controller, then that would be mapped in the same way as any other I2C controller would. This is exactly what you have with Pi5 having all the I2C controllers and GPIOs connected to RP1 over PCIe.

If your situation is neither of those, then you'll have to give some more details.

JohnGoogl, 2024-04-22:
Hi, @6by9 
Thank you for the inputs.

Yes, NVMe interface setup to PCIe using 'hat+' and 'nvme list' working Ok, let me know if any further details required.

If possible, please share respective docs/blogs if any available.

geerlingguy, 2024-04-22:
@JohnGoogl - I think what @6by9 means is what are you trying to do?

I know there's the [NVMe-MI](https://nvmexpress.org/specification/nvme-mi-specification/) spec, are you trying to manage NVMe devices over PCIe? What for? Something else? It is impossible to help without any clear idea of what you're trying to accomplish.

Typically if you want to interact with devices using I2C, it'd be helpful to know what device, and for what purpose. And PCIe doesn't directly expose I2C, which is why I think we are a bit confused about this.

I'm going to move this to 'discussion'.