geerlingguy/raspberry-pi-pcie-devices #633: Test Pineboards HatDrive! Dual NVMe Board

geerlingguy, 2024-05-13:
Pineboards' [HatDrive! Dual NVMe Board](https://pineboards.io/products/hatdrive-dual-2xnvme-2230-2242-gen-2-for-raspberry-pi-5) includes an ASmedia ASM1182e PCIe Gen 2 switch chip to split one lane of PCIe Gen 2 from the Pi 5 to two 2230 or 2242 M.2 NVMe SSD M-key slots:

TODO: Image here.

Since https://github.com/raspberrypi/firmware/issues/1833 is now in the Raspberry Pi's firmware (a `sudo rpi-update` should get the Pi 5 to the latest version), this board should also allow booting NVMe SSDs behind the PCIe switch!

geerlingguy, 2024-05-13:
Using the latest firmware, I was able to boot off one of the two NVMe drives, though the bootup sequence takes like 30 seconds before it starts loading the OS (it only takes about 3-4 seconds with the drive connected directly). I don't think there are any other issues, but I'll try reseating everything. Maybe a power issue with the MakerDisk NVMe SSDs I'm currently testing?

Full system is only drawing 5-8W, so it seems like NVMe might not be pulling down more than the 5W or so the connector is rated for. I might swap out the drives for some Pinedrive drives.

geerlingguy, 2024-05-13:
I first tried re-seating everything, and only running the boot NVMe in slot 2, with nothing in slot 1. It still takes a pretty long time before the kernel boot process starts up, and it's still going a bit slow. Going to try with the Pinedrive now, to see if it's the MakerDisk (which didn't have any issues connected directly), the firmware, or something else...

geerlingguy, 2024-05-13:
Got some timeouts in the logs during boot:

```
[    3.647613] could not locate request for tag 0x0
[    3.652656] nvme nvme0: invalid id 0 completed on queue 1
[   34.125313] nvme nvme0: I/O 902 (I/O Cmd) QID 1 timeout, aborting
[   34.176150] nvme nvme0: Abort status: 0x0
[   64.837298] nvme nvme0: I/O 902 QID 1 timeout, reset controller
[   64.941397] nvme0n1: I/O Cmd(0x2) @ LBA 1318912, 32 blocks, I/O Error (sct 0x3 / sc 0x71) 
[   64.950099] I/O error, dev nvme0n1, sector 1318912 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[   64.987988] nvme nvme0: 4/0/0 default/read/poll queues
...
[   66.052071] could not locate request for tag 0x0
[   66.057275] nvme nvme0: invalid id 0 completed on queue 4
...
[   67.110524] could not locate request for tag 0x0
[   67.110533] nvme nvme0: invalid id 0 completed on queue 4
[   67.170655] nvme0n1: I/O Cmd(0x2) @ LBA 1087352, 144 blocks, I/O Error (sct 0x2 / sc 0x82) 
[   67.170665] protection error, dev nvme0n1, sector 1087352 op 0x0:(READ) flags 0x80700 phys_seg 18 prio class 2
[   67.170685] nvme0n1: I/O Cmd(0x2) @ LBA 1087232, 112 blocks, I/O Error (sct 0x2 / sc 0x82) 
[   67.170689] protection error, dev nvme0n1, sector 1087232 op 0x0:(READ) flags 0x80700 phys_seg 14 prio class 2
...
[   97.605320] nvme nvme0: I/O 128 (I/O Cmd) QID 4 timeout, aborting
[   97.649624] nvme nvme0: Abort status: 0x0
[  128.325309] nvme nvme0: I/O 128 QID 4 timeout, reset controller
[  128.425422] nvme0n1: I/O Cmd(0x2) @ LBA 1273320, 64 blocks, I/O Error (sct 0x3 / sc 0x71) 
[  128.425428] I/O error, dev nvme0n1, sector 1273320 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 2
[  128.425439] nvme0n1: I/O Cmd(0x2) @ LBA 2614328, 8 blocks, I/O Error (sct 0x3 / sc 0x71) 
[  128.425442] I/O error, dev nvme0n1, sector 2614328 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[  128.439981] nvme nvme0: 4/0/0 default/read/poll queues
```

geerlingguy, 2024-05-13:
When I tried running some disk benchmarks behind the switch:

```
[  289.638013] could not locate request for tag 0x0
[  289.638025] nvme nvme0: invalid id 0 completed on queue 3
[  289.927384] could not locate request for tag 0x0
[  289.927387] nvme nvme0: invalid id 0 completed on queue 3
[  289.927388] could not locate request for tag 0x0
[  289.927390] nvme nvme0: invalid id 0 completed on queue 3
[  289.927394] could not locate request for tag 0x0
[  289.927395] nvme nvme0: invalid id 0 completed on queue 3
```

geerlingguy, 2024-05-13:
Swapped in the Pineboard. First boot on a freshly-flashed Pi OS Lite did the normal cycle, but second boot was being weird.

Spying on things with my Debug Probe, I see:

```
Boot mode: NVME (06) order f14
PCI1 init
PCI1 reset
PCIe scan 00001b21:00001182
PCIe scan 00001b21:00001182
PCIe scan 00001b21:00001182
PCIe scan 00001b21:00001182
PCIe scan 00001b21:00001182
PCIe scan 00001987:00005013
Found device on bridge 4 with vid 1987 enable 1
PCIe scan 00001b21:00001182
PCIe scan 00001987:00005013
VID 0x1987 MN SSD94-2563CG-PB                        
NVME on 0
MBR: 0x00002000, 1048576 type: 0x0c
MBR: 0x00102000,499061424 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261116 (4)
rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
FAT32 clusters 261116
[nvme] autoboot.txt not found
Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
Trying partition: 0
type: 32 lba: 8192 'mkfs.fat' ' bootfs     ' clusters 261115 (4)
rsc 32 fat-sectors 2040 root dir cluster 2 sectors 0 entries 0
FAT32 clusters 261115
Read config.txt bytes     1213 hnd 0x199
[nvme] pieeprom.upd not found
usb_max_current_enable default 0 max-current 5000
Read bcm2712-rpi-5-b.dtb bytes    77755 hnd 0xe6
dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712

NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
NOTICE:  BL31: Built : 14:26:57, Jun 22 2023
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
```

Then after a few seconds, it starts stalling out with nvme errors:

```
[    4.513325] systemd[1]: Started systemd-journald.service - Journal Service.
[    4.612647] could not locate request for tag 0x0
[    4.617854] nvme nvme0: invalid id 0 completed on queue 1
[   36.759208] nvme nvme0: controller is down; will reset: CSTS=0xffffffff, PCI_STATUS=0xffff
[   36.768075] nvme nvme0: Does your device have a faulty power saving mode enabled?
[   36.776147] nvme nvme0: Try "nvme_core.default_ps_max_latency_us=0 pcie_aspm=off" and report a bug
[   50.418384] nvme0n1: I/O Cmd(0x2) @ LBA 1273448, 136 blocks, I/O Error (sct 0x3 / sc 0x71) 
[   50.427333] I/O error, dev nvme0n1, sector 1273448 op 0x0:(READ) flags 0x80700 phys_seg 5 prio class 2
[   50.427345] nvme0n1: I/O Cmd(0x2) @ LBA 2668112, 64 blocks, I/O Error (sct 0x3 / sc 0x71) 
[   50.446109] I/O error, dev nvme0n1, sector 2668112 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 2
[   50.446145] nvme0n1: I/O Cmd(0x2) @ LBA 2667024, 64 blocks, I/O Error (sct 0x3 / sc 0x71) 
[   50.446147] nvme0n1: I/O Cmd(0x2) @ LBA 1057840, 8 blocks, I/O Error (sct 0x3 / sc 0x71) 
[   50.446153] I/O error, dev nvme0n1, sector 1057840 op 0x0:(READ) flags 0x83700 phys_seg 1 prio class 2
[   50.446157] I/O error, dev nvme0n1, sector 2667024 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 2
[   50.492912] nvme0n1: I/O Cmd(0x2) @ LBA 1057888, 8 blocks, I/O Error (sct 0x3 / sc 0x71) 
```

Maybe I just have a bad board? I'll try again with a HatDrive Commander or whatever the PCIe switch board is, to see if that's the issue.

geerlingguy, 2024-05-13:
On a separate HatDrive! Commander board (which I've tested with a SATA controller + 2.5G Ethernet in my first SATA Pi NAS build—before I went for a 2.5G USB Ethernet adapter), I am getting the same problem...

```
[    3.342066] Run /init as init process
[    3.370367] mmc1: new ultra high speed DDR50 SDIO card at address 0001
[    3.605276] could not locate request for tag 0x0
[    3.610316] nvme nvme0: invalid id 0 completed on queue 1
[   34.117399] nvme nvme0: I/O 64 (I/O Cmd) QID 1 timeout, aborting
[   34.168148] nvme nvme0: Abort status: 0x0
```

It seems to have issues a few seconds into boot, and then it stalls out. If I leave it for a really long time it eventually boots. But it's highly unstable, and running a disk benchmark eventually crashes the thing with tons of NVMe errors.

Going to try digging up another PCIe switch to see if that fixes anything.

geerlingguy, 2024-05-13:
Same thing with 3rd PCIe switch. And the 2nd and 3rd are both switches I have never had an issue with in the past :/

```
[    3.373708] mmc1: new ultra high speed DDR50 SDIO card at address 0001
[    3.605280] could not locate request for tag 0x0
[    3.610319] nvme nvme0: invalid id 0 completed on queue 3
```

geerlingguy, 2024-05-13:
Just to reset my sanity, I replugged everything direct (dropping the switch out of the equation).

It boots up with no problem whatsoever, and the filesystem mounts right away, nicely:

```
[    3.205616] mmc1: new ultra high speed DDR50 SDIO card at address 0001
[    3.582639] EXT4-fs (nvme0n1p2): mounted filesystem 93c89e92-8f2e-4522-ad32-68faed883d2f ro with ordered data mode. Quota mode: none.
[    3.704346] NET: Registered PF_INET6 protocol family
```

Disk benchmarks run as normal, so either there's a firmware issue with NVMe boot behind the switch... or... something. Maybe I should try some other NVMe SSDs! (I've tried two: Cytron MakerDisk 2242 and Pineboards NVMe 2242).

geerlingguy, 2024-05-13:
I don't think it's a power issue, but never say never.

[image]

geerlingguy, 2024-05-13:
Tried with just one Pinedrive inserted, booting off microSD card. Still getting NVMe errors:

```
[    3.606303] could not locate request for tag 0x0
[    3.611346] nvme nvme0: invalid id 0 completed on queue 1
[   34.107080] nvme nvme0: I/O 512 (I/O Cmd) QID 1 timeout, aborting
[   34.156550] nvme nvme0: Abort status: 0x0
[   64.827093] nvme nvme0: I/O 512 QID 1 timeout, reset controller
[   64.935196] nvme0n1: I/O Cmd(0x2) @ LBA 500118016, 32 blocks, I/O Error (sct 0x3 / sc 0x71) 
[   64.944071] I/O error, dev nvme0n1, sector 500118016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[   64.981789] nvme nvme0: 4/0/0 default/read/poll queues
```

I may try going to older firmware to see if it works better, or if it's a hardware issue.

mikegapinski, 2024-05-13:
Did you disable aspm? Ive had someone with two HatDrive bottoms and the commander and it solved the PCIe errors

geerlingguy, 2024-05-13:
I'm [benchmarking the drive with `fio`](https://www.jeffgeerling.com/blog/2024/quick-nvme-performance-testing-fio) and it seems to be working okay when I just have one drive, unformatted, running in the carrier. Going to reboot a couple more times and see if it's stable.

mikegapinski, 2024-05-13:
If you want to rule out power try a bottom with a commander. You can connect the power header to GPIO and it'll offload the ribbon cable. 

I doubt that's it, the ribbon can handle 4 drives without issues 😅

geerlingguy, 2024-05-13:
Yeah I know lol. I will try that at some point, but going to try `pcie_aspm=off` in `/boot/firmware/cmdline.txt` next. Just need to try booting into NVMe so I can edit it. Or maybe pull the drive, connect to my Mac, edit the file, pop it back on, lol.

Someone needs to make a nice pop-in-pop-out M.2 slot for quicker swaps :)

geerlingguy, 2024-05-13:
I pulled the drive, added `pcie_aspm=off` to `cmdline.txt` over on my Mac, then rebooted...

And now it's booting up _perfectly_, every time. Wouldn't you know!?

On the downside—`fio`'s only giving me 215 MiB/sec now, it was giving me 450 MiB/sec with just one drive on the bus before...?

mikegapinski, 2024-05-13:
ASPM could also be disabled with a dtoverlay; I'll see if I can figure it out tomorrow.

I don't know what's wrong with the aspm implementation though, I also saw people disable it for Coral.

In 95% of all the issues I've been dealing with supporting our users in the past few months it makes all the pcie error go away.

geerlingguy, 2024-05-14:
@mikegapinski mentioned over in the pi firmware issue that the phison controller doesn't play nice with power states behind the bridge, and recommended:

  1. Edit `/boot/firmware/config.txt`
  2. Add the line `dtparam=pciex1_no_l0s=on`
  3. Save and reboot

Now I'm getting PCIe Gen 2 speeds, and no crashing or hanging issues:

```
pi@pi5test:~ $ sudo fio fio.fio
nvme0: (g=0): rw=read, bs=(R) 256KiB-256KiB, (W) 256KiB-256KiB, (T) 256KiB-256KiB, ioengine=libaio, iodepth=32
fio-3.33
Starting 1 process
Jobs: 1 (f=1): [R(1)][100.0%][r=435MiB/s][r=1741 IOPS][eta 00m:00s]
nvme0: (groupid=0, jobs=1): err= 0: pid=920: Tue May 14 17:13:19 2024
  read: IOPS=1739, BW=435MiB/s (456MB/s)(12.8GiB/30019msec)
    slat (nsec): min=5036, max=67483, avg=5780.38, stdev=1710.71
    clat (usec): min=18114, max=35613, avg=18385.56, stdev=244.30
     lat (usec): min=18135, max=35618, avg=18391.34, stdev=244.29
    clat percentiles (usec):
     |  1.00th=[18482],  5.00th=[18482], 10.00th=[18482], 20.00th=[18482],
     | 30.00th=[18482], 40.00th=[18482], 50.00th=[18482], 60.00th=[18482],
     | 70.00th=[18482], 80.00th=[18482], 90.00th=[18482], 95.00th=[18482],
     | 99.00th=[18482], 99.50th=[18482], 99.90th=[18482], 99.95th=[20579],
     | 99.99th=[32637]
   bw (  KiB/s): min=436736, max=446464, per=100.00%, avg=445801.33, stdev=1228.82, samples=60
   iops        : min= 1706, max= 1744, avg=1741.27, stdev= 4.77, samples=60
  lat (msec)   : 20=100.01%, 50=0.05%
  cpu          : usr=0.45%, sys=1.54%, ctx=53738, majf=0, minf=9
  IO depths    : 1=0.0%, 2=0.0%, 4=0.0%, 8=0.0%, 16=0.0%, 32=100.0%, >=64=0.0%
     submit    : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.0%, 64=0.0%, >=64=0.0%
     complete  : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.1%, 64=0.0%, >=64=0.0%
     issued rwts: total=52211,0,0,0 short=0,0,0,0 dropped=0,0,0,0
     latency   : target=0, window=0, percentile=100.00%, depth=32

Run status group 0 (all jobs):
   READ: bw=435MiB/s (456MB/s), 435MiB/s-435MiB/s (456MB/s-456MB/s), io=12.8GiB (13.7GB), run=30019-30019msec

Disk stats (read/write):
  nvme1n1: ios=60595/0, merge=0/0, ticks=1127880/0, in_queue=1127880, util=100.00%
```

geerlingguy, 2024-05-14:
Just because people have been asking a bit... power consumption stats with two Pinedrive 2242 NVMe SSDs installed:

  - Idle: 4.4W
  - fio against one drive: 5.7W (brief peak at 6.7W)
  - fio against two drives: 7W (brief peak at 8.3W)

[image]

During the two-drive testing, each drive individually pegged at 216 MB/sec, which illustrates a tiny bit of slowdown when accessing both drives, like in RAID.

closed 2024-05-24 by geerlingguy

mikegapinski, 2024-06-18:
@geerlingguy 

I added a kernel patch that disables ASPM without having to use any overlays: https://github.com/raspberrypi/linux/pull/6229

If you ever stumble upon a drive that is throwing errors like this in Gen 3 or behind a switch write down the ID from lspci -nn. Adding this quirk to more drives takes 5 seconds :)