geerlingguy/raspberry-pi-pcie-devices #637: 52Pi N07 M.2 2280 PCIe to NVMe Bottom

yoyojacky, 2024-05-15:
Hi geerlingguy, 
Nice to meet you, it's me again. lol, we've got new pcie device and i am going to introduce it for you.

# 52Pi  N07 M.2 2280 PCIe to NVMe Bottom
* SKU: EP-0221
* Wiki: [wiki](https://wiki.52pi.com/index.php?title=EP-0211)
## Description
N07 is an bottom-mounted NVME M.2 SSD(NVMe 2230, 2242, 2260, 2280 GEN 3) PIP (PCIe Peripheral Board) for the Raspberry Pi 5 that uses the new PCIE interface of the Raspberry Pi 5 to utilise the NVME M2 SSDs for fast data transfers and ultra-fast booting.

N07 is compatible with the standard HAT size of Raspberry Pi and supports quad sizes of NVMe SSD: 2230, 2242, 2260 and 2280. The included SSD mounting screws make it convenient for users to switch between the two sizes at any time.

## Features
* Designed for RPI 5: Dedicated to the latest single-board computer, the Raspberry Pi 5.
* NVMe Capability: Supports installing NVMe (M-key) drives in M.2 format sizes 2230, 2242, 2260 and 2280, extra custom CNC SSD mount screw, no soldering required.
* PCIe Support: PCIe x1 interface in both Gen2 & Gen3 standards.
* Ventilation hole design: Ventilation hole design provides excellent ventilation airflow for cooling.
* Gold-plated PCB: Gold-plated PCBs offer superior electrical performance.
* PCIe ultra-short signal lines: The short trace routing of PCIe is more reliable and faster, fully meeting the signal requirements of PCIe 3.0.
* LED Indicators: Features LED light for M.2 disk activity (“ACT”).
* M.2 Standard-Compliant Power System: Integrated voltage regulator delivering up to 3A for the 3.3V power rail, compliant with M.2 (NGFF) standard.
* All Necessary Accessories Included: Comes with a FPC PCIe ribbon, M2 and M2.5 screws, spacers, and extra custom CNC SSD mount screw.

## Gallery
[image]
[image]
[image]

[image]

samip5, 2024-10-03:
If only one could also find a case that would fit with the long pillars.

yoyojacky, 2024-10-21:
> If only one could also find a case that would fit with the long pillars.

You can replace the long pillars with short pillars, that's  can make it easier to get a fitable case.

samip5, 2024-10-21:
> You can replace the long pillars with short pillars

True, but it only comes with the long ones. :)

yoyojacky, 2024-10-22:
> > You can replace the long pillars with short pillars
> 
> True, but it only comes with the long ones. :)

Oh, what the length do you want to use? please send the e-mail to me, i will arrange my teamates send a copper pillar pack to you。

yoyojacky, 2024-10-22:
> > You can replace the long pillars with short pillars
> 
> True, but it only comes with the long ones. :)

BTW, send e-mail to jacky.li@52pi.com will be ok.