geerlingguy/raspberry-pi-pcie-devices #641: 52Pi B12 RPi 5 Double FPC PCIe HAT

yoyojacky, 2024-05-15:
Hi geerlingguy, 
Hope you are well. Today I am going to introduce a new hat for Tiny NVMe SSD expansion board .

# 52Pi B12 RPi 5 Double FPC PCIe HAT
* SKU：EP-0230
* WIKI: [wiki](https://wiki.52pi.com/index.php?title=EP-0230)

## Description
Introducing B12 Raspberry Pi HAT: Expand Your PCIe Connectivity Elevate your Raspberry Pi 5 experience with the B12 Raspberry Pi HAT, a versatile accessory that transforms your FPC PCIe interface into two, enabling seamless connection to multiple PCIe devices. With the ability to daisy-chain multiple B12 units, users can create a multi-level PCIe setup for enhanced flexibility and functionality. Explore the standout features of the B12 below:

## Features
* Dual PCIe Expansion: Effortlessly expand your Raspberry Pi 5's FPC PCIe interface into two, providing users with the convenience of connecting multiple PCIe devices simultaneously for increased productivity and versatility.
* Daisy-Chaining Capability: Connect multiple B12 units in a cascading fashion to create a multi-tiered PCIe configuration, allowing for expanded connectivity options and customized setups tailored to your specific needs.
* Seamless Integration:Designed for easy integration, the B12 Raspberry Pi HAT offers a user-friendly experience, making it simple to set up and configure for immediate use with your Raspberry Pi 5.
* Enhanced Connectivity: Unlock the potential to connect a wide range of PCIe devices, such as SSDs, GPUs, network cards, and more, expanding the capabilities of your Raspberry Pi 5 for diverse applications and projects.
* Robust and Reliable Design: Built with quality and durability in mind, the B12 Raspberry Pi HAT ensures stable and reliable performance, providing a dependable solution for your PCIe expansion needs.
* Transform your Raspberry Pi 5 into a powerhouse of connectivity with the B12 Raspberry Pi HAT. Experience seamless PCIe expansion, enhanced flexibility, and endless possibilities for your projects and applications. Elevate your setup with the B12 and unleash the full potential of your Raspberry Pi 5.
* Dual Board Expansion: This feature allows for the connection of two additional Pineberry Pi boards, providing users with the flexibility to expand their projects' scope and functionality.
* ASMedia PCIe Switch Integration: The board incorporates an ASMedia PCIe Switch (Gen2 variant), facilitating reliable and efficient data transfer between connected devices.
* Included FPC Ribbon Cables: Three FPC ribbon cables are provided, simplifying the connection process and offering flexibility in setup and configuration.
* Compatible Form Factor: Board format is compatible with the original Raspberry Pi HAT dimensions of 65 x 56.50 mm.
* 40-Pin Raspberry Pi HAT Connector Support: Supports the standard 40-pin Raspberry Pi HAT connector.

## Gallery
* Product Outlook
[image]
[image]

[image]

[image]
[image]

Givy, 2024-10-29:
Hi,
Is there going to be any power supply issues if we add, a CPU fan, an "AI + Hat 26" and a "NVMe.2 SSD" card to this RPi-5?.  Thank you.

Junii55, 2024-12-06:
Did you figure it out?

Givy, 2024-12-08:
Hi,
I purchased a "Raspberry Pi 5 NVME PCIe SSD Base NVPI5-2280B" board which has an additional "5 v" input on it, in case NVME SSD needs it.  But I have not gone through with it yet and I need to get the above board as well.  Now, My issue is, where on the RPi-5 to connect the additional 5-v to?  Also, is there a such a low power NVMe SSD that can work with it!

Junii55, 2024-12-10:
I tried it, it didn't work

Givy, 2024-12-14:
Hi,
Are you able to send me some pictures? I would like to see how you made these connections to your RPi-5?  What happened?  It didn't boot up or did you get any error messages etc.?
On my issue, I am going to use the power off of the RPi GPIO.  I am also going to purchase a SK hynix Gold P31 M.2 NVMe SSD, which has low power consumption.

Thank you,
GIVI

yoyojacky, 2025-02-06:
> Hi, I purchased a "Raspberry Pi 5 NVME PCIe SSD Base NVPI5-2280B" board which has an additional "5 v" input on it, in case NVME SSD needs it. But I have not gone through with it yet and I need to [...]

Raspberry Pi 5's GPIO may provide 5V and GND.

dijitalxyz, 2025-02-20:
I bought B12 through my friend
but there are some socket compatibility problems
copper socket ends are in different directions
official raspi m2 hat+ socket FBC cable ends are in different directions
same way AI HAT+ socket ends are in opposite directions

waveshare m2 hat there 2 sides are in the same direction
52Pi B12 FBC HAT 2 sides are in the same direction

since the socket directions are different, there will be trouble in connecting different devices.

When I connect waveshare m2 with the new FBC cable, I can work but AI HAT+ insists on not working

RobbieIoT, 2025-02-26:
> I bought B12 through my friend but there are some socket compatibility problems copper socket ends are in different directions official raspi m2 hat+ socket FBC cable ends are in different [...]

Hi there,

Which AI Hat+ did you try?
Which AI module did you try?

I am using a dual M2 M-Key HAT with an NVMe at this moment, where the Coral Dual TPU does not fit on (E-key instead of M-key).
So I found a compatible HAT for the Coral Dual TPU and was thinking of combining the two HAT's with the B12 (or B14), but very curious if that might work.
The dual M2 M-Key HAT is using the same PCIe 2.0 switch chip ASM1182E, so wondering if these two HAT's can be stacked too, because the the website 52pi.com mentions the following: 'With the ability to daisy-chain multiple B12 units'.
Would this mean, as long as it's the same AMS1182 chip, it should/could be possible?

Curious to find out!

Thanks in advance :-)

dijitalxyz, 2025-03-11:
@RobbieIoT

Normal setup without EP-0230
official Raspberry Pi AI HAT+ (26T)
official Raspberry Pi 5 M.2 HAT+ Mkey
AI HAT+ ve M2 HAT+ not work with EP-0230

I still can't think of any use for the halio accelerator other than using it with the camera
https://wiki.geekworm.com/I bought the C790 to use it by inputting an HDMI signal instead of a camera, but that's a separate problematic device

It appears that someone is making these devices and someone is testing them.

celide-pi, 2025-03-31:
Recently I've tested the 52pi B12 expansion module and encountered performance issues:
- NVMe drive speeds drop from 1.1 GB/s to370-530 MB/s when connected through the module
- Intel BE200 WiFi 7 card shows firmware errors when used with Mcuzone's HAT (errors don't occur with direct RPi connection)

Both components function properly when connected directly to Raspberry Pi's PCIe interface, indicating potential compatibility issues with the expansion hardware. Especially, PCI Gen 3 does not work. 

Could you recommend alternative HATs which are able to give me gen 3 speeds for to different components (wifi card and nmve)?

Hedda, 2025-10-03:
@yoyojacky any chance you could make a similarly functioning but "stacked" Double FPC PCIe M.2 Adapter (i.e. a stacking dual M.2 PCIe switch adapter) that would work with the [Raspberry Pi Compute Module 5 IO Board](https://www.raspberrypi.com/products/compute-module-5-io-board/) (and the [Raspberry Pi Compute Module 5 Development Kit](https://www.raspberrypi.com/products/cm5-dev-kit/) and [Home Assistant Yellow](https://www.home-assistant.io/yellow/) to allow users to have an M.2 AI accelerator and an M.2 SSD at the same time?   Another Reaspberry Pi Compute Module 5 IO carrier board that need the same type of stacked M.2 PCIe switch adapter solution is the "[CM5 PoE BASE A](https://www.waveshare.com/cm5-poe-base-a.htm)" (Waveshare Raspberry Pi Compute Module 5 IO Board With PoE for all variants of CM5) .

If you look at the pictures of these two pictures bellow then I think you will at least get an idea of the form-factor the concept where you would instead have new dual adapter that allows the user to add NVMe SSD underneath in the stack and the AI accelerator stacked on top of it via the dual M.2 slots that would need to be stacked with an off-set for that. 

[image]

[image]

The point with that is it would be awesome for users of the Raspberry Pi Compute Module 5 IO Board and Home Asssistant Yellow if we could use a M.2 AI accelerator card it without sacrificing the NVMe SSD storage M.2 slot, thus the need for some kind of dual M.2 PCIe switch adapter in a nice slim format that stacks the M.2 adapters on-top of each other.

Like your adapter above I know that function-wise the are at dual M.2 HAT(s)/Shield(s) with PCIe switch chip for the regular Raspberry Pi 5 that allow you to use a [Hailo-8 AI accelerator (Raspberry Pi AI Kit) ](https://www.raspberrypi.com/documentation/accessories/ai-kit.html) and a M.2 NVMe SSD at the same time? See example Geekworm X1005 PCIe to M.2 NVMe SSD PIP Bottom for Raspberry Pi 5 which by reports looks to also support Hailo-8 AI Accelerator but that adapter does not have an M.2 input interface:

* https://www.cnx-software.com/2024/02/03/add-two-m-2-nvme-ssds-to-raspberry-pi-5-with-geekworm-x1004-hat/?amp=1

Perhaps Nabu Casa could make and sell such dual M.2 adapter with an onboard PCIe switch IC for the Home Assistant Yellow so we can use both an NVMe SSD and an AI accererator card at the same time?

I have at least not seen anyone else who sells exactly such an "Single slot to dual M.2 adapter" adapter but found this "NVME to dual SSD adapter" which does not have the same functionality since it is only converts one NVMe slot to two SATA module slots but think it might still be good as a visual reference of a  stacked form-factor concept:

* https://www.tindie.com/products/crimier/m2-dual-drive-adapter-nvme-solderable-sata/

Again, there are otherwise plenty of other single-to-dual adapter products (or even single-to-three or single-to-four) with PCIe switches for the Raspberry Pi that split one M.2 slot into two, but none that I found have this type of stacked form factor which would allow us to fit two cards in the Home Assistant Yellow-.