geerlingguy/raspberry-pi-pcie-devices #647: Add Radxa Dual 2.5G Router HAT

akgnah, 2024-05-24:
Radxa' [Dual 2.5G Router HAT](https://docs.radxa.com/en/accessories/dual-2.5-router-hat) features a PCIe Gen3 switch, the [ASM2806](https://www.asmedia.com.tw/product/012Yq70sX2zI9xN5/7c5YQ79xz8urEGr1), enhancing its connectivity capabilities. 

This HAT has 2x PCIe Gen 3 uplinks (RPi5 can only use 1x, some Radxa boards can use 2x) and 4x PCIe Gen 3 downlinks:

- 1x 2.5G Ethernet,
- 1x 2.5G Ethernet,
- 1x M.2 M Key,
- 1x RPi5 PCIe FPC

[image]

geerlingguy, 2024-05-24:
Indeed! I have one of these boards to test—I think there may have been some concerns over the interfaces working correctly on the Pi 5, and I'm planning to test it soon!

geerlingguy, 2024-05-24:
On the site here: https://pipci.jeffgeerling.com/hats/radxa-dual-2.5g-router.html

geerlingguy, 2024-05-24:
Just booting it the first time, a few observations:

1. When I plugged in my 12V barrel plug, I got a nice little spark out of the connector. Always a surprise :D
2. `lspci` is empty on first boot, though I have an older EEPROM on this Pi. Updating and testing again...
3. After upgrading to latest stable and rebooting, still nothing.

After adding the following to `/boot/firmware/config.txt`:

```
dtparam=pciex1
dtparam=pciex1_gen=3
```

And then rebooting, I get:

```
pi@pi5:~ $ lspci
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0000:01:00.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0000:02:00.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0000:02:02.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0000:02:06.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0000:02:0e.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0000:03:00.0 Non-Volatile memory controller: Phison Electronics Corporation PS5013 E13 NVMe Controller (rev 01)
0000:05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0000:06:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0001:01:00.0 Ethernet controller: Device 1de4:0001
```

For full output:

<details><summary>Click to expand full lspci output</summary>

```
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 38
	Bus: primary=00, secondary=01, subordinate=06, sec-latency=0
	Memory behind bridge: 00000000-002fffff [size=3M] [32-bit]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr+ FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <1us, L1 <2us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s (overdriven), Width x1
			TrErr- Train- SlotClk+ DLActive- BWMgmt+ ABWMgmt+
		RootCap: CRSVisible+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd+
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
[... 772 lines cut ...]
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [b0] MSI-X: Enable+ Count=61 Masked-
		Vector table: BAR=0 offset=00000000
		PBA: BAR=0 offset=00002000
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: rp1
```
</details>

I don't see the NVMe drive with `lsblk` though, so I wonder if I'm getting an error like I was with boot-behind-a-PCIe-switch... checking on that now.

geerlingguy, 2024-05-24:
If I unplug the NVMe drive, so that socket is empty on the top, the 2nd 2.5G Ethernet port isn't working. I see the 1st port, but port 2 doesn't seem to be showing up. If I swap my Ethernet cable from port 2 to port 1, port 1 lights up.

```
[    3.759676] r8169 0000:05:00.0: enabling device (0000 -> 0002)
[    3.819660] r8169 0000:05:00.0: error -EIO: PCI read failed
[    3.826310] r8169: probe of 0000:05:00.0 failed with error -5
[    3.833494] r8169 0000:06:00.0: enabling device (0000 -> 0002)
[    3.836708] brcmstb-i2c 107d508200.i2c:  @97500hz registered in interrupt mode
[    3.852841] brcmstb-i2c 107d508280.i2c:  @97500hz registered in interrupt mode
[    3.864572] r8169 0000:06:00.0 eth1: RTL8125B, 00:e0:4c:68:00:04, XID 641, IRQ 173
[    3.872764] r8169 0000:06:00.0 eth1: jumbo features [frames: 9194 bytes, tx checksumming: ko]
```

When I put the NVMe drive back in place, both 2.5G Ethernet ports work. Odd!

geerlingguy, 2024-05-24:
I also tested a Pineboards HatDrive! M.2 HAT plugged into the FFC on the board, and it didn't light up the PWR LED at all, nor did the drive show up at all under `lspci`.

I've also tried using PCIe Gen 2, and the NVMe attached to the board, and it still shows up with `lspci`, but doesn't list under `lsblk`.

Even turning off ASPM (with `pcie_aspm=off` inside `/boot/firmware/cmdline.txt`), it doesn't mount the drive.

geerlingguy, 2024-05-24:
Testing with an Inland 256GB 2280 NVMe SSD, I see it with lspci:

```
0000:03:00.0 Non-Volatile memory controller: Silicon Motion, Inc. SM2263EN/SM2263XT SSD Controller (rev 03)
```

But again, no `lsblk`. And again, it says enabling device, but no more past that:

```
pi@pi5:~ $ dmesg | grep nvme
[    2.569973] nvme nvme0: pci function 0000:03:00.0
[    2.575002] nvme 0000:03:00.0: enabling device (0000 -> 0002)
```

akgnah, 2024-05-24:
> If I unplug the NVMe drive, so that socket is empty on the top, the 2nd 2.5G Ethernet port isn't working. I see the 1st port, but port 2 doesn't seem to be showing up. If I swap my Ethernet cable [...]

Odd! I kept the NVMe drive plugged in during the test, so I didn't notice the problem.

akgnah, 2024-05-24:
> I also tested a Pineboards HatDrive! M.2 HAT plugged into the FFC on the board, and it didn't light up the PWR LED at all, nor did the drive show up at all under `lspci`.

In the current version, x1.0 (x for evaluation), the FPC seat is incorrectly rotated 180 degrees.

If you have a heat gun, you can remove it and re-solder it. We can also send you a corrected board.

akgnah, 2024-05-25:
> I've also tried using PCIe Gen 2, and the NVMe attached to the board, and it still shows up with `lspci`, but doesn't list under `lsblk`.

The NVMe device show up in `lspci` but not in `lsblk`, which has been troubling me. I've done a lot of searching and read many posts on Raspberry Pi forums, but nothing has helped.

But this HAT works well on Radxa's boards, such as the [ROCK 5C](https://radxa.com/products/rock5/5c), so I feel like it might not be a hardware issue.

geerlingguy, 2024-05-25:
@akgnah - Yeah, it's definitely an odd situation—usually there's at least an error message in `dmesg`, but here... nothing! I wonder if the switch is throwing something in the Pi for a loop, this is the first time I've tried a Gen 3 switch (even at Gen 2 speed) on a Pi at all.

If anyone had access to a PCIe debugger that could probably help, but I am far away from being able to have one of those :D

Regarding the FPC, I may take a shot at reversing it, I finally have my hot air station set up and got power to it yesterday. I'll maybe have a go next week.

To be clear, if I remove it from the board, rotate the entire connector 180°, and re-solder it there, it should be wired up correctly? Thanks for the help!

akgnah, 2024-05-25:
> To be clear, if I remove it from the board, rotate the entire connector 180°, and re-solder it there, it should be wired up correctly?

Yes, if it's rotated 180 degrees, its signal is correct.

I connected a M.2 HAT to the corrected connector, but the NVMe device still only appears in `lspci`.

geerlingguy, 2024-05-26:
This might be something worth opening an issue for in the https://github.com/raspberrypi/firmware repo—it seems like it could be related to https://github.com/raspberrypi/firmware/issues/1833

klakiti, 2024-06-01:
Hi!
Does the NIC functionality work without external power? A Raspberry Pi 5 with dual 2.5gbps would be an amazing OPNSense box, and if a low profile hat could provide that, it would be a dream come true.

geerlingguy, 2024-06-03:
I have been testing powering the Pi via the barrel jack on the HAT, so there's still just one power connection.

DanaGoyette, 2024-06-04:
I'd be curious to see if `nvme list` (from the `nvme-cli` package) shows anything.  If `/dev/nvme0` shows up, you should be able to try some of the following:
`nvme id-ctrl -H /dev/nvme0`
`nvme list-ns`
`nvme id-ns -H /dev/nvme0 -n 1`

Most drives don't support more than one namespace, but perhaps that one namespace is missing or somehow not detected by the kernel?

hipboi, 2024-06-05:
@DanaGoyette 

If you are interested in troubleshooting this, we can send you a sample to have some debugging fun. Please send me an email(tom@radxa.com) if you are interested.

tltangliang, 2024-06-18:
I speculate that this is a compatibility issue with the ASMedia chip. 
When testing the Intel Optane SSD, I found that the Intel Optane SSD works fine directly on Raspberry Pi 5, but after passing through the ASM1184E bridge chip, it cannot function properly. It cannot be detected in lsblk, but other regular NVMe drives have no issues. 

This board seems to use the ASM2806 as the main chip, and there may be some compatibility bugs. 

Further testing is needed to find out. It is not sold on the official website, and it seems there are no other channels to purchase it from.

geerlingguy, 2024-06-19:
@tltangliang - Yeah, it sounds like Radxa are not wanting to release the board until they are certain it can be made to work (which I'm 100% in support of!), so this issue is currently more of a 'debug party' and I will continue to plug away as I can.

4MeInfo, 2024-06-20:
The FCC seat brings nearly 5 volts but no  accessories  is detected.

I tested the M.2 Slot with the  Realtek R8125 driver and WD Blue SN570 NVMe 

07: PCI 300.0: 0108 Non-Volatile memory controller (NVM Express)
  [Created at pci.386]
  Unique ID: svHJ.BG8T6CV0dd8
  Parent ID: B35A.lPUYI3PvRI6
  SysFS ID: /devices/platform/axi/1000110000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0
  SysFS BusID: 0000:03:00.0
  Hardware Class: storage
  Model: "Sandisk Non-Volatile memory controller"
  Vendor: pci 0x15b7 "Sandisk Corp"
  Device: pci 0x501a 
  SubVendor: pci 0x15b7 "Sandisk Corp"
  SubDevice: pci 0x501a 
  Memory Range: 0x1b00000000-0x1b00003fff (rw,non-prefetchable)
  Memory Range: 0x1b00004000-0x1b000040ff (rw,non-prefetchable)
  IRQ: 38 (no events)
  Module Alias: "pci:v000015B7d0000501Asv000015B7sd0000501Abc01sc08i02"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #9 (PCI bridge)

and a Samsung NVMe 980
07: PCI 300.0: 0108 Non-Volatile memory controller (NVM Express)
  [Created at pci.386]
  Unique ID: svHJ.hLDNcFqboN7
  Parent ID: B35A.lPUYI3PvRI6
  SysFS ID: /devices/platform/axi/1000110000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0/0000:03:00.0
  SysFS BusID: 0000:03:00.0
  Hardware Class: storage
  Model: "Samsung Electronics Non-Volatile memory controller"
  Vendor: pci 0x144d "Samsung Electronics Co Ltd"
  Device: pci 0xa809 
  SubVendor: pci 0x144d "Samsung Electronics Co Ltd"
  SubDevice: pci 0xa801 
  Memory Range: 0x1b00000000-0x1b00003fff (rw,non-prefetchable)
  IRQ: 38 (no events)
  Module Alias: "pci:v0000144Dd0000A809sv0000144Dsd0000A801bc01sc08i02"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #9 (PCI bridge)

but nothing changed.

some output from hwinfo:
09: PCI 200.0: 0604 PCI bridge (Normal decode)
  [Created at pci.386]
  Unique ID: B35A.lPUYI3PvRI6
  Parent ID: VCu0.lPUYI3PvRI6
  SysFS ID: /devices/platform/axi/1000110000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/0000:02:00.0
  SysFS BusID: 0000:02:00.0
  Hardware Class: bridge
  Model: "ASMedia PCI bridge"
  Vendor: pci 0x1b21 "ASMedia Technology Inc."
  Device: pci 0x2806 
  Revision: 0x01
  Driver: "pcieport"
  IRQ: 39 (no events)
  Module Alias: "pci:v00001B21d00002806sv00000000sd00000000bc06sc04i00"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #10 (PCI bridge) 

and pcicrawler

00:00.0 root_port, speed 8GT/s, width x1
 └─01:00.0 upstream_port, ASMedia Technology Inc. (1b21), device 2806
    ├─02:00.0 downstream_port, slot 0, device present, speed 8GT/s, width x1
    │  └─03:00.0 endpoint, Samsung Electronics Co Ltd (144d) NVMe SSD Controller 980 (a809)
    ├─02:02.0 downstream_port, slot 0, speed 2.5GT/s, width x1
    ├─02:06.0 downstream_port, slot 0, device present, speed 5GT/s, width x1
    │  └─05:00.0 endpoint, Realtek Semiconductor Co., Ltd. (10ec) RTL8125 2.5GbE Controller (8125)
    └─02:0e.0 downstream_port, slot 0, device present, speed 5GT/s, width x1
       └─06:00.0 endpoint, Realtek Semiconductor Co., Ltd. (10ec) RTL8125 2.5GbE Controller (8125)
0001:00:00.0 root_port, speed 5GT/s, width x4
 └─0001:01:00.0 endpoint, 1de4:0001

I didn't  find any kernel modules or drivers for the asm2806. I compiled my own kernel with all pcie and nvme option on but also nothing changed. I also think the is a problem with the compatibility of the ASM2806

hipboi, 2024-06-26:
The dual 2.5G HAT and NVMe SSD work without any issue on ROCK 5C and 5A. We are out of clue here, maybe this is released?

https://github.com/raspberrypi/linux/pull/6229

geerlingguy, 2024-06-26:
I do think that has something to do with it, but I'm staring at the HAT right now sitting there tempting me to test it again... I just got back from two weeks of travel, so I will try to pick it up again soon—but alternatively, if there's someone else who I could send my prototype board to, I'd be willing to do that!

I may not have time for a week or two to pick it up again.

mikegapinski, 2024-06-28:
@geerlingguy check the revision of the ASMedia chip itself (date code). Some older revisions are not playing well with the Pi 5.

geerlingguy, 2024-07-03:
@mikegapinski 

From `lspci`:

```
0000:01:00.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
```

And on the chip itself:

ASM2806
B3NG7727A1 2022

[image]

Also @DanaGoyette I ran the `nvme` commands:

```
pi@pi5:~ $ sudo apt install -y nvm-cli

pi@pi5:~ $ nvme id-ctrl -H /dev/nvme0
/dev/nvme0: No such file or directory

pi@pi5:~ $ nvme list-ns
pi@pi5:~ $ nvme id-ns -H /dev/nvme0 -n 1
... same ...
```

The NVMe SSD I'm testing with uses a Silicon Motion controller SM2263EN/SM2263XT:

```
0000:03:00.0 Non-Volatile memory controller: Silicon Motion, Inc. SM2263EN/SM2263XT SSD Controller (rev 03) (prog-if 02 [NVM Express])
	Subsystem: Silicon Motion, Inc. SM2263EN/SM2263XT (DRAM-less) NVMe SSD Controllers
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 38
	Region 0: Memory at 1b00000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L1 <8us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [b0] MSI-X: Enable- Count=16 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00002100
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [158 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [178 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [180 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
```

Do you want me to run this with the drive directly attached to a Pi instead? It's not mounting the device anywhere.

I've also tried using various combinations of PCIe compatibility modes:

```
# In /boot/firmware/cmdline.txt
pcie_aspm=off
```

```
# In /boot/firmware/config.txt
dtoverlay=pineboards-hat-ai
# or
dtoverlay=pciex1-compat-pi5,no-mip
# or
dtparam=pciex1_gen=1
```

mikegapinski, 2024-07-03:
Chips made in 2024 (Q2+) are OK.

I don't know if it applies to the Gen 3 switch as well, but some changes were made to the Gen 2 ones recently.

mikegapinski, 2024-07-03:
If the drive you have doesn't work without the switch (on a HatDrive Bottom V3 or newer) it won't work.

On your lspci it seems like the endpoint is detected but the NVMe driver is not loaded. That's why you can't write to the drive 😅

If it does work standalone it is possible that an updated pcie switch solve everything for you. I have devices that only work on current switches regardless of what overlays are used. It's a mess overall, thankfully we've caught that in time. Dealing with Chinese chip makers is not always straightforward. 

Wasn't sure about sharing that information but if our competitors are reading this: Source your components directly or ask for a photo of the reel 😅

geerlingguy, 2024-07-03:
> Source your components directly or ask for a photo of the reel 😅

Ha, I'm guessing otherwise they might ship some old stock to clear it out! After all, on some systems and in many use cases, it's not a big deal and won't cause problems. I know I've seen a lot of USB expansion boards using the ASM2806.

The drive does work direct, or actually also with the ASM1182e in the Geekworm X1011 — see https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/618#issuecomment-2087888943

mikegapinski, 2024-07-03:
It's just the mix of specific devices with specific revisions on specific firmware. Everyone is improving constantly and you can exactly OTA internal firmware / silicon changes. 

Generally speaking manufacturers don't have old inventory, you're ordering from current batches usually but MOQs are higher than when you go through a reseller (and they tend to have older revisions in stock)

KeyserSoze1, 2024-07-10:
I have a Rev 1.1 of this board. Changes appear to be the FPC connector fix and a copper heatsink on the switch.

I've only begun setting mine up but thought I would post my initial troubleshooting.  All testing thus far has been with an updated version of Bookworm OS with a bootloader dated Wed  5 Jun 15:41:49 UTC 2024 (1717602109).

I get the below output from lspci as others have already noted
```
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0000:01:00.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:00.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:02.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:06.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:0e.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0000:06:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0001:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
```

**2.5 GB Adapters**
Only 1 functional adapter using the r8169 driver.  I get the same issue as @geerlingguy mentions above with the 2nd adapter 
`[    3.488345] r8169: probe of 0000:05:00.0 failed with error -5`
I get this error regardless if I have the M.2 slot populated.  So far I've only tested one M.2 drive which may just be incompatible?

Speed of 2.5gb connection is good on the working adapter and showing 2.5gb link speed.
```
Settings for eth1:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Auto-negotiation: on
	master-slave cfg: preferred slave
	master-slave status: slave
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: external
	MDI-X: Unknown
	Supports Wake-on: pumbg
	Wake-on: d
	Link detected: yes
```
I hope to test the r8125 driver vs the r8169 to see if there are differences there.

**M.2 Drive**
Only tested a Samsung SM961 which failed to appear in any log.  I have verified I could see the drive on another PC.  This drive didn't appear to be recognized and wasn't shown in lspci output.  I plan to test additional drives in the near future.  I suspect this is the reason I can't use the 2nd 2.5gb ethernet as it appears to be linked somehow to the M.2 slot based on previous finding.

**FPC Connection**
Untested at this point but appears to be "fixed" as the connector is reversed from other images.

Hopefully more to follow in the coming days.

KeyserSoze1, 2024-07-11:
I've attempted to use the r8125 driver and found some interesting observations....

Not sure if I missed this above, and apologies if so, but after some more testing I found that I can't get both ethernet ports to work regardless of M.2 if using Gen3.  If I drop to Gen2 then it's possible to get both ethernets IF I have a M.2 drive installed.

Below is a table that is the results of getting both 2.5g ethernet ports enabled.  This goes for both the r8125 and r8169 drivers.

|                 | Gen 2          | Gen 3                      |
| :----------- | :--------------: | :-------------------------: |
| M.2 Not Installed | NO | NO |
| M.2 Installed | YES | NO |

The behavior for when it doesn't work is the same except the drivers behave differently.

r8169
-
Behavior from NO cells in table.
Only one of the 2.5g Ethernet interface work while the other one gets the 
`[    3.488345] r8169: probe of 0000:05:00.0 failed with error -5` error.  
The working interface seems to behave normally as a 2.5gb NIC. M.2 is not detected in Gen 3.

M.2 is detected in Gen 2 but beyond being detected is not usable.  Like mentioned in prevous comments it is not listed from lsblk.

r8125
-
Behavior from NO cells in table.
Neither 2.5g Ethernet is available. 
```
[    5.868162] r8125: loading out-of-tree module taints kernel.
[    5.871554] r8125 Ethernet controller driver 9.013.02-NAPI loaded
[    5.871662] r8125 0000:05:00.0: enabling device (0000 -> 0002)
[    5.923551] unknown chip version (7c800000)
```
That is the only thing in the logs and no interface becomes available to use. Behavior of M.2 is the same as with the r8169 driver.

The output from the YES cell is
```
[    5.647918] r8125: loading out-of-tree module taints kernel.
[    5.676861] r8125 Ethernet controller driver 9.013.02-NAPI loaded
[    5.676938] r8125 0000:05:00.0: enabling device (0000 -> 0002)
[    5.709959] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    5.712052] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
[    5.712155] r8125 Ethernet controller driver 9.013.02-NAPI loaded
[    5.712223] r8125 0000:06:00.0: enabling device (0000 -> 0002)
[    5.764709] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    5.766734] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
```

I'm not sure what to make of all of this.  It seems that some of you have both NICS working with Gen3 and an M.2 but so far that has not been the case for me with the Samsung SM961.  I might try a different drive in a bit to see if it changes the above results in Gen3.

Is it possible the IPEX to FPC Cable is the issue here?  I received 2 and haven't tried the 2nd one but that's honestly a bit of a stab in the dark at this point.

KeyserSoze1, 2024-07-12:
Apologies for the update spamming.  I'm using this partially to document my progress as I work through this and if anyone has issues with it please let me know and I'll try to condense these.  I'm hoping if I take a break from this it will potentially be helpful to someone else also.

Today I started looking into why the NVME was not loading properly.  
```
[   11.731965] nvme nvme0: pci function 0000:03:00.0
[   11.736710] nvme 0000:03:00.0: enabling device (0000 -> 0002)
[   11.793991] probe of 0000:03:00.0 returned 19 after 62120 usecs
```
The probe for NVME is failing with error code 19 (ENODEV).  Spending just a bit of time looking through the nvme kernel source I believe this is the culprit based on the messages preceding it. https://github.com/torvalds/linux/blob/43db1e03c086ed20cc75808d3f45e780ec4ca26e/drivers/nvme/host/pci.c#L2491
Not sure why that is failing however. 

Next I tried to enable Gen3 to see what differences I could discover.
NVME is not discovered at all using Gen3 and the first RTL8125 Ethernet fails instead.
Here is something to note.  The first thing scanned on the ASMedia 2806 (NVME in Gen2 and 1st RTL8125 in Gen3) fails.
Gen 2 log
`[   11.434525] pci 0000:03:00.0: BAR 0: assigned [mem 0x1b00000000-0x1b00003fff 64bit]`
Gen 3 log
`[   11.197574] pci 0000:05:00.0: BAR 2: assigned [mem 0x1b00000000-0x1b0000ffff 64bit]`

The Gen 2 log is referencing the NVME and notice the start address.  The Gen 3 is referencing the 1st RTL8125 and notice it has the same start address as the NVME when using Gen 2.  Both of these devices fail to start in their respective boots.  This could be circumstantial but the fact that I believe there is a read error in those memory spaces could be the reason they fail.  Cause is still unknown however.  I would like to try the FPC connector to see if this theory holds but that will have to wait a bit as I don't have another device on hand that I can try easily for a week or two.  I believe the scan should come in this order 1. NVME, 2. FPC, 3. RTL8125, 4. RTL8125.  So the FPC should fail instead of the 1st RTL if my theory holds.

Just a few bread crumbs and nothing concrete at this point.

KeyserSoze1, 2024-07-14:
A positive update today.  I have been able to get gen3 working when using a different NVME drive.  When swapping the Samsung SM961 for a cheap Sk Hynix PC711 I am able to detect the drive while using Gen3.

Poking at my last hypothesis about the memory region failing to read being an issue, I put together a hacked kernel to skip this region of memory when assigning BAR to the PCI devices. And it was successful!  I now have both RTL8125's and an NVME drive accessible.  All 3 devices seem to be functioning normally.  This is obviously a hack and doesn't fix the root of the problem but it does show the device can work.  I'm reaching the limit of my understanding on what could cause this however.  If anyone is interested in my kernel source hack/patch let me know and I can post it somewhere.

Here's the fio report for the Sk Hynix
```
nvme0: (g=0): rw=read, bs=(R) 256KiB-256KiB, (W) 256KiB-256KiB, (T) 256KiB-256KiB, ioengine=libaio, iodepth=32
fio-3.33
Starting 1 process
Jobs: 1 (f=1): [R(1)][100.0%][r=851MiB/s][r=3405 IOPS][eta 00m:00s]
nvme0: (groupid=0, jobs=1): err= 0: pid=1505: Sat Jul 13 22:44:31 2024
  read: IOPS=3403, BW=851MiB/s (892MB/s)(24.9GiB/30010msec)
    slat (nsec): min=8407, max=52371, avg=10176.20, stdev=466.58
    clat (usec): min=2951, max=20340, avg=9390.19, stdev=311.07
     lat (usec): min=2961, max=20350, avg=9400.36, stdev=311.06
    clat percentiles (usec):
     |  1.00th=[ 8979],  5.00th=[ 9372], 10.00th=[ 9372], 20.00th=[ 9372],
     | 30.00th=[ 9372], 40.00th=[ 9372], 50.00th=[ 9372], 60.00th=[ 9372],
     | 70.00th=[ 9372], 80.00th=[ 9372], 90.00th=[ 9372], 95.00th=[ 9372],
     | 99.00th=[11338], 99.50th=[11338], 99.90th=[12649], 99.95th=[13435],
     | 99.99th=[17171]
   bw (  KiB/s): min=868864, max=874196, per=100.00%, avg=871721.25, stdev=1762.17, samples=60
   iops        : min= 3394, max= 3414, avg=3405.07, stdev= 6.88, samples=60
  lat (msec)   : 4=0.01%, 10=98.56%, 20=1.46%, 50=0.01%
  cpu          : usr=0.80%, sys=4.37%, ctx=102067, majf=0, minf=9
  IO depths    : 1=0.0%, 2=0.0%, 4=0.0%, 8=0.0%, 16=0.0%, 32=100.0%, >=64=0.0%
     submit    : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.0%, 64=0.0%, >=64=0.0%
     complete  : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.1%, 64=0.0%, >=64=0.0%
     issued rwts: total=102131,0,0,0 short=0,0,0,0 dropped=0,0,0,0
     latency   : target=0, window=0, percentile=100.00%, depth=32

Run status group 0 (all jobs):
   READ: bw=851MiB/s (892MB/s), 851MiB/s-851MiB/s (892MB/s-892MB/s), io=24.9GiB (26.8GB), run=30010-30010msec

Disk stats (read/write):
  nvme0n1: ios=119055/0, merge=0/0, ticks=1118850/0, in_queue=1118850, util=99.73%
```

geerlingguy, 2024-07-15:
@KeyserSoze1 - oh wow! I just ordered a PC711 SSD to confirm on my end - do you have the kernel patch you applied to get it working? I wonder if it could be another quirk in the bus that could be worked around in the device tree?

KeyserSoze1, 2024-07-15:
Here's the kernel patch as applied to the latest 6.6.y branch.

https://www.dropbox.com/scl/fi/9k0iuo44gxn3df79ggs61/radxa_2.5_kernel.patch?rlkey=z41jzlil2vnvzypmxyhpi120m&dl=0

I'm not sure how to best handle this quirk.  Hopefully someone else more knowledgeable can jump in and help create a more manageable fix to this than my absolute hack I did to verify the issue.  I'm not sure why the initial memory region causes issues to begin with. It shouldn't from my understanding but obviously something is not working as intended there.

On first attempt I was not able to boot from NVME but I haven't tried any of the additional config.txt tweaks yet.

KeyserSoze1, 2024-07-20:
I have not successfully been able to boot from NVME.  I have tried to change the boot order also so NVME is first but the disk doesn't appear to be detected.  I can get it to use the NVME as root if I leave an SD card in there and change the cmdline.txt to change the root to the NVME partition.

Here's the output from lsblk after I change the root on the SD card cmdline.txt.  

```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  59.7G  0 disk 
├─mmcblk0p1 179:1    0   512M  0 part 
└─mmcblk0p2 179:2    0  59.2G  0 part 
nvme0n1     259:0    0 238.5G  0 disk 
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
└─nvme0n1p2 259:2    0   238G  0 part /
```

mikegapinski, 2024-07-20:
It makes perfect sense since the patch you’ve made doesn’t affect the initial PCIe enumeration that is done in the bootloader. 

I’d suggest creating an issue in the firmware repo at Raspberry Pi, it shouldn’t be that hard for them to fix and they are mostly likely unaware about this right now (I recon most of the testing was done on a 2-port Gen 2 switch).

@geerlingguy do you still have that 4-port GEN 2 monstrosity with pogo pins? It is worth checking out if booting from NVMe works on this board with the current bootloader. You might need to be a little creative with finding the right drive to try if the pcie switch is not fresh (2024 datecode, older versions throw ASPM errors)

KeyserSoze1, 2024-07-21:
> It makes perfect sense since the patch you’ve made doesn’t affect the initial PCIe enumeration that is done in the bootloader.
>
> I’d suggest creating an issue in the firmware repo at Raspberry Pi, it [...]

So you believe the bootloader issue is related to this being a 4 port switch vs 2?  Do you think this also then leads into the issue my patch is working around?

geerlingguy, 2024-07-21:
> @geerlingguy do you still have that 4-port GEN 2 monstrosity with pogo pins? It is worth checking out if booting from NVMe works on this board with the current bootloader. You might need to be a [...]

@mikegapinski - I do still have that board, I will try to dig it out and throw a known-good boot NVMe on each of the ports to see what happens with the latest firmware on one of my Pi 5s.

@KeyserSoze1 - I think an issue on the firmware repo is definitely in order—at minimum we can track this issue more centrally (I know some of the Pi engineering team checks on these issues, but we're getting pretty deep on this issue and I wouldn't blame them if they lost visibility on it!).

KeyserSoze1, 2024-07-23:
In the issue I just opened up in the firmware repo (moved to linux repo) there was a suggestion to try a dtoverlay https://github.com/raspberrypi/linux/issues/6278#issuecomment-2244544771

This works in allowing all devices to initialize and is 100% better than the kernel hack patch.  It doesn't solve the bootloader issue so hoping we can still get to that.

I did notice one thing when trying `dtoverlay=pciex1-compat-pi5,mmio-hi` that I started to get these messages 
`r8125 0000:05:00.0: no MSI/MSI-X. Back to INTx.`
Probably not an issue. Not sure if using INTx is ultimately better for the NIC vs MSI?  However, I was able to remove the message by adding no-mip to the dtoverlay `dtoverlay=pciex1-compat-pi5,no-mip,mmio-hi`

I'm using the r8125 driver as I have found that to be significantly better depending on the setup and will update my results on that later.

KeyserSoze1, 2024-07-23:
The current situation appears to be an ASMedia issue with being assigned low mmio (memory mapped I/O) addresses. My understanding is they can use those addresses for internal purposes and not pass them on to the devices downstream. The overlay gets around this by starting assignment at 2 GB instead of 0.  This seems to work well for the device once booted but does not solve the bootloader on booting from NVME through the switch.

I'm curious for anyone with a Rock SBC with this HAT to see the dmesg output to see what assignments happen during boot for that.  If anyone has one and could post some logs that would be useful to verify that mmio assignment is happening differently on those which is why they work. Maybe @hipboi knows someone who could help in that regard.

I'm currently thinking of submitting a Feature Request to the firmware repo to start at the top of the I/O address range and work down as brought up here https://github.com/raspberrypi/linux/issues/6278#issuecomment-2245885931
This would definitely be a workaround as the ASMedia should work with any valid mmio address but that doesn't appear to be the case.  There are however other devices that have similar issues which is why the overlay was created in the first place. I'm not as familiar with the Pi dev community. Does anyone else have any thoughts on this idea and if this would be considered?

RadxaMitchell, 2024-07-24:
**Here are some logs from using the Radxa Dual 2.5G Router HAT on the ROCK-5C.**

**I've got SSDs hooked up to both the M.2 port and the FFC port on the Radxa Dual 2.5G Router HAT**

**Using the `lspci` command after startup displays the following:**
```
radxa@rock-5c:~$ lspci
0004:40:00.0 PCI bridge: Fuzhou Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0004:43:00.0 Non-Volatile memory controller: Yangtze Memory Technologies Co.,Ltd Device 0071 (rev 01)
0004:44:00.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0004:45:00.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0004:45:02.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0004:45:06.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0004:45:0e.0 PCI bridge: ASMedia Technology Inc. Device 2806 (rev 01)
0004:46:00.0 Non-Volatile memory controller: Kingston Technology Company, Inc. OM3PDP3 NVMe SSD (rev 01)
0004:4a:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0004:4b:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```

**When I use the `lsblk` command I can see that there are `nvme0n1` , `nvme1n1` devices**
```
radxa@rock-5c:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0  14.7G  0 disk
├─mmcblk0p1  179:1    0    16M  0 part /config
├─mmcblk0p2  179:2    0   300M  0 part /boot/efi
└─mmcblk0p3  179:3    0  14.4G  0 part /
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
zram0        254:0    0   1.9G  0 disk [SWAP]
nvme0n1      259:0    0 953.9G  0 disk
nvme1n1      259:1    0 119.2G  0 disk
```

**Here is the dmesg log**
```
radxa@rock-5c:~$ sudo dmesg | grep nvme
[    9.847659] nvme nvme0: pci function 0004:43:00.0
[    9.851703] nvme 0004:43:00.0: enabling device (0000 -> 0002)
[    9.851802] nvme nvme1: pci function 0004:46:00.0
[    9.866792] nvme nvme0: missing or invalid SUBNQN field.
[    9.867186] nvme 0004:46:00.0: enabling device (0000 -> 0002)
[    9.894246] nvme nvme1: missing or invalid SUBNQN field.
[   10.161651] nvme nvme0: allocated 32 MiB host memory buffer.
[   10.217665] nvme nvme0: 8/0/0 default/read/poll queues
[   10.332702] nvme nvme1: 8/0/0 default/read/poll queues
```

**The two 2.5 Gbps ports can also be viewed using the `ip a` command.**
```
radxa@rock-5c:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 5e:dc:13:e1:70:78 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.119/24 brd 192.168.2.255 scope global dynamic noprefixroute eth0
       valid_lft 2123sec preferred_lft 2123sec
    inet6 240e:3b7:324b:dc80::b3f/128 scope global dynamic noprefixroute
       valid_lft 2125sec preferred_lft 2125sec
    inet6 fd02::b3f/128 scope global dynamic noprefixroute
       valid_lft 2125sec preferred_lft 2125sec
    inet6 fd02::7618:a73c:60e6:775e/64 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 240e:3b7:324b:dc80:5e75:448:75e2:e88f/64 scope global dynamic noprefixroute
       valid_lft 241423sec preferred_lft 155023sec
    inet6 fe80::2e5e:1b56:a5fb:294a/64 scope link noprefixroute
[... 217 lines cut ...]
    clat (usec): min=10305, max=53591, avg=11064.25, stdev=2323.08
     lat (usec): min=10465, max=53865, avg=11297.79, stdev=2329.59
    clat percentiles (usec):
     |  1.00th=[10290],  5.00th=[10421], 10.00th=[10421], 20.00th=[10421],
     | 30.00th=[10421], 40.00th=[10421], 50.00th=[10421], 60.00th=[10421],
     | 70.00th=[10421], 80.00th=[10421], 90.00th=[14222], 95.00th=[14615],
     | 99.00th=[22938], 99.50th=[23200], 99.90th=[32113], 99.95th=[53740],
     | 99.99th=[53740]
   bw (  KiB/s): min=163840, max=376832, per=99.99%, avg=362338.46, stdev=39976.43, samples=39
   iops        : min=   40, max=   92, avg=88.46, stdev= 9.76, samples=39
  lat (msec)   : 20=98.02%, 50=1.92%, 100=0.06%
  cpu          : usr=2.44%, sys=4.78%, ctx=2299, majf=0, minf=19
  IO depths    : 1=100.0%, 2=0.0%, 4=0.0%, 8=0.0%, 16=0.0%, 32=0.0%, >=64=0.0%
     submit    : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.0%, 64=0.0%, >=64=0.0%
     complete  : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.0%, 64=0.0%, >=64=0.0%
     issued rwts: total=0,1770,0,0 short=0,0,0,0 dropped=0,0,0,0
     latency   : target=0, window=0, percentile=100.00%, depth=4

Run status group 0 (all jobs):
  WRITE: bw=354MiB/s (371MB/s), 354MiB/s-354MiB/s (371MB/s-371MB/s), io=7080MiB (7424MB), run=20007-20007msec

Disk stats (read/write):
  nvme1n1: ios=0/29847, merge=0/0, ticks=0/168041, in_queue=168042, util=99.39%

```

RadxaMitchell, 2024-07-24:
**Sorry, I forgot to add some kernel logs for using the Radxa Dual 2.5G Router HAT on the ROCK-5C, which I've added below**

Related to pcie:
```
radxa@rock-5c:~$ sudo dmesg | grep pci
[    8.281823] vcc3v3_pcie: 3300 mV, disabled
[    8.281870] reg-fixed-voltage vcc3v3-pcie: Looking up vin-supply from device tree
[    8.281878] vcc3v3_pcie: supplied by vcc5v0_sys
[    8.281913] reg-fixed-voltage vcc3v3-pcie: vcc3v3_pcie supplying 3300000uV
[    8.969781] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
[    8.969806] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
[    8.975491] rk-pcie fe190000.pcie: missing legacy IRQ resource
[    8.975524] rk-pcie fe190000.pcie: IRQ msi not found
[    8.975539] rk-pcie fe190000.pcie: use outband MSI support
[    8.975556] rk-pcie fe190000.pcie: Missing *config* reg space
[    8.975586] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
[    8.975621] rk-pcie fe190000.pcie:      err 0x00f4000000..0x00f40fffff -> 0x00f4000000
[    8.975652] rk-pcie fe190000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[    8.975684] rk-pcie fe190000.pcie:      MEM 0x00f4200000..0x00f4ffffff -> 0x00f4200000
[    8.975712] rk-pcie fe190000.pcie:      MEM 0x0a00000000..0x0a3fffffff -> 0x0a00000000
[    8.975761] rk-pcie fe190000.pcie: Missing *config* reg space
[    8.975803] rk-pcie fe190000.pcie: invalid resource
[    9.183082] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x3
[    9.208618] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x7
[    9.291938] rk-pcie fe190000.pcie: PCIe Link up, LTSSM is 0x30011
[    9.292110] rk-pcie fe190000.pcie: PCI host bridge to bus 0004:40
[    9.292121] pci_bus 0004:40: root bus resource [bus 40-4f]
[    9.292130] pci_bus 0004:40: root bus resource [??? 0xf4000000-0xf40fffff flags 0x0]
[    9.292143] pci_bus 0004:40: root bus resource [io  0x0000-0xfffff] (bus address [0xf4100000-0xf41fffff])
[    9.292150] pci_bus 0004:40: root bus resource [mem 0xf4200000-0xf4ffffff]
[    9.292156] pci_bus 0004:40: root bus resource [mem 0xa00000000-0xa3fffffff pref]
[    9.292199] pci 0004:40:00.0: [1d87:3588] type 01 class 0x060400
[    9.292227] pci 0004:40:00.0: reg 0x38: [mem 0x00000000-0x0000ffff pref]
[    9.292302] pci 0004:40:00.0: supports D1 D2
[    9.292308] pci 0004:40:00.0: PME# supported from D0 D1 D3hot
[    9.302936] pci 0004:40:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    9.303196] pci 0004:41:00.0: [1b21:2806] type 01 class 0x060400
[    9.303426] pci 0004:41:00.0: enabling Extended Tags
[    9.303850] pci 0004:41:00.0: PME# supported from D0 D3hot D3cold
[    9.304008] pci 0004:41:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0004:40:00.0 (capable of 15.752 Gb/s with 8.0 GT/s PCIe x2 link)
[    9.321894] pci 0004:41:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    9.322263] pci 0004:42:00.0: [1b21:2806] type 01 class 0x060400
[    9.322443] pci 0004:42:00.0: enabling Extended Tags
[    9.322762] pci 0004:42:00.0: PME# supported from D0 D3hot D3cold
[    9.323814] pci 0004:42:02.0: [1b21:2806] type 01 class 0x060400
[    9.323996] pci 0004:42:02.0: enabling Extended Tags
[    9.324314] pci 0004:42:02.0: PME# supported from D0 D3hot D3cold
[    9.325982] pci 0004:42:06.0: [1b21:2806] type 01 class 0x060400
[    9.326163] pci 0004:42:06.0: enabling Extended Tags
[    9.326487] pci 0004:42:06.0: PME# supported from D0 D3hot D3cold
[    9.329422] pci 0004:42:0e.0: [1b21:2806] type 01 class 0x060400
[    9.329604] pci 0004:42:0e.0: enabling Extended Tags
[    9.329922] pci 0004:42:0e.0: PME# supported from D0 D3hot D3cold
[    9.335916] pci 0004:42:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    9.335946] pci 0004:42:02.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    9.335976] pci 0004:42:06.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    9.336004] pci 0004:42:0e.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    9.336404] pci 0004:43:00.0: [1e49:0071] type 00 class 0x010802
[    9.336487] pci 0004:43:00.0: reg 0x10: [mem 0x00000000-0x00003fff 64bit]
[    9.337286] pci 0004:43:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0004:40:00.0 (capable of 63.012 Gb/s with 16.0 GT/s PCIe  x4 link)
[    9.355320] pci_bus 0004:43: busn_res: [bus 43-4f] end is updated to 43
[    9.355667] pci 0004:44:00.0: [1b21:2806] type 01 class 0x060400
[    9.355862] pci 0004:44:00.0: enabling Extended Tags
[    9.356213] pci 0004:44:00.0: PME# supported from D0 D3hot D3cold
[    9.356364] pci 0004:44:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0004:40:00.0 (capable of 15.752 Gb/s with 8.0 GT/s PCIe x2 link)
[    9.375313] pci 0004:44:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    9.375697] pci 0004:45:00.0: [1b21:2806] type 01 class 0x060400
[    9.375897] pci 0004:45:00.0: enabling Extended Tags
[    9.376253] pci 0004:45:00.0: PME# supported from D0 D3hot D3cold
[    9.377313] pci 0004:45:02.0: [1b21:2806] type 01 class 0x060400
[... 113 lines cut ...]
About the r8125:
```
radxa@rock-5c:~$ sudo dmesg | grep r8125
[   13.322520] r8125: loading out-of-tree module taints kernel.
[   13.343517] r8125 2.5Gigabit Ethernet driver 9.011.00-NAPI-RSS loaded
[   13.343548] r8125 0004:4a:00.0: enabling device (0000 -> 0003)
[... cut ...]
[   13.428076] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   13.435492] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com>
[   13.442470] r8125 0004:4b:00.0 enP4p75s0: renamed from eth2
[   13.505858] r8125 0004:4a:00.0 enP4p74s0: renamed from eth1
[   19.799126] r8125: enP4p75s0: link up
[  654.781023] r8125: enP4p74s0: link up
```

**I've attached the full log file, hope that helps!**

- [kernel.log](https://github.com/user-attachments/files/16359442/rock-5c_kernel.log)

KeyserSoze1, 2024-07-24:
Thanks @RadxaMitchell that confirms my suspicions.

Rock-5C log
```
[    8.975586] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
[    8.975621] rk-pcie fe190000.pcie:      err 0x00f4000000..0x00f40fffff -> 0x00f4000000
[    8.975652] rk-pcie fe190000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[    8.975684] rk-pcie fe190000.pcie:      MEM 0x00f4200000..0x00f4ffffff -> 0x00f4200000
[    8.975712] rk-pcie fe190000.pcie:      MEM 0x0a00000000..0x0a3fffffff -> 0x0a00000000
```
**Memory Mapped IO Starts at address 0x00f4200000 for the Rock-5C**

Pi5 with Overlay Log
```
[    1.493108] brcm-pcie 1000110000.pcie: host bridge /axi/pcie@110000 ranges:
[    1.500113] brcm-pcie 1000110000.pcie:   No bus range found for /axi/pcie@110000, using [bus 00-ff]
[    1.509210] brcm-pcie 1000110000.pcie:      MEM 0x1b80000000..0x1bfffffffb -> 0x0080000000
[    1.517515] brcm-pcie 1000110000.pcie:      MEM 0x1800000000..0x1b7fffffff -> 0x0400000000
[    1.525819] brcm-pcie 1000110000.pcie:   IB MEM 0x0000000000..0x0fffffffff -> 0x1000000000
```
**Memory Mapped IO starts at address 0x0080000000 for the Pi5 with the dtoverlay in config.txt**

Pi5 No Overlay
```
[    1.647929] brcm-pcie 1000110000.pcie: host bridge /axi/pcie@110000 ranges:
[    1.655009] brcm-pcie 1000110000.pcie:   No bus range found for /axi/pcie@110000, using [bus 00-ff]
[    1.664107] brcm-pcie 1000110000.pcie:      MEM 0x1b00000000..0x1bfffffffb -> 0x0000000000
[    1.672411] brcm-pcie 1000110000.pcie:      MEM 0x1800000000..0x1affffffff -> 0x0400000000
[    1.680715] brcm-pcie 1000110000.pcie:   IB MEM 0x0000000000..0x0fffffffff -> 0x1000000000
```
**Memmory Mapped IO starts at address 0x0000000000 for the Pi5 by default**

This aligns with the idea the ASMedia chip does not handle low mmio addresses well.

mikegapinski, 2024-07-24:
> Data: środa, 24 lipca 2024 o 12:39
> Do: geerlingguy/raspberry-pi-pcie-devices ***@***.***>
> DW: Michał Gapiński ***@***.***>, Mention ***@***.***>

This could be potentially handled in brcm-pcie without having to apply the overlay. Fallback on error with mmio init in both the kernel and the bootloader (for NVMe boot).

Potentially it can be also discussed with ASMedia since they can fix it in a newer revision of the switch (they used to improve the Gen 2 chips over the years and newer ones don’t have any issues with the Pi 5).

[    1.500113] brcm-pcie 1000110000.pcie:   No bus range found for ***@***.***, using [bus 00-ff]

[    1.655009] brcm-pcie 1000110000.pcie:   No bus range found for ***@***.***, using [bus 00-ff]

KeyserSoze1, 2024-07-24:
> This could be potentially handled in brcm-pcie without having to apply the overlay. Fallback on error with mmio init in both the kernel and the bootloader (for NVMe boot). 
Potentially it can be also discussed with ASMedia since they can fix it in a newer revision of the switch (they used to improve the Gen 2 chips over the years and newer ones don’t have any issues with the Pi 

Does the brcm-pcie know it failed? This appears to me as if the assignment succeeds and the pcie goes along thinking everything is good until one of the downstream devices attempts to use one of the low addresses and things don't go as planned. I'm not sure anything upstream of the ASMedia switch is ever aware of the issue from what I can tell at least. Maybe there is a way to detect this though as that would be a good way to handle this condition.

Not sure how to bring this to ASMedia's attention but this does seem like something they should at a minimum be aware of.

KeyserSoze1, 2024-07-27:
Here's my results of bonding the 2.5gb NIC's together.
I have a cheap managed switch that supports LACP so thought I would give that a try.
These were all collected using with config.txt items
```
dtparam=pciex1
dtparam=pciex1_gen=3
dtoverlay=pciex1-compat-pi5,mmio-hi
```

Network bonding options mode=802.3ad and xmit_hash_policy=layer3+4 were used. I used iperf3 with 2 parallel connections to collect results. Results were sometimes run multiple times depending on the two ports chosen by iperf3 they might hash to same NIC based on layer3+4 hash scheme.

**r8169** driver results
----------------------------------

**Transmit results**
`[SUM]   0.00-15.00  sec  6.38 GBytes  3.65 Gbits/sec    0             sender`
Performance is not great at 3.65 Gbps.  Better than 2.5 but not a huge increase. This capped one processor at 100% which I believe is the bottleneck here.
<details>
  <summary>iperf3 Tx Results (Click to expand)</summary>

```
iperf3 -c 192.168.1.100 --time 15 -P 2
Connecting to host 192.168.1.100, port 5201
[  5] local 192.168.1.101 port 36976 connected to 192.168.1.100 port 5201
[  7] local 192.168.1.101 port 36990 connected to 192.168.1.100 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   217 MBytes  1.82 Gbits/sec    0   1.29 MBytes       
[  7]   0.00-1.00   sec   223 MBytes  1.87 Gbits/sec    0   1.31 MBytes       
[SUM]   0.00-1.00   sec   440 MBytes  3.69 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec   214 MBytes  1.80 Gbits/sec    0   1.29 MBytes       
[  7]   1.00-2.00   sec   228 MBytes  1.91 Gbits/sec    0   1.37 MBytes       
[SUM]   1.00-2.00   sec   441 MBytes  3.71 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   2.00-3.00   sec   212 MBytes  1.78 Gbits/sec    0   1.29 MBytes       
[  7]   2.00-3.00   sec   225 MBytes  1.89 Gbits/sec    0   1.37 MBytes       
[SUM]   2.00-3.00   sec   438 MBytes  3.67 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   3.00-4.00   sec   209 MBytes  1.75 Gbits/sec    0   1.29 MBytes       
[  7]   3.00-4.00   sec   222 MBytes  1.87 Gbits/sec    0   1.37 MBytes       
[SUM]   3.00-4.00   sec   431 MBytes  3.62 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   4.00-5.00   sec   210 MBytes  1.76 Gbits/sec    0   1.29 MBytes       
[  7]   4.00-5.00   sec   222 MBytes  1.87 Gbits/sec    0   1.37 MBytes       
[SUM]   4.00-5.00   sec   432 MBytes  3.63 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   5.00-6.00   sec   206 MBytes  1.73 Gbits/sec    0   1.29 MBytes       
[  7]   5.00-6.00   sec   218 MBytes  1.82 Gbits/sec    0   1.37 MBytes       
[SUM]   5.00-6.00   sec   424 MBytes  3.55 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   6.00-7.00   sec   215 MBytes  1.80 Gbits/sec    0   1.29 MBytes       
[  7]   6.00-7.00   sec   225 MBytes  1.89 Gbits/sec    0   1.37 MBytes       
[SUM]   6.00-7.00   sec   440 MBytes  3.69 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   7.00-8.00   sec   211 MBytes  1.77 Gbits/sec    0   1.29 MBytes       
[  7]   7.00-8.00   sec   224 MBytes  1.88 Gbits/sec    0   1.37 MBytes       
[SUM]   7.00-8.00   sec   435 MBytes  3.65 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   8.00-9.00   sec   215 MBytes  1.80 Gbits/sec    0   1.35 MBytes       
[  7]   8.00-9.00   sec   222 MBytes  1.87 Gbits/sec    0   1.37 MBytes       
[SUM]   8.00-9.00   sec   438 MBytes  3.67 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   9.00-10.00  sec   214 MBytes  1.79 Gbits/sec    0   1.35 MBytes       
[  7]   9.00-10.00  sec   221 MBytes  1.86 Gbits/sec    0   1.37 MBytes       
[SUM]   9.00-10.00  sec   435 MBytes  3.65 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]  10.00-11.00  sec   212 MBytes  1.78 Gbits/sec    0   1.35 MBytes       
[  7]  10.00-11.00  sec   222 MBytes  1.87 Gbits/sec    0   1.37 MBytes       
[SUM]  10.00-11.00  sec   435 MBytes  3.65 Gbits/sec    0             
- - - - - - - - - - - - - - - - - - - - - - - - -
[... 312 lines cut ...]
[  5][TX-C]  10.00-11.00  sec   280 MBytes  2.35 Gbits/sec    0   1.03 MBytes       
[  7][RX-C]  10.00-11.00  sec   277 MBytes  2.32 Gbits/sec                  
[  5][TX-C]  11.00-12.00  sec   281 MBytes  2.36 Gbits/sec    0   1.03 MBytes       
[  7][RX-C]  11.00-12.00  sec   277 MBytes  2.33 Gbits/sec                  
[  5][TX-C]  12.00-13.00  sec   280 MBytes  2.35 Gbits/sec    0   1.03 MBytes       
[  7][RX-C]  12.00-13.00  sec   232 MBytes  1.95 Gbits/sec                  
[  5][TX-C]  13.00-14.00  sec   281 MBytes  2.36 Gbits/sec    0   1.03 MBytes       
[  7][RX-C]  13.00-14.00  sec   270 MBytes  2.26 Gbits/sec                  
[  5][TX-C]  14.00-15.00  sec   280 MBytes  2.35 Gbits/sec    0   1.03 MBytes       
[  7][RX-C]  14.00-15.00  sec   265 MBytes  2.23 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-15.00  sec  4.11 GBytes  2.35 Gbits/sec    0             sender
[  5][TX-C]   0.00-15.00  sec  4.11 GBytes  2.35 Gbits/sec                  receiver
[... cut ...]
I did find some issues with the r8125 driver however that I want to note.  I used the driver from here https://github.com/awesometic/realtek-r8125-dkms

I noticed some receive results were especially bad even with one connection instead of two. I would only get a few 100mpbs of throughput. I found disabling ASPM in the driver solved this. If using the driver from the repo above I modified src/Makefile and changed the line to
`CONFIG_ASPM = n`

Not sure if anyone wanted to see this but it does show using Gen3 you can get ~5gb performance from bonding the NICs.

KeyserSoze1, 2024-08-06:
Posted bootloader request to firmware repo. https://github.com/raspberrypi/firmware/issues/1912

Not much interest thus far.

netacti, 2024-11-06:
I was wondering if the lack of progress since August is due to an issue with the ASM2806 or simply because of low interest from demand?

hipboi, 2024-11-06:
@netacti Simply because we are quite busy with a lot of interesting projects going on and the priority for the accessories are lower. If there is demand, surely we will speed up and push it to the market.

geerlingguy, 2024-11-06:
@hipboi - heh, we share that same problem! Radxa's had a lot going on. I still have this HAT sitting about 30cm away from my keyboard, and one of these days I'll have a few hours in my schedule to finally get it back up and running for more testing!

I'm still excited to get it going for a dual 2.5 Gbps router for Pi 5! I even bought a couple more NVMe SSDs to test with it, just—lack of time right now :(

geerlingguy, 2025-01-07:
FYI the rpi-eeprom issue was fixed that was preventing boot behind this PCIe switch (supposedly): https://github.com/raspberrypi/rpi-eeprom/issues/598

I should probably test again!

KeyserSoze1, 2025-01-07:
I've been using this device with no issues for months and it's been rock solid.  The only thing I was unable to do was boot from NVME. I've been using the boot partition of a microSD card and using NVME for just data.  I saw that my issue was closed and tried to boot straight from NVME and it was a SUCCESS!

I rsync'd my microSD boot partition to the NVME boot partition first so I had a fully updated NVME boot partition.

I turned on PCIE_PROBE=1 (not sure if this is needed) and I had to edit my fstab to stop mounting the boot partition of the microsd card and mount the NVME boot partition instead.  No SD card is currently inserted and I'm up and running. 

Would love to hear if it works as well for others.

geerlingguy, 2025-01-30:
Annoyingly... I finally got the parts together to do some more work with this HAT, and today for a while, I couldn't for the life of me get it to be recognized:

```
$ lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
```

First I checked the eeprom:

```
$ sudo rpi-eeprom-update
*** UPDATE AVAILABLE ***

Run "sudo rpi-eeprom-update -a" to install this update now.

To configure the bootloader update policy run "sudo raspi-config"

BOOTLOADER: update available
   CURRENT: Wed 27 Nov 14:28:19 UTC 2024 (1732717699)
    LATEST: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.
```

I updated it, and rebooted... still nothing.

Eventually, I realized this is one of the _very_ few HATs that doesn't automatically trigger the Pi to enable the PCIe FFC external connector.

So I had to enable it in the config.txt file:

```
[all]
dtparam=pciex1 
dtparam=pciex1_gen=3
```

And now it appears again:

```
$ lspci
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0000:01:00.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:00.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:02.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:06.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:02:0e.0 PCI bridge: ASMedia Technology Inc. ASM2806 4-Port PCIe x2 Gen3 Packet Switch (rev 01)
0000:05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0000:06:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0001:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
```

geerlingguy, 2025-01-31:
Booting from the NVMe SSD just using `sudo raspi-config` to set it up did not work, since this isn't a HAT+ device.

I had to follow [Raspberry Pi's Booting from PCIe Devices](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot-from-pcie) documentation, and run `sudo rpi-eeprom-config --edit`. Add:

```
PCIE_PROBE=1
```

After that, I'm booting off NVMe. Yay!

```
pi@pi5-router:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0  29.5G  0 disk 
├─mmcblk0p1 179:1    0   512M  0 part /media/pi/bootfs
└─mmcblk0p2 179:2    0    29G  0 part /media/pi/rootfs
nvme0n1     259:0    0 238.5G  0 disk 
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
└─nvme0n1p2 259:2    0   238G  0 part /
```

geerlingguy, 2025-01-31:
@akgnah and @hipboi - I think this HAT may be a good candidate for "The ULTIMATE Raspberry Pi 5 Router", and I've been slowly working towards a video on that topic...

What are the chances this HAT could go to more wide-scale production, and do you know a price for it when it would?

Also, I will attempt reversing the FPC on my HAT, hopefully it doesn't end up turning into a destructive operation :) — but if you have an additional Dual 2.5G Router HAT available I'd love to test it, as it sounds like there may be another tweak in terms of the heat sinking? (Not sure if there are any other PCB tweaks to that version...).

It would also be interesting—if a design tweak is required—to see if you could get this board to be 'HAT+' compatible so a user wouldn't be required to add the tweaks to the EEPROM or config.txt :)

hipboi, 2025-02-12:
The dual 2.5G Router HAT is ready for MP now, I think it will be on the market in the next 4 weeks. We will send you a MP ready sample. We still have few MP ready samples available, if anyone in this thread is interested, drop me an email - tom at radxa.com.

The HAT has reserved eeprom on board and we have considered the HAT+ specification, we will check how to write the content to the eerpom.

geerlingguy, 2025-02-12:
@hipboi - That's awesome news!

If I don't get to reversing the FFC on mine before I receive the production sample I'll just switch my testing to your newer version. Hoping to get something like an OpenWRT build for it set up too... we'll see how much work that entails.

geerlingguy, 2025-02-21:
I received the new samples, and everything is working... except I can't seem to get power to the external FFC connection—I've tried my Pineboards cable in both orientations (not sure if pin side should be facing outside edge or inside edge of the PCB...)

[image]

geerlingguy, 2025-02-21:
Quickly testing the dual 2.5 Gbps interfaces with `iperf3`, I have two servers with 10 Gbps NICs, running iperf3 server (`iperf3 -s`). Each one independently can hit up to 2.3 Gbps or so from the Pi (and about 1.5 Gbps back).

But together:

```
iperf3 --no-delay -c 10.0.2.51 --bind-dev eth1 --interval 0 --parallel 1 --time 30 &
iperf3 --no-delay -c 10.0.2.15 --bind-dev eth2 --interval 0 --parallel 1 --time 30 &
...
Connecting to host 10.0.2.51, port 5201
[  5] local 10.0.2.245 port 53816 connected to 10.0.2.51 port 5201
Connecting to host 10.0.2.15, port 5201
[  5] local 10.0.2.233 port 37982 connected to 10.0.2.15 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.00  sec  4.12 GBytes  1.18 Gbits/sec    0   1.03 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-30.00  sec  4.12 GBytes  1.18 Gbits/sec    0             sender
[  5]   0.00-30.00  sec  4.12 GBytes  1.18 Gbits/sec                  receiver

iperf Done.
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.00  sec  4.12 GBytes  1.18 Gbits/sec    0   1.49 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-30.00  sec  4.12 GBytes  1.18 Gbits/sec    0             sender
[  5]   0.00-30.04  sec  4.11 GBytes  1.18 Gbits/sec                  receiver
```

It seems like they're still splitting up their resources as if they were one interface... `atop` isn't showing any bottlenecks :/

[image]

Is there some easy way to overcome this rate limit? Or some other better way to test maximum bandwidth through both interfaces at once? I also tried setting `--affinity` to 0/1, 1/2, and 2/3 to see if that would make any differnce.

KeyserSoze1, 2025-02-21:
What driver are you using?  I have better performance using the r8125 driver in my testing.

geerlingguy, 2025-02-24:
@KeyserSoze1 - Aaand I just realized I had both ports plugged into the network through the _same_ 2.5 Gbps switch.

If I plug one port into one switch, and one into another, now I get 2.21 Gbps on each:

```
pi@pi5-router:~ $ iperf3 --no-delay -c 10.0.2.51 --bind-dev eth1 --interval 0 --parallel 1 --time 30 &
iperf3 --no-delay -c 10.0.2.15 --bind-dev eth2 --interval 0 --parallel 1 --time 30 &
[1] 1568
[2] 1569
pi@pi5-router:~ $ Connecting to host 10.0.2.51, port 5201
[  5] local 10.0.2.239 port 44948 connected to 10.0.2.51 port 5201
Connecting to host 10.0.2.15, port 5201
[  5] local 10.0.2.235 port 48690 connected to 10.0.2.15 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.00  sec  7.71 GBytes  2.21 Gbits/sec    0   3.25 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-30.00  sec  7.71 GBytes  2.21 Gbits/sec    0             sender
[  5]   0.00-30.00  sec  7.71 GBytes  2.21 Gbits/sec                  receiver

iperf Done.
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-30.00  sec  7.71 GBytes  2.21 Gbits/sec    0   1.49 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-30.00  sec  7.71 GBytes  2.21 Gbits/sec    0             sender
[  5]   0.00-30.04  sec  7.71 GBytes  2.20 Gbits/sec                  receiver

iperf Done.
```

[19 more replies]