geerlingguy/raspberry-pi-pcie-devices #650: Add Hailo-8L

geerlingguy, 2024-06-04:
The [Hailo-8L](https://hailo.ai/products/ai-accelerators/hailo-8l-ai-accelerator-for-ai-light-applications/) is Hailo's 'Entry Level' AI accelerator, meant for edge and lighter use cases.

[image]

It runs at 13 TOPS, and 8 TOPS/W, and was announced late last year. It [comes in two M.2 form factors](https://hailo.ai/files/hailo-8l-product-brief-en/), a 2242-size B+M-key version, and a 2230-size A+E-key version.

See related: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/164

geerlingguy, 2024-06-04:
This is on the site here: https://pipci.jeffgeerling.com/cards_m2/hailo-8l-ai-module.html

geerlingguy, 2024-06-04:
FYI, Hailo maintains a set of Raspberrry Pi 5 examples here: https://github.com/hailo-ai/hailo-rpi5-examples — these were just published yesterday, as the 8L is included in Raspberry Pi's new AI Kit, announcement here: https://www.raspberrypi.com/news/raspberry-pi-ai-kit-available-now-at-70/

See full Pi documentation here: https://www.raspberrypi.com/documentation/accessories/ai-kit.html

It seems like support is rolling into [`picamera2`](https://github.com/raspberrypi/picamera2) and [`rpicam-apps`](https://github.com/raspberrypi/rpicam-apps) now as well.

I have a blog post with more details testing the 8L on a Pi 5: [Testing Raspberry Pi's AI Kit - 13 TOPS for $70](https://www.jeffgeerling.com/blog/2024/testing-raspberry-pis-ai-kit-13-tops-70)

yoyojacky, 2024-06-14:
Hi Jeff, 
I have test it on 52Pi N04 M.2 NVMe SSD adapter board with AI  co-processor hailo-8 but not 8L, it works fine! 
[image]
[image]

expect the testing result of your 8L module.

joelewing, 2024-08-07:
Hi Jeff,
Do you know where the 2230-size A+E-key version can be purchased by itself? I'm only seeing it sold with a Raspberry Pi kit, or perhaps it is behind the 'Product Inquiry' wall on Hailo's website.

geerlingguy, 2024-08-07:
@joelewing - I haven't seen any individually, but I do also see them on Pineboards: [Ai Bundle (Hailo 8L)](https://pineboards.io/products/ai-bundle-hailo-8l)