geerlingguy/raspberry-pi-pcie-devices #652: RADXA RA028_X1.0

4MeInfo, 2024-06-20:
I recived a new testing prototype from RADXA with the print RA028_X1.0 (thanks @hipboi )
It's an 4 bay M.2 NVMe Hat that supports 2230, 2242, 2260,2280 NVMe's and is  powered by a USB C port at 12 volts (you can use the official Raspberry Pi 5 power supply).
The main chip ist the ASM2806.
More about testing tomorrow!
[image]
[image]

hipboi, 2024-06-21:
The name would be **Radxa Quad M.2 SSD HAT**. 

The USB C on the HAT supports PD power supply to power the 4x SSD and Pi 5(>45W). So the PD negotiation voltage for the sample is 12V, but we will increase to 20V by default in next HW revision, fall back to 12V if the power supply doesn't support 20V.

geerlingguy, 2024-06-21:
You're burying the lede here, the ASM2806 is a PCIe Gen 3 switch! Would be a very cool board for multiple M.2 drives, or a mix of M.2 NVMe SSDs and Hailo AI accelerators :)

@hipboi any word on when this might be up for sale? I know a few people were interested in something like this.

hipboi, 2024-06-25:
I think it will be available within 4 weeks.

4MeInfo, 2024-07-10:
So now I have my 4 NVME's an started my testing. 
I used the official Raspberry Pi 5 27 Watt power Supply.

Connected to the USB C port of the Quad M.2 SSD Hat the system used 10 to 12 watts.

>pi@TestPi:~ $ sudo pcicrawler -t
00:00.0 root_port, speed 8GT/s, width x1
 └─01:00.0 upstream_port, ASMedia Technology Inc. (1b21), device 2806
    ├─02:00.0 downstream_port, slot 0, device present, speed 8GT/s, width x1
    │  └─03:00.0 endpoint, Intel Corporation (8086) SSD 600P Series (f1a5)
    ├─02:02.0 downstream_port, slot 0, speed 2.5GT/s, width x1
    ├─02:06.0 downstream_port, slot 0, speed 2.5GT/s, width x1
    └─02:0e.0 downstream_port, slot 0, device present, speed 8GT/s, width x1
       └─06:00.0 endpoint, Intel Corporation (8086) SSD 600P Series (f1a5)
0001:00:00.0 root_port, speed 5GT/s, width x4
 └─0001:01:00.0 endpoint, 1de4:0001

> pi@TestPi:~ $ ls -la /dev/nvme*
crw------- 1 root root 245, 0 10. Jul 20:56 /dev/nvme0
brw-rw---- 1 root disk 259, 0 10. Jul 20:56 /dev/nvme0n1
brw-rw---- 1 root disk 259, 1 10. Jul 21:01 /dev/nvme0n1p1

output from dmesg:

>pi@TestPi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.37-v8-16k+ (dom@buildbot) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1_22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1782 SMP PREEMPT Mon Jul  8 12:41:03 BST 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000015ffffff (327680 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd169c0..0x000000003fd16a22 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 256 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s183976 r8192 d37208 u229376
[    0.000000] pcpu-alloc: s183976 r8192 d37208 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe  smsc95xx.macaddr=D8:3A:DD:C3:91:35 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=1469f9d1-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE
[    0.000000] Dentry cache hash table entries: 1048576 (order: 9, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 8, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 521984
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbffc000-0x00000000ffffc000] (64MB)
[... 1154 lines cut ...]
[    7.401850] Registered IR keymap rc-cec
[    7.402223] rc rc1: vc4-hdmi-1 as /devices/platform/soc/107c706400.hdmi/rc/rc1
[    7.402314] input: vc4-hdmi-1 as /devices/platform/soc/107c706400.hdmi/rc/rc1/input3
[    7.409161] input: vc4-hdmi-1 HDMI Jack as /devices/platform/soc/107c706400.hdmi/sound/card1/input4
[    7.409497] vc4-drm axi:gpu: bound 107c706400.hdmi (ops vc4_hdmi_ops [vc4])
[    7.411201] vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
[    7.411347] vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
[    7.411439] vc4-drm axi:gpu: bound 107c410000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.411537] vc4-drm axi:gpu: bound 107c411000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.412554] [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 1
[    7.416426] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    7.418783] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    7.420941] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[... cut ...]
[    9.385842] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    9.388572] pps pps0: new PPS source ptp0
[    9.388696] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    9.414834] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.345711] warning: `glances' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   13.477471] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx

It seems that the Pi an the Quad HAT don't deliver enough power to use all 4 NMES's

4MeInfo, 2024-07-22:
This Kernel Patch https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/647#issuecomment-2229494817 works perfectly, all drives shown and are available. But for booting from NVME is a EEPROM update necessary.

`pi@TestPi:~ $ dmesg | grep nvme`
`[    0.549130] nvme nvme0: pci function 0000:03:00.0`
`[    0.549145] nvme 0000:03:00.0: enabling device (0000 -> 0002)`
`[    0.568604] nvme nvme0: 3/0/0 default/read/poll queues`
`[    0.628564]  nvme0n1: p1`
`[    0.628748] nvme nvme1: pci function 0000:04:00.0`
`[    0.628760] nvme 0000:04:00.0: enabling device (0000 -> 0002)`
`[    0.632153] nvme nvme1: 1/0/0 default/read/poll queues`
`[    0.633165]  nvme1n1: p1`
`[    0.633338] nvme nvme2: pci function 0000:05:00.0`
`[    0.633354] nvme 0000:05:00.0: enabling device (0000 -> 0002)`
`[    1.514298] nvme nvme2: 1/0/0 default/read/poll queues`
`[    1.517350]  nvme2n1: p1`
`[    1.517514] nvme nvme3: pci function 0000:06:00.0`
`[    1.517525] nvme 0000:06:00.0: enabling device (0000 -> 0002)`
`[    1.521004] nvme nvme3: 1/0/0 default/read/poll queues`
`[    1.522525]  nvme3n1: p1`
`[  119.402607] EXT4-fs (nvme0n1p1): recovery complete`
`[  119.403451] EXT4-fs (nvme0n1p1): mounted filesystem 5ace72b1-c66b-424b-bfda-1b4f7d8d63a8 r/w with ordered data mode. Quota mode: none.`
`[  120.722180] EXT4-fs (nvme2n1p1): recovery complete`
`[  120.722931] EXT4-fs (nvme2n1p1): mounted filesystem 385e00d9-1411-48e6-a484-9ca9c06dc003 r/w with ordered data mode. Quota mode: none.`
`[  121.541853] EXT4-fs (nvme1n1p1): mounted filesystem 0c69476a-8abb-43d1-ac14-ee0003e05852 r/w with ordered data mode. Quota mode: none.`
`[  122.394215] EXT4-fs (nvme3n1p1): mounted filesystem 1ac3258f-26f4-4a69-b672-ccb8b08561e0 r/w with ordered data mode. Quota mode: none.`

[image]

KeyserSoze1, 2024-07-25:
Glad my patch was useful but wanted to send on that there is a dtoverlay already available to do a similar effect that also works that doesn't require patching the kernel. 
https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/647#issuecomment-2245300528

iantrobinson, 2024-09-06:
Is this available for purchase yet?

coreintelligence, 2024-11-19:
Hi, I'm super interested in this as 4xNVME HAT and I'm glad there's a software path to have all drives mountable. 
- @4MeInfo, Any further testing done on this HAT since the summer 2024?
- @hipboi, Is there any news on if this will be available? It's been quiet for something that's clearly a step up compared to options from others.

hipboi, 2024-11-19:
@coreintelligence It will be available in 4 weeks.

coreintelligence, 2024-11-19:
> @coreintelligence It will be available in 4 weeks.

Awesome! I'll happily treat myself to a nice Christmas/End of Year present for my Pi 5

4MeInfo, 2024-11-19:
> Hi, I'm super interested in this 4xNVME HAT and I'm glad there's a software path to have all drives mountable. 
> 
> - @4MeInfo, Any further testing done on this HAT since the summer 2024?
> 
> - @hipboi, Is there any news on if this will be available? It's been quiet for something that's clearly a step up compared to options from others.

3/4 Nvme's working. I dont have a power supply on it but water water cooling [image]cooling

iantrobinson, 2024-11-20:
> @coreintelligence It will be available in 4 weeks. 

Not to be a downer but you said the same thing 5 months ago. Is there any confirmation on this? And any idea on price?

coreintelligence, 2025-03-22:
Hi @hipboi, I just checked the [Radxa website](https://radxa.com/products/accessories#storage-expansion) and I don't see any word or news about this hat. Is there a different place where it would be sold or any update since the message late last year?

hipboi, 2025-03-25:
@coreintelligence The main challenge is managing the power supply and heat dissipation for this HAT. We find @4MeInfo’s setup quite inspiring. The dual 2.5G router HAT will be released earlier since it utilizes a 12V DC power supply. However, the quad M.2 HAT requires USB Type-C power due to its thickness and higher total power consumption (~65W), necessitating a 20V input. This brings up a key consideration: Should we allow powering the HAT directly from the Raspberry Pi 5’s USB-C port when users connect only one or two SSDs?

ywf1, 2025-12-14:
@hipboi hello from the future! Any updates on this product? I have a half a mind to develop a solution myself (if only asmedia would reply to my emails), I'd rather spend that money on your working product, thank you! 

Best,

ywf