geerlingguy/raspberry-pi-pcie-devices #661: open HAT-PCIe covert board

tltangliang, 2024-08-08:
A open source boring board, converting the Pi's PCIe HAT into a PCIe x1 card.

[image]

I have too many Raspberry Pi hats. Occasionally, I wish to debug them on an x86 computer, which gave birth to the idea of creating an adapter board. Thus, the open HAT-PCIe was born.

The board has 4 areas: PCIe x1 gold finger, 12V to 5V DC power area, Pi 5 PCIe connector, and Pi 5 HAT installation area

The wiring definition of the FPC connector is identical to that of the Raspberry Pi 5, please pay attention to the installation direction.

### Features

- Convert Pi 5 PCIe/NVMe HAT to PCIe x1 card
- Compatible with Pi 5 PCIe HAT
- Compatible with PCIe x1 Gen 2 and 3
- Compatible with 25mm or 30mm PCI bracket

github [https://github.com/tltangliang/open-HAT-PCIe](https://github.com/tltangliang/open-HAT-PCIe)

geerlingguy, 2024-08-08:
There's one other similar card I've seen, made by the OCP-TAP team that designed the Time Card—I will try to get more info on it, but I don't think they have planned on bringing it to production, just using it for internal debugging (at least so far). Thanks for releasing the design files for this board!

tltangliang, 2024-08-08:
I took a look at the board you mentioned, the functions are completely different. The board I designed is used to connect the Pi 5 HAT to a PC's PCIe slot.

geerlingguy, 2024-08-08:
@tltangliang - it's this one: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/663 (not currently listed in their Incubation projects :(

tltangliang, 2024-08-09:
Great minds think alike.

geerlingguy, 2024-11-12:
@tltangliang - Do you have any plans on making these available for sale somewhere like on Seeed Studios, or some other place where they might hold some stock?

tltangliang, 2024-11-12:
If you think there is a market for it, I can authorize WisdPi to act as the seller, as we have a good relationship.

geerlingguy, 2024-11-12:
@tltangliang - I think the market is pretty small, but at least a few dozen would probably sell in a given year. It'd only be worth it if a batch could be made and put up as kind of an 'addon' for the few people who would want to use it for debug, or to reuse a Pi PCIe HAT inside a PC. Fairly small market, but with the AI HAT in particular—I've already seen someone buy one intending to use it in their PC, only to realize it is direct soldered to a Pi HAT, thus can't plug into any conventional PCIe slot or M.2 slot.

tltangliang, 2024-11-13:
Good and still worth producing, for the sake of research. I've also designed a version to turn m.2

wisdpi, 2024-12-19:
@geerlingguy @tltangliang this board is in stock now.
https://www.wisdpi.com/products/hat-to-pcie-m-2
[image]