geerlingguy/raspberry-pi-pcie-devices #664: GeeekPi P33 M.2 NVME M-Key PoE+ Hat

Blaradox, 2024-08-09:
GeeekPi has made a HAT for the Pi 5 that supports PoE+ and NVME SSDs through PCIe.

[image]

The product page says it also fits the Official Pi 5 Active Cooler.

[Amazon US link](https://a.co/d/4UGPUsZ)

geerlingguy, 2024-09-11:
I have one of these and will be testing it soon! Thanks for posting the link.

geerlingguy, 2024-09-11:
On site here: https://pipci.jeffgeerling.com/hats/geeekpi-p33-m2-nvme-poe-hat.html

Blaradox, 2024-09-11:
Oh fantastic! 

Very excited to see this compared to the Hacker Gadgets PoE + NVMe HAT. Would love to setup a cluster with these.

Just had the sd card on my pi 4 die on me while away from home, thank goodness wireguard runs in RAM!

geerlingguy, 2024-09-17:
Benchmark with PCIe Gen 3 enabled on a MakerDisk 512GB 2242 M-Key M.2 SSD:

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 60.87 MB/s |
| iozone 4K random write     | 270.44 MB/s |
| iozone 1M random read      | 813.15 MB/s |
| iozone 1M random write     | 758.80 MB/s |
| iozone 1M sequential read  | 815.62 MB/s |
| iozone 1M sequential write | 756.78 MB/s |

geerlingguy, 2024-09-17:
A few notes from use:

  - Plugging in the FFC was a little challenging (I had to use my little precision tweezers to get it to seat in on the Pi)
  - There are only power and activity LEDs on the board, there's no separate indication of PoE power
  - I ran into a bug with shutdown behavior: when I shut down the Pi in software, the PoE HAT seemed to reset its power line, causing the Pi to immediately power back on. Same thing happens if I press the power button on the Pi.
  - Temps are good, hottest IC is further from the fan but only got up to 50°C
  - I didn't hear any coil whine, at least not with my ear a couple inches away (yay!).
  - The way the transformer is mounted on top makes this solution a bit taller than HackerGadgets'

dirtydevver, 2024-09-20:
if a network lead is not plugged in is it still possible to power it using USB, i noticed that the instructions say never to use usb power, but would have assumed that only means if there is a network lead plugged in AND it has POE being delivered? i want to use the device but to try and avoid bugs, such as the one mentioned above around powering off and it comes back on

eclevers, 2024-10-08:
Does any one know if it fits in a specific case ?

tcdude, 2024-10-24:
I'm wondering the same as @dirtydevver.. would it suffice to leave out the 4 Pin connector to still use the USB power supply? or even leave out both through pins to err on the safe side? does the NVME get enough power through the ribbon cable?
Just got mine and I don't want to summon the magic white smoke (and also am too lazy to move it to a PoE capable switch just now...)

tehcyx, 2024-10-29:
I just ordered the HAT with this case: https://www.amazon.com/dp/B0CMZG2R73 and it fits @eclevers 

Now it's sitting on my desk waiting to get an SD card with an OS, so I can move that to SSD and boot without SD.

Velkas, 2024-11-16:
I picked up this hat for a new Pi 5 8gb I got to introduce into my home lab. So far, I'm having trouble getting the actual disk to work. I have the Pi 5 powered by a PoE+ switch and the Hat's power light does turn on. I followed the instruction booklet and the post about booting from NVMe on Jeff's website, which ultimately was the same steps overall. I just can't get the NVMe storage to show up at all so I can't progress towards copying over the SD card to then boot from directly off the drive.

Is this the place to ask for some help? Hard to find many other resources.

Steps taken:
- Fresh image of raspberry pi 64-bit lite software written to the sd card.
- Added the boot config option `dtparam=pciex1` under the [all] category.
- Edited the eeprom settings to include `PCIE_PROBE=1` and `BOOT_ORDER=0xf416` per the instructions.
- Reboot and run lsblk but only the mmc device is showing up.

geerlingguy, 2024-11-17:
@Velkas - This isn't an official support forum, but regarding your drive (SN850x), I wonder if it draws too much power... some do. I haven't tested any WD_Black NVMe SSDs except for I think one of the 700 series, and that one worked. But some drive controllers don't seem to play nice with Pis, or they require too much power for some boards that don't expect a drive to pull more than X watts (usually more than 5W).

Velkas, 2024-11-17:
Thanks @geerlingguy! That must have been it. I grabbed a spare Samsung 980 Pro and it recognized it just fine, speed seems expected and so far so stable. Appreciate the insight!

ghostersk, 2025-01-18:
Hi all, after watching your video few months back I order it, but I just find the time to try it. It works fine, seems like it. It even stays Off when i shut down the Pi and it using POE.
The case I bought is not good fit, I cannot close the top.
Does anyone have some good case for this Hat with Pi5?

theoriginalgiga, 2025-01-29:
> Hi all, after watching your video few months back I order it, but I just find the time to try it. It works fine, seems like it. It even stays Off when i shut down the Pi and it using POE. The case I [...]

I can confirm the Ri5 and the hat fit into this case though it is a touch snug front to back due to the hat board being just a touch longer.
https://www.amazon.com/dp/B0CLFYDT8Y

ghostdevv, 2025-04-05:
Results for a [`Crucial P3 Plus 1TB PCIe Gen4 2280`](https://www.amazon.co.uk/dp/B0BYWB6237)

### Script

I grabbed [Jeff's benchmark script](https://github.com/geerlingguy/pi-cluster/blob/6e443778bc16d489c6f8049ae5e1e2624e939289/benchmarks/disk-benchmark.sh) and ran it directly without the python layer, like so:

```sh
export MOUNT_PATH="/"
export TEST_SIZE="1g"

./disk-benchmark.sh
```

### PCIe Gen 2

`dtparam=pciex1`

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 71.42 MB/s |
| iozone 4K random write     | 214.83 MB/s |
| iozone 1M random read      | 441.99 MB/s |
| iozone 1M random write     | 431.29 MB/s |
| iozone 1M sequential read  | 442.15 MB/s |
| iozone 1M sequential write | 429.62 MB/s |

### PCIe Gen 3

`dtparam=pciex1_gen=3`

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 79.51 MB/s |
| iozone 4K random write     | 310.90 MB/s |
| iozone 1M random read      | 857.95 MB/s |
| iozone 1M random write     | 823.17 MB/s |
| iozone 1M sequential read  | 858.27 MB/s |
| iozone 1M sequential write | 831.53 MB/s |

adhdtiger, 2025-06-04:
In the absence of my ability to find a place for actual support or info, I've stumbled into here.  I just put one of these together and it seems as though I can't get enough power through the hat to boot the Pi.  Anyone else have similar issues?

 For reference, I've had this pi on normal power with the same Samsung 990 2T and it works fine.  I put this together and put it on a netgear GS308EPP with the port power set on unlimited and 802.3at .  The hat light comes on but the pi never boots. I have previously been using a Pi4 w/ PoE+ hat on this switch so I know it at least has some functionality. I have another kit for my other Pi5 but haven't had time to swap out parts.

Anyway, thought I'd ask just in case.  Thanks!!

DavidXanatos, 2025-07-27:
@geerlingguy do you know how to fix the reset on power off issue?

fabio-rigato, 2025-12-24:
thank you @dmillertime for your answer!
From my understanding Patriot P320 NVMe drive it is a DRAM-less NVMe :( 
The main reason why I need an NVMe with DRAM is because I will need that node to work as a db in my k3s cluster.
From my research, KIOXIA Exceria G2 could be a good fit but I was wandering if any one of you had different point of view? :)