geerlingguy/raspberry-pi-pcie-devices #670: Intel BE200NGW WiFi 7 M.2 Card

geerlingguy, 2024-09-11:
Intel's [BE200NGW WiFi 7 M.2 Card](https://amzn.to/3B02v0r) is an inexpensive gateway into fast up-to-5 Gbps WiFi, using the 360 MHz channel width afforded on the 6 GHz band...

[image]

I have ordered a card and hope to test it soon!

geerlingguy, 2024-09-11:
Note: A few other users, like @thewade and @wdebowski, have been testing the BE200 on Pi 5 already.

See: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/120#issuecomment-2059957637

geerlingguy, 2024-10-05:
@thewade [got it working on Ubuntu 24.04](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/120#issuecomment-2059957637), and `iwlwifi` is included out of the box. (So no need for a kernel recompile.)

But on Pi OS, I don't think it is (I can check). If not, have to recompile the kernel with `iwlwifi` enabled:

```
 Symbol: IWLWIFI [=n]                                                                                            │   
  │ Type  : tristate                                                                                                │   
  │ Prompt: Intel Wireless WiFi Next Gen AGN - Wireless-N/Advanced-N/Ultimate-N (iwlwifi)                           │   
  │   Location:                                                                                                     │   
  │     -> Device Drivers                                                                                           │   
  │       -> Network device support (NETDEVICES [=y])                                                               │   
  │         -> Wireless LAN (WLAN [=y])                                                                             │   
  │ (1)       -> Intel devices (WLAN_VENDOR_INTEL [=y])                                                             │   
  │   Defined at drivers/net/wireless/intel/iwlwifi/Kconfig:2                                                       │   
  │   Depends on: NETDEVICES [=y] && WLAN [=y] && WLAN_VENDOR_INTEL [=y] && PCI [=y] && HAS_IOMEM [=y] && CFG80211  │   
  │   Selects: FW_LOADER [=y] 
```

Then you can check `dmesg` with the WiFi chip connected (`dmesg | grep iwlwifi`, it should spit out what firmware it's looking for—I think. Otherwise glance inside [the git tree for linux-firmware](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/) and find the right files.

For example, if you need version `89`:

```
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/iwlwifi-ty-a0-gf-a0-89.ucode
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/iwlwifi-ty-a0-gf-a0.pnvm
```

Stash those two files inside `/lib/firmware` and reboot.

thewade, 2024-10-05:
I saw your live stream and you mentioned you might want to use the Intel chip in access point mode with 6GHz.

Unfortunately, that might prove to be tricky due to the way regulatory works with Intel cards. In order to determine if you are in a regulatory region that supports the 6GHz band it scans nearby networks to pull the country info from them. The term they use for this is Location Aware Regulatory.

When using the chip in access point mode however it cannot do the scan and the 6GHz band remains locked.

Someone proposed a patch that allows for the scan but it hasn’t been accepted:
http://lists.infradead.org/pipermail/hostap/2024-July/042824.html

Even with the fix it isn’t great solution, cause if there are no nearby wireless networks broadcasting the country info then it will never determine the regulatory region.

I have never tried to build an OpenWRT access point myself. The OpenWRT community might have other tricks and hacks to get around this.

geerlingguy, 2024-12-02:
@BennyE posted some info for getting the card up and running in #690, and the card is now up on the site: https://pipci.jeffgeerling.com/cards_network/intel-be200-wifi-7.html

Archfx, 2025-02-23:
I was able to get the most out of BE200NGW with CM4 on Debian. I am using [RaspAP](https://github.com/RaspAP/raspap-webgui) for the router functionalities. So far so good including the Wifi repeater mode (WiFi Client mode and AP mode at once).

I have documented my process  [here](https://archfx.me/posts/2025/02/router/) in my blog if anyone is curious.

geerlingguy, 2025-03-03:
I've also noticed over in #709 that the iwlwifi module is now included in Pi OS by default, at least as of January this year, so no need for a kernel recompile, just need the firmware!

geerlingguy, 2025-03-10:
I've updated the BE200 page with instructions for getting it going more easily with the latest version of Pi OS: https://pipci.jeffgeerling.com/cards_network/intel-be200-wifi-7.html

geerlingguy, 2025-03-11:
FYI, this card _apparently_ supports PCIe Gen 4x1 speeds. I benchmarked my connection on Gen 2x1 (5 GT/s, in linked issue on the Waveshare card, at 1.2-1.4 Gbps), and then again with PCIe Gen 3 (via `dtparam=pciex1_gen=3` in boot config)... and now I'm touching on 2 Gbps!

```
$ iperf3 -c 10.0.2.15
Connecting to host 10.0.2.15, port 5201
[  5] local 10.0.2.245 port 33668 connected to 10.0.2.15 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   194 MBytes  1.63 Gbits/sec    0   3.13 MBytes       
[  5]   1.00-2.00   sec   219 MBytes  1.84 Gbits/sec    0   3.13 MBytes       
[  5]   2.00-3.00   sec   220 MBytes  1.85 Gbits/sec    0   3.32 MBytes       
[  5]   3.00-4.00   sec   220 MBytes  1.85 Gbits/sec    0   3.32 MBytes       
[  5]   4.00-5.00   sec   211 MBytes  1.77 Gbits/sec    0   3.32 MBytes       
[  5]   5.00-6.00   sec   206 MBytes  1.73 Gbits/sec    0   3.54 MBytes       
[  5]   6.00-7.00   sec   219 MBytes  1.84 Gbits/sec    0   3.54 MBytes       
[  5]   7.00-8.00   sec   220 MBytes  1.85 Gbits/sec    0   3.54 MBytes       
[  5]   8.00-9.00   sec   238 MBytes  1.99 Gbits/sec    0   3.54 MBytes       
[  5]   9.00-10.00  sec   238 MBytes  1.99 Gbits/sec    0   3.54 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.13 GBytes  1.83 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  2.13 GBytes  1.83 Gbits/sec                  receiver
```

Even bidirectional gets consistently over 1 Gbps with PCIe Gen 3 speeds:

```
$ iperf3 -c 10.0.2.15 --bidir
Connecting to host 10.0.2.15, port 5201
[  5] local 10.0.2.245 port 33504 connected to 10.0.2.15 port 5201
[  7] local 10.0.2.245 port 33512 connected to 10.0.2.15 port 5201
[ ID][Role] Interval           Transfer     Bitrate         Retr  Cwnd
[  5][TX-C]   0.00-1.00   sec   171 MBytes  1.44 Gbits/sec    0   2.79 MBytes       
[  7][RX-C]   0.00-1.00   sec  18.0 MBytes   151 Mbits/sec                  
[  5][TX-C]   1.00-2.00   sec   180 MBytes  1.51 Gbits/sec    0   2.97 MBytes       
[  7][RX-C]   1.00-2.00   sec  35.3 MBytes   296 Mbits/sec                  
[  5][TX-C]   2.00-3.00   sec   124 MBytes  1.04 Gbits/sec    0   2.97 MBytes       
[  7][RX-C]   2.00-3.00   sec  57.1 MBytes   480 Mbits/sec                  
[  5][TX-C]   3.00-4.00   sec   104 MBytes   870 Mbits/sec    0   2.97 MBytes       
[  7][RX-C]   3.00-4.00   sec  72.3 MBytes   607 Mbits/sec                  
[  5][TX-C]   4.00-5.00   sec   149 MBytes  1.25 Gbits/sec    0   3.37 MBytes       
[  7][RX-C]   4.00-5.00   sec  59.6 MBytes   500 Mbits/sec                  
[  5][TX-C]   5.00-6.00   sec   168 MBytes  1.41 Gbits/sec    0   3.37 MBytes       
[  7][RX-C]   5.00-6.00   sec  56.1 MBytes   471 Mbits/sec                  
[  5][TX-C]   6.00-7.00   sec   156 MBytes  1.31 Gbits/sec    0   3.37 MBytes       
[  7][RX-C]   6.00-7.00   sec  65.6 MBytes   550 Mbits/sec                  
[  5][TX-C]   7.00-8.00   sec   161 MBytes  1.35 Gbits/sec    0   3.37 MBytes       
[  7][RX-C]   7.00-8.00   sec  62.7 MBytes   526 Mbits/sec                  
[  5][TX-C]   8.00-9.00   sec   156 MBytes  1.31 Gbits/sec    0   3.37 MBytes       
[  7][RX-C]   8.00-9.00   sec  65.2 MBytes   547 Mbits/sec                  
[  5][TX-C]   9.00-10.00  sec   155 MBytes  1.30 Gbits/sec    0   3.37 MBytes       
[  7][RX-C]   9.00-10.00  sec  69.5 MBytes   583 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID][Role] Interval           Transfer     Bitrate         Retr
[  5][TX-C]   0.00-10.00  sec  1.49 GBytes  1.28 Gbits/sec    0             sender
[  5][TX-C]   0.00-10.00  sec  1.49 GBytes  1.28 Gbits/sec                  receiver
[  7][RX-C]   0.00-10.00  sec   563 MBytes   472 Mbits/sec                  sender
[  7][RX-C]   0.00-10.00  sec   562 MBytes   471 Mbits/sec                  receiver

iperf Done.
```

Very impressive!

geerlingguy, 2025-03-11:
Checking monitor mode at [Ryan Ziolko's suggestion](https://bsky.app/profile/mediumfidelity.bsky.social/post/3lk536wi67s2v):

```
# ip link set wlan0 down
# iw wlan0 set type monitor
# ip link set wlan0 up

# iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0 (self-managed)
country US: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(5170 - 5190 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS
	(5190 - 5210 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS
	(5210 - 5230 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS
	(5230 - 5250 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS
	(5250 - 5270 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5270 - 5290 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5290 - 5310 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5310 - 5330 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5490 - 5510 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5510 - 5530 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5530 - 5550 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5550 - 5570 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5570 - 5590 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5590 - 5610 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5610 - 5630 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
	(5630 - 5650 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
	(5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
	(5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
	(5735 - 5755 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS
	(5755 - 5775 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS
	(5775 - 5795 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS
	(5795 - 5815 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS
	(5815 - 5835 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS
	(5835 - 5855 @ 160), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(5855 - 5875 @ 160), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(5875 - 5895 @ 160), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(5945 - 5965 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(5965 - 5985 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(5985 - 6005 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(6005 - 6025 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(6025 - 6045 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(6045 - 6065 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(6065 - 6085 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(6085 - 6105 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(6105 - 6125 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(6125 - 6145 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(6145 - 6165 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(6165 - 6185 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(6185 - 6205 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
	(6205 - 6225 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
	(6225 - 6245 @ 1180), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
[... 902 lines cut ...]
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ MU_MIMO_AIR_SNIFFER ]: MU-MIMO sniffer
		* [ SCAN_START_TIME ]: scan start timestamp
		* [ BSS_PARENT_TSF ]: BSS last beacon/probe TSF
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ FILS_MAX_CHANNEL_TIME ]: FILS max channel attribute override with dwell time
		* [ ACCEPT_BCAST_PROBE_RESP ]: accepts broadcast probe response
		* [ OCE_PROBE_REQ_HIGH_TX_RATE ]: probe request TX at high rate (at least 5.5Mbps)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
		* [ BEACON_PROTECTION ]: beacon protection support
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ PROTECTED_TWT ]: protected Target Wake Time (TWT) support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

</details>

geerlingguy, 2025-03-12:
@thewade - It looks like `firmware-iwlwifi` is indeed in the repo, so I will rebuild my Pi and see if that is the easier solution than manually grabbing the firmware files. Thanks!

https://www.raspberryconnect.com/raspbian-packages/76-raspbian-kernel-packages

geerlingguy, 2025-03-12:
Indeed; Bookworm's `iwlwifi-firmware` package doesn't contain the firmware for the BE200 whatsoever:

https://packages.debian.org/bookworm/all/firmware-iwlwifi/filelist

It contains the following versions for the AX210:

```
/lib/firmware/iwlwifi-ty-a0-gf-a0-59.ucode
/lib/firmware/iwlwifi-ty-a0-gf-a0-66.ucode
/lib/firmware/iwlwifi-ty-a0-gf-a0-72.ucode
```

thewade, 2025-03-12:
The version of the iwlwifi firmware that is supported is related to the Linux kernel. When you update the kernel often you need to download new firmware.

For a typical distro I would expect their various firmware packages to align with the supported kernel. It doesn't have to be the exact file either. For iwlwifi you can also have multiple firmware files and it will automatically pick the best version that is supported with the kernel.

You might be able to open a bug against Debian to add the files. For Ubuntu they maintain their own [linux-firmware](https://launchpad.net/ubuntu/+source/linux-firmware) package and in the past I had opened a bug to request they add updated firmware for iwlwifi.

Debian is more strict on open source then Ubuntu, so they might also be less interested in maintaining these packages.

geerlingguy, 2025-03-12:
> For a typical distro I would expect their various firmware packages to align with the supported kernel. It doesn't have to be the exact file either. For iwlwifi you can also have multiple firmware files and it will automatically pick the best version that is supported with the kernel.

Yeah, I had run `sudo apt update && sudo apt upgrade -y` earlier today, so both the kernel and the `firmware-iwlwifi` packages should be in sync. It has version 86, which is newer than the 83, and I presume I might be the first person who attempted a BE200 at least on the Pi 5 with Pi OS 12 and this package?

I'm not sure where the packages are built for Pi OS, I believe it's a separate place than Debian proper (Debian 12 is a bit further behind on the kernel—it's default is 6.1...).

thewade, 2025-03-13:
Regarding performance, you might be able to use Multi-Link Operation (MLO) to improve performance by spreading the connection over multiple bands at once (e.g. 5GHz/6GHz).

From what I read online it should work with Intel and linux kernel 6.11+ and a wireless router that support the feature:
https://community.intel.com/t5/Wireless/Which-special-driver-version-need-to-support-be200-mlo-features/m-p/1619965

Haven't had a chance to test it myself yet.

hbiyik, 2025-09-26:
guys i am using this card daily in 6e network (never had a wifi7 network to test) in even rockchip 6.1 kernel.

Intel provides latest iwlwifi driver here and you can compile as dkms. This way will always get mac80211, cfg80211, and iwlwifi* drivers regardsless of your kernel. That means basically all kernel wireless stack.

the downside is, most likely you wont be able to use another wireless drivers when those are in use.

Here is the backports source:
https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/backport-iwlwifi.git/

here is the arch package (it is basically a script, you modify it or run manually by looking at the compilations steps as reference):
https://github.com/hbiyik/agrrepo/blob/master/dkms/iwlwifi-backport-dkms-git/PKGBUILD

only trick in the kernel is you need to make sure, MAC80211 & CFG80211 is compiled as module (not as part of kernel image), so that backports MAC80211 & CFG80211 dkms, can overrride them.

hope this helps

thewade, 2025-10-21:
@travlytle if you are having trouble with monitor mode some things to try:
- disabling network manager
- setting your wireless country using `sudo raspi-config`
- disabling the onboard wifi interface with `dtoverlay=disable-wifi` in your boot config
- make sure to set the channel width to match your access point

Be sure to check the output of `iw reg get` to make sure it detected the country for your phy. Make sure it also lists the frequencies for the channels you want to scan.

For the Raspberry Pi once you set the country with `raspi-config` I am not sure if you need the workaround for LAR.

For your reference here is my output for creating the monitor and doing the trace:

```
pi@raspberrypi:~ $ sudo -s
root@raspberrypi:/home/pi# systemctl stop NetworkManager
root@raspberrypi:/home/pi# iw dev wlan0 info
Interface wlan0
	ifindex 3
	wdev 0x1
	addr 90:65:84:80:11:af
	type managed
	wiphy 0
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
root@raspberrypi:/home/pi# iw phy0 interface add mon0 type monitor
root@raspberrypi:/home/pi# ip link set wlan0 up
root@raspberrypi:/home/pi# iw dev wlan0 scan > /dev/null
root@raspberrypi:/home/pi# ip link set mon0 up
root@raspberrypi:/home/pi# ip link set wlan0 down
root@raspberrypi:/home/pi# iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0 (self-managed)
country US: DFS-UNSET
	(2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
	(2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
	(2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
	(5170 - 5190 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-320MHZ
	(5190 - 5210 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-320MHZ
	(5210 - 5230 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-320MHZ
	(5230 - 5250 @ 160), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-320MHZ
	(5250 - 5270 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ
	(5270 - 5290 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ
	(5290 - 5310 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ
	(5310 - 5330 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ
	(5490 - 5510 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ
	(5510 - 5530 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ
	(5530 - 5550 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ
	(5550 - 5570 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ
	(5570 - 5590 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ
	(5590 - 5610 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ
	(5610 - 5630 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-320MHZ
	(5630 - 5650 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-320MHZ
	(5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, NO-320MHZ
	(5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, NO-320MHZ
	(5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, NO-320MHZ
	(5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, NO-320MHZ
[... 34 lines cut ...]
0002:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)
0002:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge

root@raspberrypi:/home/pi# ethtool -i wlan0
driver: iwlwifi
version: 6.12.20+rpt-rpi-2712
firmware-version: 92.67ce4588.0 gl-c0-fm-c0-92.uc
expansion-rom-version: 
bus-info: 0001:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

root@raspberrypi:/home/pi# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
root@raspberrypi:/home/pi# 
```

If you continue to have issues, please let me know exactly what you are seeing with the same commands as I did.