geerlingguy/raspberry-pi-pcie-devices #672: Positron CM4 Touchscreen & Wifi head unit.

nomadsgalaxy, 2024-09-13:
Designed for the Positron V3.2, this head unit includes a spot for a CM4 Chip, has built in Wi-fi, a micro SD Card slot, and a USB C port.

https://kb-3d.com/store/positron/1555-ldo-35-controller-touchscreen-for-positron-v32-3d-printer-1725634442949.html

geerlingguy, 2024-09-13:
It looks like it's a cheaper alternative to something like the [Raspberry Pad 5](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/405) from BigTreeTech.

nomadsgalaxy, 2024-09-13:
I plan on making it work with The Rook printer, but yeah, it's essentially the same but with less IO

emilcarr, 2025-08-05:
I’ve ordered one of these to use with my Ender 3 and Octoprint. Unfortunately, I could only find it with shipping from the States -> UK. Will report back once I’ve tried it out.
I’ll need to find someone with a CM4 IO board I can borrow, to flash the eMMC.

emilcarr, 2025-08-15:
So the display panel itself is from waveshare: https://www.waveshare.com/product/3.5inch-dpi-lcd.htm

The USB-C port is missing CC resistors, so can only be powered by a USB A-C cable.