geerlingguy/raspberry-pi-pcie-devices #673: Add GeeekPi N16 Quad M-Key Raspberry Pi 5 M.2 NVMe HAT

geerlingguy, 2024-09-13:
GeeekPi released an updated [N16 Quad M.2 M-Key NVMe SSD HAT for Raspberry Pi 5](https://amzn.to/47y676c).

[image]

This is similar to the [Geekworm X1011](https://pipci.jeffgeerling.com/hats/geekworm-x1011-4-drive-nvme.html), but uses it's own USB-C PD circuit to provide enough power for all the M.2 drives and the Pi, without use of POGO pins. Might be a little more stable for some people if they use higher powered M.2 drives, but there are still some caveats with drive compatibility and overall speed being limited by the ASMedia PCIe Gen 2 switch.

The overall RAID0 speed (maximum you can get with multiple drives on here) is limited to 400-500 MB/sec, versus the 800+ MB/sec you can get with a Gen 3 switch.

crankyoldbugger, 2024-10-01:
This is very interesting, but I have to ask, what power supply would you use here?  They say it can take an input of "9V-20V PD" in what looks like a USB-C port, and they provide a USB cable to run from the M16 to the Pi5, but they don't seem to provide any sort of power supply for the N16 that I can see.

brucetony, 2024-10-27:
Just set mine up and it runs off the official raspberry pi 5 power adapter just fine. GeeekPi includes a USB-C jumper to run between the Pi5 and the HAT, and then the HAT has a separate USB-C for power input. I ended up damaging one of the 2 included PCIe ribbon cables, but other than that, the HAT functions wonderfully and was able to setup a RAIDZ1 pool on the 4 drives I used

crankyoldbugger, 2024-10-27:
Thanks, brucetony, that answers that question.
If you don't mind another question, are you able to get the OS to boot from the NVMe/HAT setup, or do you still need an SD card?

brucetony, 2024-10-27:
> Thanks, brucetony, that answers that question.
> If you don't mind another question, are you able to get the OS to boot from the NVMe/HAT setup, or do you still need an SD card?
> 

I don't see why not. I use a sata SSD as my boot drive (via USB 3), but one of the NVME drives should be able to be used instead

crankyoldbugger, 2024-10-28:
Thanks, again, brucetony.  I'm going to grab one of these things and start experimenting.

int15a, 2024-11-21:
I had intermittent problems with my SSD drives (Silicon Power) just stopping working when using the PCIe Gen 3 config, but going back to Gen2 seemed to have fixed the problem.

masseyy, 2025-01-30:
@int15a How many drives are you using. Do you have all 4 in? I am having issues with 4 nvmes at once. It all gets recognised but as soon as I get IO activity happening on all drives i get controller down will reset messages. Doesn’t matter if Gen 3 is on or off

int15a, 2025-01-30:
@masseyy I have all four in there. I had to reboot the unit from time to time (lost connection to the SSDs) and had to do a hard reset once.

It generally works fine though. I never bothered looking at the logd to see what was happening yet.

masseyy, 2025-01-30:
@int15a Thanks. Would you mind sharing your config and cmdline.txt?And anymore of your setup. 

Are you also using it in Raid?

I wanted this ultimately to have one redundancy disk.

int15a, 2025-01-30:
@masseyy 

cmdline.txt: console=serial0,115200 console=tty1 root=PARTUUID=e17e5358-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=CA

What config are you looking for?

masseyy, 2025-01-30:
@int15a /boot/firmware/config.txt

int15a, 2025-01-30:
@masseyy Right, stupid question!
==============================
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
dtparam=pciex1
dtparam=pciex1_gen=2

dtparam=pciex2
dtparam=pciex2_gen=2

dtparam=pciex3
dtparam=pciex3_gen=2

dtparam=pciex4
dtparam=pciex4_gen=2

masseyy, 2025-01-31:
Thanks @int15a Raid 5 been working fine? Or how do you use the storage?

int15a, 2025-01-31:
@masseyy All SSds are independent, I didn't add any redundancy.

int15a, 2025-02-01:
@masseyy All right, weird timing but I had to sudo reboot the Pi because SSDs 2, 3 and 4 disappeared. Webmin only saw #1.

masseyy, 2025-02-01:
@int15a From my testing and others I have came across online - this board only works "properly" with two NVMEs. 3 or 4 works (if you can even say that) as it doesn't work properly.
I created a new topic on Raspberry Pi Forums - https://forums.raspberrypi.com/viewtopic.php?t=383391

You won't be able to see it until a moderator has approved this for me, you will see until then - "The requested topic does not exist."

masseyy, 2025-02-06:
@int15a How's the drives and board been behaving now?

int15a, 2025-02-08:
@masseyy 

Overall it's pretty good, but I still need to soft reboot the Pi every once in a while. Even had to hard reset once. Here's my drive config:

#1: 1TB
#2: 4TB
#3: 4TB
#4: 4TB

The funny thing is that #1 never has a problem. Not sure what the issue is at the moment.

int15a, 2025-02-19:
@masseyy Yes, I am using the same board.

Does your 100w power supply supports the PD protocol? Here's the last I heard from their support:

"You can replace it with a 65W power supply with PD protocol, which can ensure that both the N16 disk and the Raspberry Pi can be powered well. 

You can refer to our Wiki information page for this product: https://wiki.52pi.com/index.php?title=EP-0180

Our N16 Quad M.2 has been tested with a 65W or 100W power supply, such as this power supply:" (picture attached)

[image]

int15a, 2025-02-19:
Also, I am currently using a 65x PD power supply as per their previous suggestion. It looked that it fixed the issue at first, but I'll give the 100w a go. If I need to remove drives to make it work, I'll have to find a different strategy altogether.

int15a, 2025-02-22:
@masseyy Well, still no luck with the 100w. I tried pulling a drive out to see if that would help, but the Pi doesn't boot if all four aren't there for some reason.

int15a, 2025-02-24:
@masseyy 

**Quick update - Problem Solved!**

Turns out the issue was not with power, but with heat. I'm ashamed I didn't make the connection earlier, but the HAT was functionning properly, until I got the matching metal case.

I took the Pi out of the case and it's been running smoothly for over 48 hours, and counting.

It sounds counterintuitive, but the heatsinks are causing the issue. Because the SSDs are located at the bottom, the heatsinks do not leave enough room for heat dissipation. Plus, they are apparently not that good.

I removed them and put the Pi back in the case. Unless I encounter more issues, I'll consider my case closed.

TLDR; Power is OK, heat was the issue.

geerlingguy, 2025-02-24:
> I removed them and put the Pi back in the case. Unless I encounter more issues, I'll consider my **case closed**.

Nice :D

satmanuk2, 2025-03-02:
just to give my 2 cents, i received one of these N16 Quad boards a few days ago & have encountered the same random controller reset issues when 3 or 4 drives are used at the same time.

- it is not an external power supply issue, i have tried multiple PSU's including the official RPI5 5A PSU & other branded PD PSU's up to 20v 100W) - when measured at the wall peak consumption rarely exceeds 10-12W under load (idle around 6-7w) so no where near the PSU limits.
- it is not a heat issue - it happens casesless & even when the board are not physically attached to each other (laid out side by side & forcibly cooled). it also happens after a cold startup
- tried multiple RPI's (even borrowed the homeassistant PI that has been working without issue for ages)
- tried multiple brands and size of NVME, samsung, WD etc.
- tried my own USBC uplink cable (to rule out the uplink jumper that was supplied)
- tried my own FPC cables
- even put the board behind another pcie switch (52pi double hat) to provide an intentional bottleneck.

im no expert, but it would appear to be a power brownout within the boards own regulator or PD circuitry.

returning it to Amazon for refund

masseyy, 2025-03-02:
@satmanuk2 Sounds about right! Please if you don't mind adding this information to this raspberry pi thread? As they (52pi Support) think I am making this up. Mine was purchased directly from the manufacturer.

This is a design fault with their insufficient 5v 8a regulator being unable to supply power to all. 

https://forums.raspberrypi.com/viewtopic.php?p=2295322

satmanuk2, 2025-03-02:
@masseyy done & waiting moderation.

I also mentioned that the same 4 wide drive config were tested working in a CM3588 & N150 pocket NAS (which run way hotter & are still reliable!). im not 100% sure but think both those have a dedicated 1x lane for each slot, so no asmedia switch in the middle (might be wrong though, i havent had any reason to dig into it, as they just worked).
 
as mentioned in the post, there is no recourse with the amazon fulfilled order for me to open dialogue for replacement etc, only to return it.

jkary, 2025-03-24:
I'm working on designing a case.  I saw some comments above about heat issues.  I currently have my PI/Hat setup with double height hex screws on the NVMe side and single height standoffs between the Pi and N16.

I'm assuming this will give enough room for proper airflow and cooling.  I didn't fully understand the heat issues.  Was the solution to remove the heat sync and fan assembly off the Pi?  Should I design in active cooling?

yoyojacky, 2025-04-01:
I'm not sure why your drives keep having issues on N16. It seems like you might be using Patriot P-series SSDs. [Here are the specs for the 256 GB version](https://www.techpowerup.com/ssd-specs/patriot-p300-256-gb.d636), and [here are the specs for the 512 GB version](https://www.techpowerup.com/ssd-specs/patriot-p300-512-gb.d1024). You can see different controllers and NAND chips across different capacities, which feels quite messy. The website also warns that "Multiple hardware versions found." It seems like this product has poor compatibility. We'll try to check the heat issue and give you some feedback.

yoyojacky, 2025-04-01:
> [@int15a](https://github.com/int15a) Is yours the same board?
> 
> You could try the same as me and test just using two.The issues should go away. See my last few posts.
> 
> They have sent me a replacement but I am still seeing controller resets when using 3 or 4 ssds. Currently using a 100w anker nano power supply and see same issues.

Could you please show me your configuration and steps you have done?

int15a, 2025-04-14:
Ok, so I am still having the issue, roughly once every 2 to 5 days. Not sure about the cause because there is no extra load on the system that I can think of.

Geeekpi's support suggested I try with different drives. So, I decided that I wouldn't replace $700 worth of M.2s to make a $54 board happy.

Instead I took the plunge and pre-ordered a N2. The N1 had good reviews, and the ventilation and ethernet speed "issues" were addressed with the N2.

https://www.kickstarter.com/projects/lincplus/redefining-nas-style-power-and-usability-with-lincplus

On top of being able to re-use my 4 current M.2 drives, there is also two 2.5 inch SATA SSD slots.

naturalv8, 2025-07-23:
does this support Hailo Acceleration Module?

Tuxplayers, 2025-11-12:
> Würde das funktionieren?
>  https://makerworld.com/en/models/477106-raspberry-pi-5-m-2-hat-case?from=search#profileId-388126
> […](#)

Leider Nein  [](https://m.media-amazon.com/images/I/71z11zFI4fL._AC_SY355_.jpg) 👈