geerlingguy/raspberry-pi-pcie-devices #679: Raspberry Pi SSD

geerlingguy, 2024-10-24:
Raspberry Pi is now selling the [Raspberry Pi SSD](https://www.raspberrypi.com/products/ssd/), a 2230-size NVMe SSD rated at PCIe Gen 3 speed.

[image]

I have a 256 GB model to test, and they are also releasing a 512 GB model. The product data sheet doesn't list certain specs like MTBF, TBW, shock ratings, or how many lanes of PCIe Gen 3 are supported (assuming you put it on a device other than a Pi 5).

I would like very much to find those things out, so once I get it hooked up, I'll post the full `lspci` output here.

geerlingguy, 2024-10-24:
On the site here: https://pipci.jeffgeerling.com/cards_m2/raspberry-pi-ssd.html

geerlingguy, 2024-10-24:
lspci output:

```
0000:01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller 980 (prog-if 02 [NVM Express])
	Subsystem: Samsung Electronics Co Ltd NVMe SSD Controller 980 (DRAM-less)
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 38
	Region 0: Memory at 1b80000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/32 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [70] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [b0] MSI-X: Enable+ Count=13 Masked-
		Vector table: BAR=0 offset=00003000
		PBA: BAR=0 offset=00002000
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap+ MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [148 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [158 v1] Power Budgeting <?>
	Capabilities: [168 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [188 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [190 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=26016ns
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: nvme
```
[... 115 lines cut ...]
Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
critical_warning			: 0
temperature				: 25°C (298 Kelvin)
available_spare				: 100%
available_spare_threshold		: 10%
percentage_used				: 0%
endurance group critical warning summary: 0
Data Units Read				: 10 (5.12 MB)
Data Units Written			: 0 (0.00 B)
host_read_commands			: 138
host_write_commands			: 0
controller_busy_time			: 0
power_cycles				: 2
power_on_hours				: 0
unsafe_shutdowns			: 0
media_errors				: 0
num_err_log_entries			: 0
Warning Temperature Time		: 0
Critical Composite Temperature Time	: 0
Temperature Sensor 1           : 25°C (298 Kelvin)
Thermal Management T1 Trans Count	: 0
Thermal Management T2 Trans Count	: 0
Thermal Management T1 Total Time	: 0
Thermal Management T2 Total Time	: 0
```

geerlingguy, 2024-10-24:
Benchmarking... here's a listing on [harddrivebenchmark.com](https://www.harddrivebenchmark.net/hdd.php?hdd=SAMSUNG+MZ9LQ256HBJD-00BVL) for the Samsung.

And on the Pi 5, running my [disk-benchmark.sh](https://raw.githubusercontent.com/geerlingguy/pi-cluster/master/benchmarks/disk-benchmark.sh):

| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 58.76 MB/s |
| iozone 4K random write     | 186.04 MB/s |
| iozone 1M random read      | 748.70 MB/s |
| iozone 1M random write     | 724.94 MB/s |
| iozone 1M sequential read  | 800.15 MB/s |
| iozone 1M sequential write | 728.29 MB/s |

Steve-Tech, 2024-10-26:
Hey Jeff, I was just curious if this supported 4K Native Advanced Format.

You can check using either `nvme id-ns -H /dev/nvme0n1` or `smartctl -c /dev/nvme0n1`, and it'll be down the bottom as LBA Format or Sizes.

For example my WD Black (which does) has this:
```
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0x2 Good
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0x1 Better (in use)
```

leander091, 2024-11-30:
hi @Steve-Tech, 

Have some RPI SSD's myself and it does not like it support 4k blocks as shown below:
`LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
`

Steve-Tech, 2024-11-30:
Hi @leander091, alright, thanks for checking!

ned14, 2025-01-22:
Here is a newly purchased Raspberry Pi 512Gb SSD. It no longer claims to be a Samsung, but is rather is a Biwin SSD. It was purchased from a big brand well known very respectable supplier, and if it is a fake, it's an extremely good one. The packaging and label all look perfect.

I wouldn't personally associate Biwin SSDs nor its "KingSpec" controller with quality like I would with Samsung, but if this is genuine I assume the RaspPi Foundation know what they're doing.

I'll tell you this: power consumption is remarkably low with this SSD. Like, crazy low.

PCIe 2.0:
```
dd if=/dev/zero of=/dev/nvme0n1 bs=1M count=10000 oflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 26.7882 s, 391 MB/s
```
Write load is 0.5 watts. Wow.
```
dd of=/dev/null if=/dev/nvme0n1 bs=1M count=10000 iflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 24.0597 s, 436 MB/s
```
Read load is 0.7 watts.

PCIe 2.0 throughput performance is very competitive.

PCIe 3.0:
```
dd if=/dev/zero of=/dev/nvme0n1 bs=1M count=10000 oflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 14.5891 s, 719 MB/s
```
Write load is 0.9 watts.

```
dd of=/dev/null if=/dev/nvme0n1 bs=1M count=10000 iflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 13.4262 s, 781 MB/s
```
Read load is 0.8 watts.

PCIe 3.0 throughput performance is nothing to be sniffed at. DRAMful SSDs have a bit faster reads.

Other details:
```
0000:01:00.0 Non-Volatile memory controller: Biwin Storage Technology Co., Ltd. KingSpec NX series NVMe SSD (DRAM-less) (rev 01) (prog-if 02 [NVM Express])
        Subsystem: Biwin Storage Technology Co., Ltd. KingSpec NX series NVMe SSD (DRAM-less)
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 41
        Region 0: Memory at 1b00020000 (64-bit, non-prefetchable) [size=16K]
        Expansion ROM at 1b00000000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
                Address: 0000000000000000  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
                        MaxPayload 512 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x4, ASPM L1, Exit Latency L1 <64us
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s (downgraded), Width x1 (downgraded)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 17s to 64s, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
[... 143 lines cut ...]
temperature                             : 51 °C (324 K)
available_spare                         : 100%
available_spare_threshold               : 10%
percentage_used                         : 0%
endurance group critical warning summary: 0
Data Units Read                         : 21 (10.75 MB)
Data Units Written                      : 0 (0.00 B)
host_read_commands                      : 494
host_write_commands                     : 0
controller_busy_time                    : 0
power_cycles                            : 1
power_on_hours                          : 0
unsafe_shutdowns                        : 1
media_errors                            : 0
num_err_log_entries                     : 0
Warning Temperature Time                : 0
Critical Composite Temperature Time     : 0
Temperature Sensor 1           : 59 °C (332 K)
Temperature Sensor 2           : 51 °C (324 K)
Thermal Management T1 Trans Count       : 0
Thermal Management T2 Trans Count       : 0
Thermal Management T1 Total Time        : 0
Thermal Management T2 Total Time        : 0
```
The claimed power use states do seem to very much correlate with empirically measured power consumption.

ned14, 2025-01-22:
`disk_benchmark.sh` for this Biwin based official Raspberry Pi SSD, running on a Raspberry Pi 5:

```
| Benchmark                  | Result |
| -------------------------- | ------ |
| iozone 4K random read      | 57.91 MB/s |
| iozone 4K random write     | 166.35 MB/s |
| iozone 1M random read      | 754.69 MB/s |
| iozone 1M random write     | 677.44 MB/s |
| iozone 1M sequential read  | 753.37 MB/s |
| iozone 1M sequential write | 679.47 MB/s |
```

That is a bit less than the Samsung based SSD.

In case it was a fake and I've been diddlied, I tried filling the drive to full:

```
dd if=/dev/zero of=testfile bs=1M
dd: error writing 'testfile': No space left on device
502905593856 bytes (503 GB, 468 GiB) copied, 2039.26 s, 247 MB/s
```
No i/o errors. Looks true sized. Obviously it ran out of SLC cache, that made it slow down to about one quarter peak write speed. I noticed that power consumption went up to 2.7 watts too, so running out of SLC cache does increase write power consumption considerably.
```
smartctl -a /dev/nvme0
smartctl 7.4 2023-08-01 r5530 [aarch64-linux-6.8.0-1017-raspi] (local build)
Copyright (C) 2002-23, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       BIWIN CE430T5D100-512G
Serial Number:                      2446143801757
Firmware Version:                   1.4.7.67
PCI Vendor/Subsystem ID:            0x1dee
IEEE OUI Identifier:                0x50c68e
Total NVM Capacity:                 512,110,190,592 [512 GB]
Unallocated NVM Capacity:           0
Controller ID:                      0
NVMe Version:                       1.4
Number of Namespaces:               1
Namespace 1 Size/Capacity:          512,110,190,592 [512 GB]
Namespace 1 Formatted LBA Size:     512
Local Time is:                      Wed Jan 22 01:12:36 2025 GMT
Firmware Updates (0x18):            4 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x0014):     DS_Mngmt Sav/Sel_Feat
Log Page Attributes (0x02):         Cmd_Eff_Lg
Maximum Data Transfer Size:         32 Pages
Warning  Comp. Temp. Threshold:     85 Celsius
Critical Comp. Temp. Threshold:     87 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     3.00W       -        -    0  0  0  0      100     600
 1 +     2.80W       -        -    1  1  1  1      150     700
 2 +     2.70W       -        -    2  2  2  2      200    1000
 3 -   0.2100W       -        -    3  3  3  3     1000   13000
 4 -   0.0090W       -        -    4  4  4  4     2000   19000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         3

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02)
Critical Warning:                   0x00
Temperature:                        43 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    0%
Data Units Read:                    115,054 [58.9 GB]
Data Units Written:                 1,152,533 [590 GB]
Host Read Commands:                 1,212,961
Host Write Commands:                5,279,038
Controller Busy Time:               35
Power Cycles:                       2
Power On Hours:                     1
Unsafe Shutdowns:                   1
Media and Data Integrity Errors:    0
Error Information Log Entries:      0
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               61 Celsius
Temperature Sensor 2:               43 Celsius

Error Information (NVMe Log 0x01, 16 of 64 entries)
No Errors Logged

Self-test Log (NVMe Log 0x06)
Self-test status: No self-test in progress
No Self-tests Logged
```
That also agrees that the entire drive was filled.

I see from searching the web that the Biwin-made drives appear to have become the default OEM for official RaspPi SSDs from Nov-Dec last year onwards. Multiple people report the same. I guess it's legit so.

geerlingguy, 2025-01-23:
@ned14 - Before raising the visibility further, I'd like to hear back in the pi/linux repo, see if it might be something overlooked in the kernel, or something that could be fixed elsewhere.

Not supporting TRIM is certainly not a wonderful 'feature' of the new flash controller...

The official Pi page for the SSD (and the Datasheet) don't mention TRIM support at all, but it'd be a bit of a cop-out to say that means they don't have to support it—because the initial batch of Samsung drives _did_... and online databases like [This Techpowerup page for the Raspberry Pi SSD](https://www.techpowerup.com/ssd-specs/raspberry-pi-ssd-256-gb.d2226) also include that in the spec (because that's what everyone had access to when the disks were first introduced).

So I would definitely like to 'raise a little hell', so to speak, if people who buy newer Pi SSDs are getting an objectively worse piece of hardware than the original that I tested. If the performance were a couple percentage points different, or thermals were slightly different, that's one thing. But TRIM is a pretty essential feature for flash! I even [wrote a blog post on it for Pi back in 2020](https://www.jeffgeerling.com/blog/2020/enabling-trim-on-external-ssd-on-raspberry-pi)!

ned14, 2025-01-24:
> nvme list uses the Identify Namespace command to extract this information. Depending on the namespace feature THINP this value can be static and equal to the namespace size:
> NVMe specification 1.4c s6.1.7:
> 
> When the THINP bit in the NSFEAT field is cleared to ‘0’, the controller:
> • shall report a value in the Namespace Capacity field that is equal to the Namespace Size; and
> • may report a value in the Namespace Utilization field that is always equal to the value in the Namespace Capacity field.
>
> Deallocate support is mandatory, and as your testing shows, causes data to be rendered unrecoverable.

That above is the official response.

I guess somebody ought to do some write perf benchmarking to see if things slow right down once the disc reports full. If they do, then Biwin appears to have implemented TRIM as `memset(0)`. Fabulous.