geerlingguy/raspberry-pi-pcie-devices #681: Add i226 TimeNIC V1

geerlingguy, 2024-11-08:
A new addition to the TimeCard lineup, the [TimeNIC V1](https://github.com/opencomputeproject/Time-Appliance-Project/tree/master/Incubation/Hardware/TimeNIC) is a tiny (and cute) Intel i226 PCI NIC with SDP (Session Description Protocol) input and output (via u.FL), and external SDP input and output via SMA connectors:

[image]

I plan on testing this NIC on a Pi 5 and seeing how easy it is to get the Intel NIC to operate at 2.5 Gbps with PTP timing as well...

proteus-cpi, 2025-01-02:
The ECAD.7z seems to be password protected!

geerlingguy, 2025-01-02:
I'll ask about that... maybe that was accidental?

ahmadexp, 2025-01-02:
> The ECAD.7z seems to be password protected!

There is a license related issue with one of the parts on the PCB and we are waiting for that to be resolved. Adding @julian to this as he is dealing with it.