geerlingguy/raspberry-pi-pcie-devices #684: Test AMD RX 7600 GPU

geerlingguy, 2024-11-18:
Following on my testing with the AMD Radeon Pro W7700, I would like to see if an [ASRock Challenger AMD Radeon RX 7600 8GB](https://amzn.to/3OboSn4) runs on the Pi 5.

TODO: Photo here.

geerlingguy, 2024-11-18:
Got an Oops! @Coreforge hunts these down like it's deer season.

```
[    4.688899] amdgpu 0000:03:00.0: amdgpu: RAS: optional ras ta ucode is not available
[    4.705344] amdgpu 0000:03:00.0: amdgpu: RAP: optional rap ta ucode is not available
[    4.705357] amdgpu 0000:03:00.0: amdgpu: SECUREDISPLAY: securedisplay ta ucode is not available
[    4.705405] amdgpu 0000:03:00.0: amdgpu: smu driver if version = 0x00000035, smu fw if version = 0x0000003c, smu fw program = 0, smu fw version = 0x00524500 (82.69.0)
[    4.705408] amdgpu 0000:03:00.0: amdgpu: SMU driver if version not matched
[    4.745485] amdgpu 0000:03:00.0: amdgpu: SMU is initialized successfully!
[    4.745963] [drm] Display Core v3.2.247 initialized on DCN 3.2.1
[    4.745968] [drm] DP-HDMI FRL PCON supported
[    4.748579] [drm] DMUB hardware initialized: version=0x07001900
[    5.009654] [drm] kiq ring mec 3 pipe 1 q 0
[    5.014614] [drm] VCN decode and encode initialized successfully(under DPG Mode).
[    5.015078] amdgpu 0000:03:00.0: [drm:jpeg_v4_0_hw_init [amdgpu]] JPEG decode initialized successfully.
[    5.016362] amdgpu 0000:03:00.0: amdgpu: SE 2, SH per SE 2, CU per SH 8, active_cu_number 32
[    5.016803] amdgpu 0000:03:00.0: amdgpu: ring gfx_0.0.0 uses VM inv eng 0 on hub 0
[    5.016806] amdgpu 0000:03:00.0: amdgpu: ring comp_1.0.0 uses VM inv eng 1 on hub 0
[    5.016808] amdgpu 0000:03:00.0: amdgpu: ring comp_1.1.0 uses VM inv eng 4 on hub 0
[    5.016809] amdgpu 0000:03:00.0: amdgpu: ring comp_1.2.0 uses VM inv eng 6 on hub 0
[    5.016811] amdgpu 0000:03:00.0: amdgpu: ring comp_1.3.0 uses VM inv eng 7 on hub 0
[    5.016812] amdgpu 0000:03:00.0: amdgpu: ring comp_1.0.1 uses VM inv eng 8 on hub 0
[    5.016814] amdgpu 0000:03:00.0: amdgpu: ring comp_1.1.1 uses VM inv eng 9 on hub 0
[    5.016816] amdgpu 0000:03:00.0: amdgpu: ring comp_1.2.1 uses VM inv eng 10 on hub 0
[    5.016817] amdgpu 0000:03:00.0: amdgpu: ring comp_1.3.1 uses VM inv eng 11 on hub 0
[    5.016819] amdgpu 0000:03:00.0: amdgpu: ring sdma0 uses VM inv eng 12 on hub 0
[    5.016820] amdgpu 0000:03:00.0: amdgpu: ring sdma1 uses VM inv eng 13 on hub 0
[    5.016822] amdgpu 0000:03:00.0: amdgpu: ring vcn_unified_0 uses VM inv eng 0 on hub 8
[    5.016823] amdgpu 0000:03:00.0: amdgpu: ring jpeg_dec uses VM inv eng 1 on hub 8
[    5.016825] amdgpu 0000:03:00.0: amdgpu: ring mes_kiq_3.1.0 uses VM inv eng 14 on hub 0
[    5.017964] Unable to handle kernel paging request at virtual address ffffffc08208d000
[    5.026457] Mem abort info:
[    5.029409]   ESR = 0x0000000096000061
[    5.029412]   EC = 0x25: DABT (current EL), IL = 32 bits
[    5.029414]   SET = 0, FnV = 0
[    5.029415]   EA = 0, S1PTW = 0
[    5.029416]   FSC = 0x21: alignment fault
[    5.029417] Data abort info:
[    5.029417]   ISV = 0, ISS = 0x00000061, ISS2 = 0x00000000
[    5.029418]   CM = 0, WnR = 1, TnD = 0, TagAccess = 0
[    5.029420]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    5.029421] swapper pgtable: 4k pages, 39-bit VAs, pgdp=000000000137f000
[    5.029423] [ffffffc08208d000] pgd=1000000100165003, p4d=1000000100165003, pud=1000000100165003, pmd=100000010f037003, pte=00680001029a0f0f
[    5.029428] Internal error: Oops: 0000000096000061 [#1] PREEMPT SMP
[    5.029431] Modules linked in: bnep binfmt_misc spidev brcmfmac_wcc amdgpu(+) hci_uart btbcm aes_ce_blk aes_ce_cipher ghash_ce gf128mul bluetooth sha2_ce sha256_arm64 sha1_ce ecdh_generic vc4 ecc raspberrypi_hwmon libaes rpivid_hevc(C) pisp_be v4l2_mem2mem brcmfmac snd_soc_hdmi_codec videobuf2_dma_contig snd_usb_audio brcmutil cec i2c_brcmstb drm_dma_helper spi_bcm2835 cfg80211 snd_soc_core amdxcp snd_hwdep snd_usbmidi_lib drm_exec snd_compress videobuf2_memops i2c_algo_bit snd_rawmidi drm_buddy videobuf2_v4l2 snd_pcm_dmaengine drm_suballoc_helper snd_seq_device v3d drm_display_helper gpu_sched gpio_keys videodev snd_pcm rfkill drm_shmem_helper drm_ttm_helper snd_timer ttm videobuf2_common pwm_fan drm_kms_helper hid_apple mc snd raspberrypi_gpiomem rp1_adc nvmem_rmem joydev hid_multitouch sg uio_pdrv_genirq uio uinput cuse i2c_dev ledtrig_pattern drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    5.029491] CPU: 1 PID: 336 Comm: (udev-worker) Tainted: G         C         6.6.60-v8-AMDGPU+ #1
[    5.029494] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    5.029496] pstate: 40400009 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    5.029498] pc : __memset+0x16c/0x188
[    5.029505] lr : amdgpu_mes_ctx_alloc_meta_data+0x68/0xc0 [amdgpu]
[    5.029865] sp : ffffffc08093b520
[    5.029866] x29: ffffffc08093b520 x28: ffffff8102c904c8 x27: 0000000000000000
[    5.029869] x26: ffffff8102cb67b8 x25: ffffff8102c904c8 x24: ffffff8102cc0000
[    5.029871] x23: 0000000000000001 x22: 0000000000008000 x21: ffffff8102c80000
[    5.029874] x20: 0000000000000000 x19: ffffffc08093b628 x18: ffffffffffffffff
[    5.029876] x17: ffffff81fef97250 x16: ffffffd084ced580 x15: 0000000000000000
[    5.029878] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[    5.029880] x11: 0000000000000000 x10: ffffff81fef855c0 x9 : 0000000000000000
[    5.029883] x8 : ffffffc08208d000 x7 : 0000000000000000 x6 : 000000000000003f
[    5.029885] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 0000000000000004
[    5.029887] x2 : 0000000000033fc0 x1 : 0000000000000000 x0 : ffffffc08208d000
[    5.029890] Call trace:
[    5.029891]  __memset+0x16c/0x188
[... cut ...]
[    5.310353]  load_module+0x1dd0/0x2080
[    5.310354]  __do_sys_init_module+0x19c/0x1e0
[    5.310356]  __arm64_sys_init_module+0x24/0x38
[    5.310357]  invoke_syscall+0x50/0x128
[    5.310360]  el0_svc_common.constprop.0+0xc8/0xf0
[    5.310363]  do_el0_svc+0x24/0x38
[    5.310365]  el0_svc+0x40/0xe8
[    5.310368]  el0t_64_sync_handler+0x100/0x130
[    5.310369]  el0t_64_sync+0x190/0x198
[    5.310372] Code: 91010108 54ffff4a 8b040108 cb050042 (d50b7428) 
[    5.310374] ---[ end trace 0000000000000000 ]---
```

geerlingguy, 2024-11-19:
@DanaGoyette - For decoding the stack traces, I'm trying to use `decode_stacktrace.sh`, but it seems to be failing a little:

```
pi@pi5-pcie:~/linux/scripts $ ./decode_stacktrace.sh /boot/firmware/kernel_2712.img < log.txt
[    5.726629] __memset+0x16c/0x188 
readelf: Error: Not an ELF file - it has the wrong magic bytes at the start
WARNING! Modules path isn't set, but is needed to parse this symbol
[    5.726632] amdgpu_mes_self_test+0xc0/0x450 amdgpu
...
```

I assumed that's my vmlinuz as I compile and copy it with:

```
make -j6 Image.gz modules dtbs
sudo cp arch/arm64/boot/Image.gz /boot/firmware/$KERNEL.img
```

Or am I missing something obvious?

Ah... same thing with the right vmlinux:

```
pi@pi5-pcie:~/linux $ ./scripts/decode_stacktrace.sh /usr/lib/modules/6.6.60-v8-AMDGPU+/build/vmlinux /home/pi/linux < scripts/log.txt
[    5.726629] __memset (??:?) 
readelf: Error: Not an ELF file - it has the wrong magic bytes at the start
WARNING! Modules path isn't set, but is needed to parse this symbol
```

(that dir is just a symlink back to `build -> /home/pi/linux`)

geerlingguy, 2024-11-19:
Ah... looking at https://github.com/torvalds/linux/blob/master/scripts/decode_stacktrace.sh, I noticed it's not unzipping (well, un-`xz`ing) the module files, so it's plain `readelf` calls are erroring out.

```
# Doesn't work
$ readelf -WS /lib/modules/6.6.60-v8-AMDGPU+/kernel/drivers/gpu/drm/amd/amdgpu/amdgpu.ko.xz 
readelf: Error: Not an ELF file - it has the wrong magic bytes at the start

# Works
$ sudo unxz /lib/modules/6.6.60-v8-AMDGPU+/kernel/drivers/gpu/drm/amd/amdgpu/amdgpu.ko.xz
$ readelf -WS /lib/modules/6.6.60-v8-AMDGPU+/kernel/drivers/gpu/drm/amd/amdgpu/amdgpu.ko
There are 63 section headers, starting at offset 0xae1e50:
...
```

Thanks to [this LKML message hinting at that](https://lore.kernel.org/lkml/CALTww28Fz5Y11q1UE8r6qRrwOHgKOWw=iX3AisoSFGh_VhAbJA@mail.gmail.com/T/).

Module compression is a compile-time option, and my Kernel was configured with `MODULE_COMPRESS_XZ`. In `menuconfig` I changed it to `Enable loadable module support` > `Module compression mode` > `None` and will recompile.

geerlingguy, 2024-11-19:
Hmm... closer this time, but still no dice:

```
pi@pi5-pcie:~/linux $ ./scripts/decode_stacktrace.sh /home/pi/linux/vmlinux /lib/modules/6.6.60-v8-AMDGPU+ < trace.txt
[    6.527910] __memset (??:?) 
WARNING! Modules path isn't set, but is needed to parse this symbol
[    6.531234] amdgpu_mes_self_test+0xc0/0x450 amdgpu
WARNING! Modules path isn't set, but is needed to parse this symbol
[    6.536646] mes_v11_0_late_init+0x58/0x78 amdgpu
```

Coreforge, 2024-11-20:
I just pushed all of the 7000 series stuff, hopefully this oops should be fixed too.

geerlingguy, 2024-11-20:
@Coreforge - Thanks! I assume it's all in your https://github.com/Coreforge/linux/tree/rpi-6.6.y-gpu branch? (I see latest commit 45 min ago). I will still do some testing for this issue before pulling that code because I'd like to get the 'paste the stack trace, get the line of code' debugging working :D

geerlingguy, 2024-11-20:
That worked!

```
pi@pi5-pcie:~/linux $ ./scripts/decode_stacktrace.sh /home/pi/linux/vmlinux /lib/modules/6.6.60-v8-AMDGPU+ < trace.txt
[    5.542863] Call trace:
[    5.545310] __memset (/home/pi/linux/arch/arm64/lib/memset.S:197) 
[    5.548634] amdgpu_mes_self_test (/home/pi/linux/drivers/gpu/drm/amd/amdgpu/amdgpu_mes.c:1367) amdgpu
[    5.553950] mes_v11_0_late_init (/home/pi/linux/drivers/gpu/drm/amd/amdgpu/mes_v11_0.c:1320) amdgpu
[    5.559071] amdgpu_device_ip_late_init (/home/pi/linux/drivers/gpu/drm/amd/amdgpu/amdgpu_device.c:2650) amdgpu
[    5.564886] amdgpu_device_init (/home/pi/linux/drivers/gpu/drm/amd/amdgpu/amdgpu_device.c:3858) amdgpu
[    5.570262] amdgpu_driver_load_kms (/home/pi/linux/drivers/gpu/drm/amd/amdgpu/amdgpu_kms.c:147) amdgpu
[    5.575724] amdgpu_pci_probe (/home/pi/linux/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c:2145) amdgpu
[    5.580749] local_pci_probe (/home/pi/linux/drivers/pci/pci-driver.c:325) 
[    5.584508] pci_device_probe (/home/pi/linux/drivers/pci/pci-driver.c:392 /home/pi/linux/drivers/pci/pci-driver.c:417 /home/pi/linux/drivers/pci/pci-driver.c:460) 
[    5.588439] really_probe (/home/pi/linux/drivers/base/dd.c:579 /home/pi/linux/drivers/base/dd.c:658) 
[    5.592110] __driver_probe_device (/home/pi/linux/drivers/base/dd.c:800) 
[    5.596478] driver_probe_device (/home/pi/linux/drivers/base/dd.c:830) 
[    5.600671] __driver_attach (/home/pi/linux/drivers/base/dd.c:1217 /home/pi/linux/drivers/base/dd.c:1156) 
[    5.604515] bus_for_each_dev (/home/pi/linux/drivers/base/bus.c:370) 
[    5.608360] driver_attach (/home/pi/linux/drivers/base/dd.c:1234) 
[    5.611942] bus_add_driver (/home/pi/linux/drivers/base/bus.c:675) 
[    5.615699] driver_register (/home/pi/linux/drivers/base/driver.c:246) 
[    5.619544] __pci_register_driver (/home/pi/linux/drivers/pci/pci-driver.c:1463) 
[    5.623824] amdgpu_init (/home/pi/linux/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c:2315) amdgpu
[    5.628327] do_one_initcall (/home/pi/linux/init/main.c:1238) 
[    5.632172] do_init_module (/home/pi/linux/kernel/module/main.c:2535) 
[    5.635931] load_module (/home/pi/linux/kernel/module/main.c:2986) 
[    5.639687] init_module_from_file (/home/pi/linux/kernel/module/main.c:3176) 
[    5.643967] __arm64_sys_finit_module (/home/pi/linux/kernel/module/main.c:3187 /home/pi/linux/kernel/module/main.c:3214 /home/pi/linux/kernel/module/main.c:3197 /home/pi/linux/kernel/module/main.c:3197) 
[    5.648684] invoke_syscall (/home/pi/linux/./arch/arm64/include/asm/current.h:19 /home/pi/linux/arch/arm64/kernel/syscall.c:56) 
[    5.652442] el0_svc_common.constprop.0 (/home/pi/linux/arch/arm64/kernel/syscall.c:141) 
[    5.657160] do_el0_svc (/home/pi/linux/arch/arm64/kernel/syscall.c:154) 
[    5.660481] el0_svc (/home/pi/linux/./arch/arm64/include/asm/daifflags.h:28 /home/pi/linux/arch/arm64/kernel/entry-common.c:133 /home/pi/linux/arch/arm64/kernel/entry-common.c:144 /home/pi/linux/arch/arm64/kernel/entry-common.c:679) 
[    5.663540] el0t_64_sync_handler (/home/pi/linux/arch/arm64/kernel/entry-common.c:697) 
[    5.667908] el0t_64_sync (/home/pi/linux/arch/arm64/kernel/entry.S:599) 
[ 5.671578] Code: 91010108 54ffff4a 8b040108 cb050042 (d50b7428)
All code
========
   0:	91010108 	add	x8, x8, #0x40
   4:	54ffff4a 	b.ge	0xffffffffffffffec  // b.tcont
   8:	8b040108 	add	x8, x8, x4
   c:	cb050042 	sub	x2, x2, x5
  10:*	d50b7428 	dc	zva, x8		<-- trapping instruction

Code starting with the faulting instruction
===========================================
   0:	d50b7428 	dc	zva, x8
[    5.677692] ---[ end trace 0000000000000000 ]---
```

geerlingguy, 2024-11-20:
As an aside, when the driver gets locked up in whatever state it's in, the system won't shut down (for at least 5-10 minutes), and a `sudo modprobe amdgpu` hangs... all the while, the card is sucking down _160W_ of power, with the fans running full blast :)

[image]

geerlingguy, 2024-11-20:
New error:

```
[    8.084752] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    8.090603] pstate: 80400009 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    8.097590] pc : sdma_v6_0_mqd_init+0xe4/0x148 [amdgpu]
[    8.103206] lr : sdma_v6_0_mqd_init+0xcc/0x148 [amdgpu]
[    8.108674] sp : ffffffc080983200
[    8.111994] x29: ffffffc080983200 x28: ffffff8110fc0da0 x27: ffffff81010cba00
[    8.119156] x26: ffffff81010cbb30 x25: ffffff81010cbb38 x24: ffffff81010cbb28
[    8.126318] x23: ffffff8110fc0c00 x22: ffffff81010cbb00 x21: ffffff8110f80000
[    8.133480] x20: ffffffc080983250 x19: ffffffc08106d000 x18: ffffffffffffffff
[    8.140642] x17: 0000000000001fd0 x16: ffffffd084ced580 x15: 0000000000000100
[    8.147804] x14: ffffff81a85f832f x13: 0000000000000000 x12: 7720322e322e3836
[    8.154966] x11: 0000000000000040 x10: ffffff8110fc0de8 x9 : ffffffd064283acc
[    8.162128] x8 : ffffffc080983298 x7 : 0000000000000000 x6 : 0000000000000228
[    8.169290] x5 : ffffff81077f16a8 x4 : 0000000010000000 x3 : 000000000000000f
[    8.176452] x2 : 0000000000004000 x1 : 0000000000000a00 x0 : 0000000000000000
[    8.183614] Call trace:
[    8.186062]  sdma_v6_0_mqd_init+0xe4/0x148 [amdgpu]
[    8.191181]  amdgpu_mes_queue_init_mqd+0xb4/0x190 [amdgpu]
[    8.196908]  amdgpu_mes_add_hw_queue+0x208/0x4e0 [amdgpu]
[    8.202548]  amdgpu_mes_add_ring+0x1ec/0x3a8 [amdgpu]
[    8.207837]  amdgpu_mes_self_test+0x1f8/0x450 [amdgpu]
[    8.213213]  mes_v11_0_late_init+0x58/0x78 [amdgpu]
[    8.218328]  amdgpu_device_ip_late_init+0x68/0x308 [amdgpu]
[    8.224139]  amdgpu_device_init+0x2038/0x2208 [amdgpu]
[    8.229514]  amdgpu_driver_load_kms+0x20/0x1a8 [amdgpu]
[    8.234975]  amdgpu_pci_probe+0x154/0x420 [amdgpu]
[    8.239999]  local_pci_probe+0x48/0xb8
[    8.243758]  pci_device_probe+0xac/0x1c8
[    8.247690]  really_probe+0x150/0x2c0
[    8.251360]  __driver_probe_device+0x80/0x140
[    8.255728]  driver_probe_device+0x44/0x170
[    8.259921]  __driver_attach+0x9c/0x1b0
[    8.263765]  bus_for_each_dev+0x80/0xe8
[    8.267611]  driver_attach+0x2c/0x40
[    8.271194]  bus_add_driver+0xec/0x218
[    8.274950]  driver_register+0x68/0x138
[    8.278795]  __pci_register_driver+0x54/0x68
[    8.283075]  amdgpu_init+0x6c/0xff8 [amdgpu]
[    8.287577]  do_one_initcall+0x60/0x2c0
[    8.291423]  do_init_module+0x60/0x218
[    8.295182]  load_module+0x1dd0/0x2080
[    8.298938]  init_module_from_file+0x8c/0xd8
[    8.303218]  __arm64_sys_finit_module+0x150/0x330
[    8.307935]  invoke_syscall+0x50/0x128
[    8.311693]  el0_svc_common.constprop.0+0xc8/0xf0
[    8.316412]  do_el0_svc+0x24/0x38
[    8.319733]  el0_svc+0x40/0xe8
[    8.322792]  el0t_64_sync_handler+0x100/0x130
[    8.327159]  el0t_64_sync+0x190/0x198
[    8.330830] Code: 52880002 528001e3 b9402a81 52800000 (f803c27f) 
[    8.336945] ---[ end trace 0000000000000000 ]---
```

geerlingguy, 2024-11-21:
Interesting. I recompiled with your latest commit, and I'm getting display output.

But `glmark2-es2` won't run, it starts to open a window then it stops (no `dmesg` errors or anything):

```
pi@pi5-pcie:~ $ DISPLAY=:0 glmark2-es2
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      AMD
    GL_RENDERER:    AMD Radeon RX 7600 (gfx1102, LLVM 15.0.6, DRM 3.54, 6.6.60-v8-AMDGPU+)
    GL_VERSION:     OpenGL ES 3.2 Mesa 23.2.1-1~bpo12+rpt3
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false: FPS: 37 FrameTime: 27.721 ms
=======================================================
                                  glmark2 Score: 36 
=======================================================
```

GravityMark works, and at 1440p windowed, I got a score of `20,809`. ([Result here](https://gravitymark.tellusim.com/report/?id=087cb184e1a05543270e1b10b31be2ec32977296), compared to [W7700 result of `31,982`](https://gravitymark.tellusim.com/report/?id=830b7e896d07b2702012ba48096bf54b9924c602))

geerlingguy, 2025-09-29:
FYI I'm re-testing this card with @yanghaku's patch (see #764), and it is working pretty much flawlessly! Much simpler to set it up this way, maybe it'll be upstreamable!