geerlingguy/raspberry-pi-pcie-devices #686: Distinguish CM compatibility for CM Boards (CM4 vs CM5)

geerlingguy, 2024-11-27:
Right now the site just assumes 'Compute Module 4' for all the boards listed on the [Compute Module Boards and Projects](https://pipci.jeffgeerling.com/boards_cm), but now that the Compute Module 5 is out, I should have some sort of compatibility indication, both on the overall listing, and the Board page layout, to indicate CM4 vs CM5 compatibility.

In the update, I would also need to set CM5 to 'Unknown' for all boards initially, except for those I've confirmed work (e.g. through my video [This tiny computer changes EVERYTHING](https://www.youtube.com/watch?v=X4blR5Ua3S0), or through my blog, e.g. [Home Assistant Yellow](https://www.jeffgeerling.com/blog/2024/home-assistant-yellow-instant-2x-iot-speedup-cm5).

Confirmed working:

  - Home Assistant Yellow
  - Compute Blade (Dev)
  - CM4 IO Board
  - CM5 IO Board (not on site yet)
  - NEC MPi4 MediaPlayer TV Kit
  - Nebra CM4 Flash Stick
  - DeskPi Super6C
  - Axzez Interceptor 2.0
  - Seaberry
  - Waveshare CM4 IO Base

bakadave, 2025-01-05:
Looking forward to seeing if the Taco will end up working with the CM5. Would love faster speeds and the NVMe boot.

nekomatic, 2025-02-23:
I got lost of mixed results from my googling... So could you clarify if the DeskPi Super6C - does Raspberry PI CM5 work correctly with both network and nvme storage on this board? As well as with all the 'Module 1' IOs?

geerlingguy, 2025-02-24:
@nekomatic - AFACT yes, I was able to get network, NVMe, etc. all working on the 3-node Pi 5 cluster I set up on a board. I didn't have more than 3 CM5s at the time I recorded my video testing it.

nekomatic, 2025-02-24:
Thanks @geerlingguy, have you tried this board with a Radxa CM5 by any chance?

ADevAJ, 2025-03-06:
Hi, wondering if you can confirm if the "Waveshare CM4-NANO-B" board + "CM5 Pi" combination produces audio through the 3.5mm jack when played on the Pi? This would be perfect for my setup but I wanted to confirm.

mohit3112, 2025-03-15:
It will be helpful to know if cm4 nano A also works with cm5 , thanks

Fordi, 2025-05-27:
Watching this thread to know when to migrate my Radxa Taco NAS from a CM4.  A 16G/64G CM5 came in the mail today, and I'm itching to make it useful.

If you can feed me test procedures (like, I know plug the board into the board and turn on, but I want to know what other data you'd like to have if I get it booted), I can try them together.

philletourneau, 2025-09-04:
+1 would love to see this compatibility listed on the page!