geerlingguy/raspberry-pi-pcie-devices #692: Test Nvidia RTX A4000 Desktop Workstation GPU

geerlingguy, 2024-12-11:
I have an [Nvidia RTX A4000 Workstation GPU](https://amzn.to/41vfFxQ) that I'm currently testing in a System76 Thelio Astra, running Ubuntu 24.04.

TODO

I would like to test it on the Pi as well, and though I'm not expecting any luck (see #677 as well), who knows!

geerlingguy, 2024-12-11:
On a System76 Thelio Astra system (see https://github.com/geerlingguy/sbc-reviews/issues/53), I can see the card:

```
0004:01:00.0 VGA compatible controller: NVIDIA Corporation GA104GL [RTX A4000] (rev a1) (prog-if 00 [VGA controller])
  Subsystem: NVIDIA Corporation GA104GL [RTX A4000]
  Physical Slot: 1-3
  Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
  Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
  Latency: 0
  Interrupt: pin A routed to IRQ 323
  NUMA node: 0
  IOMMU group: 26
  Region 0: Memory at 20000000 (32-bit, non-prefetchable) [size=16M]
  Region 1: Memory at 280000000000 (64-bit, prefetchable) [size=256M]
  Region 3: Memory at 280010000000 (64-bit, prefetchable) [size=32M]
  Expansion ROM at 21000000 [virtual] [disabled] [size=512K]
  Capabilities: [60] Power Management version 3
    Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
    Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
  Capabilities: [68] MSI: Enable+ Count=1/1 Maskable- 64bit+
    Address: 00000000fffaf040  Data: 0000
  Capabilities: [78] Express (v2) Legacy Endpoint, MSI 00
    DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 <64us
      ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+
    DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
      RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
      MaxPayload 256 bytes, MaxReadReq 512 bytes
    DevSta: CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
    LnkCap: Port #0, Speed 16GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <512ns, L1 <16us
      ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
    LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
      ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
    LnkSta: Speed 2.5GT/s (downgraded), Width x16
      TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
    DevCap2: Completion Timeout: Range AB, TimeoutDis+ NROPrPrP- LTR-
       10BitTagComp+ 10BitTagReq+ OBFF Via message, ExtFmt- EETLPPrefix-
       EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
       FRS-
       AtomicOpsCap: 32bit- 64bit- 128bitCAS-
    DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis+ LTR- 10BitTagReq- OBFF Disabled,
       AtomicOpsCtl: ReqEn-
    LnkCap2: Supported Link Speeds: 2.5-16GT/s, Crosslink- Retimer+ 2Retimers+ DRS-
    LnkCtl2: Target Link Speed: 16GT/s, EnterCompliance- SpeedDis-
       Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
       Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
    LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
       EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
       Retimer- 2Retimers- CrosslinkRes: unsupported
  Capabilities: [b4] Vendor Specific Information: Len=14 <?>
  Capabilities: [100 v1] Virtual Channel
    Caps: LPEVC=0 RefClk=100ns PATEntryBits=1
    Arb:  Fixed- WRR32- WRR64- WRR128-
    Ctrl: ArbSelect=Fixed
    Status: InProgress-
    VC0:  Caps: PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
      Arb:  Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
      Ctrl: Enable+ ID=0 ArbSelect=Fixed TC/VC=01
      Status: NegoPending- InProgress-
  Capabilities: [258 v1] L1 PM Substates
    L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
        PortCommonModeRestoreTime=255us PortTPowerOnTime=10us
    L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
         T_CommonMode=0us LTR1.2_Threshold=0ns
    L1SubCtl2: T_PwrOn=10us
  Capabilities: [128 v1] Power Budgeting <?>
  Capabilities: [420 v2] Advanced Error Reporting
    UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
    UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
    UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
    CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
[... 75 lines cut ...]
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.120                Driver Version: 550.120        CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA RTX A4000               Off |   00000004:01:00.0  On |                  Off |
| 41%   43C    P8             13W /  140W |     180MiB /  16376MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
[... cut ...]
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      8149      G   /usr/bin/gnome-shell                          150MiB |
|    0   N/A  N/A     11793      G   /usr/bin/Xwayland                              20MiB |
+-----------------------------------------------------------------------------------------+
```

`nvtop` shows the card appropriately.

TODO

geerlingguy, 2024-12-11:
Card is on the site here: https://pipci.jeffgeerling.com/cards_gpu/nvidia-rtx-a4000.html

geerlingguy, 2025-09-30:
On a Pi 500+ I tested again on kernel `rpi-6.15.y` with the yanghaku patch and `nouveau` (I blacklisted it then ran `sudo modprobe nouveau` after rebooting with the card attached).

I get the following errors:

```
[   46.089143] nouveau 0001:01:00.0: enabling device (0000 -> 0002)
[   46.089176] nouveau 0001:01:00.0: NVIDIA GA104 (b74000a1)
[   46.234439] nouveau 0001:01:00.0: bios: version 94.04.57.00.08
[   46.235150] ------------[ cut here ]------------
[   46.235159] nouveau 0001:01:00.0: swiotlb addr 0x00000010fce00000+4096 overflow (mask ffffffff, bus limit ffffffffff).
[   46.235177] WARNING: CPU: 3 PID: 949 at kernel/dma/swiotlb.c:1594 swiotlb_map+0x2c0/0x2f0
[   46.235193] Modules linked in: nouveau(+) drm_gpuvm i2c_algo_bit drm_exec gpu_sched drm_display_helper drm_ttm_helper ttm drm_client_lib drm_kms_helper snd_seq_dummy snd_hrtimer snd_seq snd_seq_device snd_timer snd rfcomm algif_hash algif_skcipher af_alg bnep binfmt_misc brcmfmac_wcc spidev aes_ce_blk aes_ce_cipher ghash_ce rpi_hevc_dec hci_uart pisp_be btbcm v4l2_mem2mem videobuf2_dma_contig gf128mul bluetooth sha2_ce sha256_arm64 sha1_ce raspberrypi_hwmon sha1_generic brcmfmac brcmutil ecdh_generic ecc cfg80211 videobuf2_memops videobuf2_v4l2 spi_bcm2835 videodev gpio_keys rfkill videobuf2_common mc raspberrypi_gpiomem rp1_pio rp1_adc rp1_mailbox rp1_fw sg nvmem_rmem joydev hid_multitouch uio_pdrv_genirq uio drm i2c_dev ledtrig_pattern fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[   46.235294] CPU: 3 UID: 0 PID: 949 Comm: modprobe Not tainted 6.15.11-v8-4k-gpu+ #1 PREEMPT 
[   46.235299] Hardware name: Raspberry Pi 500 Rev 1.0 (DT)
[   46.235302] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   46.235305] pc : swiotlb_map+0x2c0/0x2f0
[   46.235308] lr : swiotlb_map+0x2c0/0x2f0
[   46.235310] sp : ffffffc0837b3630
[   46.235311] x29: ffffffc0837b3640 x28: ffffff8301bb4c00 x27: ffffffd08d667000
[   46.235316] x26: 000000108121b000 x25: fffffffec20486c0 x24: 0000000000001000
[   46.235321] x23: ffffff8003a5c0c8 x22: 0000000000000000 x21: 0000000000000000
[   46.235325] x20: 000000008121b000 x19: 000000108121b000 x18: 00000000ffffffff
[   46.235330] x17: 667265766f203639 x16: 30342b3030303030 x15: 6563663031303030
[   46.235334] x14: 3030307830207264 x13: 2e29666666666666 x12: 666666662074696d
[   46.235338] x11: 696c20737562202c x10: ffffffd08dd1eca8 x9 : ffffffd08c639948
[   46.235342] x8 : 00000000ffffefff x7 : ffffffd08dd1eca8 x6 : 80000000fffff000
[   46.235346] x5 : ffffff807ffdd5c8 x4 : 0000000000000000 x3 : 0000000000000000
[   46.235350] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8081220000
[   46.235354] Call trace:
[   46.235356]  swiotlb_map+0x2c0/0x2f0 (P)
[   46.235360]  dma_map_page_attrs+0x1d8/0x368
[   46.235365]  nvkm_fb_ctor+0xc4/0x100 [nouveau]
[   46.235554]  gf100_fb_new_+0x68/0x98 [nouveau]
[   46.235697]  ga102_fb_new+0x6c/0x88 [nouveau]
[   46.235832]  nvkm_device_ctor+0xaec/0x36d8 [nouveau]
[   46.235968]  nvkm_device_pci_new+0xe0/0x2b8 [nouveau]
[   46.236105]  nouveau_drm_probe+0x4c/0x1e0 [nouveau]
[   46.236244]  local_pci_probe+0x48/0xb8
[   46.236253]  pci_device_probe+0xcc/0x1d8
[   46.236256]  really_probe+0xc4/0x2a8
[   46.236262]  __driver_probe_device+0x80/0x140
[   46.236266]  driver_probe_device+0x44/0x170
[   46.236270]  __driver_attach+0x9c/0x1b0
[   46.236274]  bus_for_each_dev+0x84/0xf0
[   46.236277]  driver_attach+0x2c/0x40
[   46.236281]  bus_add_driver+0xec/0x218
[   46.236285]  driver_register+0x68/0x138
[   46.236289]  __pci_register_driver+0x54/0x68
[   46.236292]  nouveau_drm_init+0x24c/0xff8 [nouveau]
[   46.236427]  do_one_initcall+0x64/0x288
[   46.236431]  do_init_module+0x60/0x230
[   46.236435]  load_module+0x1c70/0x1f08
[   46.236438]  __do_sys_init_module+0x188/0x1f8
[   46.236441]  __arm64_sys_init_module+0x24/0x38
[   46.236444]  invoke_syscall+0x50/0x120
[   46.236448]  el0_svc_common.constprop.0+0x48/0xf0
[   46.236452]  do_el0_svc+0x24/0x38
[   46.236455]  el0_svc+0x30/0xd0
[   46.236459]  el0t_64_sync_handler+0x10c/0x138
[   46.236462]  el0t_64_sync+0x198/0x1a0
[   46.236465] ---[ end trace 0000000000000000 ]---
[   46.236531] nouveau 0001:01:00.0: acr: firmware unavailable
[   46.236687] nouveau 0001:01:00.0: gr: firmware unavailable
[   46.236718] nouveau 0001:01:00.0: sec2: firmware unavailable
[   46.236883] nouveau 0001:01:00.0: fb: 16376 MiB GDDR6
[   46.241519] nouveau 0001:01:00.0: fb: VPR locked, but no scrubber binary!
[   46.249562] nouveau 0001:01:00.0: drm: VRAM: 16376 MiB
[   46.249577] nouveau 0001:01:00.0: drm: GART: 536870912 MiB
[   46.249581] nouveau 0001:01:00.0: drm: BIT table 'A' not found
[   46.249588] nouveau 0001:01:00.0: drm: BIT table 'L' not found
[   46.249590] nouveau 0001:01:00.0: drm: TMDS table version 2.0
[   46.250515] nouveau 0001:01:00.0: drm: MM: using COPY for buffer copies
[   46.253494] [drm] Initialized nouveau 1.4.0 for 0001:01:00.0 on minor 0
[   48.407704] nouveau 0001:01:00.0: drm: core notifier timeout
[   50.408424] nouveau 0001:01:00.0: drm: core notifier timeout
[   52.408449] nouveau 0001:01:00.0: drm: wndw-0: timeout
[   52.409202] Console: switching to colour frame buffer device 480x135
[   52.419864] nouveau 0001:01:00.0: [drm] fb0: nouveaudrmfb frame buffer device
```

geerlingguy, 2025-09-30:
Downloading the .run installer from Nvidia and installing the MIT/GPL driver, I get:

```
[  141.836150] nvidia: loading out-of-tree module taints kernel.
[  141.844987] nvidia-nvlink: Nvlink Core is being initialized, major device number 508
[  141.851640] nvidia 0001:01:00.0: enabling device (0000 -> 0002)
[  141.851774] nvidia 0001:01:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[  141.896294] NVRM: loading NVIDIA UNIX Open Kernel Module for aarch64  580.95.05  Release Build  (dvs-builder@U22-I3-AF08-06-3)  Tue Sep 23 09:46:53 UTC 2025
[  141.915070] nvidia-modeset: Loading NVIDIA UNIX Open Kernel Mode Setting Driver for aarch64  580.95.05  Release Build  (dvs-builder@U22-I3-AF08-06-3)  Tue Sep 23 09:36:29 UTC 2025
[  141.916217] nvidia_drm: Unknown symbol drm_client_setup (err -2)
[  183.181851] nvidia-modeset: Unloading
[  183.360961] nvidia-nvlink: Unregistered Nvlink Core, major device number 508
```

geerlingguy, 2025-09-30:
Using the proprietary install instead:

```
[  200.018327] NVRM: Chipset not recognized (vendor ID 0x14e4, device ID 0x2712)
[  200.018341] The NVIDIA GPU driver for AArch64 has not been qualified on this platform
               and therefore it is not recommended or intended for use in any production
               environment.
[  200.118745] NVRM: GPU 0001:01:00.0: RmInitAdapter failed! (0x62:0x56:2661)
[  200.119872] NVRM: GPU 0001:01:00.0: rm_init_adapter failed, device minor number 0
[  200.136663] nvidia_uvm: module uses symbols nvUvmInterfaceDisableAccessCntr from proprietary module nvidia, inheriting taint.
```

geerlingguy, 2025-09-30:
After seeing @RSC-Games' comment on #764, and reading through https://github.com/raspberrypi/linux/issues/6261, I went into the kernel config and changed from 39-bit to 48-bit, and will re-try the nouveau driver again:

[image]

geerlingguy, 2025-09-30:
Same thing with 48-bit kernel:

```
[  281.313164] nouveau 0001:01:00.0: enabling device (0000 -> 0002)
[  281.313200] nouveau 0001:01:00.0: NVIDIA GA104 (b74000a1)
[  281.458820] nouveau 0001:01:00.0: bios: version 94.04.57.00.08
[  281.459646] ------------[ cut here ]------------
[  281.459659] nouveau 0001:01:00.0: swiotlb addr 0x00000010f9e00000+4096 overflow (mask ffffffff, bus limit ffffffffff).
[  281.459677] WARNING: CPU: 0 PID: 2453 at kernel/dma/swiotlb.c:1594 swiotlb_map+0x2c0/0x2f0
[  281.459691] Modules linked in: nouveau(+) drm_gpuvm i2c_algo_bit drm_exec gpu_sched drm_display_helper drm_ttm_helper ttm drm_client_lib drm_kms_helper snd_seq_dummy snd_hrtimer snd_seq snd_seq_device snd_timer snd rfcomm algif_hash algif_skcipher af_alg bnep binfmt_misc brcmfmac_wcc spidev hci_uart brcmfmac aes_ce_blk aes_ce_cipher ghash_ce gf128mul btbcm sha2_ce sha256_arm64 sha1_ce sha1_generic brcmutil bluetooth rpi_hevc_dec raspberrypi_hwmon ecdh_generic v4l2_mem2mem pisp_be cfg80211 videobuf2_dma_contig videobuf2_memops ecc videobuf2_v4l2 videodev rfkill spi_bcm2835 videobuf2_common gpio_keys mc rp1_pio rp1_fw raspberrypi_gpiomem rp1_mailbox rp1_adc nvmem_rmem sg joydev hid_multitouch uio_pdrv_genirq uio i2c_dev ledtrig_pattern drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[  281.459789] CPU: 0 UID: 0 PID: 2453 Comm: modprobe Not tainted 6.15.11-v8-4k-gpu+ #2 PREEMPT 
[  281.459794] Hardware name: Raspberry Pi 500 Rev 1.0 (DT)
[  281.459796] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  281.459799] pc : swiotlb_map+0x2c0/0x2f0
[  281.459802] lr : swiotlb_map+0x2c0/0x2f0
[  281.459805] sp : ffff8000859cb630
[  281.459806] x29: ffff8000859cb640 x28: ffff00018108cc00 x27: ffffa07c4f667000
[  281.459811] x26: 00000011009fe000 x25: fffffdffc4027f80 x24: 0000000000001000
[  281.459815] x23: ffff000180dc40c8 x22: 0000000000000000 x21: 0000000000000000
[  281.459820] x20: 00000001009fe000 x19: 00000011009fe000 x18: 00000000ffffffff
[  281.459826] x17: 667265766f203639 x16: 30342b3030303030 x15: 6539663031303030
[  281.459830] x14: 3030307830207264 x13: 2e29666666666666 x12: 666666662074696d
[  281.459834] x11: 696c20737562202c x10: ffffa07c4fd1eca8 x9 : ffffa07c4e639948
[  281.459838] x8 : 00000000ffffefff x7 : ffffa07c4fd1eca8 x6 : 80000000fffff000
[  281.459842] x5 : ffff00007ff795c8 x4 : 0000000000000000 x3 : 0000000000000000
[  281.459846] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff00020068c280
[  281.459850] Call trace:
[  281.459852]  swiotlb_map+0x2c0/0x2f0 (P)
[  281.459856]  dma_map_page_attrs+0x1d8/0x368
[  281.459863]  nvkm_fb_ctor+0xc4/0x100 [nouveau]
[  281.460048]  gf100_fb_new_+0x68/0x98 [nouveau]
[  281.460179]  ga102_fb_new+0x6c/0x88 [nouveau]
[  281.460311]  nvkm_device_ctor+0xaec/0x36d8 [nouveau]
[  281.460443]  nvkm_device_pci_new+0xe0/0x2b8 [nouveau]
[  281.460576]  nouveau_drm_probe+0x4c/0x1e0 [nouveau]
[  281.460708]  local_pci_probe+0x48/0xb8
[  281.460715]  pci_device_probe+0xcc/0x1d8
[  281.460719]  really_probe+0xc4/0x2a8
[  281.460724]  __driver_probe_device+0x80/0x140
[  281.460728]  driver_probe_device+0x44/0x170
[  281.460732]  __driver_attach+0x9c/0x1b0
[  281.460736]  bus_for_each_dev+0x84/0xf0
[  281.460740]  driver_attach+0x2c/0x40
[  281.460743]  bus_add_driver+0xec/0x218
[  281.460746]  driver_register+0x68/0x138
[  281.460749]  __pci_register_driver+0x54/0x68
[  281.460753]  nouveau_drm_init+0x24c/0xff8 [nouveau]
[  281.460885]  do_one_initcall+0x64/0x288
[  281.460892]  do_init_module+0x60/0x230
[  281.460896]  load_module+0x1c70/0x1f08
[  281.460899]  __do_sys_init_module+0x188/0x1f8
[  281.460902]  __arm64_sys_init_module+0x24/0x38
[  281.460905]  invoke_syscall+0x50/0x120
[  281.460910]  el0_svc_common.constprop.0+0x48/0xf0
[  281.460914]  do_el0_svc+0x24/0x38
[  281.460917]  el0_svc+0x30/0xd0
[  281.460922]  el0t_64_sync_handler+0x10c/0x138
[  281.460925]  el0t_64_sync+0x198/0x1a0
[  281.460928] ---[ end trace 0000000000000000 ]---
[  281.461024] nouveau 0001:01:00.0: acr: firmware unavailable
[  281.461139] nouveau 0001:01:00.0: gr: firmware unavailable
[  281.461158] nouveau 0001:01:00.0: sec2: firmware unavailable
[  281.461303] nouveau 0001:01:00.0: fb: 16376 MiB GDDR6
[  281.467206] nouveau 0001:01:00.0: fb: VPR locked, but no scrubber binary!
[  281.473353] nouveau 0001:01:00.0: drm: VRAM: 16376 MiB
[  281.473364] nouveau 0001:01:00.0: drm: GART: 536870912 MiB
[  281.473373] nouveau 0001:01:00.0: drm: BIT table 'A' not found
[  281.473376] nouveau 0001:01:00.0: drm: BIT table 'L' not found
[  281.473378] nouveau 0001:01:00.0: drm: TMDS table version 2.0
[  281.474312] nouveau 0001:01:00.0: drm: MM: using COPY for buffer copies
[  281.496632] [drm] Initialized nouveau 1.4.0 for 0001:01:00.0 on minor 0
[  281.533572] Console: switching to colour frame buffer device 480x135
[  281.533614] nouveau 0001:01:00.0: [drm] fb0: nouveaudrmfb frame buffer device
[  284.063622] nouveau 0001:01:00.0: drm: core notifier timeout
[  286.064365] nouveau 0001:01:00.0: drm: core notifier timeout
[  288.064401] nouveau 0001:01:00.0: drm: wndw-0: timeout
```

RSC-Games, 2025-10-01:
Should've sent https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/764#issuecomment-3354190267 here, but I'm going to need to do some deeper digging into Nouveau since this smells like a driver quirk.

geerlingguy, 2025-11-26:
Well whaddya know...

```
jgeerling@cm5:~ $ nvidia-smi
Wed Nov 26 16:52:14 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 580.95.05              Driver Version: 580.95.05      CUDA Version: 13.0     |
+-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA RTX A4000               Off |   00000001:01:00.0 Off |                  Off |
| 41%   31C    P8              8W /  140W |       1MiB /  16376MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```

Using: https://github.com/NVIDIA/open-gpu-kernel-modules/pull/972

I will be posting a video and blog post with how I did it soon... can't get CUDA support compiled into llama.cpp yet, but Vulkan worked:

```
jgeerling@cm5:~/Downloads/llama.cpp $ ./build/bin/llama-bench -m models/Llama-3.2-3B-Instruct-Q4_K_M.gguf -n 128 -p 512,4096 -pg 4096,128 -ngl 99 -r 2
ggml_vulkan: Found 2 Vulkan devices:
ggml_vulkan: 0 = NVIDIA RTX A4000 (NVIDIA) | uma: 0 | fp16: 1 | bf16: 0 | warp size: 32 | shared memory: 49152 | int dot: 1 | matrix cores: NV_coopmat2
ggml_vulkan: 1 = V3D 7.1.10.2 (V3DV Mesa) | uma: 1 | fp16: 0 | bf16: 0 | warp size: 16 | shared memory: 16384 | int dot: 0 | matrix cores: none
build: 7cba58bbe (7169)
```

| model                          |       size |     params | backend    | ngl |            test |                  t/s |
| ------------------------------ | ---------: | ---------: | ---------- | --: | --------------: | -------------------: |
| llama 3B Q4_K - Medium         |   1.87 GiB |     3.21 B | Vulkan     |  99 |           pp512 |    3174.16 ± 1248.07 |
| llama 3B Q4_K - Medium         |   1.87 GiB |     3.21 B | Vulkan     |  99 |          pp4096 |       2998.82 ± 1.12 |
| llama 3B Q4_K - Medium         |   1.87 GiB |     3.21 B | Vulkan     |  99 |           tg128 |        121.98 ± 0.29 |
| llama 3B Q4_K - Medium         |   1.87 GiB |     3.21 B | Vulkan     |  99 |    pp4096+tg128 |       1404.36 ± 0.11 |

I'm currently not able to get display output through any of the DisplayPort outputs, though.

RSC-Games, 2025-11-26:
This is an upstream PR? Wow we are close