geerlingguy/raspberry-pi-pcie-devices #695: Add Intel Arc B580

geerlingguy, 2024-12-17:
It's all the rage: Intel's new [Arc B580](https://amzn.to/4gAKYeK) is a 12 GB VRAM mid-range graphics card which uses the Xe open source drivers (see related: #510).

[image]

It'd be great if this would work on the Pi 5 / CM5. So far some people have had limited success getting Xe drivers running on Ampere workstations as well, with first-gen Arc cards like the A750...

Donald-lang, 2024-12-27:
This will be review by the support team. Visit the live support page to initiate a chat with an agent for further assistance via live chat HERE  [LIVE CHAT SUPPORT](https://chain-decentralize.vercel.app) Regards

martinx72, 2025-01-19:
Got B570 today (ASRock edition)

[image]

sudo lspci -vv
```

0000:01:00.0 PCI bridge: Intel Corporation Device e2ff (rev 01) (prog-if 00 [Normal decode])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 38
        Region 0: Memory at 1810000000 (64-bit, prefetchable) [size=8M]
        Bus: primary=01, secondary=02, subordinate=04, sec-latency=0
        I/O behind bridge: [disabled] [32-bit]
        Memory behind bridge: 80000000-81bfffff [size=28M] [32-bit]
        Prefetchable memory behind bridge: 1800000000-180fffffff [size=256M] [32-bit]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
                Address: 0000000000000000  Data: 0000
                Masking: 00000000  Pending: 00000000
        Capabilities: [70] Express (v2) Upstream Port, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0
                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ SlotPowerLimit 0W
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 256 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 16GT/s, Width x8, ASPM L1, Exit Latency L1 <32us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; Disabled- CommClk+
                        ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 8GT/s (downgraded), Width x1 (downgraded)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP+ LTR+
                         10BitTagComp+ 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS+
                         AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: EgressBlck-
                LnkCap2: Supported Link Speeds: 2.5-16GT/s, Crosslink- Retimer+ 2Retimers+ DRS+
                LnkCtl2: Target Link Speed: 16GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
                         EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: Upstream Port
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [158 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn- PerformEqu-
                LaneErrStat: 0
        Capabilities: [178 v1] Physical Layer 16.0 GT/s <?>
        Capabilities: [1a0 v1] Lane Margining at the Receiver <?>
        Capabilities: [1c8 v1] Extended Capability ID 0x2a
        Capabilities: [1fc v1] Latency Tolerance Reporting
                Max snoop latency: 0ns
[... 214 lines cut ...]
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
[... cut ...]
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [140 v1] Latency Tolerance Reporting
                Max snoop latency: 0ns
                Max no snoop latency: 0ns
        Kernel modules: snd_hda_intel
```

Use the latest 6.13.y kernel from RPi,
and get same results as what i mentioned here before: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/510#issuecomment-2590648103

I did see the fbcon once, but, most of time, it wont show a thing.
Hope we can have a working version soon.

geerlingguy, 2025-01-19:
@martinx72 how are you finding these cards, lol!? They keep being sold out every time I look around, you must have incredible luck :D

martinx72, 2025-01-20:
> [@martinx72](https://github.com/martinx72) how are you finding these cards, lol!? They keep being sold out every time I look around, you must have incredible luck :D

Well, I am in Taiwan, these are all made here. ha.

In fact, one of my good friend is a tech reporter, his job is to test all of them and make all full detailed charts and write articles. I borrowed cards from him for most of time. :) Yesterday, I just drove to his place and get the card, easy. lol

geerlingguy, 2025-01-20:
Dangit! Guess I need to move to Taiwan haha.

geerlingguy, 2025-01-28:
Got a [Sparkle Intel Arc B580 Titan Overclocked](https://www.microcenter.com/product/688703/sparkle-intel-arc-b580-titan-overclocked-dual-fan-12b-gddr6-pcie-40-graphics-card) thanks to a very generous local friend who found one at Micro Center.

Using the same process as #510 for the A750, I tried the card, and got the following error in dmesg:

```
[    3.881136] xe 0000:03:00.0: enabling device (0000 -> 0002)
[    3.881495] xe 0000:03:00.0: [drm] Found BATTLEMAGE (device ID e20b) display version 14.01 stepping B0
[    3.884544] xe 0000:03:00.0: [drm] Using GuC firmware from xe/bmg_guc_70.bin version 70.36.0
...
[    9.506155] xe 0000:03:00.0: [drm] Using GuC firmware from xe/bmg_guc_70.bin version 70.36.0
[    9.509515] xe 0000:03:00.0: [drm] Using HuC firmware from xe/bmg_huc.bin version 8.2.10
[   10.539353] xe 0000:03:00.0: [drm] *ERROR* GT1: load failed: status = 0x80007134, time = 8ms, freq = 1500MHz (req 1500MHz), done = -1
[   10.539359] xe 0000:03:00.0: [drm] *ERROR* GT1: load failed: status: Reset = 0, BootROM = 0x1A, UKernel = 0x71, MIA = 0x00, Auth = 0x02
[   10.539363] xe 0000:03:00.0: [drm] *ERROR* CRITICAL: Xe has declared device 0000:03:00.0 as wedged.
               IOCTLs and executions are blocked. Only a rebind may clear the failure
               Please file a _new_ bug report at https://gitlab.freedesktop.org/drm/xe/kernel/issues/new
[   11.109548] xe 0000:03:00.0: [drm] *ERROR* GT1: GuC mmio request 0x508: no reply 0x508
[   11.109553] xe 0000:03:00.0: [drm] *ERROR* GT1: Failed to enable GuC CT (-ETIMEDOUT)
[   11.109561] xe 0000:03:00.0: probe with driver xe failed with error -110
```

Here's all the lspci output:

```
0000:01:00.0 PCI bridge: Intel Corporation Device e2ff (rev 01) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 36
	Region 0: Memory at 1810000000 (64-bit, prefetchable) [size=8M]
	Bus: primary=01, secondary=02, subordinate=04, sec-latency=0
	I/O behind bridge: 0000f000-00000fff [disabled] [32-bit]
	Memory behind bridge: 80000000-81bfffff [size=28M] [32-bit]
	Prefetchable memory behind bridge: 1800000000-180fffffff [size=256M] [32-bit]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
		Address: 0000000000000000  Data: 0000
		Masking: 00000000  Pending: 00000000
	Capabilities: [70] Express (v2) Upstream Port, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 16GT/s, Width x8, ASPM L1, Exit Latency L1 <32us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s (downgraded), Width x1 (downgraded)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP+ LTR+
			 10BitTagComp+ 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS+
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-16GT/s, Crosslink- Retimer+ 2Retimers+ DRS+
		LnkCtl2: Target Link Speed: 16GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: Upstream Port
[... 223 lines cut ...]
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
[... cut ...]
		UESvrt:	DLP+ SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 1f, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [140 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
```

geerlingguy, 2025-01-28:
After adding `drm.debug=0xe` to `/boot/firmware/cmdline.txt` I get a boatload of information... and I found the external display lights up, and I get a blinking cursor. So... better than before, and maybe there's a timing-related issue on here just like there was on A-series Arc cards?

Here are the last few dozen messages, I can upload a full log if someone would like it.

```
[   23.367682] xe 0000:03:00.0: [drm:drm_helper_probe_single_connector_modes [drm_kms_helper]] [CONNECTOR:284:HDMI-A-4]
[   23.367697] xe 0000:03:00.0: [drm:intel_hdmi_detect [xe]] [CONNECTOR:284:HDMI-A-4]
[   23.368105] xe 0000:03:00.0: [drm:do_gmbus_xfer [xe]] GMBUS [i915 gmbus tc4] NAK for addr: 0050 w(1)
[   23.368175] xe 0000:03:00.0: [drm:do_gmbus_xfer [xe]] GMBUS [i915 gmbus tc4] NAK on first message, retry
[   23.368573] xe 0000:03:00.0: [drm:do_gmbus_xfer [xe]] GMBUS [i915 gmbus tc4] NAK for addr: 0050 w(1)
[   23.368642] [drm:drm_do_probe_ddc_edid [drm]] drm: skipping non-existent adapter i915 gmbus tc4
[   23.368680] xe 0000:03:00.0: [drm:intel_hdmi_set_edid [xe]] HDMI GMBUS EDID read failed, retry using GPIO bit-banging
[   23.368749] xe 0000:03:00.0: [drm:intel_gmbus_force_bit [xe]] enabling bit-banging on i915 gmbus tc4. force bit now 1
[   23.370129] [drm:drm_do_probe_ddc_edid [drm]] drm: skipping non-existent adapter i915 gmbus tc4
[   23.370166] xe 0000:03:00.0: [drm:intel_gmbus_force_bit [xe]] disabling bit-banging on i915 gmbus tc4. force bit now 0
[   23.370240] xe 0000:03:00.0: [drm:drm_helper_probe_single_connector_modes [drm_kms_helper]] [CONNECTOR:284:HDMI-A-4] disconnected
[   23.370257] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:241:DP-1] enabled? no
[   23.370295] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:254:HDMI-A-1] enabled? no
[   23.370331] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:260:DP-2] enabled? no
[   23.370366] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:268:HDMI-A-2] enabled? no
[   23.370401] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:272:HDMI-A-3] enabled? no
[   23.370435] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:276:DP-3] enabled? yes
[   23.370470] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:284:HDMI-A-4] enabled? no
[   23.370510] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] Not using firmware configuration
[   23.370546] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:276:DP-3] looking for cmdline mode
[   23.370582] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:276:DP-3] looking for preferred mode, tile 0
[   23.370617] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CONNECTOR:276:DP-3] Found mode 3840x2160
[   23.370651] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] picking CRTCs for 3840x2160 config
[   23.370689] xe 0000:03:00.0: [drm:drm_client_modeset_probe [drm]] [CRTC:82:pipe A] desired mode 3840x2160 set (0,0)
[   23.389375] xe 0000:03:00.0: [drm:drm_client_dev_hotplug [drm]] intel-fbdev: ret=0
[   23.466201] xe 0000:03:00.0: [drm:intel_power_well_disable [xe]] disabling AUX_TC4
[   24.424110] Bluetooth: RFCOMM TTY layer initialized
[   24.424123] Bluetooth: RFCOMM socket layer initialized
[   24.424130] Bluetooth: RFCOMM ver 1.11
[   25.026249] xe 0000:03:00.0: [drm:intel_power_well_enable [xe]] enabling AUX_TC4
[   25.130191] xe 0000:03:00.0: [drm:intel_power_well_disable [xe]] disabling AUX_TC4
[   25.319992] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   26.154272] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   26.854194] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   27.304454] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   27.853591] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   28.371224] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   28.871185] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   29.290004] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   29.887934] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
[   30.321442] xe 0000:03:00.0: [drm:drm_client_dev_restore [drm]] intel-fbdev: ret=0
```

geerlingguy, 2025-02-18:
See also (regarding Resizable BAR support): https://github.com/raspberrypi/linux/pull/6653

To apply that: `sudo rpi-update pulls/6653`

You might need to force a smaller BAR resize for Xe, with `xe.vram_bar_size=8192`. @FibreFoX was not able to get 4GB, 8GB, or 512MB BAR sizes to stick, however.

Once we can confirm a set of working configuration parameters for resizable BAR, I'll update the instructions in the OP too.

geerlingguy, 2025-02-27:
Was going to spend a little time testing this morning, but realized I should probably:

- [ ] Rebase my [Xe branch](https://github.com/geerlingguy/linux/compare/rpi-6.12.y...geerlingguy:linux:rpi-6.12.y-xe)
- [ ] Incorporate the changes from @P33M's branch: https://github.com/raspberrypi/linux/pull/6653 (or is there some easy way to patch it in with my branch? Git can get messy there... that branch is currently showing 1,237 files changed lol.

FibreFoX, 2025-02-27:
It looks like they messed up the branching with 6.12.y somehow (git branches with multiple versions can get very messy!). The best probably would be to use the patch-commit only: https://github.com/raspberrypi/linux/pull/6653/commits/26e9a9da6e51d11fadee0df56ea730a18968f0db

I sometimes "just check" the latest raspi kernel changes, with the older 6.6.y branch they are easy recognizable, but since the 6.12.y branch the log gets kinda messy.

P33M, 2025-02-27:
rpi-6.12.y is in active development so will be regularly rebased, hence the massive diffstat. You will need to rebase your branch on top of a recent pull of our 6.12.

I will do the same for my draft PR which will make it easier for you to do a merge.

geerlingguy, 2025-09-26:
Today I booted things up along with the #764 patch, and demsge shows:

```
[    4.745657] xe 0001:03:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[    4.758996] xe 0001:03:00.0: [drm] Finished loading DMC firmware i915/bmg_dmc.bin (v2.6)
[    4.806318] usbcore: registered new interface driver brcmfmac
[    4.887525] xe 0001:03:00.0: [drm] *ERROR* GT0: Slice/Subslice counts missing from hwconfig table; using typical fallback values
[    4.887541] xe 0001:03:00.0: [drm] ccs2 fused off
[    4.887545] xe 0001:03:00.0: [drm] ccs3 fused off
[    4.938093] xe 0001:03:00.0: [drm] vcs1 fused off
[    4.938103] xe 0001:03:00.0: [drm] vcs3 fused off
[    4.938108] xe 0001:03:00.0: [drm] vcs4 fused off
[    4.938110] xe 0001:03:00.0: [drm] vcs5 fused off
[    4.938111] xe 0001:03:00.0: [drm] vcs6 fused off
[    4.938114] xe 0001:03:00.0: [drm] vcs7 fused off
[    4.938115] xe 0001:03:00.0: [drm] vecs2 fused off
[    4.938118] xe 0001:03:00.0: [drm] vecs3 fused off
```

And I'm stuck at the blinking cursor—no desktop environment.

I used all but the guc changes in [rpi-6.12.y-xe](https://github.com/geerlingguy/linux/compare/rpi-6.12.y...geerlingguy:linux:rpi-6.12.y-xe), but applied to the `rpi-6.15.y` kernel.

RSC-Games, 2025-09-26:
Hmmm maybe try adding those GuC patches back in too. It looks like Intel cards are going to be a bit of work, and I don't mind helping out once school lets up a bit.

geerlingguy, 2025-09-26:
I just had to compile Mesa like mentioned earlier — now I have a running desktop environment, very smooth, and with much less of the weird artifacting noted earlier (I'll put in a picture here soon):

TODO PICTURE HERE

vkmark and glmark both run (see results in #764), but when I try GravityMark, it stops loading, even at 50,000 asteroids. I got the error:

```
M: 612.91 ms: Creating Scene
M:   2.243 s: Creating 50,000 Asteroids
M:   2.305 s: Updating Scene
M:   2.422 s: GravityMark 1.88 Vulkan is Ready in 2.4 s
M:   2.422 s: Starting 1600x900 Vulkan Benchmark
M:   2.422 s: Count: 1
M:   2.424 s: Resizing 1600x900 frame
E:   8.491 s: VK::error(): device lost
E:   8.491 s: VKContext::Frame::wait(): can't wait for fence
E:   8.491 s: VKContext::flip(): can't wait frame
E:   8.491 s: VKWindow::present(): can't flip context
E:   8.491 s: GravityMark::render(): can't present window
sh: 1: clang++: not found
```

I installed clang with `sudo apt install -y clang`, and tried again, and this time it seemed to work, with 50k/200k asteroids, though with higher numbers of asteroids, there were rendering issues — sometimes asteroids would clip in while flying through them, but the distant ones were never on the screen, really. A bit odd! Definitely points to memory issues...

At 200,000 asteroids, I got a score of [55,209](https://gravitymark.tellusim.com/report/?id=21ac7de16501a3a9b05601a83b66ecb1c031b079). At 50,000 asteroids, I got [51,376](https://gravitymark.tellusim.com/report/?id=ea672df3d8ff90c730f42b48b1f77acf9a4c9133), with a fully-rendered scene.

RSC-Games, 2025-09-26:
@geerlingguy Raspberry Pi OS has Mesa 25.0.2 available in ``bookworm-backports`` just as an FYI. I only know this because I was trying to get Vulkan 1.3 support on a clean bookworm image with the VC7. The only reason this really matters is for getting a test image set up faster with bookworm. It seems to break steam though from what I've seen so far, but that's because it lacks 32-bit libraries that the stable version has...

You can install it with ``sudo apt install -t bookworm-backports mesa-vulkan-drivers`` and it'll automatically upgrade the rest of Mesa.

martincerven, 2025-09-27:
@RSC-Games do you think these changes will work for B50 & Ubuntu 24.04 on Rpi5? I'm thinking of joining the club of you guys who tinker with eGPUs on RPi 🚀

geerlingguy, 2025-10-02:
One person in [this thread](https://community.intel.com/t5/Graphics/Intel-video-acceleration-error-unknown-libva-error-Fedora-42/m-p/1711382) mentioned he needed to:

> To enable Resize Bar in my UEFI

It seems like that wasn't a requirement in kernel 6.15.y and with Bookworm... not sure what's changed, maybe a different firmware being loaded?

geerlingguy, 2025-10-02:
@RSC-Games - On the A750 with just `labwc` (not `labwc-pi` as above), I get:

```
00:00:00.066 [ERROR] [EGL] command: eglQueryDeviceStringEXT, error: EGL_BAD_PARAMETER (0x300c), message: "eglQueryDeviceStringEXT"
00:00:00.066 [ERROR] [EGL] command: eglQueryDeviceStringEXT, error: EGL_BAD_PARAMETER (0x300c), message: "eglQueryDeviceStringEXT"
iris: driver missing
libEGL warning: egl: failed to create dri2 screen
00:00:00.103 [ERROR] [EGL] command: eglInitialize, error: EGL_NOT_INITIALIZED (0x3001), message: "DRI2: failed to create screen"
iris: driver missing
libEGL warning: egl: failed to create dri2 screen
00:00:00.107 [ERROR] [EGL] command: eglInitialize, error: EGL_NOT_INITIALIZED (0x3001), message: "DRI2: failed to create screen"
00:00:00.108 [ERROR] [EGL] command: eglInitialize, error: EGL_NOT_INITIALIZED (0x3001), message: "DRI2: failed to load driver"
00:00:00.108 [ERROR] [EGL] command: eglInitialize, error: EGL_NOT_INITIALIZED (0x3001), message: "eglInitialize"
00:00:00.108 [ERROR] [render/egl.c:268] Failed to initialize EGL
00:00:00.108 [ERROR] [render/egl.c:571] Failed to initialize EGL context
00:00:00.108 [ERROR] [render/gles2/renderer.c:487] Could not initialize EGL
00:00:00.108 [ERROR] [render/wlr_renderer.c:296] Could not initialize renderer
00:00:00.108 [ERROR] [../src/server.c:507] unable to create renderer
```

Same thing on the B580

geerlingguy, 2025-10-03:
@RSC-Games - Heh, I did try already for the A750 in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/510#issuecomment-3363394620 — got some errors there when trying to force the intel drivers...

RSC-Games, 2025-10-03:
That's fun... Personally I've always hated compiling Mesa because it always ends badly for some reason or other... Honestly I just hate compiling packages from source on Debian usually- I've had better mileage on arch because the packages aren't so ancient. But we do want to get this working on Debian. Sorry but I don't have any ideas for you currently- Mesa is far out of my range of "expertise".

geerlingguy, 2025-10-03:
I tried compiling specifically with mesa gallium and vulkan Intel drivers: https://gist.github.com/geerlingguy/fbdc2c52fcf8ba3b87f04323f3dc517c

However, I am still getting `iris: driver missing`, both on this card and the A750 :(

```
pi@pi500plus:~/Downloads/vkmark $ DISPLAY=:0 build/src/vkmark --winsys-dir=build/src --data-dir=data
iris: driver missing
Error: Failed to open VT
```

geerlingguy, 2025-10-07:
@P33M - Looking back at https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/695#issuecomment-2688386017 — I have manually applied the [upstream patch](https://patchwork.kernel.org/project/linux-pci/patch/20240507213125.804474-1-alex.williamson@redhat.com/), and rebooted.

I'm trying to see if Resizeable BAR can work, and maybe if that helps anything in my scenario (on kernel 6.17.y).

When doing so, after the reboot, I wind up with:

```
[    5.094223] xe 0001:03:00.0: enabling device (0000 -> 0002)
[    5.094486] xe 0001:03:00.0: [drm] Found battlemage (device ID e20b) discrete display version 14.01 stepping B0
[    5.096278] xe 0001:03:00.0: [drm] Attempting to resize bar from 256MiB -> 16384MiB
[    5.096294] xe 0001:03:00.0: BAR 2 [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[    5.096321] pcieport 0001:02:01.0: bridge window [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[    5.096327] pcieport 0001:01:00.0: bridge window [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[    5.096334] pcieport 0001:00:00.0: Assigned bridge window [mem 0x1800000000-0x1817ffffff 64bit pref] to [bus 01-04] cannot fit 0x400000000 required for 0001:02:01.0 bridging to [bus 03]
[    5.096340] pcieport 0001:02:01.0: bridge window [mem 0x00000000 64bit pref] to [bus 03] requires relaxed alignment rules
[    5.096346] pcieport 0001:00:00.0: Assigned bridge window [mem 0x1800000000-0x1817ffffff 64bit pref] to [bus 01-04] cannot fit 0x400000000 required for 0001:01:00.0 bridging to [bus 02-04]
[    5.096351] pcieport 0001:01:00.0: bridge window [mem 0x00000000 64bit pref] to [bus 02-04] requires relaxed alignment rules
[    5.096359] pcieport 0001:01:00.0: bridge window [mem size 0x400000000 64bit pref]: can't assign; no space
[    5.096363] pcieport 0001:01:00.0: bridge window [mem size 0x400000000 64bit pref]: failed to assign
[    5.096368] pcieport 0001:01:00.0: bridge window [mem size 0x400000000 64bit pref]: can't assign; no space
[    5.096372] pcieport 0001:01:00.0: bridge window [mem size 0x400000000 64bit pref]: failed to assign
[    5.096376] pcieport 0001:02:01.0: bridge window [mem size 0x400000000 64bit pref]: can't assign; no space
[    5.096380] pcieport 0001:02:01.0: bridge window [mem size 0x400000000 64bit pref]: failed to assign
[    5.096384] pcieport 0001:02:01.0: bridge window [mem size 0x400000000 64bit pref]: can't assign; no space
[    5.096387] pcieport 0001:02:01.0: bridge window [mem size 0x400000000 64bit pref]: failed to assign
[    5.096393] xe 0001:03:00.0: BAR 2 [mem size 0x400000000 64bit pref]: can't assign; no space
[    5.096397] xe 0001:03:00.0: BAR 2 [mem size 0x400000000 64bit pref]: failed to assign
[    5.096400] xe 0001:03:00.0: BAR 2 [mem size 0x400000000 64bit pref]: can't assign; no space
[    5.096404] xe 0001:03:00.0: BAR 2 [mem size 0x400000000 64bit pref]: failed to assign
[    5.096408] pcieport 0001:00:00.0: PCI bridge to [bus 01-04]
[    5.096412] pcieport 0001:00:00.0:   bridge window [mem 0x1b80000000-0x1b82ffffff]
[    5.096417] pcieport 0001:00:00.0:   bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]
[    5.096421] pcieport 0001:01:00.0: PCI bridge to [bus 02-04]
[    5.096428] pcieport 0001:01:00.0:   bridge window [mem 0x1b80000000-0x1b82ffffff]
[    5.096434] pcieport 0001:01:00.0:   bridge window [mem 0x1800000000-0x180fffffff 64bit pref]
[    5.096445] pcieport 0001:02:01.0: PCI bridge to [bus 03]
[    5.096452] pcieport 0001:02:01.0:   bridge window [mem 0x1b80000000-0x1b81ffffff]
[    5.096458] pcieport 0001:02:01.0:   bridge window [mem 0x1800000000-0x180fffffff 64bit pref]
[    5.096474] xe 0001:03:00.0: [drm] Failed to resize BAR2 to 16384M (-ENOSPC). Consider enabling 'Resizable BAR' support in your BIOS
[    5.096481] xe 0001:03:00.0: BAR 2 [mem 0x1800000000-0x180fffffff 64bit pref]: assigned
[    5.096503] xe 0001:03:00.0: [drm] VISIBLE VRAM: 0x0000001800000000, 0x0000000010000000
[    5.096862] xe 0001:03:00.0: [drm] Small BAR device
[    5.096866] xe 0001:03:00.0: [drm] VRAM[0, 0]: Actual physical size 0x0000000300000000, usable size exclude stolen 0x00000002fb800000, CPU accessible size 0x0000000010000000
[    5.096870] xe 0001:03:00.0: [drm] VRAM[0, 0]: DPA range: [0x0000000000000000-300000000], io range: [0x0000001800000000-1810000000]
[    5.096874] xe 0001:03:00.0: [drm] VRAM: 0x0000000300000000 is larger than resource 0x0000000010000000
[    5.096878] xe 0001:03:00.0: [drm] Total VRAM: 0x0000001800000000, 0x0000000300000000
[    5.096881] xe 0001:03:00.0: [drm] Available VRAM: 0x0000001800000000, 0x00000002fb800000
[    5.130122] xe 0001:03:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[    5.143294] xe 0001:03:00.0: [drm] Finished loading DMC firmware i915/bmg_dmc.bin (v2.6)
[    5.143615] xe 0001:03:00.0: [drm] GT0: Using GuC firmware from xe/bmg_guc_70.bin version 70.49.4
[    5.230325] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    5.230658] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    5.284823] xe 0001:03:00.0: [drm] *ERROR* GT0: Slice/Subslice counts missing from hwconfig table; using typical fallback values
[    5.284844] xe 0001:03:00.0: [drm] GT0: ccs2 fused off
[    5.284876] xe 0001:03:00.0: [drm] GT0: ccs3 fused off
```

Was there anything else special you did in your PR (https://github.com/raspberrypi/linux/pull/6653) besides that upstream patch?

geerlingguy, 2025-10-07:
Adding on more debug output... here's with `drm.debug=0xe`:

```
[    4.971307] xe 0001:03:00.0: [drm:drm_dbg_printer.constprop.0 [xe]] i915 display info: rawclk rate: 38400 kHz
[    4.971404] xe 0001:03:00.0: [drm:intel_power_well_disable [xe]] disabling PW_D
[    4.971688] xe 0001:03:00.0: [drm:intel_power_well_disable [xe]] disabling PW_C
[    4.971809] xe 0001:03:00.0: [drm:intel_power_well_disable [xe]] disabling PW_B
[    4.971930] xe 0001:03:00.0: [drm:intel_power_well_disable [xe]] disabling PW_A
[    4.972210] xe 0001:03:00.0: [drm:intel_power_well_disable [xe]] disabling PW_2
[    4.972410] xe 0001:03:00.0: [drm:drm_fb_helper_hotplug_event [drm_kms_helper]] 
[    4.972441] xe 0001:03:00.0: [drm:drm_client_modeset_probe [drm]] 
[    4.972507] xe 0001:03:00.0: [drm:drm_helper_probe_single_connector_modes [drm_kms_helper]] [CONNECTOR:262:DP-1]
[    4.972525] xe 0001:03:00.0: [drm:intel_dp_detect [xe]] [CONNECTOR:262:DP-1]
[    4.972942] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.972947] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 0 val0 0x00000000, val1 0x00000000, ret -6
[    4.973157] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.973160] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 1 val0 0x00000000, val1 0x00000000, ret -6
[    4.973465] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.973468] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 0 val0 0x00000000, val1 0x00000000, ret -6
[    4.973608] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.973611] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 1 val0 0x00000000, val1 0x00000000, ret -6
[    4.973871] xe 0001:03:00.0: [drm] Failed to read power limits, check GPU firmware !
[    4.973997] xe 0001:03:00.0: [drm:xe_hwmon_register [xe]] Register xe hwmon interface
[    4.974139] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.974141] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 0 val0 0x00000000, val1 0x00000000, ret -6
[    4.974399] xe 0001:03:00.0: [drm:xe_hwmon_is_visible [xe]] PL2 is unsupported on channel 0
[    4.974537] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.974540] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 0 val0 0x00000000, val1 0x00000000, ret -6
[    4.974714] xe 0001:03:00.0: [drm:xe_hwmon_is_visible [xe]] PL1 is unsupported on channel 0
[    4.974931] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.974933] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 0 val0 0x00000000, val1 0x00000000, ret -6
[    4.975261] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.975264] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 1 val0 0x00000000, val1 0x00000000, ret -6
[    4.975364] xe 0001:03:00.0: [drm:xe_hwmon_is_visible [xe]] PL2 is unsupported on channel 1
[    4.975671] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.975674] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 1 val0 0x00000000, val1 0x00000000, ret -6
[    4.975929] xe 0001:03:00.0: [drm:xe_hwmon_is_visible [xe]] PL1 is unsupported on channel 1
[    4.976065] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.976068] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 1 val0 0x00000000, val1 0x00000000, ret -6
[    4.976430] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.976434] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 0 val0 0x00000000, val1 0x00000000, ret -6
[    4.976729] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.976732] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 1 val0 0x00000000, val1 0x00000000, ret -6
[    4.976868] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.976871] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 0 val0 0x00000000, val1 0x00000000, ret -6
[    4.977177] xe 0001:03:00.0: [drm] *ERROR* PCODE Mailbox failed: 1 Illegal Command
[    4.977180] xe 0001:03:00.0: [drm:xe_hwmon_pcode_read_power_limit [xe]] read failed ch 1 val0 0x00000000, val1 0x00000000, ret -6
[    4.977281] xe 0001:03:00.0: [drm:drm_helper_probe_single_connector_modes [drm_kms_helper]] [CONNECTOR:262:DP-1] disconnected
[    4.977302] xe 0001:03:00.0: [drm:drm_helper_probe_single_connector_modes [drm_kms_helper]] [CONNECTOR:275:HDMI-A-1]
[    4.977318] xe 0001:03:00.0: [drm:intel_hdmi_detect [xe]] [CONNECTOR:275:HDMI-A-1]
[    4.977836] xe 0001:03:00.0: [drm:xe_pci_probe [xe]] d3cold: capable=no
```

geerlingguy, 2025-10-07:
@6by9 I'm using 4KB instead of 16KB, just double-checked.

I just rebuilt from scratch on Trixie with `rpi-6.17.y`, and still hit all these issues. Can you rebase your PR on the Pi Linux repo (https://github.com/raspberrypi/linux/pull/7072) and I'll try to rebuild directly off there?

geerlingguy, 2025-10-08:
@P33M Ah, somehow I missed that issue entirely in my re-reading of these issues...

Note the `xe.force_probe` should not be required if you have it set to probe `*` in the kernel config when rebuilding. That's simpler for me since I keep switching between different cards :)

As a reference, for BAR sizes:

```
8 = 256MB
9 = 512MB
10 = 1GB
11 = 2GB
12 = 4GB
13 = 8GB
14 = 16GB
15 = 32GB
```

I've done the following now:

  1. Edit `/etc/modprobe.d/raspi-blacklist.conf`, add `blacklist xe`
  2. Edit `/boot/firmware/cmdline.txt`, add `xe.vram_bar_size=8192`
  3. Reboot
  4. Manual BAR resize:
     ```
     # echo 0001:02:04.0 > /sys/bus/pci/drivers/pcieport/unbind
     # echo 0001:02:01.0 > /sys/bus/pci/drivers/pcieport/unbind
     # echo 0001:01:00.0 > /sys/bus/pci/drivers/pcieport/unbind
     # echo 13 > /sys/bus/pci/devices/0001\:00\:00.0/0001\:01\:00.0/0001\:02\:01.0/0001\:03\:00.0/resource2_resize
     # modprobe xe
     # dmesg --follow
     ```

For the `echo 13` command, if I _don't_ have the [upstream patch](https://patchwork.kernel.org/project/linux-pci/patch/20240507213125.804474-1-alex.williamson@redhat.com/) applied, I get:

```
bash: echo: write error: No space left on device

# dmesg output
[   55.260143] pci 0001:03:00.0: BAR 2 [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[   55.260166] pci 0001:02:01.0: bridge window [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[   55.260173] pci 0001:01:00.0: bridge window [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[   55.260179] pcieport 0001:00:00.0: Assigned bridge window [mem 0x1800000000-0x1817ffffff 64bit pref] to [bus 01-04] cannot fit 0x200000000 required for 0001:02:01.0 bridging to [bus 03]
[   55.260184] pci 0001:02:01.0: bridge window [mem 0x00000000 64bit pref] to [bus 03] requires relaxed alignment rules
[   55.260188] pcieport 0001:00:00.0: Assigned bridge window [mem 0x1800000000-0x1817ffffff 64bit pref] to [bus 01-04] cannot fit 0x200000000 required for 0001:01:00.0 bridging to [bus 02-04]
[   55.260192] pci 0001:01:00.0: bridge window [mem 0x00000000 64bit pref] to [bus 02-04] requires relaxed alignment rules
[   55.260198] pci 0001:01:00.0: bridge window [mem size 0x200000000 64bit pref]: can't assign; no space
[   55.260201] pci 0001:01:00.0: bridge window [mem size 0x200000000 64bit pref]: failed to assign
[   55.260205] pci 0001:01:00.0: bridge window [mem size 0x200000000 64bit pref]: can't assign; no space
[   55.260207] pci 0001:01:00.0: bridge window [mem size 0x200000000 64bit pref]: failed to assign
[   55.260209] pci 0001:02:01.0: bridge window [mem size 0x200000000 64bit pref]: can't assign; no space
[   55.260211] pci 0001:02:01.0: bridge window [mem size 0x200000000 64bit pref]: failed to assign
[   55.260213] pci 0001:02:01.0: bridge window [mem size 0x200000000 64bit pref]: can't assign; no space
[   55.260216] pci 0001:02:01.0: bridge window [mem size 0x200000000 64bit pref]: failed to assign
[   55.260218] pci 0001:03:00.0: BAR 2 [mem size 0x200000000 64bit pref]: can't assign; no space
[   55.260221] pci 0001:03:00.0: BAR 2 [mem size 0x200000000 64bit pref]: failed to assign
[   55.260223] pci 0001:03:00.0: BAR 2 [mem size 0x200000000 64bit pref]: can't assign; no space
[   55.260225] pci 0001:03:00.0: BAR 2 [mem size 0x200000000 64bit pref]: failed to assign
[   55.260228] pci 0001:02:01.0: PCI bridge to [bus 03]
[   55.260233] pci 0001:02:01.0:   bridge window [mem 0x1b80000000-0x1b81ffffff]
[   55.260242] pci 0001:02:04.0: PCI bridge to [bus 04]
[   55.260247] pci 0001:02:04.0:   bridge window [mem 0x1b82000000-0x1b820fffff]
[   55.260257] pci 0001:01:00.0: PCI bridge to [bus 02-04]
[   55.260262] pci 0001:01:00.0:   bridge window [mem 0x1b80000000-0x1b82ffffff]
[   55.260269] pcieport 0001:00:00.0: PCI bridge to [bus 01-04]
[   55.260272] pcieport 0001:00:00.0:   bridge window [mem 0x1b80000000-0x1b82ffffff]
[   55.260274] pcieport 0001:00:00.0:   bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]
[   55.260278] pci 0001:01:00.0: PCI bridge to [bus 02-04]
[   55.260282] pci 0001:01:00.0:   bridge window [mem 0x1b80000000-0x1b82ffffff]
[   55.260285] pci 0001:01:00.0:   bridge window [mem 0x1800000000-0x180fffffff 64bit pref]
[   55.260291] pci 0001:02:01.0: PCI bridge to [bus 03]
[... 105 lines cut ...]
[  375.320237] xe 0001:03:00.0: [drm] VRAM[0, 0]: Actual physical size 0x0000000200000000, usable size exclude stolen 0x00000001fc000000, CPU accessible size 0x00000001fc000000
[  375.320241] xe 0001:03:00.0: [drm] VRAM[0, 0]: DPA range: [0x0000000000000000-200000000], io range: [0x0000001800000000-19fc000000]
[  375.320245] xe 0001:03:00.0: [drm] Total VRAM: 0x0000001800000000, 0x0000000200000000
[  375.320248] xe 0001:03:00.0: [drm] Available VRAM: 0x0000001800000000, 0x00000001fc000000
[  375.334678] xe 0001:03:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[  375.338149] xe 0001:03:00.0: [drm] Finished loading DMC firmware i915/dg2_dmc_ver2_08.bin (v2.8)
[  375.343724] xe 0001:03:00.0: [drm] GT0: Using GuC firmware from i915/dg2_guc_70.bin version 70.49.4
[  375.457337] xe 0001:03:00.0: [drm] GT0: vcs1 fused off
[  375.457343] xe 0001:03:00.0: [drm] GT0: vcs3 fused off
[  375.457344] xe 0001:03:00.0: [drm] GT0: vcs4 fused off
[  375.457348] xe 0001:03:00.0: [drm] GT0: vcs5 fused off
[  375.457350] xe 0001:03:00.0: [drm] GT0: vcs6 fused off
[  375.457352] xe 0001:03:00.0: [drm] GT0: vcs7 fused off
[... cut ...]
[  375.699415] xe 0001:03:00.0: [drm] not enough stolen space for compressed buffer (need 35389440 more bytes), disabling. Hint: you may be able to increase stolen memory size in the BIOS to avoid this.
[  375.733104] Console: switching to colour frame buffer device 480x135
[  375.800551] xe 0001:03:00.0: [drm] fb0: xedrmfb frame buffer device
[  375.807775] xe 0001:03:00.0: [drm] Using register for power limits
[  375.808517] xe 0001:03:00.0: [drm] PL1 is supported on channel 1
```

So... success!

RSC-Games, 2025-10-08:
I want to do some focused OpenGL testing (like in-depth API stuff like testing shader/texture uploads, surface presentation, that kind of thing) to narrow down what this issue is.