geerlingguy/raspberry-pi-pcie-devices #700: Samsung SM961 MZ-VPW2560

ned14, 2025-01-16:
https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/647 reports that the SM961 doesn't work, however I did not find this myself with the latest eeprom:

```
rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Tue Jan 14 00:16:48 UTC 2025 (1736813808)
    LATEST: Tue Jan 14 00:16:48 UTC 2025 (1736813808)
   RELEASE: beta (/lib/firmware/raspberrypi/bootloader-2712/beta)
            Use raspi-config to change the release.
```

The SM961 works absolutely fine in both PCIe 2.0 and PCIe 3.0 on two M.2 adapters I tried. Mine came from eBay for €16 and was heavily used, but still has years of lifespan left in it. It has an inhouse Samsung manufactured controller with Samsung flash and I _think_ it has DRAM, which makes it especially speedy under random i/o on the Pi:

https://www.techpowerup.com/ssd-specs/samsung-sm961-512-gb.d1214

This is running at PCIe 2.0:

```
dd if=/dev/zero of=/dev/nvme0n1 bs=1M count=10000 oflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 28.1119 s, 373 MB/s
```
Idle is 0.3 watts, write load is +3 watts.

```
dd of=/dev/null if=/dev/nvme0n1 bs=1M count=10000 iflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 24.0549 s, 436 MB/s
```
Idle is 0.3 watts, read load is +3 watts.

This is running at PCIe 3.0:

```
dd if=/dev/zero of=/dev/nvme0n1 bs=1M count=10000 oflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 15.1569 s, 692 MB/s
```
Idle is 0.3 watts, write load is +3.6 watts.

```
dd of=/dev/null if=/dev/nvme0n1 bs=1M count=10000 iflag=direct
10485760000 bytes (10 GB, 9.8 GiB) copied, 12.6483 s, 829 MB/s
```
Idle is 0.3 watts, read load is +3.3 watts.

I've given this drive a sound battering with constant i/o load over a week at PCIe 3.0 speeds and there have been zero errors or anything in the Linux kernel log. Good drive, and great value for €16.

KeyserSoze1, 2025-01-31:
I should try to test this again with the latest eeprom but my suspicion for this not working was more related to compatibility with ASMedia 2806 PCIE switch on that specific HAT than directly with the Pi.  If memory serves it worked with gen 2 but not gen 3 with that specific device.

ned14, 2025-01-31:
Mine is running at PCIe 3.0 just fine. I tried two different HATs, neither has a PCIe switch.

I gave mine a real battering over a week. Zero issues noticed. I gave it a blast with ZFS which detects any bit flips due to ZFS checksumming. Also perfect. I was on a Dec 2024 EEPROM, which may have helped compatibility.

It's an old drive off eBay, about 10k hours on it and maybe 100 Tb written. It's been sent off to the colocation datacentre as the principle server. Believe or not, it'll cost me under €3/month each Pi for a 100Mbit uplink. You'd wonder how they can make money, but they're one of the biggest Pi colocation providers in Europe.