geerlingguy/raspberry-pi-pcie-devices #705: CM5 MINIMA REV1 and REV2 - OSHW minimal CM5 boards

geerlingguy, 2025-02-05:
The [CM5 MINIMA REV1](https://github.com/piecol/CM5_MINIMA_REV1) is a tiny almost-CM-sized board with a RAK3172 LoRaWAN chip from Rak Wireless integrated on the top side, and an integrated SHTC3 Sensirion temperature and humidity sensor:

[image]

The [CM5 MINIMA REV2](https://github.com/piecol/CM5_MINIMA_REV2) is a tiny almost-CM-sized board with an M.2 M-key 2230-size slot for an SSD (or any other PCIe device or adapter that will fit), and an integrated SHTC3 Sensirion temperature and humidity sensor:

[image]

Both offer an array of interfaces, a fan header, USB-C PD power input, and a low-profile Ethernet connector to keep the board quite compact, even in the vertical axis.

@piecol sent over one of each of these for me to test, I will hopefully get a CM5 plugged in soon!

cmd0s, 2025-04-01:
I pre-ordered the MINIMA REV2 from https://www.elecrow.com/cm5-minima-r-2.html  
The package is on its way, and I seriously can’t wait to get it.

I’ve been following the development of this CM5 board since the beginning on [LinkedIn](https://www.linkedin.com/in/pierluigi-colangeli/recent-activity/all/) and just wanted to say - awesome work! 
I truly admire the design and execution. I’m rooting for the project, and while I may be a bit of a geek… I’m *really* excited to finally get the board in my hands.

I plan to run a full Ethereum node on it using [Web3 Pi](https://www.web3pi.io).
This is a very demanding use case - both for the CPU and the storage. During the initial sync, the node uses 100% of the processor for around 24 hours and writes approximately 1.2TB of data to the NVMe drive. It's going to be a tough and interesting stress test for this board.  
For NVMe storage, I’ll be using a [Micron 2TB M.2 2230](https://www.techpowerup.com/ssd-specs/micron-2400-2-tb.d810) drive. It’s fast, energy-efficient, runs cool compared to full-size SSDs, and fits perfectly in the 2230 form factor. Even better - it's single-sided, so all components are on one side of the PCB.

[image]

Additionally, I’m going to connect a [Waveshare 1.69-inch LCD Module (SKU:24382)](https://www.waveshare.com/1.69inch-lcd-module.htm). The MINIMA REV2 has a [dedicated connector for it](https://www.linkedin.com/posts/pierluigi-colangeli_lcd-minima-kicad-activity-7306027987030761474-Rg1-), and [Web3 Pi](https://www.web3pi.io) supports it natively. System status appears on-screen just 30 seconds after boot.

[image]

Usually, I use a standard **Raspberry Pi 5 8GB** with a microSD card, so I’m curious how the MINIMA REV2 will perform. Both Web3 Pi and [Web3 Pi Imager](https://www.web3pi.io/products/imager) (a customized Raspberry Pi Imager) support booting directly from NVMe. I’m hoping to boot from NVMe right away - but if that doesn't work, I’ll fall back to the onboard eMMC.

I’ll post an update once it arrives and I get it up and running.  
The package shipped from China and is currently in the Netherlands, so assuming no customs issues - it should be in my hands soon. Can't wait!

cmd0s, 2025-04-07:
Today my [MINIMA REV2](https://github.com/piecol/CM5_MINIMA_REV2) board finally arrived! 🎉  

[image]

This board is designed by Italian electronics engineer [Pierluigi Colangeli](https://www.linkedin.com/in/pierluigi-colangeli/), and as far as I understand, it's manufactured and distributed by [Elecrow](https://www.elecrow.com/cm5-minima-r-2.html).

### Ordering and Shipping

I pre-ordered it for $65 + $2.65 for tracked YunExpress shipping.  
It took 12 days to arrive from mainland China to Poland. Everything was well-packaged and arrived without any damage.

The board came in an anti-static zip-lock bag, wrapped in bubble wrap. No extra packaging, instructions, or box - just the board, plain and simple.

### First Impressions

The PCB is made from black laminate with white silkscreen. It looks great and feels very solid.  
The soldering and overall build quality are excellent.

[image] [image]

### First Boot

I flashed the latest [Web3 Pi](https://www.web3pi.io/) image onto a [Micron 2230 NVMe](https://www.techpowerup.com/ssd-specs/micron-2400-2-tb.d810) using the [Web3 Pi Imager](https://www.web3pi.io/products/imager).  
Then I plugged the SSD into the MINIMA REV2 board.

[image]

On the other side, I installed a [Raspberry Pi CM5008032](https://www.mouser.pl/ProductDetail/Raspberry-Pi/SC1571?qs=3vio67wFuYq%252Bb8K7iGUxVA%3D%3D) along with the [EDA Technology ED-CM5ACOOLER](https://www.mouser.pl/ProductDetail/EDATEC/ED-CM5ACooler?qs=7%2F6SraaimPQ26TcfHFYapQ%3D%3D) active cooler. 

[image] [image]
 
The fan connector fits perfectly into the fan header on the board, and the cable length is sufficient. However, only two of the four cooler mounting holes aligned properly. The other two are blocked by the Ethernet and HDMI ports, so I could only loosely secure the cooler.

[image] [image]

I connected HDMI for monitoring, Ethernet for internet (via DHCP), and power via USB-C using the [official RPi 5 power supply](https://www.mouser.pl/ProductDetail/Raspberry-Pi/SC1157?qs=HoCaDK9Nz5fNeKL9V81LRA%3D%3D).  
I flipped the rpiboot switch to OFF to boot from NVMe and not use the onboard eMMC or microSD.

After powering on, the board booted automatically. Two green LEDs lit up.  

[image]

The [Web3 Pi](https://www.web3pi.io/) installer launched without any issues, and after about 10 minutes, MINIMA REV2 became a fully running Ethereum node.  
Synchronization started immediately, and it will be under heavy 100% load for the next 24 hours or so.

[image]

### Summary

Everything I needed worked on the first try - zero issues.  
I’ll report back after full sync is completed and let you know whether [MINIMA REV2](https://github.com/piecol/CM5_MINIMA_REV2) handled the stress of a full Ethereum node sync.

I haven’t yet tested additional features like the I2C display connector or the second USB port. I’ll try those later and update this post.

So far, I’m really happy with the board.  
It’s small but packed with useful ports and features. Build quality is fantastic.  
A very cool [open-source](https://github.com/piecol/CM5_MINIMA_REV2) project!

If I had more time, I’d love to design a custom enclosure for it with an LCD display acting as a dashboard. I already have the [display code ready](https://github.com/Web3-Pi/rpi-lcd-dashboard) - it just needs a proper case :)

geerlingguy, 2025-04-08:
Awesome! Thanks for posting your experience with this tiny board :)

cmd0s, 2025-04-09:
[image]

Synchronization completed without any issues. 24 hours of continuous maximum load was no problem at all.

buildacubesat, 2025-04-13:
I've been using a MINIMA REV2.0 from Pier's prototype run for a few weeks now, and it's a great little board. Plus, he's just a lovely guy who deserves all the support. I'd definitely recommend getting one for any robotics, media streaming, or edge AI application.

Also I'm attempting a respin based on the MINIMA and made a quick video about the process: https://youtu.be/0_Mj95_X1gc