geerlingguy/raspberry-pi-pcie-devices #712: Test RX VEGA56 8G

juliosnovaes, 2025-03-05:
Hey, I’m trying to test this board, but I have no idea what else to do. In my tests, I followed the material [link to the material], but the board isn’t displaying any video. I think it might just be a small detail.

[image]

I used the same settings to test a different board, the RX 580, and that one worked fine.

dmesg:
~~~
[    0.000000] Linux version 6.6.78-v8-16k-amdgpu+ (julionovaes@raspberrypi) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Tue Mar  4 21:20:58 GMT 2025
[    0.659374] usb usb1: Manufacturer: Linux 6.6.78-v8-16k-amdgpu+ xhci-hcd
[    0.659651] usb usb2: Manufacturer: Linux 6.6.78-v8-16k-amdgpu+ xhci-hcd
[    0.660611] usb usb3: Manufacturer: Linux 6.6.78-v8-16k-amdgpu+ xhci-hcd
[    0.660844] usb usb4: Manufacturer: Linux 6.6.78-v8-16k-amdgpu+ xhci-hcd
[    4.239955] [drm] amdgpu kernel modesetting enabled.
[    4.240137] amdgpu 0000:03:00.0: enabling device (0000 -> 0002)
[    4.788947] amdgpu 0000:03:00.0: amdgpu: Fetched VBIOS from ROM BAR
[    4.788959] amdgpu: ATOM BIOS: 113-376XL-UO2
[    4.915116] amdgpu 0000:03:00.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
[    4.915125] amdgpu 0000:03:00.0: amdgpu: PCIE atomic ops is not supported
[    4.981821] amdgpu 0000:03:00.0: amdgpu: MEM ECC is not presented.
[    4.981828] amdgpu 0000:03:00.0: amdgpu: SRAM ECC is not presented.
[    4.981872] amdgpu 0000:03:00.0: BAR 2: releasing [mem 0x1810000000-0x18101fffff 64bit pref]
[    4.981877] amdgpu 0000:03:00.0: BAR 0: releasing [mem 0x1800000000-0x180fffffff 64bit pref]
[    4.981936] amdgpu 0000:03:00.0: BAR 0: assigned [mem 0x1800000000-0x19ffffffff 64bit pref]
[    4.981949] amdgpu 0000:03:00.0: BAR 2: assigned [mem 0x1a00000000-0x1a001fffff 64bit pref]
[    4.982007] amdgpu 0000:03:00.0: amdgpu: VRAM: 8176M 0x000000F400000000 - 0x000000F5FEFFFFFF (8176M used)
[    4.982010] amdgpu 0000:03:00.0: amdgpu: GART: 512M 0x0000000000000000 - 0x000000001FFFFFFF
[    4.982013] amdgpu 0000:03:00.0: amdgpu: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF
[    4.982136] [drm] amdgpu: 8176M of VRAM memory ready
[    4.982139] [drm] amdgpu: 3972M of GTT memory ready.
[    4.982760] amdgpu: hwmgr_sw_init smu backed is vega10_smu
[    5.394852] Modules linked in: bnep hid_logitech_hidpp spidev rp1_pio joydev brcmfmac_wcc amdgpu(+) hci_uart btbcm bluetooth brcmfmac aes_ce_blk vc4 aes_ce_cipher ghash_ce gf128mul sha2_ce sha256_arm64 snd_soc_hdmi_codec sha1_ce cec ecdh_generic brcmutil cfg80211 drm_dma_helper ecc raspberrypi_hwmon snd_soc_core libaes snd_compress snd_pcm_dmaengine i2c_brcmstb snd_pcm amdxcp spi_bcm2835 rpivid_hevc(C) pisp_be drm_exec v4l2_mem2mem videobuf2_dma_contig i2c_algo_bit rfkill drm_buddy drm_suballoc_helper gpio_keys videobuf2_memops videobuf2_v4l2 drm_display_helper v3d snd_timer drm_ttm_helper ttm videodev pwm_fan gpu_sched drm_shmem_helper videobuf2_common snd mc drm_kms_helper hid_logitech_dj binfmt_misc rp1 rp1_mailbox raspberrypi_gpiomem rp1_adc nvmem_rmem uio_pdrv_genirq uio i2c_dev drm ledtrig_pattern fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    5.476537] CPU: 2 PID: 315 Comm: (udev-worker) Tainted: G         C         6.6.78-v8-16k-amdgpu+ #1
[    5.502306] lr : gfx_v9_0_hw_init+0x19fc/0x2980 [amdgpu]
[    5.588679]  amdgpu_device_init+0x1be4/0x2208 [amdgpu]
[    5.594070]  amdgpu_driver_load_kms+0x20/0x1a8 [amdgpu]
[    5.599544]  amdgpu_pci_probe+0x154/0x420 [amdgpu]
[    5.647654]  amdgpu_init+0x6c/0xff8 [amdgpu]
~~~

 lspci: 
~~~
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)  
0000:01:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge (rev c3)  
0000:02:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Vega 10 PCIe Bridge  
0000:03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XL/XT [Radeon RX Vega 56/64\] (rev c3)  
0000:03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 HDMI Audio [Radeon Vega 56/64]  
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 21)  
0001:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
~~~

geerlingguy, 2025-03-06:
For your dmesg logs, can you use `dmesg` and scroll through to the part where it says `amdgpu_device_init+0x1be4/0x2208` and copy out that whole area (not just the lines that have `amdgpu`)?

juliosnovaes, 2025-03-06:
I was using grep and didn't see the rest of the output

[    5.077682]  amdgpu_device_init+0x1be4/0x2208 [amdgpu]
[    5.083071]  amdgpu_driver_load_kms+0x20/0x1a8 [amdgpu]
[    5.088543]  amdgpu_pci_probe+0x154/0x420 [amdgpu]
[    5.093579]  local_pci_probe+0x48/0xb8
[    5.097337]  pci_device_probe+0xac/0x1c8
[    5.101268]  really_probe+0x150/0x2c0
[    5.104938]  __driver_probe_device+0x80/0x140
[    5.109305]  driver_probe_device+0x44/0x170
[    5.113497]  __driver_attach+0x9c/0x1b0
[    5.117341]  bus_for_each_dev+0x80/0xe8
[    5.121186]  driver_attach+0x2c/0x40
[    5.124768]  bus_add_driver+0xec/0x218
[    5.128523]  driver_register+0x68/0x138
[    5.132367]  __pci_register_driver+0x54/0x68
[    5.136647]  amdgpu_init+0x6c/0xff8 [amdgpu]
[    5.141159]  do_one_initcall+0x60/0x2c0
[    5.145004]  do_init_module+0x60/0x218
[    5.148762]  load_module+0x1dd0/0x2080
[    5.152519]  __do_sys_init_module+0x19c/0x1e0
[    5.156888]  __arm64_sys_init_module+0x24/0x38
[    5.161344]  invoke_syscall+0x50/0x128
[    5.165102]  el0_svc_common.constprop.0+0xc8/0xf0
[    5.169819]  do_el0_svc+0x24/0x38
[    5.173139]  el0_svc+0x38/0xd0
[    5.176198]  el0t_64_sync_handler+0x100/0x130
[    5.180565]  el0t_64_sync+0x190/0x198
[    5.184235] Code: 91010108 54ffff4a 8b040108 cb050042 (d50b7428) 
[    5.190348] ---[ end trace 0000000000000000 ]---
[   20.749954] detected fb_set_par error, error code: -16

geerlingguy, 2025-03-06:
Just to get it all - can you clip all the output from the part that's like `--- [cut here] --- ` above that?

Usually that entire chunk of output helps to find the exact part of the code that's failing. In this case, probably just a line that needs tweaking inside the Vega 56 driver series.

juliosnovaes, 2025-03-06:
Sorry, but I don't feel that line. I'm sending the full log.
Could it have something to do with the page size that I changed to 4k as recommended for running box86?

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.78-v8-16k-amdgpu+ (julionovaes@raspberrypi) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Tue Mar  4 21:20:58 GMT 2025
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000001c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000001c00000..0x0000000005bfffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165e0..0x000000003fd16617 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x1fef973c0-0x1fef99fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 2048 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 30 pages/cpu s85928 r8192 d28760 u122880
[    0.000000] pcpu-alloc: s85928 r8192 d28760 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: SSBS not fully self-synchronizing
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  smsc95xx.macaddr=2C:CF:67:1D:2F:56 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=42d07cd6-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=BR
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0'
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2062336
[    0.000000] Policy zone: Normal
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbfff000-0x00000000fffff000] (64MB)
[    0.000000] Memory: 8048228K/8380416K available (13504K kernel code, 2208K rwdata, 4292K rodata, 4864K init, 1086K bss, 266652K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43565 entries in 171 pages
[    0.000000] ftrace: allocated 171 pages with 5 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[... 684 lines cut ...]
[    5.963746]  invoke_syscall+0x50/0x128
[    5.967504]  el0_svc_common.constprop.0+0xc8/0xf0
[    5.972221]  do_el0_svc+0x24/0x38
[    5.975541]  el0_svc+0x38/0xd0
[    5.978599]  el0t_64_sync_handler+0x100/0x130
[    5.982966]  el0t_64_sync+0x190/0x198
[    5.986636] Code: 91010108 54ffff4a 8b040108 cb050042 (d50b7428) 
[    5.992749] ---[ end trace 0000000000000000 ]---
[... cut ...]
[   15.704140] logitech-hidpp-device 0003:046D:4023.0003: HID++ 2.0 device connected.
[  216.646848] warning: `ThreadPoolForeg' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[  218.197650] systemd[1]: Starting upower.service - Daemon for power management...
[  218.316103] systemd[1]: Started upower.service - Daemon for power management.
[  273.267097] systemd[1]: systemd-journald.service: State 'stop-watchdog' timed out. Killing.
[  273.275570] systemd[1]: systemd-journald.service: Killing process 289 (systemd-journal) with signal SIGKILL.

geerlingguy, 2025-03-07:
Thanks! If @coreforge could take a quick gander, is it just one more memcpy or something that's missing for Vega?

Coreforge, 2025-03-07:
It'll need the same fixes the other generations needed too (that memset and maybe a few others, and the structs). I should be able to hopefully add that tomorrow if I remember to

juliosnovaes, 2025-03-11:
Sorry for the delay, I just managed to do the new build now. I still can't get video with Vega. Now the fans start and stop right away.

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.12.9-v8-16k-vega+ (julionovaes@raspberrypi) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Tue Mar 11 01:49:09 GMT 2025
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000005ffffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165e0..0x000000003fd16617 (0 KiB) nomap non-reusable nvram@0
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x1fefa2e80-0x1fefa5b7f]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 1024 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 33 pages/cpu s95000 r8192 d31976 u135168
[    0.000000] pcpu-alloc: s95000 r8192 d31976 u135168 alloc=33*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: SSBS not fully self-synchronizing
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  smsc95xx.macaddr=2C:CF:67:1D:2F:56 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=42d07cd6-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=BR
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] mempolicy: NUMA default policy overridden to 'interleave:0'
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Fallback order for Node 0: 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2096128
[    0.000000] Policy zone: Normal
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbfff000-0x00000000fffff000] (64MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 44978 entries in 176 pages
[    0.000000] ftrace: allocated 176 pages with 3 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[... 712 lines cut ...]
[    9.621294] x29: ffffffc08315bbf0 x28: 0000000000000000 x27: 0000000000000004
[    9.628455] x26: 00000000000001fc x25: 00000000000001fa x24: 00000000ffe00000
[    9.635616] x23: 0000000000000000 x22: 0000000000000000 x21: 0000000000000004
[    9.642777] x20: 0000000000000028 x19: ffffff8110a18818 x18: 0000000000000000
[... cut ...]
[    9.721947]  kthread+0x11c/0x128
[    9.725180]  ret_from_fork+0x10/0x20
[    9.728764] Code: a97e2c8a a97f348c f101005f 540000e8 (a9001c06) 
[    9.734877] ---[ end trace 0000000000000000 ]---
[   19.420943] amdgpu 0000:03:00.0: amdgpu: Dumping IP State
[   19.429503] amdgpu 0000:03:00.0: amdgpu: Dumping IP State Completed
[   19.435852] amdgpu 0000:03:00.0: amdgpu: ring gfx timeout, signaled seq=3, emitted seq=4
[   19.443985] amdgpu 0000:03:00.0: amdgpu: Process information: process labwc pid 847 thread labwc:cs0 pid 984

MollySophia, 2025-05-03:
I've got exactly the same crash trace as your latest one. You can try this patch on @Coreforge 's `rpi-6.12.y-gpu` branch:
```
From 778e3925ee72d995b9548f8fe25798817eb38742 Mon Sep 17 00:00:00 2001
From: Molly Sophia <mollysophia379@gmail.com>
Date: Sat, 3 May 2025 19:27:59 +0800
Subject: [PATCH] another gfx9 fix

---
 drivers/gpu/drm/amd/amdgpu/amdgpu_ring.h | 5 +++--
 1 file changed, 3 insertions(+), 2 deletions(-)

diff --git a/drivers/gpu/drm/amd/amdgpu/amdgpu_ring.h b/drivers/gpu/drm/amd/amdgpu/amdgpu_ring.h
index f93f51002..11f0cf09f 100644
--- a/drivers/gpu/drm/amd/amdgpu/amdgpu_ring.h
+++ b/drivers/gpu/drm/amd/amdgpu/amdgpu_ring.h
@@ -28,6 +28,7 @@
 #include <drm/gpu_scheduler.h>
 #include <drm/drm_print.h>
 #include <drm/drm_suballoc.h>
+#include <linux/io.h>
 
 struct amdgpu_device;
 struct amdgpu_ring;
@@ -402,12 +403,12 @@ static inline void amdgpu_ring_write_multiple(struct amdgpu_ring *ring,
 	chunk2 <<= 2;
 
 	if (chunk1)
-		memcpy(dst, src, chunk1);
+		memcpy_toio(dst, src, chunk1);
 
 	if (chunk2) {
 		src += chunk1;
 		dst = (void *)ring->ring;
-		memcpy(dst, src, chunk2);
+		memcpy_toio(dst, src, chunk2);
 	}
 
 	ring->wptr += count_dw;
-- 
2.49.0

```

This makes my MI50 working on rpi5.

[image]

[image]