geerlingguy/raspberry-pi-pcie-devices #715: Add Switchberry - A Raspberry Pi CM 1 Gbps Managed Switch

geerlingguy, 2025-03-20:
The Switchberry is an OCP-TAP open source hardware project to build an open 1 Gbps managed Ethernet switch with PTP capabilities:

> The PCB integrates a managed 7-port Gigabit Ethernet switch, exposing various interfaces and peripherals for a versatile and cost-effective 1G network switch with precision time protocol (PTP) capabilities.

TODO: Photo here.

I have one on hand and will be testing it soon, now that I have the Time HAT up and running on a Pi 5: #674

geerlingguy, 2025-03-20:
Copying over the Getting Started guide from the Switchberry README:

## Getting Started

1. **Hardware Setup**:
   - Insert the Compute Module 4 (CM4) into the PCB.
   - Connect peripherals such as HDMI, Ethernet, and USB as needed.
   - Connect network cables to the RJ45 connectors and SFP slot for the desired network configuration.

2. **Software Setup**:
   - Flash the Raspberry Pi CM4 with a compatible OS (such as Raspberry Pi OS).
   - Set up the SPI interface to manage the KSZ9567 switch via CM4.
   - Configure PTP settings as required for your application.

3. **PTP Configuration**:
   - Configure the switch’s Transparent Clock or Boundary Clock mode to enable precise time synchronization.
   - Use PTP software tools on the CM4 to manage and monitor network timing.

My plan is to go from that to either a full Ansible playbook to configure everything, or at least to have the shell commands so you could set it up by hand!

geerlingguy, 2025-05-22:
Setup steps I've taken so far:

  1. Flash Pi OS to a CM5 with eMMC (haven't tested alternative boot methods yet)
  2. Boot the CM5
  3. Enable SPI: `sudo raspi-config nonint do_spi 0` (yes, [`0` enables SPI](https://www.raspberrypi.com/documentation/computers/configuration.html#spi-nonint), by adding `dtparam=spi=on` in `/boot/firmware/config.txt`)
  4. Reboot the CM5

At this point, I see SPI loaded on the CM5:

```
$ lsmod | grep spi
spidev                 49152  0
spi_bcm2835            49152  0
spi_dw_mmio            49152  0
spi_dw                 49152  1 spi_dw_mmio

$ ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 May 22 13:27 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 May 22 13:27 /dev/spidev0.1
crw-rw---- 1 root spi 153, 2 May 22 13:27 /dev/spidev10.0
```

I'm thinking a Python program using [spidev](https://learn.sparkfun.com/tutorials/raspberry-pi-spi-and-i2c-tutorial/all) would be necessary to configure things, but that seems like riding a bicycle with your hands.

Microchip has an [Application Note for SPI configuration of their switch chips](https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ApplicationNotes/ApplicationNotes/00002577A.pdf), but the use is a little obtuse to this SPI noob.

And they refer to a web GUI for one of their evaluation boards that runs on... Ubuntu 14.04 :D — see https://github.com/Microchip-Ethernet/EVB-KSZ9897/releases/tag/LAN7801_KSZ9897_V_1.1

I also found this newer repository [Microchip-Ethernet/EVB-KSZ9477](https://github.com/Microchip-Ethernet/EVB-KSZ9477) with a Web GUI, but it seems that, too, is meant to build a relatively ancient custom Linux ISO for an evaluation board...

redat00, 2025-05-23:
Very very cool looking board !

I've been working on the subject of having my own little switch for a while now, with a board similar to this one. 

During my research I learned about [switchdev](https://docs.kernel.org/networking/switchdev.html) which is a solution built-in the Kernel module to allow the communication with a switch IC just as the one present on this board more easily. It's a really cool way to do it, as it then allow the port to be managed your Linux commands such as `ifconfig` etc.. 

Unfortunately it seems like the switch IC used on this board is not supported, as no driver exist.. 😔

If such driver existed (like for the [KSZ9477](https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/net/dsa/microchip/ksz9477.c))  then the board would be perfect, I think! 

If someone knows how to write such drivers, it's their time to shine !

ahmadexp, 2025-05-23:
@redat00 Come join us and let's make this board your dream board too. Julian already wrote the DSA kernel driver for it.

geerlingguy, 2025-05-23:
Full diff of the bcm2711 defconfig defaults from the latest kernel to the `.config` provided by Julian:

```patch
< # Linux/arm64 6.12.22 Kernel Configuration
---
> # Linux/arm64 6.12.30 Kernel Configuration
20a21
> CONFIG_LD_CAN_USE_KEEP_IN_OVERLAY=y
32c33
< CONFIG_LOCALVERSION="-v8-DSA"
---
> CONFIG_LOCALVERSION="-v8"
1515,1539c1516
< CONFIG_NET_DSA=m
< CONFIG_NET_DSA_TAG_NONE=m
< CONFIG_NET_DSA_TAG_AR9331=m
< CONFIG_NET_DSA_TAG_BRCM_COMMON=m
< CONFIG_NET_DSA_TAG_BRCM=m
< CONFIG_NET_DSA_TAG_BRCM_LEGACY=m
< CONFIG_NET_DSA_TAG_BRCM_PREPEND=m
< CONFIG_NET_DSA_TAG_HELLCREEK=m
< CONFIG_NET_DSA_TAG_GSWIP=m
< CONFIG_NET_DSA_TAG_DSA_COMMON=m
< CONFIG_NET_DSA_TAG_DSA=m
< CONFIG_NET_DSA_TAG_EDSA=m
< CONFIG_NET_DSA_TAG_MTK=m
< CONFIG_NET_DSA_TAG_KSZ=m
< CONFIG_NET_DSA_TAG_OCELOT=m
< CONFIG_NET_DSA_TAG_OCELOT_8021Q=m
< CONFIG_NET_DSA_TAG_QCA=m
< CONFIG_NET_DSA_TAG_RTL4_A=m
< CONFIG_NET_DSA_TAG_RTL8_4=m
< CONFIG_NET_DSA_TAG_RZN1_A5PSW=m
< CONFIG_NET_DSA_TAG_LAN9303=m
< CONFIG_NET_DSA_TAG_SJA1105=m
< CONFIG_NET_DSA_TAG_TRAILER=m
< CONFIG_NET_DSA_TAG_VSC73XX_8021Q=m
< CONFIG_NET_DSA_TAG_XRS700X=m
---
> # CONFIG_NET_DSA is not set
1655c1632
< CONFIG_DCB=y
---
> # CONFIG_DCB is not set
1678c1655
< CONFIG_NET_SWITCHDEV=y
---
> # CONFIG_NET_SWITCHDEV is not set
1753a1731
> CONFIG_BT_MTK=m
1758c1736
< # CONFIG_BT_HCIBTUSB_MTK is not set
---
> CONFIG_BT_HCIBTUSB_MTK=y
1852d1829
< CONFIG_NET_IEEE8021Q_HELPERS=y
2611,2659d2587
< 
< #
< # Distributed Switch Architecture drivers
< #
< CONFIG_B53=m
< CONFIG_B53_SPI_DRIVER=m
< CONFIG_B53_MDIO_DRIVER=m
< CONFIG_B53_MMAP_DRIVER=m
< CONFIG_B53_SRAB_DRIVER=m
< CONFIG_B53_SERDES=m
< CONFIG_NET_DSA_BCM_SF2=m
< CONFIG_NET_DSA_LOOP=m
< CONFIG_NET_DSA_LANTIQ_GSWIP=m
< CONFIG_NET_DSA_MT7530=m
< CONFIG_NET_DSA_MT7530_MDIO=m
< CONFIG_NET_DSA_MT7530_MMIO=m
< CONFIG_NET_DSA_MV88E6060=m
< CONFIG_NET_DSA_MICROCHIP_KSZ_COMMON=m
< CONFIG_NET_DSA_MICROCHIP_KSZ9477_I2C=m
< CONFIG_NET_DSA_MICROCHIP_KSZ_SPI=m
< CONFIG_NET_DSA_MICROCHIP_KSZ_PTP=y
< CONFIG_NET_DSA_MICROCHIP_KSZ8863_SMI=m
< CONFIG_NET_DSA_MV88E6XXX=m
< CONFIG_NET_DSA_MV88E6XXX_PTP=y
< CONFIG_NET_DSA_MSCC_FELIX_DSA_LIB=m
< CONFIG_NET_DSA_MSCC_OCELOT_EXT=m
< CONFIG_NET_DSA_MSCC_SEVILLE=m
< CONFIG_NET_DSA_AR9331=m
< CONFIG_NET_DSA_QCA8K=m
< CONFIG_NET_DSA_QCA8K_LEDS_SUPPORT=y
< CONFIG_NET_DSA_SJA1105=m
< CONFIG_NET_DSA_SJA1105_PTP=y
< CONFIG_NET_DSA_XRS700X=m
< CONFIG_NET_DSA_XRS700X_I2C=m
< CONFIG_NET_DSA_XRS700X_MDIO=m
< CONFIG_NET_DSA_REALTEK=m
< CONFIG_NET_DSA_REALTEK_MDIO=y
< CONFIG_NET_DSA_REALTEK_SMI=y
< CONFIG_NET_DSA_REALTEK_RTL8365MB=m
< CONFIG_NET_DSA_REALTEK_RTL8366RB=m
< CONFIG_NET_DSA_REALTEK_RTL8366RB_LEDS=y
< CONFIG_NET_DSA_SMSC_LAN9303=m
< CONFIG_NET_DSA_SMSC_LAN9303_I2C=m
< CONFIG_NET_DSA_SMSC_LAN9303_MDIO=m
< CONFIG_NET_DSA_VITESSE_VSC73XX=m
< CONFIG_NET_DSA_VITESSE_VSC73XX_SPI=m
< CONFIG_NET_DSA_VITESSE_VSC73XX_PLATFORM=m
< # end of Distributed Switch Architecture drivers
< 
2762d2689
< CONFIG_IXGBE_DCB=y
2766d2692
< CONFIG_I40E_DCB=y
2775d2700
< CONFIG_ADIN1110=m
2786,2787d2710
< CONFIG_PRESTERA=m
< CONFIG_PRESTERA_PCI=m
2805,2808c2728
< CONFIG_LAN966X_SWITCH=m
< CONFIG_LAN966X_DCB=y
< CONFIG_VCAP=y
< CONFIG_FDMA=y
---
> # CONFIG_VCAP is not set
2810,2811d2729
< CONFIG_MSCC_OCELOT_SWITCH_LIB=m
< CONFIG_MSCC_OCELOT_SWITCH=m
2860d2777
< CONFIG_ROCKER=m
2950,2951c2867
< CONFIG_MEDIATEK_GE_PHY=m
< # CONFIG_MEDIATEK_GE_SOC_PHY is not set
---
> # CONFIG_MEDIATEK_GE_PHY is not set
2981c2897
< CONFIG_VITESSE_PHY=m
---
> # CONFIG_VITESSE_PHY is not set
3040c2956
< CONFIG_MDIO_GPIO=m
---
> # CONFIG_MDIO_GPIO is not set
3043c2959
< CONFIG_MDIO_MSCC_MIIM=m
---
> # CONFIG_MDIO_MSCC_MIIM is not set
3059,3061c2975
< CONFIG_PCS_XPCS=m
< CONFIG_PCS_LYNX=m
< CONFIG_PCS_MTK_LYNXI=m
---
> # CONFIG_PCS_XPCS is not set
3274a3189
> CONFIG_MT7925_COMMON=m
3276c3191
< # CONFIG_MT7925U is not set
---
> CONFIG_MT7925U=m
4229c4144
< # CONFIG_SENSORS_POWERZ is not set
---
> CONFIG_SENSORS_POWERZ=m
4321c4236
< # CONFIG_SENSORS_ADS7828 is not set
---
> CONFIG_SENSORS_ADS7828=m
4504c4419
< CONFIG_MFD_OCELOT=m
---
> # CONFIG_MFD_OCELOT is not set
4646a4562
> CONFIG_REGULATOR_WAVESHARE_TOUCHSCREEN=m
5186c5102
< # CONFIG_VIDEO_IMX335 is not set
---
> CONFIG_VIDEO_IMX335=m
5637a5554,5555
> CONFIG_DRM_CLIENT_SELECTION=y
> CONFIG_DRM_CLIENT_SETUP=y
5787a5706
> CONFIG_DRM_PANEL_WAVESHARE_TOUCHSCREEN_V2=m
6572a6492
> # CONFIG_HID_UNIVERSAL_PIDFF is not set
8030c7950
< # CONFIG_TCS3472 is not set
---
> CONFIG_TCS3472=m
8040c7960
< # CONFIG_VEML6040 is not set
---
> CONFIG_VEML6040=m
8589c8509,8520
< # CONFIG_EROFS_FS is not set
---
> CONFIG_EROFS_FS=m
> # CONFIG_EROFS_FS_DEBUG is not set
> CONFIG_EROFS_FS_XATTR=y
> CONFIG_EROFS_FS_POSIX_ACL=y
> CONFIG_EROFS_FS_SECURITY=y
> CONFIG_EROFS_FS_BACKED_BY_FILE=y
> CONFIG_EROFS_FS_ZIP=y
> # CONFIG_EROFS_FS_ZIP_LZMA is not set
> # CONFIG_EROFS_FS_ZIP_DEFLATE is not set
> # CONFIG_EROFS_FS_ZIP_ZSTD is not set
> # CONFIG_EROFS_FS_ONDEMAND is not set
> # CONFIG_EROFS_FS_PCPU_KTHREAD is not set
8718c8649,8650
< # CONFIG_UNICODE is not set
---
> CONFIG_UNICODE=m
> # CONFIG_UNICODE_NORMALIZATION_SELFTEST is not set
9041c8973
< CONFIG_PACKING=y
---
> # CONFIG_PACKING is not set
9064c8996
< CONFIG_CRYPTO_ARCH_HAVE_LIB_CHACHA=m
---
> CONFIG_CRYPTO_ARCH_HAVE_LIB_CHACHA=y
9065a8998
> CONFIG_CRYPTO_LIB_CHACHA_INTERNAL=m
9067a9001
> CONFIG_CRYPTO_LIB_CURVE25519_INTERNAL=m
9071c9005
< CONFIG_CRYPTO_ARCH_HAVE_LIB_POLY1305=m
---
> CONFIG_CRYPTO_ARCH_HAVE_LIB_POLY1305=y
9072a9007
> CONFIG_CRYPTO_LIB_POLY1305_INTERNAL=m
9449c9384
< # CONFIG_FTRACE_SYSCALLS is not set
---
> CONFIG_FTRACE_SYSCALLS=y
```

geerlingguy, 2025-05-26:
After going through all the instructions from @julianstj1 (listed a few comments earlier), I _believe_ I have the switch set up as a transparent clock finally, with hardware forwarding.

Next up... how to confirm the settings? (And for future use: how to make it easy to switch between boundary / transparent clock, or change other common settings for the switch?)

geerlingguy, 2025-05-26:
Fun aside, and probably just some strange quirk on this V2 prototype board I'm testing...

Every time I cold boot (have it powered off, and then power it on) with a CM4 installed, the CM4 will not begin powering up.

A CM5 powers up from power off just fine. But a CM4 will not.

However, after trying everything under the sun, I figured out if I unplug it and slowly plug it in, pushing down from the 'back' (the Pi's power/WiFi side) towards the 'front' (the Pi's RAM side), it will power up just fine, and all interfaces work.

It'll even power up and run without the front fully pushed down, but then Ethernet and some other high speed interfaces seem to not work (but they come up fine as I rock the whole thing into place.

Strangest behavior from the Hirose B2B connectors I've ever seen!

hayzamjs, 2025-06-20:
@geerlingguy Thanks so much for your detailed comments above they’ve been incredibly helpful!

I’ve built a similar PCB based on the SwitchBerry design, but I’m still having trouble getting it to work as anything other than a basic L2 switch. Just wondering if you’ve made any further progress since your last update?

geerlingguy, 2026-02-19:
I have received a couple 'V6' units, both of which are fully built out but don't include a built-in GPS module. I am testing one, and I'll be leaving some first time use notes here:

https://github.com/Time-Appliances-Project/Switchberry/issues/2