geerlingguy/raspberry-pi-pcie-devices #717: pi5 support the talos linux

mohang6770, 2025-04-10:
Hi Raspberry Pi Team,

First of all, thank you for everything you’ve built—Raspberry Pi 5 is an amazing piece of hardware, and we’re truly excited about its potential.

I’m currently working on a big client project where we’re using Kubernetes, and Talos Linux is our OS of choice because of its security and simplicity. Unfortunately, Talos doesn’t yet work on the Pi 5, mainly due to bootloader and hardware compatibility issues.

It would mean a lot if you could support or guide the Talos community with what’s needed to get Pi 5 working—whether that’s details around the boot process, device tree support, or anything else that could help.

This support would be a huge help to us and many others looking to use Raspberry Pi 5 in production environments with modern OS options like Talos.

Really appreciate your time and everything you’ve done for the community 🙏

6by9, 2025-04-10:
Wrong repo - this is Jeff's repo for testing PCIe devices on Pi4/5.

https://github.com/raspberrypi/linux/issues if you want a real response from Raspberry Pi.

geerlingguy, 2025-04-10:
Closing and recommending posting this elsewhere as I don't control Talos Linux support on the Pi ;)