geerlingguy/raspberry-pi-pcie-devices #718: Add 52Pi / GeeekPi U2500 M.2 NVME Dual USB3.0 to 2.5g Ethernet for Raspberry pi 5

idcrook, 2025-04-13:
# 52Pi U2500 M.2 NVME Dual USB3.0 to 2.5g Ethernet for Raspberry pi 5
 
- SKU: EP-0235
- Wiki: [EP-0235 - 52Pi Wiki](https://wiki.52pi.com/index.php?title=EP-0235)

## Description

The U2500 M.2 NVMe & dual 2.5G ethernet for Raspberry Pi 5. The Raspberry Pi HAT (Hardware Attached on Top) board you have is a versatile expansion module designed to enhance the capabilities of your Raspberry Pi 5.

[image]

idcrook, 2025-04-14:
Silkscreen on PCB  has `U2500D v1.2` and `w02`

## NVMe

```
$ lspci -v -nn -s 0001:01:00
0001:01:00.0 Non-Volatile memory controller [0108]: Phison Electronics Corporation PS5019-E19 PCIe4 NVMe Controller (DRAM-less) [1987:5019] (rev 01) (prog-if 02 [NVM Express])
	Subsystem: Phison Electronics Corporation PS5019-E19 PCIe4 NVMe Controller (DRAM-less) [1987:5019]
	Flags: bus master, fast devsel, latency 0, IRQ 38
	Memory at 1b00000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: <access denied>
	Kernel driver in use: nvme
```

## USB / Ethernet

```
$ lsusb
Bus 004 Device 002: ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0665:6000 Cypress Semiconductor Gaming keyboard
Bus 003 Device 004: ID 093a:2521 Pixart Imaging, Inc. Optical Mouse
Bus 003 Device 003: ID 0b95:6804 ASIX Electronics Corp. AX68004
Bus 003 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

USB ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN

```
$ dmesg | grep -A2 8156
[    1.082069] usb 2-1: New USB device found, idVendor=0bda, idProduct=8156, bcdDevice=31.04
[    1.082076] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[    1.082079] usb 2-1: Product: USB 10/100/1G/2.5G LAN
--
[    1.129268] usb 4-1: New USB device found, idVendor=0bda, idProduct=8156, bcdDevice=31.04
[    1.129273] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[    1.129275] usb 4-1: Product: USB 10/100/1G/2.5G LAN
```

idcrook, 2025-04-14:
Ethernet links at `Speed: 2500Mb/s`

```
$ ethtool eth1 && ethtool eth2 && ethtool eth0
Settings for eth1:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: MII
	PHYAD: 32
	Transceiver: internal
netlink error: Operation not permitted
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
Settings for eth2:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	                        2500baseT/Full
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        2500baseT/Full
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: MII
	PHYAD: 32
	Transceiver: internal
netlink error: Operation not permitted
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
Settings for eth0:
	Supported ports: [ TP	 MII ]
	Supported link modes:   10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	Supported pause frame use: Transmit-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full
	                        100baseT/Half 100baseT/Full
	                        1000baseT/Half 1000baseT/Full
	Advertised pause frame use: Transmit-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	Link partner advertised pause frame use: Symmetric
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Auto-negotiation: on
	master-slave cfg: preferred slave
	master-slave status: slave
	Port: Twisted Pair
	PHYAD: 1
	Transceiver: external
	MDI-X: Unknown
netlink error: Operation not permitted
	Link detected: yes
```

geerlingguy, 2025-04-14:
Nice looking HAT! I love the functionality, reminds me a lot of the Radxa Dual 2.5G HAT, though with a quite different architecture — #647 

Have you done any testing with both interfaces running full blast through iperf3 at the same time? Seems like this architecture could work out okay.

Taking up both USB 3.0 ports is a little tricky but if you just need a 2.5 Gbps router... this doesn't sacrifice the PCIe bandwidth for that!

geerlingguy, 2025-04-14:
The board is now on the site here: https://pipci.jeffgeerling.com/hats/52pi-nvme-dual-2.5g-eth-hat-u2500-w02.html

idcrook, 2025-04-14:
I haven't done much (any?) testing yet.  I created a custom OpenWRT image using their webpage and will be using that as router testbed setup.

Any pointers to what kinda of testing you'd like to see would be welcome.

geerlingguy, 2025-04-14:
@idcrook - I'd love to see if you can have two computers (or one with a 10+ gbps connection) be `iperf3 -s` servers, and then start two `iperf3 -c [ip address of server]` connections out from the Pi, one over each interface. Setting up the networking for that can be a burden though, because Linux tends to like sending packets over one connection... so you'd get a total of 2.3 Gbps no matter what.

In the past I would set up two separate local networks, or just direct connect the two Ethernet ports to two different computers, with custom IPs for each, so I could be sure packets would flow through the individual ports and not get re-routed since they'd be on the same network.

idcrook, 2025-04-14:
@geerlingguy OK tested using iperf3 on Raspberry Pi OS Desktop 64-bit. here's the details:

# testing 52Pi NVMe + dual 2.5G Eth HAT

I have no 10Gbit Ethernet hardware, so testing is has been done using generic 2.5 Gbit switch.

- Raspberry Pi 5 4GB with NVMe SSD
- Windows, Linux hosts with RTL8125 PCIe cards
- M1 Mac mini with onboard 1G
	-  2.5G in a sonnettech thunderbolt docking station

# iperf3 summaries

- single port  onboard 1G: **944 Mbits/sec** (full speed)
- single port 2.5G : -**2.34 Gbits/sec** (full speed)
- dual port 2.5G : ~**1.0 - 1.3 Gbits/sec** on each
	- all 2.5G ports on the same 2.5G switch
- dual port 2.5G plus 1G onboard:
	- 1G onboard :  **944 Mbits/sec** (full speed)
	- 2.5G: ~**400-900 Mbits/sec** 

**strange observation**
Onboard 1G to macOS iperf (2.5G Eth in docking station)
	- **1.85 Gbits/sec** (double theoretical peak 1G speed)

Guess  Linux IP stack maybe isn't always faithful when it sends packets...

# single port on-board 1G

### Onboard 1G to macOS iperf (1G onboard Mac mini)

a different 1G switch for 1G connection

```
 $ iperf3 -B 192.168.50.38 -c 192.168.50.7
Connecting to host 192.168.50.7, port 5201
[  5] local 192.168.50.38 port 51645 connected to 192.168.50.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   115 MBytes   964 Mbits/sec    0    878 KBytes
[  5]   1.00-2.00   sec   112 MBytes   944 Mbits/sec    0    878 KBytes
[  5]   2.00-3.00   sec   112 MBytes   944 Mbits/sec    0    878 KBytes
[  5]   3.00-4.00   sec   112 MBytes   944 Mbits/sec    0    878 KBytes
[  5]   4.00-5.00   sec   111 MBytes   933 Mbits/sec    0    878 KBytes
[  5]   5.00-6.00   sec   112 MBytes   944 Mbits/sec    0    878 KBytes
[  5]   6.00-7.00   sec   112 MBytes   944 Mbits/sec    0    878 KBytes
[  5]   7.00-8.00   sec   112 MBytes   944 Mbits/sec    0    878 KBytes
[  5]   8.00-9.00   sec   111 MBytes   933 Mbits/sec    0    878 KBytes
[  5]   9.00-10.00  sec   112 MBytes   944 Mbits/sec    0    878 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.10 GBytes   944 Mbits/sec    0             sender
[  5]   0.00-10.01  sec  1.10 GBytes   941 Mbits/sec                  receiver
```

# single port (USB to 2.5GB)

### HAT 2.5G port 1 to Windows iperf

```
.\iperf3.exe -B  192.168.50.33 --server
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.50.39, port 56309
[  5] local 192.168.50.33 port 5201 connected to 192.168.50.39 port 52613
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   278 MBytes  2.33 Gbits/sec
[  5]   1.00-2.00   sec   278 MBytes  2.34 Gbits/sec
[  5]   2.00-3.00   sec   278 MBytes  2.34 Gbits/sec
[  5]   3.00-4.00   sec   278 MBytes  2.34 Gbits/sec
[... 159 lines cut ...]
```

# Onboard 1G to macOS iperf (2.5G Eth in docking station)

```
$ iperf3 -B 192.168.50.38 -c 192.168.50.2
Connecting to host 192.168.50.2, port 5201
[  5] local 192.168.50.38 port 55375 connected to 192.168.50.2 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   222 MBytes  1.87 Gbits/sec   13    445 KBytes
[  5]   1.00-2.00   sec   221 MBytes  1.86 Gbits/sec    0    717 KBytes
[  5]   2.00-3.00   sec   220 MBytes  1.85 Gbits/sec   23    427 KBytes
[  5]   3.00-4.00   sec   220 MBytes  1.85 Gbits/sec    0    707 KBytes
[  5]   4.00-5.00   sec   221 MBytes  1.86 Gbits/sec   25    510 KBytes
[  5]   5.00-6.00   sec   220 MBytes  1.85 Gbits/sec    0    724 KBytes
[  5]   6.00-7.00   sec   221 MBytes  1.86 Gbits/sec    0    732 KBytes
[  5]   7.00-8.00   sec   220 MBytes  1.85 Gbits/sec   22    532 KBytes
[  5]   8.00-9.00   sec   220 MBytes  1.85 Gbits/sec    0    725 KBytes
[  5]   9.00-10.00  sec   221 MBytes  1.86 Gbits/sec    2    619 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.16 GBytes  1.85 Gbits/sec   85             sender
[  5]   0.00-10.00  sec  2.15 GBytes  1.85 Gbits/sec                  receiver

```

geerlingguy, 2025-04-14:
Okay thank you! It's at least showing that the individual ports are capable of 2.5 Gbps each, so I think the main bottlenecks will probably either be people's own networks/switches, or if doing any routing/filtering, the firewall running on the Pi's SoC.

I'm thinking one of these boards would be fun for testing.

The one downside is I'm looking into building a dual 2.5 Gbps router with NVMe storage and an M.2 E-key WiFi slot—or failing that, a USB 3.0 WiFi 6E or 7 stick capable of running a small AP (just a few users).

I'm nearly there with the Radxa HAT, but have had some trouble with the extra PCIe switch on that HAT.

Darktron, 2025-04-22:
I have the older model hat mine is called U5000_v1.1
I have moved to the Radxa Dual 2.5G Router Hat since this one let me down due to USB connections for the RTL8125BG chips theres just many interrupt that can't be redirected to all cores permanently.