geerlingguy/raspberry-pi-pcie-devices #728: RX 6800XT (XfX Speedster SWFT 319 Radeon RX 6800XT)

pigong, 2025-05-21:
I am very impressed how well the driver provided by CoreForge  (6.12.y-gpu) works for my card. I just checked out and built following your suggestions and everything worked. I am using a very similar setup to what you use in the videos, the only difference is that I use a PoE/m.2 combi-hat. The graphics card feels totally stable, I haven't had a single crash (no output whatsoever in ˋdmesg --followˋ) while trying out games, benchmarks and utilities. Thanks very much! I am happy to post stuff like ˋlspci -vvvvˋ-output if there is interest, but as mature as things seem there is no need for that any more? Let me know.

I have one problem, however: no sound, and I have no idea what to do about it. Lspci does show the audio device on the graphics-card: ˋAudio device: Advanced Micro Devices, Inc. [AMD/ATI] Navi 21/23 HDMI/DP Audio Controlerˋ. ˋaplay -lˋ gives two hdmi-related output-lines - not sure if they mean anything: card 0/1: vc4hdmi0/1 [vc4-hdmi-0/1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]. Other than that, I am not sure, where to look. The rpios desktop shows no audio-control in the status-bar.

Am I in the right place to ask for help here, or should I go somewhere else? :) 

Best
.p.

geerlingguy, 2025-05-22:
@pigong - Could you post a picture of your setup, would love to see it! (If not, no big deal!). Would also love a `sudo lspci -vvv` with the output for the card, especially with the audio device. Someday I would like to see what's preventing audio over HDMI on the AMD GPUs, as I haven't seen it work on the Pi, but it works on other Arm systems like my Ampere systems and the Orion O6.

@coreforge also hadn't gotten it working, though nobody's really tested that thoroughly yet. Could just be a missing setting in the kernel build or something. https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/6#issuecomment-1824363174

Coreforge, 2025-05-22:
Audio uses the `snd_hda_intel` driver, which also needs to be enabled. It loads (and prints some errors, but they don't seem to block it). I'm not sure anymore if I got my monitor to show up as an audio device, but I also didn't test it much further than loading the driver.

bagong, 2025-05-22:
I am pigong on my pi and bagong by default. I'll post two quick shots for now, just to show I am not fibbing, and I'll post the lspci as pigong next. It's all still proof of concept stage yet, so ugly pictures for now. I am hoping to assemble a feature rich proof of concept nextcloud server (including transcoding and a chatbot integrated into nextcloud office) using three pi's and your kernel 😉
[image]

[image]

geerlingguy, 2025-05-22:
That's a nice looking card!

If you use @Coreforge's branches, that should be enough—most of the other referenced patches are just based off his branch, meant to be applied to the bare raspberrypi kernel tree.

Coreforge, 2025-05-22:
Another thing, transcoding probably won't quite work (on the GPU). I haven't figured out so far why it's so unstable, but depending on the GPU and mesa version, I've only been able to either encode or decode properly (if any of those), if it doesn't work, I either got corrupted data (ffmpeg was able to mostly play some of the corrupted files) or the GPU VM fault.

pigong, 2025-05-23:
As Coreforge assumed, transcoding - actually simply video playback - is pretty bad: vlc instantly crashes on any video I through at it (it doesn't do so with the internal graphics card). mpv is a bit more tolerant, it plays a old dv-video in avi container, but crashes on a mp4 generated from that video in Handbrake. When playing around with codec alternatives in VLC, one can get it to play the video dv/avi-video, but mp4 always crashes.

Vaainfo produces clean-looking output. So I wonder if there is anything I could do to enhance things? Maybe compile a mediaplayer adjusted to the environment? (I remember it took the pi-people a while to get vlc working well with the internal graphics)... Or is it hopeless because driver support is still too buggy? Can I do anything to help debug?

Coreforge, 2025-05-23:
I've had a little more luck with mesa built from source, but even then I never got it working reliably. I don't really have a good approach to debugging this either unfortunately.

Caojia520Suiyng, 2025-05-27:
Is it faster to use Jetson Orin Super 8G for inference of QWEN models or Raspberry Pi external AMD RX6800XT for inference speed, and deploy large models externally using Orama? Please give some suggestions. Thank you very much