geerlingguy/raspberry-pi-pcie-devices #729: Test GPU (AMD Radeon R9 380X)

420noscope-exe, 2025-05-22:
I just recently got my new 16GB cm5 lite and upgraded my ARM server.  I had an R9 380 sitting in the server, disconnected for quite some time, waiting for the day I could use it.  I would like a server for testing custom game engine builds and compiling ARM binaries for them.  I want vulkan support so I decided to use the amdgpu driver.

[image]

I didn't get it to work, but I think I got pretty far along:

```
sudo apt install git

git clone --depth=1 --branch rpi-6.6.y-gpu https://github.com/Coreforge/linux.git

cd linux

wget https://gist.githubusercontent.com/Coreforge/91da3d410ec7eb0ef5bc8dee24b91359/raw/b4848d1da9fff0cfcf7b601713efac1909e408e8/memcpy_unaligned.c

gcc -shared -fPIC -o memcpy.so memcpy_unaligned.c
sudo mv memcpy.so /usr/local/lib/memcpy.so
sudo nano /etc/ld.so.preload
```

Then I put the following line inside ld.so.preload:
	`/usr/local/lib/memcpy.so`

```

sudo apt install bc bison flex libssl-dev make raspberrypi-kernel-headers

KERNEL=kernel_2712
make bcm2712_defconfig

sudo apt-get install libncurses-dev
make menuconfig

```
Kernel Features > Kernel support for 32-bit EL0 > Fix up misaligned multi-word loads and stores in user space
Kernel Features > Fix up misaligned loads and stores from userspace for 64bit code
Device Drivers > Graphics support > AMD GPU , as well as SI/CIK support
Device Drivers > Graphics support > AMD GPU>Display Engine Configuration>AMD DC Support for Southern Islands ASICs
Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > Force Architecture can write-combine memory

saved as .config

```
make -j6 Image.gz modules dtbs
sudo make -j6 modules_install

sudo cp /boot/firmware/$KERNEL.img /boot/firmware/$KERNEL-backup.img
sudo cp arch/arm64/boot/Image.gz /boot/firmware/$KERNEL.img
sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/firmware/
sudo cp arch/arm64/boot/dts/overlays/*.dtb* /boot/firmware/overlays/
sudo cp arch/arm64/boot/dts/overlays/README /boot/firmware/overlays/

sudo apt install firmware-amd-graphics

sudo reboot
```

I didn't get a display output, but I did see some errors with `dmesg | grep amd`

```
[    3.441096] [drm] amdgpu kernel modesetting enabled.
[    3.441160] amdgpu 0000:01:00.0: of_irq_parse_pci: failed with rc=134
[    3.441301] amdgpu 0000:01:00.0: Unable to change power state from D3cold to D0, device inaccessible
[   15.446930] amdgpu: Doesn't get ack from pf.
[   15.446935] amdgpu: Doesn't get ack from pf, give up
[   15.446940] amdgpu 0000:01:00.0: amdgpu: Fatal error during GPU init
[   15.446943] amdgpu 0000:01:00.0: amdgpu: amdgpu: finishing device.
[   27.447168] amdgpu: Doesn't get ack from pf.
[   39.447397] amdgpu: Doesn't get ack from pf.
[   39.447403] [drm:amdgpu_device_fini_hw [amdgpu]] *ERROR* failed to release exclusive mode on fini
[   39.447675] amdgpu: probe of 0000:01:00.0 failed with error -62
```

```
pasquale@PASARM01:~/linux $ lspci -v
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 30) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 38
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	Memory behind bridge: 80000000-800fffff [size=1M] [32-bit]
	Prefetchable memory behind bridge: 1800000000-1817ffffff [size=384M] [32-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

0000:01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Tonga PRO [Radeon R9 285/380] (rev f1) (prog-if 00 [VGA controller])
	Subsystem: Tul Corporation / PowerColor Tonga PRO [Radeon R9 285/380]
	!!! Unknown header type 7f
	Memory at 1800000000 (64-bit, prefetchable) [size=256M]
	Memory at 1810000000 (64-bit, prefetchable) [size=2M]
	Memory at 1b80000000 (32-bit, non-prefetchable) [size=256K]
	Expansion ROM at 1b80040000 [disabled] [size=128K]
	Kernel modules: amdgpu

0000:01:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Tonga HDMI Audio [Radeon R9 285/380]
	Subsystem: Tul Corporation / PowerColor Tonga HDMI Audio [Radeon R9 285/380]
	!!! Unknown header type 7f
	Memory at 1b80060000 (64-bit, non-prefetchable) [size=16K]

0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2712 PCIe Bridge (rev 30) (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 39
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	Memory behind bridge: 00000000-005fffff [size=6M] [32-bit]
	Prefetchable memory behind bridge: [disabled] [64-bit]
	Capabilities: <access denied>
	Kernel driver in use: pcieport

0001:01:00.0 Ethernet controller: Raspberry Pi Ltd RP1 PCIe 2.0 South Bridge
	Flags: bus master, fast devsel, latency 0, IRQ 39
	Memory at 1f00410000 (32-bit, non-prefetchable) [size=16K]
	Memory at 1f00000000 (32-bit, non-prefetchable) [virtual] [size=4M]
	Memory at 1f00400000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: rp1

```

Using `git clone --depth=1 clone --single-branch --branch rpi-6.6.y https://github.com/raspberrypi/linux` gave no display at all, no dmesg errors, `Fix up misaligned loads and stores from userspace for 64bit code` and `Force Architecture can write-combine memory` were missing as options in `menuconfig`.  Using the official raspberry pi kernel, I also manually applied CoreForge's patch.

```
wget -O amdgpu-pi5.patch https://github.com/geerlingguy/linux/pull/8.patch
git apply -v amdgpu-pi5.patch 
```

I also tried with the most recent commit as per the raspberry pi documentation.  This acted the same as the 6.6 kernel, but the patch also failed to apply.

bagong, 2025-05-22:
As someone who went through basically the same steps 2 days ago (for rpi5 & RX6800XT) with partial success (no audio), and failure (RX6700XT) may I generallize by asking two questions:

- is it necessary to apply any patches (memcopy and/or gerlingguy) if one straightforwardly used coreforges gpu branches?
- should one keep using the 6.6.y-gpu branch for certain models, and the 6.12.y-gpu branch for others?

This is a really cool project, however the info bits are a bit scattered over various texts and videos, and one isn't always sure they represent the current state of development. Thanks for your considerations!

420noscope-exe, 2025-05-22:
I agree, the info is a bit scattered across the net.

1.  As far as I know the memcopy patch is still required, even if you use the coreforge fork.

2. I did not have any luck with the official 6.6.y and 6.12.y branches.  So I cannot recommend them at all.  Use coreforge's fork.

Side note:  I swapped out the 380x for a RX 6400.  It worked immediately, display and full acceleration in chromium.

[image]

Looks like some work needs to be done for 300 series cards.

geerlingguy, 2025-05-22:
Yeah, I'm not sure anyones tried any 300 series AMD cards yet, you may be the first.

Coreforge, 2025-05-23:
6.12.y-gpu is a little bit more up to date than 6.6.y-gpu. Unless there are some issues with 6.12, I don't really see a reason to keep 6.6.

The memcpy library isn't needed anymore on the pi 5, it should now all be handled in the kernel instead. There is a chance it gives a little bit better performance, but I don't know if it really makes a difference in practice. If there are any things that only run with the memcpy library but not without, please tell me (and check dmesg), as that might mean some op codes aren't implemented yet, or aren't implemented correctly.  
Note that this doesn't apply to the pi 4, but that doesn't work with as many GPUs anyways.

It's interesting that the driver errors out and it's not an alignment fault like usually. That probably will make finding the issue a bit harder.

Coreforge, 2025-06-01:
Nice to see that it works!  
`amdgpu_top` should show you more detailed usage information of different parts of the GPU. My guess is that the framerate drops so much with VNC because encoding is probably done on the CPU, so the framebuffer needs to be transferred to the CPU. I've seen the SDMA block being the one with the highest usage a decent bit (and depending on how the copy is done, it might also run into the alignment trap a lot, which would be a big performance hit), and the load on the PCIe link from frequent framebuffer copies isn't too low either.  
If it's still unstable, you can also try disabling `Force architecture can write-combine memory` again. I haven't run into issues with it on the pi5, but that doesn't mean there can't be any.

It'd be interesting to see what happens on the CM4, I never got amdgpu to work there, only radeon. Since this card is supported by both, it might be different, but it also might not be. Debugging on the cm4 is much more annoying though, as it tends to just lock up and not even have the CPU properly reachable over JTAG anymore.