geerlingguy/raspberry-pi-pcie-devices #732: Add new Waveshare boards - CM5 NANO A and B, and CM5 Mini Base A

geerlingguy, 2025-05-22:
[image]

I received these boards in the mail recently, and have been testing them—very very tiny CM5 boards that retain almost all the IO of the official IO board (though shrunk down).

  - [CM5 Mini Base Board B](https://amzn.to/4kimBog) - Pi 5-sized board with a few welcome upgrades, like M.2 NVMe slot on the back, and full-size HDMI (There's also the 'A' with fewer IO options)
  - [CM5 Nano A](https://amzn.to/4ktapke) - very thin board with all the basic IO broken out, same height/width as the CM5
  - [CM5 Nano B](https://amzn.to/3FdnZJP) - similar to the A, but with even more IO, including Gigabit Ethernet and PCIe FFC under the CM5 itself.

geerlingguy, 2025-05-22:
They're on the site here:

  - https://pipci.jeffgeerling.com/boards_cm/waveshare-cm5-nano.html
  - https://pipci.jeffgeerling.com/boards_cm/waveshare-cm5-base.html

closed 2025-05-22 by geerlingguy

kdpuvvadi, 2026-01-22:
Has anyone used this? when i'm trying to mount m.2 ssd, screw for ssd is touching carrier board.