geerlingguy/raspberry-pi-pcie-devices #741: RX 9060XT 16G

martinx72, 2025-06-07:
Got RX9070XT and RX9060XT few days ago.
I tested RX9060XT 16G it with RPi5 16G few hours ago

Kernel is latest 6.12.y  https://github.com/raspberrypi/linux  rpi-6.12.y  cde997ebd50e5fecfa7a2676578f27deb10efd34 commit

kernel was built as RPi official guide + AMD GPU is on in menuconfig ( what i checked are as below )

Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > AMD GPU
Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > AMD GPU > Enable amdgpu support for SI parts 
Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > AMD GPU > Enable amdgpu support for CIK parts
Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > AMD GPU > Always enable userptr write support
Device Drivers > Graphics support > Direct Rendering Manager (XFree86 4.1.0 and higher DRI support) > AMD GPU > Display Engine Configuration > AMD DC support for Southern Islands ASICs

lspci -vv result (AMD partso only)
```
0001:01:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD/ATI] Navi 10 XL Upstream Port of PCI Express Switch (rev 25) (prog-if 00 [Normal decode])
        Subsystem: Gigabyte Technology Co., Ltd Navi 10 XL Upstream Port of PCI Express Switch
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 36
        Region 0: Memory at 1b80100000 (32-bit, non-prefetchable) [size=16K]
        Bus: primary=01, secondary=02, subordinate=03, sec-latency=0
        I/O behind bridge: 0000f000-00000fff [disabled] [32-bit]
        Memory behind bridge: 80000000-800fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: 1800000000-1817ffffff [size=384M] [32-bit]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: [48] Vendor Specific Information: Len=08 <?>
        Capabilities: [50] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [58] Express (v2) Upstream Port, MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0
                        ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ SlotPowerLimit 0W
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 256 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
                LnkCap: Port #0, Speed 32GT/s, Width x16, ASPM L1, Exit Latency L1 <64us
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 8GT/s (downgraded), Width x1 (downgraded)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR+
                         10BitTagComp+ 10BitTagReq+ OBFF Not Supported, ExtFmt+ EETLPPrefix+, MaxEETLPPrefixes 4
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-
                         AtomicOpsCap: Routing+ 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: EgressBlck-
                LnkCap2: Supported Link Speeds: 2.5-32GT/s, Crosslink- Retimer+ 2Retimers+ DRS-
                LnkCtl2: Target Link Speed: 32GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
                         EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [c0] Subsystem: Gigabyte Technology Co., Ltd Navi 10 XL Upstream Port of PCI Express Switch
        Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
[... 190 lines cut ...]
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [2a0 v1] Access Control Services
                ACSCap: SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl- DirectTrans-
                ACSCtl: SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl- DirectTrans-
        Kernel driver in use: snd_hda_intel
        Kernel modules: snd_hda_intel

```

geerlingguy, 2025-06-08:
@martinx72 - Thanks for the details! Looks like maybe the 90-series cards are unhappy because those versions don't have the little tweaks we've had to make for the previous series in @Coreforge's patches.

`amdgpu_irq_put` seems like maybe there's a bad memory access in there.

I don't yet have a 9070 or 9060 to test with. I should probably try to pick one up, now that they're not so unobtanium!

Coreforge, 2025-06-09:
Can you retry it with my 6.12 branch?

martinx72, 2025-06-18:
> Can you retry it with my 6.12 branch?

Sure, will do it now.
Sorry for my super late reply. got covid once again somehow somewhere i dont know. 
just getting better today.

martinx72, 2025-06-18:
> Can you retry it with my 6.12 branch?

Here is the result

```
[    8.055908] sr 1:0:0:0: Attached scsi CD-ROM sr0
[    8.269729] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[    8.270546] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[    8.274859] [drm] amdgpu kernel modesetting enabled.
[    8.275066] amdgpu 0001:03:00.0: enabling device (0000 -> 0002)
[    8.275086] [drm] initializing kernel modesetting (IP DISCOVERY 0x1002:0x7590 0x1458:0x2429 0xC0).
[    8.275101] [drm] register mmio base: 0x00000000
[    8.275102] [drm] register mmio size: 524288
[    8.279629] [drm] add ip block number 0 <soc24_common>
[    8.279635] [drm] add ip block number 1 <gmc_v12_0>
[    8.279636] [drm] add ip block number 2 <ih_v7_0>
[    8.279638] [drm] add ip block number 3 <psp>
[    8.279640] [drm] add ip block number 4 <smu>
[    8.279642] [drm] add ip block number 5 <dm>
[    8.279644] [drm] add ip block number 6 <gfx_v12_0>
[    8.279646] [drm] add ip block number 7 <sdma_v7_0>
[    8.279648] [drm] add ip block number 8 <vcn_v5_0_0>
[    8.279650] [drm] add ip block number 9 <jpeg_v5_0_0>
[    8.279651] [drm] add ip block number 10 <mes_v12_0>
[    8.288544] amdgpu 0001:03:00.0: amdgpu: Fetched VBIOS from ROM BAR
[    8.288556] amdgpu: ATOM BIOS: 113-R906XTGO-F1
[    8.353768] amdgpu 0001:03:00.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
[    8.353786] amdgpu 0001:03:00.0: amdgpu: PCIE atomic ops is not supported
[    8.353803] [drm] GPU posting now...
[    8.353911] [drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
[    8.353931] amdgpu 0001:03:00.0: BAR 2 [mem 0x1810000000-0x18101fffff 64bit pref]: releasing
[    8.353935] amdgpu 0001:03:00.0: BAR 0 [mem 0x1800000000-0x180fffffff 64bit pref]: releasing
[    8.353958] pcieport 0001:02:00.0: bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]: releasing
[    8.353961] pcieport 0001:01:00.0: bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]: releasing
[    8.353963] pcieport 0001:00:00.0: bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]: releasing
[    8.353972] pcieport 0001:00:00.0: bridge window [mem size 0x600000000 64bit pref]: can't assign; no space
[    8.353974] pcieport 0001:00:00.0: bridge window [mem size 0x600000000 64bit pref]: failed to assign
[    8.353977] pcieport 0001:01:00.0: bridge window [mem size 0x600000000 64bit pref]: can't assign; no space
[    8.353979] pcieport 0001:01:00.0: bridge window [mem size 0x600000000 64bit pref]: failed to assign
[    8.353981] pcieport 0001:02:00.0: bridge window [mem size 0x600000000 64bit pref]: can't assign; no space
[    8.353983] pcieport 0001:02:00.0: bridge window [mem size 0x600000000 64bit pref]: failed to assign
[    8.353986] amdgpu 0001:03:00.0: BAR 0 [mem size 0x400000000 64bit pref]: can't assign; no space
[    8.353987] amdgpu 0001:03:00.0: BAR 0 [mem size 0x400000000 64bit pref]: failed to assign
[    8.353989] amdgpu 0001:03:00.0: BAR 2 [mem size 0x00200000 64bit pref]: can't assign; no space
[    8.353991] amdgpu 0001:03:00.0: BAR 2 [mem size 0x00200000 64bit pref]: failed to assign
[    8.353993] pcieport 0001:00:00.0: PCI bridge to [bus 01-03]
[    8.353996] pcieport 0001:00:00.0:   bridge window [mem 0x1b00000000-0x1b001fffff]
[    8.353999] pcieport 0001:00:00.0: PCI bridge to [bus 01-03]
[    8.354001] pcieport 0001:00:00.0:   bridge window [mem 0x1b00000000-0x1b001fffff]
[    8.354002] pcieport 0001:00:00.0:   bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]
[    8.354005] pcieport 0001:01:00.0: PCI bridge to [bus 02-03]
[    8.354009] pcieport 0001:01:00.0:   bridge window [mem 0x1b00000000-0x1b000fffff]
[    8.354012] pcieport 0001:01:00.0:   bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]
[    8.354018] pcieport 0001:02:00.0: PCI bridge to [bus 03]
[    8.354022] pcieport 0001:02:00.0:   bridge window [mem 0x1b00000000-0x1b000fffff]
[    8.354025] pcieport 0001:02:00.0:   bridge window [mem 0x1800000000-0x1817ffffff 64bit pref]
[    8.354037] [drm] Not enough PCI address space for a large BAR.
[    8.354040] amdgpu 0001:03:00.0: BAR 0 [mem 0x1800000000-0x180fffffff 64bit pref]: assigned
[    8.354050] amdgpu 0001:03:00.0: BAR 2 [mem 0x1810000000-0x18101fffff 64bit pref]: assigned
[    8.354060] amdgpu 0001:03:00.0: amdgpu: VRAM: 16304M 0x0000008000000000 - 0x00000083FAFFFFFF (16304M used)
[    8.354063] amdgpu 0001:03:00.0: amdgpu: GART: 512M 0x0000000000000000 - 0x000000001FFFFFFF
[    8.354066] [drm] Detected VRAM RAM=16304M, BAR=256M
[    8.354068] [drm] RAM width 128bits GDDR6
[    8.354206] [drm] amdgpu: 16304M of VRAM memory ready
[    8.354209] [drm] amdgpu: 8109M of GTT memory ready.
[    8.354229] [drm] GART: num cpu pages 32768, num gpu pages 131072
[    8.354314] amdgpu 0001:03:00.0: amdgpu: PCIE GART of 512M enabled (table at 0x0000008000000000).
[    8.362661] [drm] Loading DMUB firmware via PSP: version=0x00010E00
[    8.378077] Bluetooth: hci0: BCM: chip id 107
[    8.378278] Bluetooth: hci0: BCM: features 0x2f
[... 61 lines cut ...]
[    8.796672]  el0t_64_sync+0x190/0x198
[    8.796676] Code: 91010108 54ffff4a 8b040108 cb050042 (d50b7428)
[    8.796677] ---[ end trace 0000000000000000 ]---
[    8.859998] Process accounting resumed
[    8.927772] Adding 524272k swap on /var/swap.  Priority:-2 extents:6 across:2211840k SS
[... cut ...]
[    9.396018] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    9.406661] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    9.422320] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   13.510052] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   15.095457] Bluetooth: RFCOMM TTY layer initialized
[   15.100381] Bluetooth: RFCOMM socket layer initialized
[   15.105551] Bluetooth: RFCOMM ver 1.11
pi@raspberrypi:~ $
```

kennedyshead, 2025-10-04:
# My findings

I sometimes get over-exited, just tell me if some info is to much or if something is not wanted 😉

Been running for a couple of days now, everything seems "about" stable, there are some strange behavior when something uses the on board GPU while the eGPU is accessed 🤷 (it spikes all cores and swaps out, have not looked deep here I made sure they are not accessed at the same time).

dmesg when this happens:
```
pcieport 0001:01:00.0: Unable to change power state from D3hot to D0, device inaccessible
```

## Setup

### Hardware
* [Pi5 HAT poe+pci](https://www.amazon.com/dp/B0DF712DL8?ref=ppx_yo2ov_dt_b_fed_asin_title)
* [PCIE Occulink](https://www.amazon.com/dp/B0BZVW9VWL?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1)
* [Pi5](https://www.amazon.com/Raspberry-Pi-8GB-SC1112-Quad-core/dp/B0CK2FCG1K/)
* [PSU](https://www.amazon.com/dp/B0C3M56Y13?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1)
* Power to Pi5 supplied over network (increases the heat for the pi, but even though its overclocked it stays ~65°-70° C)

### `/boot/firmvare/config.txt` changes (Do NOT overclock the PI without active cooling, like given by the HAT in this setup!)
```
dtparam=pciex1
dtparam=pciex1_gen=3
arm_freq=3000
gpu_freq=1100
arm_boost=1
```

### kernel build
* [rpi-kernel-repo](https://github.com/raspberrypi/linux.git) `origin/rpi-6.15.y` `b07d612aae961e99fdb0b828ce6d54c183d1833d`
* Added path: [ARM SoCs Lack I/O Coherency](https://github.com/geerlingguy/raspberry-pi-pcie-devices/discussions/756) from yanghaku
* config changes:
[image]
[image]

### amd-firmware
Here I had some issues, so I ended up downloading all *.bin in [linux-firmware-amdgpu](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/) and just overwriting the `/usr/lib/firmware/amdgpu` dir.

### llama.cpp build
```bash
git clone https://github.com/ggml-org/llama.cpp.git
cd llama.cpp
cmake -B build -DGGML_VULKAN=1
cmake --build build -j --config Release
```

## Results

### dmesg 
```
[    7.434181] [drm] amdgpu kernel modesetting enabled.
[    7.434388] amdgpu 0001:03:00.0: enabling device (0000 -> 0002)
[    7.434399] [drm] initializing kernel modesetting (IP DISCOVERY 0x1002:0x7590 0x1043:0x0625 0xC0).
[    7.434416] [drm] register mmio base: 0x80000000
[    7.434426] [drm] register mmio size: 524288
[    7.438549] amdgpu 0001:03:00.0: amdgpu: detected ip block number 0 <soc24_common>
[    7.438553] amdgpu 0001:03:00.0: amdgpu: detected ip block number 1 <gmc_v12_0>
[    7.438556] amdgpu 0001:03:00.0: amdgpu: detected ip block number 2 <ih_v7_0>
[    7.438559] amdgpu 0001:03:00.0: amdgpu: detected ip block number 3 <psp>
[    7.438561] amdgpu 0001:03:00.0: amdgpu: detected ip block number 4 <smu>
[    7.438563] amdgpu 0001:03:00.0: amdgpu: detected ip block number 5 <dm>
[    7.438565] amdgpu 0001:03:00.0: amdgpu: detected ip block number 6 <gfx_v12_0>
[    7.438567] amdgpu 0001:03:00.0: amdgpu: detected ip block number 7 <sdma_v7_0>
[    7.438569] amdgpu 0001:03:00.0: amdgpu: detected ip block number 8 <vcn_v5_0_0>
[    7.438571] amdgpu 0001:03:00.0: amdgpu: detected ip block number 9 <jpeg_v5_0_0>
[    7.438573] amdgpu 0001:03:00.0: amdgpu: detected ip block number 10 <mes_v12_0>
[    7.447726] amdgpu 0001:03:00.0: amdgpu: Fetched VBIOS from ROM BAR
[    7.447732] amdgpu: ATOM BIOS: 115-G264BP20-100
[    7.554581] amdgpu 0001:03:00.0: amdgpu: Trusted Memory Zone (TMZ) feature not supported
[... 354 lines cut ...]

gmake[2]: *** [ggml/src/ggml-vulkan/CMakeFiles/ggml-vulkan.dir/build.make:223: ggml/src/ggml-vulkan/CMakeFiles/ggml-vulkan.dir/ggml-vulkan.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:1414: ggml/src/ggml-vulkan/CMakeFiles/ggml-vulkan.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

### frigate

This is the vigilante using the on-board GPU, hence I need to stop it when using llama.cpp. I use a coral for OD and GPU for ffmpeg-hwaccel but using the `preset-vaapi` flag makes frigate fail to start and the `-hwaccel drm` uses on-board and kills the system when used at the same time as `llama.cpp`

## Of topic (sorry, tell me if you want me to remove)

I think I will get a 16G pi5 as the 8G is swapping a lot (I am pushing it with all services I put on it though 😄). I really would love to have only 1 Pi running all my always on stuff, but that is probably not possible!

Anyhow, this one is running

### Docker compose:
* zigbee2mqtt
* frigate (when not using the amdgpu)
* wyoming-whisper-api-client
* OpenWeb

### system.d
* llama.cpp server (when not using frigate)

geerlingguy, 2025-10-07:
Heh, that's a bit of heavy stuff—I tend to restrict my heavier processing apps like Frigate, whisper, or llama to a single node, though I have plenty of Pis to go around...

kennedyshead, 2025-10-07:
> Heh, that's a bit of heavy stuff—I tend to restrict my heavier processing apps like Frigate, whisper, or llama to a single node, though I have plenty of Pis to go around...

I like to push them to absolute max :D I also have a hard cap on how many Watts are allowed to be used for the automation system :P

## Off topic again and last time. Started a discord thread on home assistant hardware for anyone interested in the topic: [Pi5 with AMDGPU](https://discord.com/channels/330944238910963714/1423983138484260874)

This branch of whisper.cpp works to build [a8d002c](https://github.com/ggml-org/whisper.cpp/commit/a8d002cfd879315632a579e73f0148d06959de36), its actually a lot faster than I expected! Why this builds and not Head is out of scope atm. Might look into it at a later time :)

Building with
```bash
git clone https://github.com/ggml-org/whisper.cpp.git
cd whisper.cpp/
git reset a8d002c
sh ./models/download-ggml-model.sh base.en
cmake -B build -DGGML_VULKAN=1
cmake --build build -j --config Release
```

Current max setup for a pi5 with eGPU (This is stable, anything more than this and system dies :D)

System.d, made a script that loads and unloads the models between runs so I don't go OOM.
* whisper.cpp
* llama.cpp (important to make sure that it uses the GPU as much as possible and not the CPU by setting the `--gpu-layers` flag)

docker-compose (will move these to system.d and see if I can cramp in something else on this setup)
* z2m
* [wyoming-whisper-api-client](https://github.com/rhasspy/wyoming-whisper-cpp) (wyoming protocol for whisper.cpp server, just noticed its archived from 6th October)

It's been real fun, I am thinking about buying a couple of more cards and try to get them running. But for now I thank you for not scolding me for being overenthusiastic <3

simonpm, 2025-11-09:
I also have the RX9060XT 16G and am trying to find the correct firmware versions for it from linux sources. Does this GPU use psp13 or psp14? Could someone be so kind and list all the firmware files needed? Many thanks!

geerlingguy, 2025-11-09:
@simonpm - The easiest way is to install the `firmware-amd-graphics` package. Then reboot, and check `dmesg` for which files are missing. Then download those files, and reboot again. Then see if any others are missing. Rinse and repeat a few times and usually you end up with the right list.

I usually note all the files I had to download and list them in one of these issues for the benefit of my future self, next time I need to re-download them all!

geerlingguy, 2025-12-09:
@robzedd note that the link speed reported by `lspci` will go up and down dynamically with more or less activity on the card. If you max it out, and use `nvtop`, does it show different link speeds?

robzedd, 2025-12-10:
hi @geerlingguy 

Thanks for getting back to me. I just ran the GravityMark 1.89 benchmark and watched the GPU activity in nvtop like you suggested. The GPU stayed at 100% usage for about 160 seconds, and the PCIe link remained at Gen 2 the whole time.

kennedyshead, 2025-12-11:
There are a couple of quirks even with a new kernel and driver. I've managed to get the GPU "almost" stable by using these kernel orptions:
`amdgpu.ppfeaturemask=0xffffffff  amdgpu.runpm=0 pcie_port_pm=off pcie_aspm.policy=performance pci=noaer`

Explanation:
* `amdgpu.ppfeaturemask=0xffffffff `
    - This is for overclocking and some other features on the card, it actually seems to get more stable when its set, have no idea why. A plus is that the OC possibillities on this card seems very promising (As always be careful <3)
* `amdgpu.runpm=0 pcie_port_pm=off pcie_aspm.policy=performance` 
    - If you have a problem with power management on pcie port this is how I turned it of. I dont know if they are all needed but for me it looks like if I remove any of these things get unstable to the point that the card gets dropped from the port. I suggest adding and removing them one by one and experiment until you dont get a `D3Cold > D0` error.
* `pci=noaer`
    - There are a lot of "crap" errors for me on pcie port, could be hardware given my PWM problems 🤷 This however turns of the reporting of the crap errors and that makes a lot of "fuzz" disappear from dmesg.

I think the key for a stable GPU is turning off kernel params and features until dmesg is less "chrismassy" and then turn them back on one by one until they are as few as possible. But test should be done over a days usage if they seem stable IMO because sometimes the actual error takes some courage-building before they appear 😄

robzedd, 2025-12-15:
Hi,

I picked up new OCuLink and PCIe flex cables and double-checked that everything’s seated correctly, but still no luck. I’m going to try a different HAT next to see if that changes anything. Any recommendations?

kennedyshead, 2025-12-15:
Also, I put it to close to radio at one point and that caused interference

robzedd, 2025-12-18:
Quick update: I bought a new NVMe HAT (GeeekPi N04 M.2 M-Key NVMe SSD Shield for Raspberry Pi 5). The product description says:
"PCIe x1 interface in both Gen2 & Gen3 standards. The short trace routing of PCIe is more reliable and faster, fully meeting the signal requirements of PCIe 3.0."

Unfortunately that did not help at all.

I then tried to force the PCIe root port to retrain at Gen 3 by running the following commands:

VAL=$(sudo setpci -s 0001:00:00.0 CAP_EXP+0x30.W)
NEW=$(printf "%04x" $(( (0x$VAL & ~0xF) | 0x3 )))
sudo setpci -s 0001:00:00.0 CAP_EXP+0x30.W=$NEW
LC=$(sudo setpci -s 0001:00:00.0 CAP_EXP+0x10.W)
LCNEW=$(printf "%04x" $((0x$LC | 0x20)))
sudo setpci -s 0001:00:00.0 CAP_EXP+0x10.W=$LCNEW
sleep 1
sudo lspci -vv -s 0001:00:00.0 | egrep "LnkSta:|LnkCap:"

That didn’t help either- it looks like I’m stuck at Gen 2. I also tried different switch combinations on the GPU dock:

[image]

No luck...

At this point, I’m fairly convinced my specific Raspberry Pi 5 simply can’t maintain a stable Gen 3 link.