geerlingguy/raspberry-pi-pcie-devices #752: Add StarTech.com 3 Port 2b 1a 1394 Mini PCI Express FireWire Card Adapter

geerlingguy, 2025-08-29:
[image]

This FireWire adapter slots into a mini PCIe slot, and supplies two FireWire 800 ports, and a single FireWire 400.

User @RobertMCortese built [Open MRU](https://www.reddit.com/r/tapeless/comments/1n3aqsk/comment/nbc8f4g/), a Raspberry Pi 5-based FireWire camera capture device, using this card and a [GeeekPi PCIe to Mini PCIe HAT](https://amzn.to/47U0vFj).

Inside `/boot/firmware/config.txt`, he added:

```
dtparam=pciex1
dtoverlay=pcie-32bit-dma
```

And it seems to work! He recorded footage using [DVgrab](https://linux.die.net/man/1/dvgrab), and uploaded it to a few threads on Reddit. As someone who owns a working Canon GL2 (which is similar to the GL1 and XL1 I used for my earliest YouTube videos), I am very interested in replicating this setup, so I've ordered the hardware as well :)

See related discussion: https://github.com/geerlingguy/raspberry-pi-pcie-devices/discussions/751

RobertMCortese, 2025-09-02:
Hey Jeff, just a quick note.  The included power adapter is not needed if you're only using the 6 pin Firewire.

geerlingguy, 2025-09-02:
@RobertMCortese - So that's only for the FW 800 ports? Nice!

RobertMCortese, 2025-09-05:
So last night I looked up what’s sent down the firewire when you press the record button on your camera.  I’ll try to get this setup to start recording when that happens.

geerlingguy, 2026-03-03:
The setup:

[image]

To attempt some video capture via DVgrab:

```
$ sudo apt install -y dvgrab

$ dvgrab
Found AV/C device with GUID 0x000085000014e35a
libiec61883 error: Failed to get channels available.
Waiting for DV...
Capture Started
^C"dvgrab-002.dv":    45.89 MiB 401 frames timecode 00:00:00.00 date 2067.02.15 22:26:25
Capture Stopped
```

And here's the result of that recording (wow the tape mechanism is LOUD! Will have to take out the tape for DV capture through the Pi. Or use an external mic):

https://github.com/user-attachments/assets/87be87d3-0007-4ba5-a493-d38af51bd019

You can also run DVgrab interactively:

```
$ dvgrab -i
Found AV/C device with GUID 0x000085000014e35a
libiec61883 error: Failed to get channels available.
Going interactive. Press '?' for help.
q=quit, p=play, c=capture, Esc=stop, h=reverse, j=backward scan, k=pause        
l=forward scan, a=rewind, z=fast forward, 0-9=trickplay, <space>=play/pause
```

Very cool, could script grabbing video from tapes this way...