geerlingguy/raspberry-pi-pcie-devices #755: TEST GPU (MSI Radeon HD 7870 GHz Edition)

RSC-Games, 2025-09-03:
This is going to be a very short issue. I freshly compiled the kernel and the GPU literally _just works_. Supertuxkart at maximum settings is getting 40 FPS (this card is old and not amazingly powerful and even PRIME offloading is working perfectly. The iGPU is still active, and while the GPU's video outputs don't output anything, it is outputting a blank signal.

[image]

Apparently both the iGPU and dGPU _can_ coexist? For any "gaming RPi laptops" this could be interesting since PRIME offloading could be used and the dGPU powered down otherwise. The main issue is PCIe bandwidth, but that's already known.

``lspci`` confirms the drivers are loaded:
```
0001:01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Pitcairn XT [Radeon HD 7870 GHz Edition] (prog-if 00 [VGA controller])
	Subsystem: Micro-Star International Co., Ltd. [MSI] Pitcairn XT [Radeon HD 7870 GHz Edition]
	Flags: bus master, fast devsel, latency 0, IRQ 187
	Memory at 1800000000 (64-bit, prefetchable) [size=256M]
	Memory at 1b00000000 (64-bit, non-prefetchable) [size=256K]
	Expansion ROM at 1b00040000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: amdgpu
	Kernel modules: amdgpu

0001:01:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Oland/Hainan/Cape Verde/Pitcairn HDMI Audio [Radeon HD 7000 Series]
	Subsystem: Micro-Star International Co., Ltd. [MSI] Oland/Hainan/Cape Verde/Pitcairn HDMI Audio [Radeon HD 7000 Series]
	Flags: fast devsel, IRQ 255
	Memory at 1b00060000 (64-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: <access denied>
```

Here's ``vulkaninfo`` output:
```
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.239

Instance Extensions: count = 23
-------------------------------
VK_EXT_acquire_drm_display             : extension revision 1
VK_EXT_acquire_xlib_display            : extension revision 1
VK_EXT_debug_report                    : extension revision 10
VK_EXT_debug_utils                     : extension revision 2
VK_EXT_direct_mode_display             : extension revision 1
VK_EXT_display_surface_counter         : extension revision 1
VK_EXT_headless_surface                : extension revision 1
VK_EXT_surface_maintenance1            : extension revision 1
VK_EXT_swapchain_colorspace            : extension revision 5
VK_KHR_device_group_creation           : extension revision 1
VK_KHR_display                         : extension revision 23
VK_KHR_external_fence_capabilities     : extension revision 1
VK_KHR_external_memory_capabilities    : extension revision 1
VK_KHR_external_semaphore_capabilities : extension revision 1
VK_KHR_get_display_properties2         : extension revision 1
VK_KHR_get_physical_device_properties2 : extension revision 2
VK_KHR_get_surface_capabilities2       : extension revision 1
VK_KHR_portability_enumeration         : extension revision 1
VK_KHR_surface                         : extension revision 25
VK_KHR_surface_protected_capabilities  : extension revision 1
VK_KHR_wayland_surface                 : extension revision 6
VK_KHR_xcb_surface                     : extension revision 6
VK_KHR_xlib_surface                    : extension revision 6

Instance Layers: count = 2
--------------------------
VK_LAYER_MESA_device_select Linux device selection layer 1.4.303  version 1
VK_LAYER_MESA_overlay       Mesa Overlay layer           1.4.303  version 1

Devices:
========
GPU0:
	apiVersion         = 1.3.305
	driverVersion      = 25.0.7
	vendorID           = 0x1002
	deviceID           = 0x6818
	deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
	deviceName         = AMD Radeon HD 7800 Series (RADV PITCAIRN)
	driverID           = DRIVER_ID_MESA_RADV
	driverName         = radv
	driverInfo         = Mesa 25.0.7-2~bpo12+1
	conformanceVersion = 1.4.0.0
	deviceUUID         = 01000000-0100-0000-0000-000000000000
	driverUUID         = 414d442d-4d45-5341-2d44-525600000000
GPU1:
	apiVersion         = 1.3.305
	driverVersion      = 25.0.7
	vendorID           = 0x14e4
	deviceID           = 0x55701c33
	deviceType         = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
	deviceName         = V3D 7.1.10.2
	driverID           = DRIVER_ID_MESA_V3DV
	driverName         = V3DV Mesa
	driverInfo         = Mesa 25.0.7-2~bpo12+1
	conformanceVersion = 1.3.8.3
	deviceUUID         = 5fd8106e-741a-cafa-e080-fdb16cf11a80
	driverUUID         = 9650e00a-0fbd-fb21-a7a9-a45106048233
GPU2:
	apiVersion         = 1.4.305
	driverVersion      = 0.0.1
	vendorID           = 0x10005
	deviceID           = 0x0000
	deviceType         = PHYSICAL_DEVICE_TYPE_CPU
	deviceName         = llvmpipe (LLVM 15.0.6, 128 bits)
	driverID           = DRIVER_ID_MESA_LLVMPIPE
	driverName         = llvmpipe
	driverInfo         = Mesa 25.0.7-2~bpo12+1 (LLVM 15.0.6)
	conformanceVersion = 1.3.1.1
	deviceUUID         = 6d657361-3235-2e30-2e37-2d327e627000
	driverUUID         = 6c6c766d-7069-7065-5555-494400000000
```

I will move on to testing steam games later to see how stable this is, but it seems pretty solid so far. Honestly, this makes me far more willing to deal with the inevitable NVIDIA issues with #754. I will report back with stability updates, and if this is rock solid, I'll host kernel binaries for people to download so they don't have to build them themselves.

RSC-Games, 2025-09-03:
Oh hey blender works right out of the box! I'm testing with the v3d driver unloaded at boot time now so this is only via the dGPU. Steam (box86?) had issues with PRIME and I wanted to switch to my 144 Hz monitor anyway.

[image]

Panning around it's moving at the full refresh rate. I'll need to find an old sample scene of mine to test further.

RSC-Games, 2025-09-04:
@geerlingguy Could you mark this as ``testing-in-progress`` please? There's not tons to test but I want to make sure it's rock solid before I close the issue and mark it as "Working." It's been remarkably stable so far but I have encountered issues with Proton specifically- DXVK works fine in normal Wine but Proton causes Lethal Company to crash (and I cannot seem to recover enough logs to figure out why). I suspect this might be down more to a Proton/box64 issue than anything else though...

I've also noticed an issue where on the dGPU a lightdm session starts and I have to manually vt-switch (which won't show up since it's rendering on the VC7) to the terminal and type ``labwc`` before I'll get a desktop session. If there's a way to fully disable the VC7 that'd be amazing, since apparently fbdev keeps running on there even without the v3d drivers loaded. Other than that, it runs great.

Things I'd still like to test before I give my results

- [ ] Proton: WineD3D Vulkan
- [ ] Proton: DXVK (BROKEN)
- [ ] WineD3D
- [ ] Standalone DXVK
- [ ] General Windows games (Probably itch.io Unity games)
- [ ] Webglsamples
- [ ] Dolphin (starts, idk how games run yet)
- [ ] Azahar (3DS games ran *okay* on the VC7 but not super well)
- [ ] Minecraft with shaders
- [ ] Furmark
- [ ] Feel free to suggest more!

Once I have tested these, if I still haven't had issues, then it's probably safe to say it *just works*.

RSC-Games, 2025-09-05:
Hell yeah- this is looking amazing!

Without shaders (mostly limited by the CPU and probably also the old HDMI standard used by this card)
[image]

With shaders (GPU Shader core complex is maxed out)
[image]

32 chunk render distance for the first, 16 chunk with Complementary Shaders for the second.

RSC-Games, 2025-09-05:
After a couple hours of use we lost the GPU but it came back perfectly usable.
Idk if it was a card hang, PCIe error, or what (maybe RADV isn't as good on these cards as you would think) but it's still working...
```
[ 7341.709966] amdgpu 0001:01:00.0: amdgpu: Dumping IP State
[ 7341.709974] amdgpu 0001:01:00.0: amdgpu: Dumping IP State Completed
[ 7341.710053] amdgpu 0001:01:00.0: amdgpu: ring gfx timeout, signaled seq=1008408, emitted seq=1008411
[ 7341.710057] amdgpu 0001:01:00.0: amdgpu: Process information: process dolphin-emu pid 29259 thread Vulkan CommandB pid 31682
[ 7341.710061] amdgpu 0001:01:00.0: amdgpu: GPU reset begin!
[ 7342.268244] amdgpu 0001:01:00.0: amdgpu: PCI CONFIG reset
[ 7342.278035] amdgpu 0001:01:00.0: amdgpu: GPU reset succeeded, trying to resume
[ 7342.278105] [drm] PCIE gen 3 link speeds already enabled
[ 7342.279075] amdgpu 0001:01:00.0: amdgpu: PCIE GART of 1024M enabled (table at 0x000000F400000000).
[ 7342.828892] [drm] UVD initialized successfully.
[ 7343.213211] [drm:amdgpu_cs_ioctl [amdgpu]] *ERROR* Failed to initialize parser -125!
[ 7343.241507] amdgpu 0001:01:00.0: amdgpu: GPU reset(1) succeeded!
```

RSC-Games, 2025-09-05:
Nvm I think that was a fluke
I rebooted last night and had the Pi running for about 12-14 hours without incident, so there was probably some weird bus issue somewhere.

RSC-Games, 2025-09-14:
In light of the new patches given in #756 I will be re-testing with the upstream RPi kernel with my own TTM patches to see how viable they are / if they solve my current marginal stability issues detailed above.

I will likely revert to Coreforge's patches since the ``_io`` functions appear to improve performance if I'm understanding correctly.

RSC-Games, 2026-10-11:
With the yanghaku TTM patch and careful routing of the OCuLink cable (and Valve's AMDGPU work) this card works perfectly and I have not run into any major hang issues. Will test again on 7.3 since that's when the SI AMDGPU patches were finished, but I will be closing this issue as there is nothing left to research.

As an additional note, multi-headed video (so display out from both the dGPU and iGPU) works perfectly now and I suspect that is both due to updates to labwc-pi and mesa.

closed 2026-10-11 by RSC-Games

geerlingguy, 2026-10-12:
@RSC-Games Excellent!