geerlingguy/raspberry-pi-pcie-devices #764: Testing yanghaku's DRM patch for cache coherency on various graphics cards

geerlingguy, 2025-09-26:
I wanted to consolidate some of my testing into a single issue, as I'd like to see how well the simple fix from https://github.com/geerlingguy/raspberry-pi-pcie-devices/discussions/756 works for various graphics cards.

I'll leave further details in the comments, but from a high level, here are cards that have been tested under Raspberry Pi OS `rpi-6.15.y` with that patch applied:

| Card | Works? | Notes |
| :-- | :-- | :-- |
| AMD Radeon RX 7900 XT 20GB | Yes | Everything runs smoothly, including vkmark and glmark |
| AMD Radeon RX 6500 XT 8GB | Yes | GUI felt choppy until restarting with `sudo systemctl restart lightdm` |
| TODO | TODO | TODO |

RSC-Games, 2025-09-26:
I don't mind testing my 3 cards later on with the new patches this weekend
I have the Radeon HD 7870, GTX 1050 Ti, then a RTX 4060

geerlingguy, 2025-09-29:
Interestingly, I was testing on a fresh Pi OS install today, and found the 7900 XT was only getting [64817](https://gravitymark.tellusim.com/report/?id=05eadce14e274c7615b230bf5837c06d95db4c9d) on PCIe Gen 2 (the default). After bumping to Gen 3, I got [65809](https://gravitymark.tellusim.com/report/?id=9aa0c3c3b00da17b1ed4ce10625cc765cbb160b0).

geerlingguy, 2025-09-29:
@6by9 That'd be great—and I will also try to test with nouveau at some point, just wanting to finish hammering out all the AMD cards I can get my hands on for testing right now.

I was also noticing on the RX 460 that @Coreforge and I both tested, his scores were mostly slightly higher than mine on GravityMark:

[image]

Could be there are some optimized paths in the amdgpu-driver-specific changes that aren't accounted for with the DRM-wide changes; but the difference is relatively small.

geerlingguy, 2025-09-30:
@6by9 - Testing with two different generations of Nvidia GPUs, I get a similar kernel dump on each with the `nouveau` driver:

```
[   46.089143] nouveau 0001:01:00.0: enabling device (0000 -> 0002)
[   46.089176] nouveau 0001:01:00.0: NVIDIA GA104 (b74000a1)
[   46.234439] nouveau 0001:01:00.0: bios: version 94.04.57.00.08
[   46.235150] ------------[ cut here ]------------
[   46.235159] nouveau 0001:01:00.0: swiotlb addr 0x00000010fce00000+4096 overflow (mask ffffffff, bus limit ffffffffff).
[   46.235177] WARNING: CPU: 3 PID: 949 at kernel/dma/swiotlb.c:1594 swiotlb_map+0x2c0/0x2f0
[   46.235193] Modules linked in: nouveau(+) drm_gpuvm i2c_algo_bit drm_exec gpu_sched drm_display_helper drm_ttm_helper ttm drm_client_lib drm_kms_helper snd_seq_dummy snd_hrtimer snd_seq snd_seq_device snd_timer snd rfcomm algif_hash algif_skcipher af_alg bnep binfmt_misc brcmfmac_wcc spidev aes_ce_blk aes_ce_cipher ghash_ce rpi_hevc_dec hci_uart pisp_be btbcm v4l2_mem2mem videobuf2_dma_contig gf128mul bluetooth sha2_ce sha256_arm64 sha1_ce raspberrypi_hwmon sha1_generic brcmfmac brcmutil ecdh_generic ecc cfg80211 videobuf2_memops videobuf2_v4l2 spi_bcm2835 videodev gpio_keys rfkill videobuf2_common mc raspberrypi_gpiomem rp1_pio rp1_adc rp1_mailbox rp1_fw sg nvmem_rmem joydev hid_multitouch uio_pdrv_genirq uio drm i2c_dev ledtrig_pattern fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[   46.235294] CPU: 3 UID: 0 PID: 949 Comm: modprobe Not tainted 6.15.11-v8-4k-gpu+ #1 PREEMPT 
[   46.235299] Hardware name: Raspberry Pi 500 Rev 1.0 (DT)
[   46.235302] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   46.235305] pc : swiotlb_map+0x2c0/0x2f0
[   46.235308] lr : swiotlb_map+0x2c0/0x2f0
[   46.235310] sp : ffffffc0837b3630
[   46.235311] x29: ffffffc0837b3640 x28: ffffff8301bb4c00 x27: ffffffd08d667000
[   46.235316] x26: 000000108121b000 x25: fffffffec20486c0 x24: 0000000000001000
[   46.235321] x23: ffffff8003a5c0c8 x22: 0000000000000000 x21: 0000000000000000
[   46.235325] x20: 000000008121b000 x19: 000000108121b000 x18: 00000000ffffffff
[   46.235330] x17: 667265766f203639 x16: 30342b3030303030 x15: 6563663031303030
[   46.235334] x14: 3030307830207264 x13: 2e29666666666666 x12: 666666662074696d
[   46.235338] x11: 696c20737562202c x10: ffffffd08dd1eca8 x9 : ffffffd08c639948
[   46.235342] x8 : 00000000ffffefff x7 : ffffffd08dd1eca8 x6 : 80000000fffff000
[   46.235346] x5 : ffffff807ffdd5c8 x4 : 0000000000000000 x3 : 0000000000000000
[   46.235350] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8081220000
[   46.235354] Call trace:
[   46.235356]  swiotlb_map+0x2c0/0x2f0 (P)
[   46.235360]  dma_map_page_attrs+0x1d8/0x368
[   46.235365]  nvkm_fb_ctor+0xc4/0x100 [nouveau]
[   46.235554]  gf100_fb_new_+0x68/0x98 [nouveau]
[   46.235697]  ga102_fb_new+0x6c/0x88 [nouveau]
[   46.235832]  nvkm_device_ctor+0xaec/0x36d8 [nouveau]
[   46.235968]  nvkm_device_pci_new+0xe0/0x2b8 [nouveau]
[   46.236105]  nouveau_drm_probe+0x4c/0x1e0 [nouveau]
[   46.236244]  local_pci_probe+0x48/0xb8
[   46.236253]  pci_device_probe+0xcc/0x1d8
[   46.236256]  really_probe+0xc4/0x2a8
[   46.236262]  __driver_probe_device+0x80/0x140
[   46.236266]  driver_probe_device+0x44/0x170
[   46.236270]  __driver_attach+0x9c/0x1b0
[   46.236274]  bus_for_each_dev+0x84/0xf0
[   46.236277]  driver_attach+0x2c/0x40
[   46.236281]  bus_add_driver+0xec/0x218
[   46.236285]  driver_register+0x68/0x138
[   46.236289]  __pci_register_driver+0x54/0x68
[   46.236292]  nouveau_drm_init+0x24c/0xff8 [nouveau]
[   46.236427]  do_one_initcall+0x64/0x288
[   46.236431]  do_init_module+0x60/0x230
[   46.236435]  load_module+0x1c70/0x1f08
[   46.236438]  __do_sys_init_module+0x188/0x1f8
[   46.236441]  __arm64_sys_init_module+0x24/0x38
[   46.236444]  invoke_syscall+0x50/0x120
[   46.236448]  el0_svc_common.constprop.0+0x48/0xf0
[   46.236452]  do_el0_svc+0x24/0x38
[   46.236455]  el0_svc+0x30/0xd0
[   46.236459]  el0t_64_sync_handler+0x10c/0x138
[   46.236462]  el0t_64_sync+0x198/0x1a0
[   46.236465] ---[ end trace 0000000000000000 ]---
```

I've added links to the two cards I've tested. I will also try the open source driver if I get some time.

RSC-Games, 2025-09-30:
It's using ``swiotlb`` oddly- this will merit a lot of research later. I was under the impression the Pi 5 didn't need swiotlb?

geerlingguy, 2025-09-30:
@RSC-Games - Going to quickly recompile here: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/692#issuecomment-3353768036

RSC-Games, 2025-10-01:
Okay after a quick google it looks like ``swiotlb`` is for bounce buffering for older devices (like cards limited to 32 bit DMA). It probably would've helped a lot with the Pi 4 PCIe issues back in those days...

For some reason Nouveau's trying to map a 64 bit address in the ``swiotlb`` range, which is limited to 32 bit addressing (as is shown here:
``[   46.235159] nouveau 0001:01:00.0: swiotlb addr 0x00000010fce00000+4096 overflow (mask ffffffff, bus limit ffffffffff)``

I'm taking stabs in the dark here as I haven't done in-depth analysis on any of these parts of the kernel, but maybe there's some compile time switch somewhere that's forcing the framebuffer memory allocator to fall back to ``swiotlb`` even though the Pi 5 is definitely capable of 64 bit DMA. This is odd and honestly smells more like a odd driver quirk than a Pi issue.

RSC-Games, 2025-10-01:
Random thought- @geerlingguy do you know if Nouveau works on an ARM system that has a coherent PCIe bus, like the Raxda or whatever it is? I'm sure you have an ARM board on hand somewhere that is fully SBSA compliant. I'm curious to see what happens there... It should work fine but if it doesn't then we have our work cut out for us.

6by9, 2025-10-01:
I'd initially made this comment on https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/2 as it was the earliest Github notification I'd had referencing nouveau WARNs. Duplicating here as it seems to be a better place.

I'm slightly intrigued how you added a PCIe device to a Pi500, but I know the hacking you (and red shirt Jeff) do :-)

swiotlb would be using bounce buffers. There should be no need for those, which makes me wonder if you still have dtoverlay=pcie-32bit-dma still loaded from testing other cards. Limiting the PCI window size to 4GB may well cause this sort of error.

Then again there is funky code in nouveau trying to set a DMA mask at https://github.com/raspberrypi/linux/blob/rpi-6.12.y/drivers/gpu/drm/nouveau/nvkm/engine/device/pci.c#L1687-L1697, except your backtrace says we're in nvkm_device_ctor which is called from line 1674, so just before this.
All most curious, and may imply that dtoverlay=pcie-32bit-dma would help if it's not already loaded (bus limit ffffffffff is 40 bits, which implies that it hasn't got a restriction, but the device does due to `mask ffffffff`).

It's probably also worth saying that DRM doesn't play nicely with bounce buffers. DRM expects that once a buffer is on display, the backing buffer can be amended by userspace and that updates what is on the screen, although updates may tear. If there are bounce buffers in the way, that falls apart.

BTW Life may be easier if you start with a Trixie image rather than Bookworm. Latest nightly build is at https://downloads.raspberrypi.com/nightlies/

RSC-Games, 2025-10-01:
It might be time to go wild with the ``printk``s in ``dma_map_page_attrs`` - if you determine the caller then it could be a quite reliable source of debug information... Could trigger an oops or a kernel stack dump in the nouveau code right before the dma mapping is made so we can tell which one is related...
https://github.com/raspberrypi/linux/blob/2f4a28199c418599ba0224186e42926b482b523c/kernel/dma/mapping.c#L155

6by9, 2025-10-01:
Looking at the callstack:

dma_map_page_attrs
nvkm_fb_ctor - https://elixir.bootlin.com/linux/v6.17/source/drivers/gpu/drm/nouveau/nvkm/subdev/fb/base.c#L286
gf100_fb_new_ - https://elixir.bootlin.com/linux/v6.17/source/drivers/gpu/drm/nouveau/nvkm/subdev/fb/gf100.c#L114

So it's likely that it's the fact that they're assuming they can `dma_map_page` on an arbitrary page allocated by [`alloc_page`](https://elixir.bootlin.com/linux/v6.17/C/ident/alloc_page)([GFP_KERNEL](https://elixir.bootlin.com/linux/v6.17/C/ident/GFP_KERNEL) | [__GFP_ZERO](https://elixir.bootlin.com/linux/v6.17/C/ident/__GFP_ZERO)); 

I haven't done any analysis of what `flush_page` is used for.
Quickest hack to test would be changing https://elixir.bootlin.com/linux/v6.17/source/drivers/gpu/drm/nouveau/nvkm/subdev/fb/gf100.c#L127 to `.sysmem.flush_page_init = NULL` so that whole block in `nvkm_fb_ctor` is skipped due to https://elixir.bootlin.com/linux/v6.17/source/drivers/gpu/drm/nouveau/nvkm/subdev/fb/base.c#L281

RSC-Games, 2025-10-01:
This'll be interesting to watch. If we're skipping the framebuffer creation, who knows what's going to happen with the full blown DRM init down the line.... might give a lot more insight though. I guess my main question is why *can't* we ``dma_map_page`` on an arbitrary page on the Pi 5?

6by9, 2025-10-01:
That gets a little further (I needed to modify drivers/gpu/drm/nouveau/nvkm/subdev/fb/gk110.c for my card), but then hits timeouts.
```
[   36.216242] nouveau 0001:01:00.0: enabling device (0000 -> 0002)
[   36.216280] nouveau 0001:01:00.0: NVIDIA GK208B (b060b0b1)
[   36.367397] nouveau 0001:01:00.0: bios: version 80.28.b8.00.05
[   36.674395] nouveau 0001:01:00.0: fb: 2048 MiB GDDR5
[   37.993376] nouveau 0001:01:00.0: drm: VRAM: 2048 MiB
[   37.993387] nouveau 0001:01:00.0: drm: GART: 1048576 MiB
[   37.993391] nouveau 0001:01:00.0: drm: TMDS table version 2.0
[   37.999981] nouveau 0001:01:00.0: drm: MM: using COPY for buffer copies
[   40.000507] nouveau 0001:01:00.0: drm: core caps notifier timeout
[   40.002003] [drm] Initialized nouveau 1.4.0 for 0001:01:00.0 on minor 2
[   40.082695] sysfs: cannot create duplicate filename '/class/graphics/fb0'
[   40.082716] CPU: 2 UID: 0 PID: 1654 Comm: modprobe Not tainted 6.17.0-v8+ #1 PREEMPT 
[   40.082721] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[   40.082723] Call trace:
[   40.082725]  show_stack+0x20/0x38 (C)
[   40.082736]  dump_stack_lvl+0x78/0x90
[   40.082740]  dump_stack+0x18/0x28
[   40.082742]  sysfs_warn_dup+0x6c/0x90
[   40.082745]  sysfs_do_create_link_sd+0xf8/0x108
[   40.082749]  sysfs_create_link+0x28/0x50
[   40.082750]  device_add+0x248/0x720
[   40.082753]  device_create_groups_vargs+0xe8/0x148
[   40.082756]  device_create+0x64/0x98
[   40.082758]  fb_device_create+0x58/0xf8
[   40.082763]  do_register_framebuffer+0x154/0x288
[   40.082766]  register_framebuffer+0x34/0x60
[   40.082768]  __drm_fb_helper_initial_config_and_unlock+0x30c/0x5a0 [drm_kms_helper]
[   40.082806]  drm_fb_helper_initial_config+0x4c/0x68 [drm_kms_helper]
[   40.082819]  drm_fbdev_client_hotplug+0x8c/0xe8 [drm_client_lib]
[   40.082824]  drm_client_register+0x60/0xb0 [drm]
[   40.082920]  drm_fbdev_client_setup+0xf0/0xc60 [drm_client_lib]
[   40.082922]  drm_client_setup+0xbc/0xe8 [drm_client_lib]
[   40.082924]  nouveau_drm_probe+0x14c/0x1d8 [nouveau]
[   40.083116]  local_pci_probe+0x48/0xd0
[   40.083119]  pci_device_probe+0xac/0x1b8
[   40.083121]  really_probe+0xc4/0x2a8
[   40.083124]  __driver_probe_device+0x80/0x140
[   40.083126]  driver_probe_device+0x48/0x170
[   40.083129]  __driver_attach+0x9c/0x1b0
[   40.083132]  bus_for_each_dev+0x7c/0xe8
[   40.083134]  driver_attach+0x2c/0x40
[   40.083136]  bus_add_driver+0xec/0x218
[   40.083138]  driver_register+0x68/0x138
[   40.083141]  __pci_register_driver+0x4c/0x60
[   40.083143]  nouveau_drm_init+0x1cc/0xff8 [nouveau]
[   40.083259]  do_one_initcall+0x4c/0x280
[   40.083262]  do_init_module+0x60/0x268
[   40.083267]  load_module+0x1d6c/0x1ef8
[   40.083270]  __do_sys_init_module+0x180/0x200
[   40.083273]  __arm64_sys_init_module+0x24/0x38
[   40.083276]  invoke_syscall+0x50/0x120
[   40.083278]  el0_svc_common.constprop.0+0x48/0xf8
[   40.083280]  do_el0_svc+0x28/0x40
[   40.083281]  el0_svc+0x34/0xf0
[   40.083284]  el0t_64_sync_handler+0xa0/0xe8
[   40.083287]  el0t_64_sync+0x198/0x1a0
[   40.083937] Unable to create device for framebuffer 0; error -17
[   40.083953] nouveau 0001:01:00.0: [drm] fb0: nouveaudrmfb frame buffer device
[   42.154952] nouveau 0001:01:00.0: drm: core notifier timeout
[   44.155062] nouveau 0001:01:00.0: drm: base-0: timeout
```
The driver shouldn't be trying to force itself to be `/class/graphics/fb0` as that is allocated via the DRM core. I'm not even sure how it's managing to do that as [`do_register_framebuffer`](https://elixir.bootlin.com/linux/v6.17/source/drivers/video/fbdev/core/fbmem.c#L437) should be keeping track of that and know that vc4 has already allocated /dev/fb0.

The timeouts are more important though. kmsprint is identifying my connected monitor correctly, and thinks it's switching to 1080p60, but no apparent output.

6by9, 2025-10-01:
Seeing as nouveau was blowing up, I've switched to an Intel Arc A380 I've got. I've picked up the extra patch to disable trying to reset the console, otherwise it throws a total wobbler.

Xe is also throwing the same kernel trace for `sysfs: cannot create duplicate filename '/class/graphics/fb0'`. That implies that something is wrong in the core, and probing either xe or nouveau is trying to evict vc4, which it shouldn't.

I suspect because of that I'm not getting a console on the display, and it also appears to be locking up my SD card. I'm also getting a kernel NULL pointer deref from fbcon_cursor, probably because of the above partially initialised framebuffer. Sorting that is probably worthwhile.

6by9, 2025-10-01:
> [@6by9](https://github.com/6by9) - Yeah, that's why I wound up commenting out `dtoverlay=vc4-kms-v3d` in `config.txt`; I couldn't get Xe to work at all with the built-in GPU trying to initialize.
> 
> Are you saying that could be a bug in the VC4 initialization that Raspberry Pi could fix?

I don't think this is a bug in vc4.
The DRM core handles allocating the /dev/fbN nodes for the emulated fbdev devices, so it shouldn't allocate duplicates. That's what the `registered_fb` array at https://elixir.bootlin.com/linux/v6.17/source/drivers/video/fbdev/core/fbmem.c#L34 is meant to be handling.

There is a call [`aperture_remove_all_conflicting_devices`](https://elixir.bootlin.com/linux/v6.17/source/drivers/gpu/drm/vc4/vc4_drv.c#L363) that vc4 (and others) call to evict the simple_framebuffer that the firmware sets up. I'm wondering if these PCIe drivers are calling something else that tries to unbind the console from under vc4, and that is throwing things off.

> I know in the past, the use case of "there being another graphics accelerator on a Pi" was seen as pretty wildly off course, but seeing how easy it is to use AMD cards now... maybe time to make the iGPU more of a standard iGPU that can easily coexist with external GPUs!

We already have multiple DRM devices in play as vc4, v3d, tinydrm, and RP1's DPI, DSI, and VEC are all independent DRM devices. Those all play nicely together, so I'm not sure yet why these PCIe cards are causing grief.
There is an [`unregister_framebuffer` call](https://elixir.bootlin.com/linux/v6.17/source/drivers/video/fbdev/core/fbmem.c#L585) which would appear to allow detaching of a framebuffer, so that'll be one of my first log points.

6by9, 2025-10-01:
> [@6by9](https://github.com/6by9) you should be able to use the bookworm-backports version of mesa. This is an alchemist card, right?

It's an Arc A380, which [wiki](https://en.wikipedia.org/wiki/Intel_Arc) says is an Alchemist card (Arc 3).

Debian appears to say that iris_dri.so isn't in any of the variants of libgl1-mesa-dri - https://packages.debian.org/bookworm-backports/arm64/libgl1-mesa-dri/filelist. `apt-file search iris_dri.so` also turns up a blank.
Have they hidden it away in some separate package?

(Nearly built from master anyway)

geerlingguy, 2025-10-01:
@6by9 on Bookworm, I followed this process and Xe was happy with both my A750 and B580: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/510#issuecomment-2605722331

So not sure if maybe the A380 could be different?

6by9, 2025-10-01:
So it's parsing some table of stuff, which is a concatenation of 
```
struct hwconfig {
   uint32_t key;
   uint32_t len;
   uint32_t val[];
};
```
entries. A bit of logging gives me
```
current 0x557d81de30 end 0x557d81e13c
current 0x557d81de30 key is 00000001 len 1, next 0x557d81de3c
current 0x557d81de3c key is 00000002 len 1, next 0x557d81de48
current 0x557d81de48 key is 00000003 len 1, next 0x557d81de54
current 0x557d81de54 key is 00000004 len 1, next 0x557d81de60
current 0x557d81de60 key is 0000003f len 1, next 0x557d81de6c
current 0x557d81de6c key is 00000040 len 0, next 0x557d81de74
current 0x557d81de74 key is 00000000 len 0, next 0x557d81de7c
current 0x557d81de7c key is 00000000 len 0, next 0x557d81de84
current 0x557d81de84 key is 00000000 len 0, next 0x557d81de8c
current 0x557d81de8c key is 00000000 len 0, next 0x557d81de94
current 0x557d81de94 key is 00000000 len 0, next 0x557d81de9c
current 0x557d81de9c key is 00000000 len 0, next 0x557d81dea4
current 0x557d81dea4 key is 00000000 len 0, next 0x557d81deac
current 0x557d81deac key is 00000000 len 1, next 0x557d81deb8
current 0x557d81deb8 key is 00000001 len 1075904512, next 0x567e05dec0
```
Key values come from https://gitlab.freedesktop.org/mesa/mesa/-/blob/main/src/intel/dev/intel_hwconfig_types.h#L17, so 0 is invalid, and that length of 1075904512 is total rubbish. I don't know if that is my card advertising rubbish, or another artifact of cache issues.

Add a `break` if key is 0, and remove `assert(current == end);` from the end, and both kmscube and labwc-pi work! labwc-pi has a few artifacts, but kmscube is fine.

......

Take that back. Rebooted with vc4-kms-v3d enabled, and the hwconfig table has changed and fails totally.

I have a suspicion that mesa isn't checking carefully enough that it has the correct DRM node for the Xe card.
- Blacklist v3d but leave vc4 and it doesn't.
- Blacklist vc4 but leave v3d and it works.
- Blacklist vc4, modprobe xe first, then modprobe vc4, and it also works, however the hwconfig list is truncated.
- Blacklist all 3, modprobe xe, then vc4, then v3d, and it works with a correct looking hwconfig list.

I've pushed my in-progress branch to https://github.com/raspberrypi/linux/pull/7072 for backup and CI builds.

6by9, 2025-10-01:
> Blacklist all 3, modprobe xe, then vc4, then v3d, and it works with a correct looking hwconfig list.

With that config I can start labwc-pi with the desktop spread across both Intel and vc4 display outputs, although there are definite composition issues.
This is becoming tasty! Time to try amdgpu seeing as you say that is working better than either of the others.

RSC-Games, 2025-10-01:
AMDGPU at this point is near perfect

geerlingguy, 2025-10-01:
@6by9 I've updated my [Mesa update / compile instructions](https://gist.github.com/geerlingguy/fbdc2c52fcf8ba3b87f04323f3dc517c) and stuck them in a Gist so I don't lose them as easily.

I just re-tested everything with kernel `rpi-6.17.y`, the nightly Trixie build, and the RX 9070 XT, and everything works _without_ a newer Mesa (the default Mesa works great now, just needed a few extra firmware files).

What's odd is I only got a score of [41,193](https://gravitymark.tellusim.com/report/?id=d6fc73192dd93f56559cf8687fdd6b94575f9ab7) (re-tested 3x) with this Trixie setup, compared to [56,944](https://gravitymark.tellusim.com/report/?id=b4330d441e83e02bf379b1d2757c44ca2434fffa) under Bookworm with Mesa compiled from source. That seems like a pretty hefty performance delta!

6by9, 2025-10-01:
Yes the stock Trixie Mesa seems to work fine with amdgpu with my RX580.
Restoring back to it after having built your own is a pain as it's split across about a number of different Debian packages. My self-built mesa was using llvmpipe instead of the hardware 3D block for some reason. I was expecting it to have been selected by default.

I haven't run benchmarks, but it's possible something in the kernel has made a difference. Otherwise there may be some optimisations disabled in mesa for the Trixie builds. 

Testing labwc across amdgpu and vc4 will have to wait until I'm in the office tomorrow - I haven't got enough displays here, and I forgot to pick up a micro-HDMI lead!

6by9, 2025-10-02:
amdgpu (with RX580) and vc4/v3d appears to all be good. labwc is quite happy rendering across the two DRM devices. I think I'll agree with RSC-Games that there is very little more to do on that.

My other AMD cards are unsupported by amdgpu, despite R5 230 supposedly being Sea Islands and hence supported :-(
(Both are `[AMD/ATI] Caicos [Radeon HD 6450/7450/8450 / R5 230 OEM]` according to lspci. One is an R5 230 and the other a 6450. Both have a VID:PID of 1002:6779, and that is listed as [unsupported in the amdgpu driver](https://elixir.bootlin.com/linux/v6.17/source/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c#L1406)).

The radeon driver works at a DRM level, but as the driver doesn't support DRM atomic operations wlroots/labwc fail. That's pretty much dead in the water for Wayland support.
kmscube works fine though.
Overall that matches with the observations in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/767

geerlingguy, 2025-10-02:
@6by9 yeah, it seems like the radeon-driver-cards will work with legacy stuff and X11, but getting them going with newer things was not happening, so I gave up on that. Still could be useful for some limited use cases! For now, I think it'd be easiest and most productive to focus on the amdgpu-driver-using cards. You can find some cheaper 6000-series cards like the 6400 and 6500 in single slot, no extra power variants, which are easiest to deploy. They still offer a decent improvement over the iGPU on the Pi, but the best bang for the buck is a little newer or more substantial VRAM cards.

geerlingguy, 2025-10-02:
Testing with Intel cards on Trixie and `rpi-6.17.y`, I ran into two different issues:

  - B580 would get PCODE and power limit errors: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/695#issuecomment-3358326842
  - A750 would seem to work fine, but just... not finish initializing: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/510#issuecomment-3361849388

I will try `rpi-6.15.y` on Trixie next, to see if it's a kernel thing.